iiab/iiab #2762: Notes: Stalled progress on nginx front.

jvonau, 2021-04-29:
Just notes
>ubuntu@box:/opt/iiab/iiab$ grep "Set 'apache_install: True'" -r roles
roles/elgg/tasks/install.yml:- name: "Set 'apache_install: True' and 'apache_enabled: True'"
roles/moodle/tasks/install.yml:- name: "Set 'apache_install: True' and 'apache_enabled: True'"
roles/pbx/tasks/install.yml:- name: "Set 'apache_install: True' and 'apache_enabled: True'"
roles/lokole/tasks/install.yml:- name: "Set 'apache_install: True' and 'apache_enabled: True'"
roles/phpmyadmin/tasks/install.yml:- name: "Set 'apache_install: True' and 'apache_enabled: True'"
roles/nodered/tasks/install.yml:- name: "Set 'apache_install: True' and 'apache_enabled: True'"

Solutions in age of appearance

moodle - [moodle-nginx.conf.j2.native](https://github.com/iiab/iiab/blob/master/roles/moodle/templates/moodle-nginx.conf.j2.native) from day one of nginx to try and should work, nobody to champion the cause [upstream usage](https://github.com/moodlebox/moodlebox/blob/master/roles/webserver/templates/etc/nginx/sites-available/default.j2)
nodered - #2154 very basic layout self tested to splash, nobody to champion the cause 
lokole - #2507 has links, nobody to champion the cause
phpmyadmin - pending PR #2712

Pending moodle feedback, should point to a solution for these:
elgg - might be handled like nextcloud or moodle-nginx.conf.j2.native?
pbx - no nginx shim, might be handled like nextcloud or moodle-nginx.conf.j2.native?

Shame all install ok, apache could go way if the effort was spent.

holta, 2021-04-29:
https://github.com/iiab/iiab/blob/master/roles/nginx/README.md should link to this ticket.

jvonau, 2021-05-02:
http://learn.elgg.org/en/stable/intro/install/nginx.html

holta, 2021-05-21:
FYI the painstaking progress in removing Apache (legacy dependencies, etc) is also being documented here:

- [roles/3-base-server/tasks/main.yml#L11-L18](https://github.com/iiab/iiab/blob/master/roles/3-base-server/tasks/main.yml#L11-L18)

holta, 2021-06-21:
Kolibri fails to display videos on iPhones, LG phones, etc -- as a result of HTTP 1.1 byte-ranges failing when Kolibri is being proxied:

https://github.com/learningequality/kolibri/issues/8164

_Thanks to anyone/everyone who might have insight into fixing this very serious failure!_

jvonau, 2021-06-28:
> Kolibri fails to display videos on iPhones, LG phones, etc -- as a result of HTTP 1.1 byte-ranges failing when Kolibri is being proxied:
> 
> [learningequality/kolibri#8164](https://github.com/learningequality/kolibri/issues/8164)
> 
> _Thanks to anyone/everyone who might have insight into fixing this very serious failure!_

This is the reason I closed this issue, the above has nothing to do with the subject of this issue, stop adding useless, unrelated entries, please remove/edit your post.

holta, 2021-07-01:
@deldesir if you choose to get involved with this important NGINX work, please see:

- @jvonau's new PR #2839
- https://www.nginx.com/resources/wiki/start/ (I too should read these, particularly any of these 100+ sections that you & others recommend!)

holta, 2021-08-09:
Huge progress: PR #2920 is done!

Thanks @jvonau for pushing, allowing this huge task to come together after 2 years.

Still there are lingering NGINX issues above, so I feel this tkt's still quite useful, to keep perspective tracking those for a bit longer.

jvonau, 2021-08-10:
And to think this all started out with #1810, which I think is complete, and #2003 where there was no nginx in sight.

holta, 2021-09-12:
Worth noting:

- https://github.com/iiab/iiab/issues/2953#issuecomment-916366174

jvonau, 2025-04-06:
Don't care if it's complete or not.

closed 2025-04-06 by jvonau