iiab/iiab #2766: NGINX advice on making http://box/cups and http://box/print work?

holta, 2021-04-30:
If a volunteer can be found to help here, @georgejhunt suggests this is a pretty easy job!  If one follows https://anthe.studio/blog/en/cups-nginx-reverse-proxy ?

Background:

- https://github.com/iiab/iiab/pull/2761
- https://github.com/iiab/iiab/tree/master/roles/nginx (README.md)

deldesir, 2021-04-30:
Good. I'll do it.

deldesir, 2021-05-04:
See https://github.com/iiab/iiab/pull/2775.

holta, 2021-06-18:
> See #2775.

Let's finalize this one way or another.  Thanks @deldesir

holta, 2021-07-14:
Good Enough for now?

http://box/print and http://box/print/ now work in all cases, thanks to PR #2858 inspired by PR #2775.

(@deldesir thanks for pushing to make this to happen.  Please re-open this ticket — or any other — to push this or any similar further!)

closed 2021-07-14 by holta