iiab/iiab #2767: Lokole 0.6.11 released, but pip install not yet available

holta, 2021-04-30:
Thanks @nzola for this install attempt readout:

```
Running WifiSetup
Traceback
pi@raspberrypi:~ $ curl -fsO https://raw.githubusercontent.com/ascoderu/lokole/master/install.py && (most recent call last):
  File "install.py", line 933, in <module>
    cli()
  File "install.py", line 929, in cli
    main(parser.parse_args(), parser.error)
  File "install.py", line 797, in main
    wifi_setup()
  File "install.py", line 83, in _call_
    result = self._run()
  File "install.py", line 274, in _run
    self._configure_wifi()
  File "install.py", line 349, in _configure_wifi
    self.sh('systemctl start hostapd.service')
  File "install.py", line 202, in sh
    raise Exception(stderr)
Exception: Job for hostapd.service failed because the control process exited with error code.
See "systemctl status hostapd.service" and "journalctl -xe" for details.
pi@raspberrypi:~ $
```

@adamsclafani & @c-w, do you know if Lokole 0.6.11 might possibly be available at https://pypi.org/project/opwen-email-client/ for IIAB in coming days?

( Based on https://github.com/ascoderu/lokole/releases/tag/0.6.11 )

jvonau, 2021-04-30:
When did Lokole gain the ability to control network services?

jvonau, 2021-04-30:
> pi@raspberrypi:~ $ curl -fsO https://raw.githubusercontent.com/ascoderu/lokole/master/install.py &&

Is this installation routine being tested outside of IIAB?
https://github.com/ascoderu/lokole/blob/master/install.py#L261-L265
Those 2 services would be installed before the role is reached in IIAB. Think this testing is missing some prerequisites before starting the install of Lokole.

nzola, 2021-04-30:
@jvonau Yes Jerry this is being tested outside of IIAB.

nzola, 2021-04-30:
@holta Please add Shaun Bathgate (me@shaunb.ca) in this conversation. He is the one who is volunteering on PR merge. @sbathgate

nzola, 2021-04-30:
> Many thanks @sbathgate <https://github.com/sbathgate>, if you have any
> time to work on this!
>
> (Before Victoria Day, or Nzola will send the Canadian Armed Forces after
> you beware ;)

LOL! I cannot win. Shaun was one of the Canadian's Navy..He knows the in
and out .LOL

holta, 2021-05-03:
Huge Thanks to @sbathgate and @c-w for the server-side work on [Lokole 0.6.12](https://github.com/ascoderu/lokole/releases/tag/0.6.12) released overnight!

1) Will PR https://github.com/ascoderu/lokole/pull/512 (Implement Terraform) also be necessary?

2) What's your general recommendation on the client side ?
(e.g. https://pypi.org/search/?o=&q=opwen_email_client does not show a release since July 2020's 0.6.10)

_Thank You to Everyone (including @adamsclafani) for your very hard work here &mdash; as everyone's now trying hard to bring this Lokole + IIAB integration to completion!_

Ref: https://github.com/ascoderu/lokole/issues/454

holta, 2021-05-13:
Lokole 0.6.13 (2 days ago) and 0.6.14 (yesterday) were both released with the pip install option restored!  As can be seen here: https://pypi.org/project/opwen-email-client/

- Should we close this ticket now?
- What's the best place to open a new ticket to track general progress here?

_Thanks @jvonau and **everybody** for the heroic + ongoing efforts!_

PS @deldesir in Haiti would like to assist in testing as this begin to converge for actual field/community use.

jvonau, 2021-05-13:
I'd file any new issues directly upstream. Keep this one open for what the OPWEN variables should be set to for IIAB purposes from #2773 and dumping apache #2507 that @deldesir might be interested in.

jvonau, 2021-05-15:
Rename the title of this issue to reflect tracking environment variables used by lokole as I'm prohibited... smh..

nzola, 2021-05-15:
I think naming or renaming is Adam H. job?

holta, 2021-05-20:
See: PR #2789 "OPWEN_SETTINGS" where progress is happening quickly.

holta, 2021-05-20:
See PR #2790 "OPWEN_SETTINGS" where progress is continuing thanks to @jvonau & @adamsclafani.

holta, 2021-05-25:
Please see PR #2800 "Lok super" which evolved from PR #2792 "Lok user"

Progress is continuing thanks to @jvonau & @adamsclafani.

A call will be held at 12 Noon PDT Tuesday (tomorrow) to review what is changing, including the upcoming [Lokole](https://github.com/ascoderu/lokole) 0.6.16 expected soon.

nzola, 2021-05-25:
Sure! Looking forward to...

holta, 2021-05-25:
Please see PR #2802 "Lok merge" which is a trimmed down version of PR #2800 "Lok super"

nzola, 2021-05-25:
Shall I do ./runrole --reinstall lokole

holta, 2021-05-25:
> Shall I do ./runrole --reinstall lokole

A more complete test would be a clean/fresh OS install including PR #2802:

```
curl d.iiab.io/install.txt | sudo bash -s 2802
```

But consult with @jvonau first if possible.

nzola, 2021-05-25:
ok

jvonau, 2021-05-25:
@nzola use `curl d.iiab.io/install.txt | sudo bash -s 2800` on your new installs going forward, remember to create local_vars.yml set with
```
lokole_sim_type: 
lokole_client_id:
```

nzola, 2021-05-26:
Thanks Adam!

holta, 2021-05-26:
> Thanks Adam!

FYI that was Jerry not me (:

nzola, 2021-05-26:
Thanks much Jerry!

jvonau, 2021-05-27:
@nzola Once https://github.com/ascoderu/lokole/pull/548 is pushed, the install instructions need a little altering:
Use just `curl d.iiab.io/install.txt | sudo bash`
Only use `lokole_sim_type: ` DO NOT USE `lokole_client_id:`

nzola, 2021-05-27:
> @nzola <https://github.com/nzola> Once ascoderu/lokole#548

@Jerry Vonau ***@***.***> Ok got it. Is it pushed yet? Thanks!

jvonau, 2021-06-01:
@nzola `/library/lokole/venv/bin/manage.py createadmin --name=admin --password=changeme`

jvonau, 2021-07-21:
close

closed 2021-07-21 by holta