iiab/iiab #2777: /var/mail/iiab-admin errors or warnings growing fast on Debian 10

holta, 2021-05-15:
@deldesir can you look into this hygiene issue in the coming week if you have time?  A copy of `/var/mail/iiab-admin` is here:

- http://sprunge.us/97EVYK?en

From your installation of PR #2776 onto 10.8.0.46, whose [iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md) output is here:

- http://sprunge.us/ZCHBzB?en

(I don't know which OS's it affects, but maybe others can tell?)

Ref: #2767

holta, 2021-05-15:
Excerpt, showing AWStats and Munin issues:

```
  55 From www-data@deb10-srv Fri May 14 22:00:01 2021
  56 Return-path: <www-data@deb10-srv>
  57 Envelope-to: root@deb10-srv
  58 Delivery-date: Fri, 14 May 2021 22:00:01 -0400
  59 Received: from www-data by box.lan with local (Exim 4.92)
  60 	(envelope-from <www-data@deb10-srv>)
  61 	id 1lhjaz-0000T2-Ky
  62 	for root@deb10-srv; Fri, 14 May 2021 22:00:01 -0400
  63 From: root@deb10-srv (Cron Daemon)
  64 To: root@deb10-srv
  65 Subject: Cron <www-data@box> [ -x /usr/share/awstats/tools/update.sh ] && /usr/share/awstats/tools/update.sh
  66 MIME-Version: 1.0
  67 Content-Type: text/plain; charset=UTF-8
  68 Content-Transfer-Encoding: 8bit
  69 X-Cron-Env: <MAILTO=root>
  70 X-Cron-Env: <SHELL=/bin/sh>
  71 X-Cron-Env: <HOME=/var/www>
  72 X-Cron-Env: <PATH=/usr/bin:/bin>
  73 X-Cron-Env: <LOGNAME=www-data>
  74 Message-Id: <E1lhjaz-0000T2-Ky@box.lan>
  75 Date: Fri, 14 May 2021 22:00:01 -0400
  76 
  77 Error while processing /etc/awstats/awstats.conf
  78 Error: SiteDomain parameter not defined in your config/domain file. You must edit it for using this version of AWStats.
  79 Setup ('/etc/awstats/awstats.conf' file, web server or permissions) may be wrong.
  80 Check config file, permissions and AWStats documentation (in 'docs' directory).
  81 
  82 From munin@deb10-srv Fri May 14 22:05:21 2021
  83 Return-path: <munin@deb10-srv>
  84 Envelope-to: root@deb10-srv
  85 Delivery-date: Fri, 14 May 2021 22:05:21 -0400
  86 Received: from munin by box.lan with local (Exim 4.92)
  87 	(envelope-from <munin@deb10-srv>)
  88 	id 1lhjg9-0003Mg-BG
  89 	for root@deb10-srv; Fri, 14 May 2021 22:05:21 -0400
  90 From: root@deb10-srv (Cron Daemon)
  91 To: root@deb10-srv
  92 Subject: Cron <munin@box> if [ -x /usr/bin/munin-cron ]; then /usr/bin/munin-cron; fi
  93 MIME-Version: 1.0
  94 Content-Type: text/plain; charset=UTF-8
  95 Content-Transfer-Encoding: 8bit
  96 X-Cron-Env: <MAILTO=root>
  97 X-Cron-Env: <SHELL=/bin/sh>
  98 X-Cron-Env: <HOME=/var/lib/munin>
  99 X-Cron-Env: <PATH=/usr/bin:/bin>
 100 X-Cron-Env: <LOGNAME=munin>
 101 Message-Id: <E1lhjg9-0003Mg-BG@box.lan>
 102 Date: Fri, 14 May 2021 22:05:21 -0400
 103 
 104 2021/05/14 22:05:02 [Warning] Could not open includedir directory /etc/munin/conf.d: No such file or directory
 105 readdir() attempted on invalid dirhandle $DIR at /usr/share/munin/munin-update line 57.
 106 closedir() attempted on invalid dirhandle $DIR at /usr/share/munin/munin-update line 58.
 107 2021/05/14 22:05:12 [Warning] Could not open includedir directory /etc/munin/conf.d: No such file or directory
 108 readdir() attempted on invalid dirhandle $DIR at /usr/share/perl5/Munin/Master/Utils.pm line 985.
 109 closedir() attempted on invalid dirhandle $DIR at /usr/share/perl5/Munin/Master/Utils.pm line 986.
 110 2021/05/14 22:05:12 [Warning] Could not open includedir directory /etc/munin/conf.d: No such file or directory
 111 readdir() attempted on invalid dirhandle $DIR at /usr/share/perl5/Munin/Master/Utils.pm line 985.
 112 closedir() attempted on invalid dirhandle $DIR at /usr/share/perl5/Munin/Master/Utils.pm line 986.
 113 2021/05/14 22:05:16 [Warning] Could not open includedir directory /etc/munin/conf.d: No such file or directory
 114 readdir() attempted on invalid dirhandle $DIR at /usr/share/perl5/Munin/Master/Utils.pm line 985.
 115 closedir() attempted on invalid dirhandle $DIR at /usr/share/perl5/Munin/Master/Utils.pm line 986.
```

jvonau, 2021-05-16:
Turn off the install of these two until you have the issue figured out... obviously something has been amiss for awhile:
>From iiab-admin@deb10-srv Wed Sep 02 13:50:31 2020

holta, 2021-05-16:
Yes, turning off AWStats and Munin is definitely an option.

My understanding is that @deldesir installed BIG-sized by accident, when he meant to install MIN-sized, while testing things like PR #2776.

Possibly a very useful mistake however &mdash; so that we begin to better understand this and similar "log bloat" issues (if this happens on all OS's, how quickly `/var/mail/iiab-admin` and similar log-like files are rapidly growing, etc).

tim-moody, 2021-05-16:
I don't see this on Ubuntu 20.04.2

/etc/awstats/awstats.conf contains SiteDomain="box.lan"

running /usr/share/awstats/tools/update.sh throws no errors and there is nothing in /var/mail/

http://192.168.3.169/awstats returns results, but it is suspicious that they are all zero, though this is just a test vm with little web activity.

Actually as I think about it, do we still have this hooked to the apache logs rather than the nginx logs? Is the trigger removing apache?

If this is broken it must be recent, because I regularly use awstats on Medical installs.

tim-moody, 2021-05-16:
http://iiab.me/awstats has stats for the current month

jvonau, 2021-05-16:
Awstats notes:
Reused old install 
>-rw-r--r-- 1 root root 317 Sep  2  2020 /etc/network/interfaces 

What you are seeing is https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=783011 in action when the deb package is first installed.

Monit notes:
https://github.com/iiab/iiab/blob/master/roles/monit/templates/monitrc#L238-L246
See the bottom of [stock-monitrc.txt](https://github.com/iiab/iiab/files/6489604/stock-monitrc.txt) 
```
###############################################################################
## Includes
###############################################################################
##
## It is possible to include additional configuration parts from other files or
## directories.
#
   include /etc/monit/conf.d/*
   include /etc/monit/conf-enabled/*
#
```
Be better to have https://github.com/iiab/iiab/blob/master/roles/monit/templates/monitrc#L116-L121) (quick scan there maybe others) as a drop-in file into /etc/monit/conf.d/ and not touch the stock monitrc file at all, in place of what is in use today.

In summary, one chicken-little report and one very long standing design flaw.

jvonau, 2021-05-16:
https://packages.debian.org/buster/all/awstats/filelist
```
ubuntu@box:/opt/iiab/iiab$ tree /etc/monit/
/etc/monit/
├── conf-available
│   ├── acpid
│   ├── apache2
│   ├── at
│   ├── cron
│   ├── mdadm
│   ├── memcached
│   ├── mysql
│   ├── nginx
│   ├── openntpd
│   ├── openssh-server
│   ├── pdns-recursor
│   ├── postfix
│   ├── rsyslog
│   ├── smartmontools
│   └── snmpd
├── conf.d
├── conf-enabled
├── monitrc
└── templates
    ├── rootbin
    ├── rootrc
    └── rootstrict

4 directories, 19 files
```

jvonau, 2021-05-16:
Change the title of this issue to track stale apps from "big"
[includedir /etc/munin/conf.d](https://github.com/iiab/iiab/blob/master/roles/munin/templates/munin.conf.j2#L27)
[munin.conf.txt](https://github.com/iiab/iiab/files/6489866/munin.conf.txt)
>includedir /etc/munin/munin-conf.d

Same issue as monit above.
```
/opt/iiab/iiab$ tree /etc/munin/
/etc/munin/
├── apache24.conf
├── munin.conf
├── munin-conf.d
├── munin-node.conf
├── plugin-conf.d [error opening dir]
├── plugins
│   ├── acpi -> /usr/share/munin/plugins/acpi
│   ├── cpu -> /usr/share/munin/plugins/cpu
│   ├── cpuspeed -> /usr/share/munin/plugins/cpuspeed
│   ├── df -> /usr/share/munin/plugins/df
│   ├── df_inode -> /usr/share/munin/plugins/df_inode
│   ├── diskstats -> /usr/share/munin/plugins/diskstats
│   ├── entropy -> /usr/share/munin/plugins/entropy
│   ├── forks -> /usr/share/munin/plugins/forks
│   ├── fw_packets -> /usr/share/munin/plugins/fw_packets
│   ├── http_loadtime -> /usr/share/munin/plugins/http_loadtime
│   ├── if_ap0 -> /usr/share/munin/plugins/if_
│   ├── if_br0 -> /usr/share/munin/plugins/if_
│   ├── if_docker0 -> /usr/share/munin/plugins/if_
│   ├── if_err_ap0 -> /usr/share/munin/plugins/if_err_
│   ├── if_err_br0 -> /usr/share/munin/plugins/if_err_
│   ├── if_err_docker0 -> /usr/share/munin/plugins/if_err_
│   ├── if_err_eth0 -> /usr/share/munin/plugins/if_err_
│   ├── if_err_tun0 -> /usr/share/munin/plugins/if_err_
│   ├── if_err_wlan0 -> /usr/share/munin/plugins/if_err_
│   ├── if_eth0 -> /usr/share/munin/plugins/if_
│   ├── if_tun0 -> /usr/share/munin/plugins/if_
│   ├── if_wlan0 -> /usr/share/munin/plugins/if_
│   ├── interrupts -> /usr/share/munin/plugins/interrupts
│   ├── irqstats -> /usr/share/munin/plugins/irqstats
│   ├── load -> /usr/share/munin/plugins/load
│   ├── memory -> /usr/share/munin/plugins/memory
│   ├── munin_stats -> /usr/share/munin/plugins/munin_stats
│   ├── netstat -> /usr/share/munin/plugins/netstat
│   ├── open_files -> /usr/share/munin/plugins/open_files
│   ├── open_inodes -> /usr/share/munin/plugins/open_inodes
│   ├── processes -> /usr/share/munin/plugins/processes
│   ├── proc_pri -> /usr/share/munin/plugins/proc_pri
│   ├── swap -> /usr/share/munin/plugins/swap
│   ├── threads -> /usr/share/munin/plugins/threads
│   ├── uptime -> /usr/share/munin/plugins/uptime
│   ├── users -> /usr/share/munin/plugins/users
│   └── vmstat -> /usr/share/munin/plugins/vmstat
├── static
│   ├── definitions.html
│   ├── dynazoom.html
│   ├── favicon.ico
│   ├── formatdate.js
│   ├── logo-h.png
│   ├── logo.png
│   ├── querystring.js
│   ├── style-1.2.css
│   ├── style.css
│   ├── style-new.css
│   └── zoom.js
└── templates
    ├── munin-categoryview.tmpl
    ├── munin-comparison-day.tmpl
    ├── munin-comparison-month.tmpl
    ├── munin-comparison-week.tmpl
    ├── munin-comparison-year.tmpl
    ├── munin-domainview.tmpl
    ├── munin-dynazoom.tmpl
    ├── munin-nodeview.tmpl
    ├── munin-overview.tmpl
    ├── munin-problemview.tmpl
    ├── munin-serviceview.tmpl
    └── partial
        ├── bottom_navigation.tmpl
        ├── footer.tmpl
        ├── generated_by.tmpl
        ├── head.tmpl
        ├── logo_navigation_comparison.tmpl
        ├── logo_navigation_problem.tmpl
        ├── logo_navigation.tmpl
        ├── logo_path.tmpl
        ├── navigation.tmpl
        └── path.tmpl

6 directories, 72 files
```

holta, 2021-05-21:
Should this ticket be closed?

Or maybe better keep it open, until AWStats is cleaned up?

(Which might be hard to pull off, given AWStats currently requires [some parts of] Apache!)

Ref: #2762 "Notes: Stalled progress on nginx front"

jvonau, 2021-05-21:
> Should this ticket be closed?
> 
> Or maybe better keep it open, until AWStats is cleaned up?
> 
> (Which might be hard to pull off, given AWStats currently requires [some parts of] Apache!)
> 
There is no requirement to install apache in install.yml

> Ref: #2762 "Notes: Stalled progress on nginx front"

No such note

> http://iiab.me/awstats has stats for the current month
> 
> and nothing in /var/mail

Close it, AWStats is running under nginx well

holta, 2021-05-21:
> There is no requirement to install apache in install.yml

Only indirectly:

https://github.com/iiab/iiab/blob/master/roles/awstats/tasks/install.yml#L12 installs pwauth which installs these 2 apt packages:

- apache2-bin
- libapache2-mod-authnz-external

This might be unavoidable with AWStats possibly??

Anyway relevant to the question in #2762 — if a clean transition away from the Apache legacy is desired.

jvonau, 2021-05-21:
> > There is no requirement to install apache in install.yml
> 
> Only indirectly:
> 
> https://github.com/iiab/iiab/blob/master/roles/awstats/tasks/install.yml#L12 installs `pwauth` which installs these 2 apt packages:

Do I really have to tell you to move pwauth one stanza down to be grouped with the other two apache dependencies?

holta, 2021-05-21:
> move pwauth one stanza down to be grouped with the other two apache dependencies?

Does apt package `openssl` work with `pwauth` — such that they should possibly stay together for Apache only (e.g. if both might not be needed when AWStats is used with NGINX ?)

cc: @tim-moody

tim-moody, 2021-05-21:
> Does apt package openssl work with pwauth — such that they should possibly stay together for Apache only (e.g. if both might not be needed when AWStats is used with NGINX ?)

My memory is that pwauth passes credentials in the clear in the http request headers, so does not use openssl.

Not sure what uses pwauth; I think nextcloud may based on the former conflict with adm cons, but not sure what it does with nginx.

holta, 2021-05-21:
> My memory is that pwauth passes credentials in the clear in the http request headers, so does not use openssl.

That seems right.

http://box/awstats was tested to work with NGINX without `pwauth` earlier this morning, so PR #2793 will likely be merged very shortly, if there are no objections / changes required there.

> Not sure what uses pwauth; I think nextcloud may based on the former conflict with adm cons, but not sure what it does with nginx.

It looks like we won't be needing `pwauth` going forward.

closed 2021-05-21 by holta