iiab/iiab #2783: Lokole failed to install on Linux Mint 20

holta, 2021-05-17:
Thanks @deldesir for having done this test earlier this morning:

> 1328 2021-05-17 09:24:11,908 p=6199 u=root n=ansible | TASK [lokole : Install 8 packages for Lokole: python3, python3-pip, python3-venv, python3-dev, python3-bcrypt, libffi-dev, libssl-dev, libopenjp2-7] ***
1329 2021-05-17 09:24:14,799 p=6199 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"cache_update_time": 1621257752, "cache_updated": false, "changed": false, "msg": "'/usr/bin/apt-get -y -o \"Dpkg::Options::=--force-confdef\" -o \"Dpkg::Options::=--force-confold\"      install 'python3-dev' 'libffi-dev' 'libjpeg-dev' 'libssl-dev'' failed: E: Unable to correct problems, you have held broken packages.\n", "rc": 100, "stderr": "E: Unable to correct problems, you have held broken packages.\n", "stderr_lines": ["E: Unable to correct problems, you have held broken packages."], "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nSome packages could not be installed. This may mean that you have\nrequested an impossible situation or if you are using the unstable\ndistribution that some required packages have not yet been created\nor been moved out of Incoming.\nThe following information may help to resolve the situation:\n\nThe following packages have unmet dependencies:\n libjpeg-dev : Depends: libjpeg8-dev but it is not going to be installed\n python3-dev : Depends: libpython3-dev (= 3.8.2-0ubuntu2) but it is not going to be installed\n               Depends: python3.8-dev (>= 3.8.2-1~) but it is not going to be installed\n", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information...", "Some packages could not be installed. This may mean that you have", "requested an impossible situation or if you are using the unstable", "distribution that some required packages have not yet been created", "or been moved out of Incoming.", "The following information may help to resolve the situation:", "", "The following packages have unmet dependencies:", " libjpeg-dev : Depends: libjpeg8-dev but it is not going to be installed", " python3-dev : Depends: libpython3-dev (= 3.8.2-0ubuntu2) but it is not going to be installed", "               Depends: python3.8-dev (>= 3.8.2-1~) but it is not going to be installed"]}
1330 2021-05-17 09:24:14,806 p=6199 u=root n=ansible | PLAY RECAP *********************************************************************
1331 2021-05-17 09:24:14,807 p=6199 u=root n=ansible | 127.0.0.1                  : ok=359  changed=203  unreachable=0    failed=1    skipped=74   rescued=0    ignored=3

iiab-diagnostics output: http://sprunge.us/m4R3Ke?en

Not sure if this helps to figure this out, but FYI:

```
root@box:~# apt list python3-dev
Listing... Done
python3-dev/focal 3.8.2-0ubuntu2 amd64
python3-dev/focal 3.8.2-0ubuntu2 i386
```

holta, 2021-05-17:
@deldesir can you check if `apt install python3-dev` repeatedly fails, when run at the command-line?

Or is it a different/underlying issue somehow related to `python_ver: 3.8` as set at https://github.com/iiab/iiab/blob/master/vars/linuxmint-20.yml#L31 possibly ?

deldesir, 2021-05-17:
It won't install. 

The following packages have unmet dependencies:
 python3-dev : Depends: libpython3-dev (= 3.8.2-0ubuntu2) but it is not going to be installed
               Depends: python3.8-dev (>= 3.8.2-1~) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

jvonau, 2021-05-17:
> @deldesir can you check if `apt install python3-dev` repeatedly fails, when run at the command-line?
>

Think this might related to e87afe3f5fb2a72aa07287f8ee02411eb2b19218 try using the full path /usr/bin/apt. 
>python3-dev/groovy,now 3.8.6-0ubuntu1 arm64 [installed]
python3.8-dev/groovy-updates,groovy-security,now 3.8.6-1ubuntu0.2 arm64 [installed,automatic]
python3.9-dev/groovy 3.9.0-5 arm64
 
While at the command prompt what does `/usr/bin/apt list python*-dev` return? Can you run `sudo apt update` then `sudo apt upgrade` successfully? 

> Or is it a different/underlying issue somehow related to `python_ver: 3.8` as set at https://github.com/iiab/iiab/blob/master/vars/linuxmint-20.yml#L31 possibly ?

No, the variable python_ver is currently only used in the nginx template, but if needed can be used in the same way as php_version is used, this is looking like the same kind of issue with versioning when there is a choice between a few deb packages is present.

deldesir, 2021-05-17:
Same problem with /usr/bin/apt.

deldesir, 2021-05-17:
/usr/bin/apt list python*-dev                       Listing... Done                                                       python-all-dev/focal 2.7.17-2ubuntu4 amd64                            python-all-dev/focal 2.7.17-2ubuntu4 i386                             python-apt-dev/focal-updates,focal-updates 2.0.0ubuntu0.20.04.4 all   python-cairo-dev/focal,focal 1.16.2-2ubuntu2 all                      python-dbus-dev/focal,focal 1.2.16-1build1 all                        python-gi-dev/focal 3.36.0-1 amd64                                    python-gi-dev/focal 3.36.0-1 i386                                     python-gobject-2-dev/focal,focal 2.28.6-14ubuntu1 all                 python-greenlet-dev/focal 0.4.15-4.1 amd64                            python-sip-dev/focal 4.19.21+dfsg-1build1 amd64
python-sip-dev/focal 4.19.21+dfsg-1build1 i386                        python-wxgtk3.0-dev/focal,focal 3.0.2.0+dfsg-8build1 all              python2-dev/focal 2.7.17-2ubuntu4 amd64                               python2-dev/focal 2.7.17-2ubuntu4 i386                                python2.7-dev/focal-updates,focal-security 2.7.18-1~20.04.1 amd64     python2.7-dev/focal-updates,focal-security 2.7.18-1~20.04.1 i386      python3-all-dev/focal 3.8.2-0ubuntu2 amd64                            python3-all-dev/focal 3.8.2-0ubuntu2 i386                             python3-cairo-dev/focal,focal 1.16.2-2ubuntu2 all                     python3-cxx-dev/focal,focal 7.1.3-0.1 all                             python3-dev/focal 3.8.2-0ubuntu2 amd64                                python3-dev/focal 3.8.2-0ubuntu2 i386                                 python3-ldb-dev/focal-updates,focal-security 2:2.0.10-0ubuntu0.20.04.3 amd64                                                                python3-ldb-dev/focal-updates,focal-security 2:2.0.10-0ubuntu0.20.04.3 i386                                                                 python3-sip-dev/focal 4.19.21+dfsg-1build1 amd64                      python3-sip-dev/focal 4.19.21+dfsg-1build1 i386                       python3-talloc-dev/focal 2.3.0-3ubuntu1 amd64                         python3-talloc-dev/focal 2.3.0-3ubuntu1 i386                          python3.8-dev/focal-updates,focal-security 3.8.5-1~20.04.2 amd64      python3.8-dev/focal-updates,focal-security 3.8.5-1~20.04.2 i386       python3.9-dev/focal-updates 3.9.0-5~20.04 amd64                       python3.9-dev/focal-updates 3.9.0-5~20.04 i386

holta, 2021-05-17:
> might [be] related to [e87afe3](https://github.com/iiab/iiab/commit/e87afe3f5fb2a72aa07287f8ee02411eb2b19218) try using the full path /usr/bin/apt.

Please try @jvonau's suggestion if you haven't yet:

```/usr/bin/apt install python3-dev```

As compared to Linux Mint's own version of apt, that's very often non-compatible and problematic:

```/usr/local/bin/apt install python3-dev```

As explained here: https://github.com/iiab/iiab-factory/blob/master/iiab#L50

deldesir, 2021-05-17:
Already did it.
iiab-admin@box:~$ sudo !!
sudo /usr/bin/apt install python3-dev
[sudo] password for iiab-admin:
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.                                        The following information may help to resolve the situation:
                                                                      The following packages have unmet dependencies:
 python3-dev : Depends: libpython3-dev (= 3.8.2-0ubuntu2) but it is not going to be installed
               Depends: python3.8-dev (>= 3.8.2-1~) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

jvonau, 2021-05-17:
```
Listing... Done 
python-all-dev/focal 2.7.17-2ubuntu4 amd64 
python-all-dev/focal 2.7.17-2ubuntu4 i386 
python-apt-dev/focal-updates,focal-updates 2.0.0ubuntu0.20.04.4 all 
python-cairo-dev/focal,focal 1.16.2-2ubuntu2 all 
python-dbus-dev/focal,focal 1.2.16-1build1 all 
python-gi-dev/focal 3.36.0-1 amd64 
python-gi-dev/focal 3.36.0-1 i386 
python-gobject-2-dev/focal,focal 2.28.6-14ubuntu1 all 
python-greenlet-dev/focal 0.4.15-4.1 amd64 
python-sip-dev/focal 4.19.21+dfsg-1build1 amd64 
python-sip-dev/focal 4.19.21+dfsg-1build1 i386 
python-wxgtk3.0-dev/focal,focal 3.0.2.0+dfsg-8build1 all 
python2-dev/focal 2.7.17-2ubuntu4 amd64 
python2-dev/focal 2.7.17-2ubuntu4 i386 
python2.7-dev/focal-updates,focal-security 2.7.18-1~20.04.1 amd64 
python2.7-dev/focal-updates,focal-security 2.7.18-1~20.04.1 i386 
python3-all-dev/focal 3.8.2-0ubuntu2 amd64 
python3-all-dev/focal 3.8.2-0ubuntu2 i386 
python3-cairo-dev/focal,focal 1.16.2-2ubuntu2 all 
python3-cxx-dev/focal,focal 7.1.3-0.1 all 
python3-dev/focal 3.8.2-0ubuntu2 amd64 
python3-dev/focal 3.8.2-0ubuntu2 i386 
python3-ldb-dev/focal-updates,focal-security 2:2.0.10-0ubuntu0.20.04.3 amd64 
python3-ldb-dev/focal-updates,focal-security 2:2.0.10-0ubuntu0.20.04.3 i386 
python3-sip-dev/focal 4.19.21+dfsg-1build1 amd64 
python3-sip-dev/focal 4.19.21+dfsg-1build1 i386 
python3-talloc-dev/focal 2.3.0-3ubuntu1 amd64 
python3-talloc-dev/focal 2.3.0-3ubuntu1 i386 
python3.8-dev/focal-updates,focal-security 3.8.5-1~20.04.2 amd64 
python3.8-dev/focal-updates,focal-security 3.8.5-1~20.04.2 i386 
python3.9-dev/focal-updates 3.9.0-5~20.04 amd64 
python3.9-dev/focal-updates 3.9.0-5~20.04 i386
```

holta, 2021-05-17:
FYI both version of `apt` claim the package is available to be installed: (falsely claim!?)

```
root@box:~# /usr/bin/apt list python3-dev
Listing... Done
python3-dev/focal 3.8.2-0ubuntu2 amd64
python3-dev/focal 3.8.2-0ubuntu2 i386

root@box:~# /usr/local/bin/apt list python3-dev
Listing... Done
python3-dev/focal 3.8.2-0ubuntu2 amd64
python3-dev/focal 3.8.2-0ubuntu2 i386
```

jvonau, 2021-05-17:
Interesting try `sudo /usr/bin/apt install python3.8-dev`

jvonau, 2021-05-17:
> FYI both versions of `apt` claim the package is available to be installed: (falsely claim!?)
> 
> ```
> root@box:~# /usr/bin/apt list python3-dev
> Listing... Done
> python3-dev/focal 3.8.2-0ubuntu2 amd64
> python3-dev/focal 3.8.2-0ubuntu2 i386
> 
> root@box:~# /usr/local/bin/apt list python3-dev
> Listing... Done
> python3-dev/focal 3.8.2-0ubuntu2 amd64
> python3-dev/focal 3.8.2-0ubuntu2 i386
> ```

missing [installed] see my example

jvonau, 2021-05-17:
Think this is Mint's issue to sort out file a bug upstream.

deldesir, 2021-05-17:
Yes and I also upgraded.
The following information may help to resolve the situation:
  ```                                                                   The following packages have unmet dependencies:                        python3.8-dev : Depends: libpython3.8-dev (= 3.8.5-1~20.04.2) but it is not going to be installed                                                           Depends: libexpat1-dev but it is not going to be installed                                                                                  Depends: zlib1g-dev but it is not going to be installed                                                                                     Recommends: libc6-dev but it is not going to be installed or                                                                                            libc-dev                                 E: Unable to correct problems, you have held broken packages.```

jvonau, 2021-05-17:
https://github.com/linuxmint

jvonau, 2021-05-17:
https://github.com/linuxmint/mintupdate might look like the place

jvonau, 2021-05-17:
https://github.com/linuxmint/mintsystem supplied the local version of apt, seems like the deb packages are pinned to a version somewhere.

jvonau, 2021-05-17:
lets try `sudo /usr/local/bin/apt held`

holta, 2021-05-17:
Someone involved with Linux Mint is now helping to investigate upstream: https://github.com/linuxmint/mintupdate/issues/680

jvonau, 2021-05-17:
`sudo /usr/bin/apt-mark showhold`

holta, 2021-05-17:
`apt install python3-dev` worked after a Linux Mint person helped downgrade `libc6` from 2.31-0ubuntu**9.3** to 2.31-0ubuntu**9.2** : https://github.com/linuxmint/mintupdate/issues/680#issuecomment-842609705

Presumably this means `/usr/local/bin/apt held` and `/usr/bin/apt-mark showhold` are no longer nec?  And might have accomplished the same thing thru a different route possibly?

In any case: @deldesir please continue testing on this VM (10.8.0.34 = 195-mint20-PR2775-0517) as necessary, and thanks all for getting to the bottom of this!

jvonau, 2021-05-18:
Keep an eye out for this on ubuntu-20.04s, that is the parent of Mint. Think the owner of the VM might have another copy around with the bad snapshot saved containing the now [retracted](https://bugs.launchpad.net/ubuntu/+source/glibc/+bug/1926918)(wth?) version of libc6-2.31-0ubuntu9.3?

holta, 2021-05-18:
Excellent question.

IIAB should not get in the business of solving every broken/legacy OS configuration.

At the same time if lots of folks really do have flaky Ubuntu 20.04 installs out there (as should become obvious in coming months?) we might flag this malignancy during IIAB's install process, intentionally stop the install, and point people very directly to something like https://github.com/linuxmint/mintupdate/issues/680 asking them to fix (or replace) their own "rather broken" OS.

holta, 2021-05-18:
FYI I'm not seeing this problem on Linux Minx 20 VM's like this one: (despite their very close provenance/lineage: the problematic VM was cloned from this one a month ago!)

```
root@mint20:~# apt list libc6
Listing... Done
libc6/focal-updates,now 2.31-0ubuntu9.2 amd64 [installed]
libc6/focal-updates 2.31-0ubuntu9.2 i386
```

Also the problem does not appear on Ubuntu 20.04 VM's that I've checked, e.g.:

```
root@u20-srv:~# apt -a list libc6
Listing... Done
libc6/focal-updates,now 2.31-0ubuntu9.2 amd64 [installed,automatic]
libc6/focal 2.31-0ubuntu9 amd64
```

```
root@u20-desk:~# apt list libc6
Listing... Done
libc6/focal-updates,now 2.31-0ubuntu9.2 amd64 [installed,automatic]
libc6/focal-updates 2.31-0ubuntu9.2 i386
```

So we can probably declare this be a rare/passing anomaly, that we'll quite likely not see again (or at least very rarely!)

holta, 2021-05-18:
@deldesir I'll leave this ticket open just in case you want to do another BIG-sized install on Linux Mint 20 in coming weeks, to further validate.

(Then we should close this, in the coming month if possible!)

deldesir, 2021-05-18:
Ok, noted. It should not take weeks however, I would rather say hours since test install is on its way :)

holta, 2021-05-20:
Resolved in conjunction with #2786.

closed 2021-05-20 by holta