iiab/iiab #2786: Corejs dependencies issue when running yarn role for Internet Archive

deldesir, 2021-05-17:
### Expected behavior and actual behavior:
Role for Yarn package manager stuck complaining about corejs version and potential security issues due to deprecation and dependency issues on Linux Mint 20.

http://sprunge.us/UPUUEI?en

holta, 2021-05-17:
Here's a bit more background on the VM being used in this testing of PRs #2775 and PR #2783 

- #2783 "Lokole failed to install on Linux Mint 20: apt package 'python3-dev' repeatedly fails to install"

holta, 2021-05-17:
Granted this VM was modified under the guidance of "official" Linux Mint people (https://github.com/linuxmint/mintupdate/issues/680 earlier today) but @jvonau asks the fair question: did that downgrade of libc6 (no matter how official) possibly contribute to this Internet Archive problem?

holta, 2021-05-17:
> http://sprunge.us/UPUUEI?en

Line 1501 from the above is regrettably long, filling almost an entire page, but here it is to highlight that error message:

> 1501 2021-05-17 17:40:31,476 p=17962 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "yarn config set child-concurrency 1 && yarn add @internetarchive/dweb-mirror", "delta": "0:01:10.760722", "end": "2021-05-17 17:40:31.418229", "msg": "non-zero return code", "rc": 1, "start": "2021-05-17 17:39:20.657507", "stderr": "warning @internetarchive/dweb-mirror > @internetarchive/dweb-archive-dist > react > fbjs > core-js@1.2.7: core-js@<3.3 is no longer maintained and not recommended for usage due to the number of issues. Because of the V8 engine whims, feature detection in old core-js versions could cause a slowdown up to 100x even if nothing is polyfilled. Please, upgrade your dependencies to the actual version of core-js.\nwarning @internetarchive/dweb-mirror > @futurepress/epubjs-reader > epubjs > @types/localforage@0.0.34: This is a stub types definition for localforage (https://github.com/localForage/localForage). localforage provides its own type definitions, so you don't need @types/localforage installed!\nwarning @internetarchive/dweb-mirror > @futurepress/epubjs-reader > epubjs > xmldom@0.3.0: Deprecated due to CVE-2021-21366 resolved in 0.5.0\nerror /opt/iiab/internetarchive/node_modules/sharp: Command failed.\nExit code: 1\nCommand: (node install/libvips && node install/dll-copy && prebuild-install) || (node-gyp rebuild && node install/dll-copy)\nArguments: \nDirectory: /opt/iiab/internetarchive/node_modules/sharp\nOutput:\ninfo sharp Downloading https://github.com/lovell/sharp-libvips/releases/download/v8.8.1/libvips-8.8.1-linux-x64.tar.gz\nprebuild-install WARN install No prebuilt binaries found (target=14.17.0 runtime=node arch=x64 libc= platform=linux)\ngyp info it worked if it ends with ok\ngyp info using node-gyp@5.1.0\ngyp info using node@14.17.0 | linux | x64\ngyp info find Python using Python version 2.7.18 found at \"/usr/bin/python2\"\ngyp info spawn /usr/bin/python2\ngyp info spawn args [\ngyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',\ngyp info spawn args   'binding.gyp',\ngyp info spawn args   '-f',\ngyp info spawn args   'make',\ngyp info spawn args   '-I',\ngyp info spawn args   '/opt/iiab/internetarchive/node_modules/sharp/build/config.gypi',\ngyp info spawn args   '-I',\ngyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',\ngyp info spawn args   '-I',\ngyp info spawn args   '/root/.cache/node-gyp/14.17.0/include/node/common.gypi',\ngyp info spawn args   '-Dlibrary=shared_library',\ngyp info spawn args   '-Dvisibility=default',\ngyp info spawn args   '-Dnode_root_dir=/root/.cache/node-gyp/14.17.0',\ngyp info spawn args   '-Dnode_gyp_dir=/usr/lib/node_modules/npm/node_modules/node-gyp',\ngyp info spawn args   '-Dnode_lib_file=/root/.cache/node-gyp/14.17.0/<(target_arch)/node.lib',\ngyp info spawn args   '-Dmodule_root_dir=/opt/iiab/internetarchive/node_modules/sharp',\ngyp info spawn args   '-Dnode_engine=v8',\ngyp info spawn args   '--depth=.',\ngyp info spawn args   '--no-parallel',\ngyp info spawn args   '--generator-output',\ngyp info spawn args   'build',\ngyp info spawn args   '-Goutput_dir=.'\ngyp info spawn args ]\ngyp info spawn make\ngyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]\nmake: Entering directory '/opt/iiab/internetarchive/node_modules/sharp/build'\n  TOUCH Release/obj.target/libvips-cpp.stamp\n  CXX(target) Release/obj.target/sharp/src/common.o\nmake: g++: Command not found\nmake: *** [sharp.target.mk:137: Release/obj.target/sharp/src/common.o] Error 127\nmake: Leaving directory '/opt/iiab/internetarchive/node_modules/sharp/build'\ngyp ERR! build error \ngyp ERR! stack Error: `make` failed with exit code: 2\ngyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)\ngyp ERR! stack     at ChildProcess.emit (events.js:376:20)\ngyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)\ngyp ERR! System Linux 5.4.0-73-generic\ngyp ERR! command \"/usr/bin/node\" \"/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js\" \"rebuild\"\ngyp ERR! cwd /opt/iiab/internetarchive/node_modules/sharp\ngyp ERR! node -v v14.17.0\ngyp ERR! node-gyp -v v5.1.0\ngyp ERR! not ok", "stderr_lines": ["warning @internetarchive/dweb-mirror > @internetarchive/dweb-archive-dist > react > fbjs > core-js@1.2.7: core-js@<3.3 is no longer maintained and not recommended for usage due to the number of issues. Because of the V8 engine whims, feature detection in old core-js versions could cause a slowdown up to 100x even if nothing is polyfilled. Please, upgrade your dependencies to the actual version of core-js.", "warning @internetarchive/dweb-mirror > @futurepress/epubjs-reader > epubjs > @types/localforage@0.0.34: This is a stub types definition for localforage (https://github.com/localForage/localForage). localforage provides its own type definitions, so you don't need @types/localforage installed!", "warning @internetarchive/dweb-mirror > @futurepress/epubjs-reader > epubjs > xmldom@0.3.0: Deprecated due to CVE-2021-21366 resolved in 0.5.0", "error /opt/iiab/internetarchive/node_modules/sharp: Command failed.", "Exit code: 1", "Command: (node install/libvips && node install/dll-copy && prebuild-install) || (node-gyp rebuild && [... cut ...] ..", "[2/4] Fetching packages...", "[3/4] Linking dependencies...", "[4/4] Building fresh packages...", "info Visit https://yarnpkg.com/en/docs/cli/add for documentation about this command."]}
1502 2021-05-17 17:40:31,479 p=17962 u=root n=ansible | PLAY RECAP *********************************************************************
1503 2021-05-17 17:40:31,480 p=17962 u=root n=ansible | 127.0.0.1                  : ok=805  changed=341  unreachable=0    failed=1    skipped=167  rescued=0    ignored=0

jvonau, 2021-05-18:
Hint, every /n is a new line

1501 2021-05-17 17:40:31,476 p=17962 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "yarn config set child-concurrency 1 && yarn add @internetarchive/dweb-mirror", "delta": "0:01:10.760722", "end": "2021-05-17 17:40:31.418229", "msg": "non-zero return code", "rc": 1, "start": "2021-05-17 17:39:20.657507", "stderr": "warning @internetarchive/dweb-mirror > @internetarchive/dweb-archive-dist > react > fbjs > core-js@1.2.7: core-js@<3.3 is no longer maintained and not recommended for usage due to the number of issues. Because of the V8 engine whims, feature detection in old core-js versions could cause a slowdown up to 100x even if nothing is polyfilled. Please, upgrade your dependencies to the actual version of core-js.\n
warning @internetarchive/dweb-mirror > @futurepress/epubjs-reader > epubjs > @types/localforage@0.0.34: This is a stub types definition for localforage (https://github.com/localForage/localForage). localforage provides its own type definitions, so you don't need @types/localforage installed!\nwarning @internetarchive/dweb-mirror > @futurepress/epubjs-reader > epubjs > xmldom@0.3.0: Deprecated due to CVE-2021-21366 resolved in 0.5.0\n
error /opt/iiab/internetarchive/node_modules/sharp: Command failed.\n
Exit code: 1\n
Command: (node install/libvips && node install/dll-copy && prebuild-install) || (node-gyp rebuild && node install/dll-copy)\n
Arguments: \n
Directory: /opt/iiab/internetarchive/node_modules/sharp\n
Output:\n
info sharp Downloading https://github.com/lovell/sharp-libvips/releases/download/v8.8.1/libvips-8.8.1-linux-x64.tar.gz\n
prebuild-install WARN install No prebuilt binaries found (target=14.17.0 runtime=node arch=x64 libc= platform=linux)\n
gyp info it worked if it ends with ok\n
gyp info using node-gyp@5.1.0\n
gyp info using node@14.17.0 | linux | x64\n
gyp info find Python using Python version 2.7.18 found at "/usr/bin/python2"\n
gyp info spawn /usr/bin/python2\n
gyp info spawn args [\n
gyp info spawn args '/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',\n
gyp info spawn args 'binding.gyp',\n
gyp info spawn args '-f',\n
gyp info spawn args 'make',\n
gyp info spawn args '-I',\n
gyp info spawn args '/opt/iiab/internetarchive/node_modules/sharp/build/config.gypi',\n
gyp info spawn args '-I',\n
gyp info spawn args '/usr/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',\n
gyp info spawn args '-I',\ngyp info spawn args '/root/.cache/node-gyp/14.17.0/include/node/common.gypi',\n
gyp info spawn args '-Dlibrary=shared_library',\ngyp info spawn args '-Dvisibility=default',\n
gyp info spawn args '-Dnode_root_dir=/root/.cache/node-gyp/14.17.0',\n
gyp info spawn args '-Dnode_gyp_dir=/usr/lib/node_modules/npm/node_modules/node-gyp',\n
gyp info spawn args '-Dnode_lib_file=/root/.cache/node-gyp/14.17.0/<(target_arch)/node.lib',\n
gyp info spawn args '-Dmodule_root_dir=/opt/iiab/internetarchive/node_modules/sharp',\n
gyp info spawn args '-Dnode_engine=v8',\n
gyp info spawn args '--depth=.',\n
gyp info spawn args '--no-parallel',\n
gyp info spawn args '--generator-output',\n
gyp info spawn args 'build',\n
gyp info spawn args '-Goutput_dir=.'\n
gyp info spawn args ]\n
gyp info spawn make\n
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]\n
make: Entering directory '/opt/iiab/internetarchive/node_modules/sharp/build'\n
 TOUCH Release/obj.target/libvips-cpp.stamp\n 
 CXX(target) Release/obj.target/sharp/src/common.o\n
make: g++: Command not found\n
make: *** [sharp.target.mk:137: Release/obj.target/sharp/src/common.o] Error 127\n
make: Leaving directory '/opt/iiab/internetarchive/node_modules/sharp/build'\n
gyp ERR! build error \n
gyp ERR! stack Error: make failed with exit code: 2\n
gyp ERR! stack at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)\n
gyp ERR! stack at ChildProcess.emit (events.js:376:20)\n
gyp ERR! stack at Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)\n
gyp ERR! System Linux 5.4.0-73-generic\n
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"\n
gyp ERR! cwd /opt/iiab/internetarchive/node_modules/sharp\n
gyp ERR! node -v v14.17.0\ngyp ERR! node-gyp -v v5.1.0\n
gyp ERR! not ok", "stderr_lines": ["warning @internetarchive/dweb-mirror > @internetarchive/dweb-archive-dist > react > fbjs > core-js@1.2.7: core-js@<3.3 is no longer maintained and not recommended for usage due to the number of issues. Because of the V8 engine whims, feature detection in old core-js versions could cause a slowdown up to 100x even if nothing is polyfilled. Please, upgrade your dependencies to the actual version of core-js.", "warning @internetarchive/dweb-mirror > @futurepress/epubjs-reader > epubjs > @types/localforage@0.0.34: This is a stub types definition for localforage (https://github.com/localForage/localForage). localforage provides its own type definitions, so you don't need @types/localforage installed!", "warning @internetarchive/dweb-mirror > @futurepress/epubjs-reader > epubjs > xmldom@0.3.0: Deprecated due to CVE-2021-21366 resolved in 0.5.0", "error /opt/iiab/internetarchive/node_modules/sharp: Command failed.", "Exit code: 1", "Command: (node install/libvips && node install/dll-copy && prebuild-install) || (node-gyp rebuild && node install/dll-copy)", "Arguments: ", "Directory: /opt/iiab/internetarchive/node_modules/sharp", "Output:", "info sharp Downloading https:/ [... cut ...]
info Visit https://yarnpkg.com/en/docs/cli/add for documentation about this command.", "stdout_lines": ["yarn config v1.22.5", "success Set "child-concurrency" to "1".", "Done in 0.05s.", "yarn add v1.22.5", "info No lockfile found.", "[1/4] Resolving packages...", "[2/4] Fetching packages...", "[3/4] Linking dependencies...", "[4/4] Building fresh packages...", "info Visit https://yarnpkg.com/en/docs/cli/add for documentation about this command."]}

Looks like a failed to build issue, keep an eye out for this popping up on other OSs

holta, 2021-05-18:
> every [ \n ] is a new line

If only there was a way to force GitHub Markdown to display newlines, instead of Ansible's cacophany of `\n` rammed together onto 1 line?

But possibly there's no automated way to do this within GitHub Markdown?

FWIW GitHub Markdown can certainly render code-like snippets with all sorts of [different colored highlighting schemes](https://github.com/github/linguist/blob/master/lib/linguist/languages.yml) as follows:

```javascript
function fancyAlert(arg) {
  if(arg) {
    $.facebox({div:'#foo'})
  }
}
```

e.g. if one types in things like:

````
```javascript
function fancyAlert(arg) {
  if(arg) {
    $.facebox({div:'#foo'})
  }
}
```

holta, 2021-05-20:
I didn't realize @deldesir had installed 195-mint-BIG-0518 = 10.8.0.30 two days ago!

We should quick-verify all links / logins on its IIAB main page: http://10.8.0.30

But so far all looks good, so @deldesir LMK if I can close this ticket?

deldesir, 2021-05-20:
> Granted this VM was modified under the guidance of "official" Linux Mint people (https://github.com/linuxmint/mintupdate/issues/680 earlier today) but @jvonau asks the fair question: did that downgrade of libc6 (no matter how official) possibly contribute to this Internet Archive problem?

Obviously it was altered by the downgrade. Since the install went well, I see no reason not to close now. 👍

closed 2021-05-20 by holta