iiab/iiab #2797: cmdsrv fails to start

Mathazzar, 2021-05-22:
### Expected behavior and actual behavior:
cmdsrv isn't starting on its own, using the command "systemctl start iiab-cmdsrv.service" also fails to start the service. Looking at the logs generated by "journalctl -xe" indicates that the iiab_lib file is missing, which I have been unable to find anywhere on github, unlike the other files in the stacktrace, which are all python scripts.

[raspbianJournald.txt](https://github.com/iiab/iiab/files/6526931/raspbianJournald.txt)

### Steps to reproduce the problem:
I followed the instructions on https://github.com/iiab/iiab/wiki/IIAB-Installation#do-everything-from-scratch from a fresh install of Raspberry OS, and completed up to step 6 before recognizing the problem when attempting to log in as described on https://github.com/iiab/iiab/wiki/IIAB-Installation#configure-the-server.

### IIAB version, OS version, hardware details:
IIAB version: 7.2
OS version: Raspbian 10.9
hadrware: Raspberry Pi Zero W

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)
[raspbianDiag.txt](https://github.com/iiab/iiab/files/6526926/raspbianDiag.txt)

holta, 2021-05-22:
From above `iiab-diagnostics` output:

> TASK [www_base : Unarchive http://download.iiab.io/packages/php7.3-stem.x64.tar to / (x64)] ***
2021-05-21 23:00:12,245 p=1515 u=root n=ansible | skipping: [127.0.0.1]
2021-05-21 23:00:12,740 p=1515 u=root n=ansible | 
TASK [www_base : Symlink /etc/php/7.3/fpm/conf.d/20-stem.ini -> /etc/php/7.3/mods-available/stem.ini] ***
2021-05-21 23:00:19,653 p=1515 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "src file does not exist, use \"force=yes\" if you really want to create the link: /etc/php/7.3/mods-available/stem.ini", "path": "/etc/php/7.3/fpm/conf.d/20-stem.ini", "src": "/etc/php/7.3/mods-available/stem.ini"}
2021-05-21 23:00:19,673 p=1515 u=root n=ansible | PLAY RECAP *********************************************************************
2021-05-21 23:00:19,679 p=1515 u=root n=ansible | 127.0.0.1                  : ok=55   changed=8    unreachable=0    failed=1    skipped=23   rescued=0    ignored=1

Relevant to Known Issue 7 of 8, among the bullet points here:

https://github.com/iiab/iiab/wiki/IIAB-7.2-Release-Notes#known-issues

jvonau, 2021-05-23:
https://github.com/iiab/iiab/pull/144/files and no version for armv6 is available for download

holta, 2021-05-24:
FYI @Mathazzar discussion of workarounds / mitigation is ongoing on PR #2799.

Aside: please note that it is extremely rare to **run** the IIAB installation process on a Raspberry Pi Zero W.  (The recommended norm is to first install IIAB onto a microSD card using a Raspberry Pi 3 or 4 — and to then later move the microSD card into a Raspberry Pi Zero W.)

Mathazzar, 2021-05-28:
PR #2799 adding a case to handle armv6l does solve the problem of cmdsrv failing to start, probably because the missing library is installed after the step where php-stem failed to be downloaded.

closed 2021-05-28 by Mathazzar

holta, 2021-07-15:
Hi @Mathazzar 

Please see PR #2837 (just merged) and if you have moment help us keep #2864 on track for the ~600 Million Spanish speakers worldwide, Thanks!