iiab/iiab #2807: inetutils-syslogd.service: Unit cannot be reloaded because it is inactive

holta, 2021-05-28:
BIG-sized fresh install onto 181-rpi4-lite-BIG-0528 = 10.8.0.58 failed in 2-common/tasks/packages.yml with these errors:

- inetutils-syslogd.service is not active, cannot reload.
- invoke-rc.d: initscript inetutils-syslogd, action \"start\" failed.
- inetutils-syslogd.service: Unit cannot be reloaded because it is inactive.
- dpkg: error processing package inetutils-syslogd (--configure):
- installed inetutils-syslogd package post-installation script subprocess returned error exit status 1

As excerpted from the last 4 lines of /opt/iiab/iiab/iiab-install.log :

> 2021-05-27 23:59:31,328 p=4644 u=root n=ansible | TASK [2-common : Install 6 deb/apt packages: avahi-daemon, exfat-fuse, exfat-utils, inetutils-syslogd, libnss-mdns, wpasupplicant (debuntu)] ***
2021-05-28 00:00:05,662 p=4644 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"cache_update_time": 1622173942, "cache_updated": false, "changed": false, "msg": "'/usr/bin/apt-get -y -o \"Dpkg::Options::=--force-confdef\" -o \"Dpkg::Options::=--force-confold\"       install 'exfat-fuse' 'exfat-utils' 'inetutils-syslogd'' failed: E: Sub-process /usr/bin/dpkg returned an error code (1)\n", "rc": 100, "stderr": "E: Sub-process /usr/bin/dpkg returned an error code (1)\n", "stderr_lines": ["E: Sub-process /usr/bin/dpkg returned an error code (1)"], "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nThe following packages will be REMOVED:\n  rsyslog\nThe following NEW packages will be installed:\n  exfat-fuse exfat-utils inetutils-syslogd\n0 upgraded, 3 newly installed, 1 to remove and 0 not upgraded.\nNeed to get 284 kB of archives.\nAfter this operation, 955 kB disk space will be freed.\nGet:1 http://raspbian.mirror.constant.com/raspbian buster/main armhf exfat-fuse armhf 1.3.0-1 [27.5 kB]\nGet:2 http://raspbian.mirror.constant.com/raspbian buster/main armhf exfat-utils armhf 1.3.0-1 [40.3 kB]\nGet:3 http://raspbian.mirror.constant.com/raspbian buster/main armhf inetutils-syslogd armhf 2:1.9.4-7+deb10u1 [216 kB]\nFetched 284 kB in 1s (284 kB/s)\n(Reading database ... \r(Reading database ... 5%\r(Reading database ... 10%\r(Reading database ... 15%\r(Reading database ... 20%\r(Reading database ... 25%\r(Reading database ... 30%\r(Reading database ... 35%\r(Reading database ... 40%\r(Reading database ... 45%\r(Reading database ... 50%\r(Reading database ... 55%\r(Reading database ... 60%\r(Reading database ... 65%\r(Reading database ... 70%\r(Reading database ... 75%\r(Reading database ... 80%\r(Reading database ... 85%\r(Reading database ... 90%\r(Reading database ... 95%\r(Reading database ... 100%\r(Reading database ... 43975 files and directories currently installed.)\r\nRemoving rsyslog (8.1901.0-1) ...\r\nSelecting previously unselected package exfat-fuse.\r\n(Reading database ... \r(Reading database ... 5%\r(Reading database ... 10%\r(Reading database ... 15%\r(Reading database ... 20%\r(Reading database ... 25%\r(Reading database ... 30%\r(Reading database ... 35%\r(Reading database ... 40%\r(Reading database ... 45%\r(Reading database ... 50%\r(Reading database ... 55%\r(Reading database ... 60%\r(Reading database ... 65%\r(Reading database ... 70%\r(Reading database ... 75%\r(Reading database ... 80%\r(Reading database ... 85%\r(Reading database ... 90%\r(Reading database ... 95%\r(Reading database ... 100%\r(Reading database ... 43917 files and directories currently installed.)\r\nPreparing to unpack .../exfat-fuse_1.3.0-1_armhf.deb ...\r\nUnpacking exfat-fuse (1.3.0-1) ...\r\nSelecting previously unselected package exfat-utils.\r\nPreparing to unpack .../exfat-utils_1.3.0-1_armhf.deb ...\r\nUnpacking exfat-utils (1.3.0-1) ...\r\nSelecting previously unselected package inetutils-syslogd.\r\nPreparing to unpack .../inetutils-syslogd_2%3a1.9.4-7+deb10u1_armhf.deb ...\r\nUnpacking inetutils-syslogd (2:1.9.4-7+deb10u1) ...\r\nSetting up exfat-utils (1.3.0-1) ...\r\nSetting up inetutils-syslogd (2:1.9.4-7+deb10u1) ...\r\ninetutils-syslogd.service is not active, cannot reload.\r\ninvoke-rc.d: initscript inetutils-syslogd, action \"start\" failed.\r\n* inetutils-syslogd.service - LSB: GNU inetutils syslog daemon\r\n   Loaded: loaded (\u001b]8;;file://box.lan/etc/init.d/inetutils-syslogd\u0007/etc/init.d/inetutils-syslogd\u001b]8;;\u0007; generated)\r\n   Active: activating (start) since Fri 2021-05-28 00:00:02 EDT; 58ms ago\r\n     Docs: \u001b]8;;man:systemd-sysv-generator(8)\u0007man:systemd-sysv-generator(8)\u001b]8;;\u0007\r\nCntrl PID: 10894 (inetutils-syslo)\r\n    Tasks: 2 (limit: 3860)\r\n   CGroup: /system.slice/inetutils-syslogd.service\r\n           |-10894 /bin/sh /etc/init.d/inetutils-syslogd start\r\n           `-10901 /bin/sh /etc/init.d/inetutils-syslogd start\r\n\r\nMay 28 00:00:02 box.lan systemd[1]: Starting LSB: GNU inetutils syslog daemon...\r\nMay 28 00:00:02 box.lan systemd[1]: \u001b[0;1;39m\u001b[0;1;31m\u001b[0;1;39minetutils-syslogd.service: Unit cannot be reloaded because it is inactive.\u001b[0m\r\ndpkg: error processing package inetutils-syslogd (--configure):\r\n installed inetutils-syslogd package post-installation script subprocess returned error exit status 1\r\nSetting up exfat-fuse (1.3.0-1) ...\r\nProcessing triggers for man-db (2.8.5-2) ...\r\nProcessing triggers for systemd (241-7~deb10u7+rpi1) ...\r\nErrors were encountered while processing:\r\n inetutils-syslogd\r\n", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information...", "The following packages will be REMOVED:", "  rsyslog", "The following NEW packages will be installed:", "  exfat-fuse exfat-utils inetutils-syslogd", "0 upgraded, 3 newly installed, 1 to remove and 0 not upgraded.", "Need to get 284 kB of archives.", "After this operation, 955 kB disk space will be freed.", "Get:1 http://raspbian.mirror.constant.com/raspbian buster/main armhf exfat-fuse armhf 1.3.0-1 [27.5 kB]", "Get:2 http://raspbian.mirror.constant.com/raspbian buster/main armhf exfat-utils armhf 1.3.0-1 [40.3 kB]", "Get:3 http://raspbian.mirror.constant.com/raspbian buster/main armhf inetutils-syslogd armhf 2:1.9.4-7+deb10u1 [216 kB]", "Fetched 284 kB in 1s (284 kB/s)", "(Reading database ... ", "(Reading database ... 5%", "(Reading database ... 10%", "(Reading database ... 15%", "(Reading database ... 20%", "(Reading database ... 25%", "(Reading database ... 30%", "(Reading database ... 35%", "(Reading database ... 40%", "(Reading database ... 45%", "(Reading database ... 50%", "(Reading database ... 55%", "(Reading database ... 60%", "(Reading database ... 65%", "(Reading database ... 70%", "(Reading database ... 75%", "(Reading database ... 80%", "(Reading database ... 85%", "(Reading database ... 90%", "(Reading database ... 95%", "(Reading database ... 100%", "(Reading database ... 43975 files and directories currently installed.)", "Removing rsyslog (8.1901.0-1) ...", "Selecting previously unselected package exfat-fuse.", "(Reading database ... ", "(Reading database ... 5%", "(Reading database ... 10%", "(Reading database ... 15%", "(Reading database ... 20%", "(Reading database ... 25%", "(Reading database ... 30%", "(Reading database ... 35%", "(Reading database ... 40%", "(Reading database ... 45%", "(Reading database ... 50%", "(Reading database ... 55%", "(Reading database ... 60%", "(Reading database ... 65%", "(Reading database ... 70%", "(Reading database ... 75%", "(Reading database ... 80%", "(Reading database ... 85%", "(Reading database ... 90%", "(Reading database ... 95%", "(Reading database ... 100%", "(Reading database ... 43917 files and directories currently installed.)", "Preparing to unpack .../exfat-fuse_1.3.0-1_armhf.deb ...", "Unpacking exfat-fuse (1.3.0-1) ...", "Selecting previously unselected package exfat-utils.", "Preparing to unpack .../exfat-utils_1.3.0-1_armhf.deb ...", "Unpacking exfat-utils (1.3.0-1) ...", "Selecting previously unselected package inetutils-syslogd.", "Preparing to unpack .../inetutils-syslogd_2%3a1.9.4-7+deb10u1_armhf.deb ...", "Unpacking inetutils-syslogd (2:1.9.4-7+deb10u1) ...", "Setting up exfat-utils (1.3.0-1) ...", "Setting up inetutils-syslogd (2:1.9.4-7+deb10u1) ...", "inetutils-syslogd.service is not active, cannot reload.", "invoke-rc.d: initscript inetutils-syslogd, action \"start\" failed.", "* inetutils-syslogd.service - LSB: GNU inetutils syslog daemon", "   Loaded: loaded (\u001b]8;;file://box.lan/etc/init.d/inetutils-syslogd\u0007/etc/init.d/inetutils-syslog [... cut ...]
> root@raspberrypi:~# apt install inetutils-syslogd
Reading package lists... Done
Building dependency tree
Reading state information... Done
inetutils-syslogd is already the newest version (2:1.9.4-7+deb10u1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up inetutils-syslogd (2:1.9.4-7+deb10u1) ...

Is there anything we can/should do...to avoid this in future?

http://sprunge.us/kxW1WH?en

holta, 2021-05-28:
@jvonau asked:

> So on a fresh install of the OS after apt updating can you install the package without  iiab being installed?

Yes, in isolation this worked &mdash; the result installing on a clean 32-bit "RaspiOS with desktop" on RPi 4 was:

```
root@raspberrypi:~# apt install inetutils-syslogd
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages will be REMOVED:
  rsyslog
The following NEW packages will be installed:
  inetutils-syslogd
0 upgraded, 1 newly installed, 1 to remove and 0 not upgraded.
Need to get 216 kB of archives.
After this operation, 1,215 kB disk space will be freed.
Do you want to continue? [Y/n]
Get:1 http://raspbian.mirror.constant.com/raspbian buster/main armhf inetutils-syslogd armhf 2:1.9.4-7+deb10u1 [216 kB]
Fetched 216 kB in 1s (300 kB/s)
(Reading database ... 163698 files and directories currently installed.)
Removing rsyslog (8.1901.0-1) ...
Selecting previously unselected package inetutils-syslogd.
(Reading database ... 163641 files and directories currently installed.)
Preparing to unpack .../inetutils-syslogd_2%3a1.9.4-7+deb10u1_armhf.deb ...
Unpacking inetutils-syslogd (2:1.9.4-7+deb10u1) ...
Setting up inetutils-syslogd (2:1.9.4-7+deb10u1) ...
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for systemd (241-7~deb10u7+rpi1) ...

jvonau, 2021-05-28:
Start looking at what versions are installed on an older install to figure out if this is a recent apt package update and/or if systemd has been updated as this a change in noted behavior. How reproducible is this?

jvonau, 2021-05-28:
Try this, boot a fresh image, `sudo apt update` `apt list --upgradable > ~/apt-list.txt` and attach the file to this issue for further review.

holta, 2021-05-28:
32-bit Lite released today (https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/2021-05-07-raspios-buster-armhf-lite.zip) installed on RPi 4 shows:

```
root@raspberrypi:~# apt list --upgradable
Listing... Done
libraspberrypi-bin/testing 1:1.20210527-1 armhf [upgradable from: 1.20210430-1]
libraspberrypi-dev/testing 1:1.20210527-1 armhf [upgradable from: 1.20210430-1]
libraspberrypi-doc/testing 1:1.20210527-1 armhf [upgradable from: 1.20210430-1]
libraspberrypi0/testing 1:1.20210527-1 armhf [upgradable from: 1.20210430-1]
libx11-6/stable 2:1.6.7-1+deb10u2 armhf [upgradable from: 2:1.6.7-1+deb10u1]
libx11-data/stable 2:1.6.7-1+deb10u2 all [upgradable from: 2:1.6.7-1+deb10u1]
linux-libc-dev/testing 1:1.20210527-1 armhf [upgradable from: 4.18.20-2+rpi1]
pi-bluetooth/testing 0.1.17 all [upgradable from: 0.1.15]
python-gpiozero/testing 1.6.2-1 all [upgradable from: 1.5.1]
python-spidev/testing 20200602~200721-1~buster armhf [upgradable from: 20190221~182651-1]
python3-gpiozero/testing 1.6.2-1 all [upgradable from: 1.5.1]
python3-spidev/testing 20200602~200721-1~buster armhf [upgradable from: 20190221~182651-1]
raspberrypi-bootloader/testing 1:1.20210527-1 armhf [upgradable from: 1.20210430-1]
raspberrypi-kernel/testing 1:1.20210527-1 armhf [upgradable from: 1.20210430-1]
rpi-eeprom/testing 12.4-1 armhf [upgradable from: 12.3-1]
```

```
root@raspberrypi:~# uname -a
Linux raspberrypi 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l GNU/Linux
```

After `apt -y dist-upgrade` and reboot it shows:

```
root@raspberrypi:~# uname -a
Linux raspberrypi 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l GNU/Linux
```

holta, 2021-05-28:
> How reproducible is this?

With today's new release of 32-bit RaspiOS Lite, the problem did not occur.  So hopefully...this is now resolved and/or very rare.

jvonau, 2021-05-28:
From kxW1WH?en
>18 =IIAB==========================================================================
  19 -rw-r--r-- 1 root root 145 Mar  4 18:04 /etc/rpi-issue
  20 
  21 Raspberry Pi reference 2021-03-04
  22 Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 461109415073d2eb67083709662ba983cc191f14, stage2
  23 
  24 -IIAB-EXPLANATION-OF-THE-ABOVE-------------------------------------------------
  25 
  26 stage2 = Raspberry Pi OS Lite
  27 stage4 = Raspberry Pi OS with desktop
  28 stage5 = Raspberry Pi OS with desktop + recommended software
  29 
  30 SEE https://github.com/RPi-Distro/pi-gen#stage-anatomy

Now you are saying an newer release doesn't have this problem. Should really follow the best practices, use the latest release offered.

holta, 2021-06-01:
Has not occurred on repeated installs on RPi 4 since 5 days ago (on 32-bit RaspiOS Lite and Ubuntu Server 21.04).

So let's assume this was an intermittent/transient issue, that for whatever reason arose just prior to RaspiOS publishing its new OS images last Friday.

closed 2021-06-01 by holta