iiab/iiab #2808: nodejs_version: 14.x -> 16.x

holta, 2021-05-28:
Some (e.g. Kiwix) have already made the jump.

IIAB can make the change here anytime:

- https://github.com/iiab/iiab/blob/master/vars/default_vars.yml#L412
- https://github.com/iiab/iiab/blob/master/roles/nodejs/defaults/main.yml#L12

But IIAB can likely wait until Node.js 16 becomes "Active LTS" on [2021-10-26](https://nodejs.org/en/about/releases/).

holta, 2021-06-16:
@mitra42 IIAB works with PHP 16.3.0 (npm 7.15.1) except for [roles/internetarchive/tasks/install.yml (Lines 44-49)](https://github.com/iiab/iiab/blob/master/roles/internetarchive/tasks/install.yml#L44-L49) &mdash; can you tell if this might be easily fixable?  Here's the command that fails:

```
yarn config set child-concurrency 1 && yarn add @internetarchive/dweb-mirror
```

FYI 1: Node.js 16 was released 2021-04-20 and becomes LTS on 2021-10-26.  Are there certain dependencies that do not yet work with Node.js 16.x ?

FYI 2: Here's the error that occurs repeatedly &mdash; e.g. during a fresh install of IIAB onto Debian 10.9 &mdash; also whenever I try `cd /opt/iiab/iiab` then `./runrole internetarchive` :

> TASK [internetarchive : Run 'yarn add @internetarchive/dweb-mirror' to download/populate /opt/iiab/internetarchive/node_modules (CAN TAKE ~5 MINUTES)] ***
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "yarn config set child-concurrency 1 && yarn add @internetarchive/dweb-mirror", "delta": "0:00:40.144089", "end": "2021-06-16 12:02:22.533963", "msg": "non-zero return code", "rc": 1, "start": "2021-06-16 12:01:42.389874", "stderr": "warning @internetarchive/dweb-mirror > @futurepress/epubjs-reader > epubjs > @types/localforage@0.0.34: This is a stub types definition for localforage (https://github.com/localForage/localForage). localforage provides its own type definitions, so you don't need @types/localforage installed!\nwarning @internetarchive/dweb-mirror > @internetarchive/dweb-archive-dist > react > fbjs > core-js@1.2.7: core-js@<3.3 is no longer maintained and not recommended for usage due to the number of issues. Because of the V8 engine whims, feature detection in old core-js versions could cause a slowdown up to 100x even if nothing is polyfilled. Please, upgrade your dependencies to the actual version of core-js.\nwarning @internetarchive/dweb-mirror > @futurepress/epubjs-reader > epubjs > xmldom@0.3.0: Deprecated due to CVE-2021-21366 resolved in 0.5.0\nerror /opt/iiab/internetarchive/node_modules/sharp: Command failed.\nExit code: 1\nCommand: (node install/libvips && node install/dll-copy && prebuild-install) || (node-gyp rebuild && node install/dll-copy)\nArguments: \nDirectory: /opt/iiab/internetarchive/node_modules/sharp\nOutput:\ninfo sharp Downloading https://github.com/lovell/sharp-libvips/releases/download/v8.8.1/libvips-8.8.1-linux-x64.tar.gz\nprebuild-install WARN install No prebuilt binaries found (target=16.3.0 runtime=node arch=x64 libc= platform=linux)\ngyp info it worked if it ends with ok\ngyp info using node-gyp@7.1.2\ngyp info using node@16.3.0 | linux | x64\ngyp info find Python using Python version 3.7.3 found at \"/usr/bin/python3\"\n(node:24529) [DEP0150] DeprecationWarning: Setting process.config is deprecated. In the future the property will be read-only.\n(Use `node --trace-deprecation ...` to show where the warning was created)\ngyp info spawn /usr/bin/python3\ngyp info spawn args [\ngyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',\ngyp info spawn args   'binding.gyp',\ngyp info spawn args   '-f',\ngyp info spawn args   'make',\ngyp info spawn args   '-I',\ngyp info spawn args   '/opt/iiab/internetarchive/node_modules/sharp/build/config.gypi',\ngyp info spawn args   '-I',\ngyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',\ngyp info spawn args   '-I',\ngyp info spawn args   '/root/.cache/node-gyp/16.3.0/include/node/common.gypi',\ngyp info spawn args   '-Dlibrary=shared_library',\ngyp info spawn args   '-Dvisibility=default',\ngyp info spawn args   '-Dnode_root_dir=/root/.cache/node-gyp/16.3.0',\ngyp info spawn args   '-Dnode_gyp_dir=/usr/lib/node_modules/npm/node_modules/node-gyp',\ngyp info spawn args   '-Dnode_lib_file=/root/.cache/node-gyp/16.3.0/<(target_arch)/node.lib',\ngyp info spawn args   '-Dmodule_root_dir=/opt/iiab/internetarchive/node_modules/sharp',\ngyp info spawn args   '-Dnode_engine=v8',\ngyp info spawn args   '--depth=.',\ngyp info spawn args   '--no-parallel',\ngyp info spawn args   '--generator-output',\ngyp info spawn args   'build',\ngyp info spawn args   '-Goutput_dir=.'\ngyp info spawn args ]\ngyp info spawn make\ngyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]\nmake: Entering directory '/opt/iiab/internetarchive/node_modules/sharp/build'\n  TOUCH Release/obj.target/libvips-cpp.stamp\n  CXX(target) Release/obj.target/sharp/src/common.o\n<command-line>: warning: \"_GLIBCXX_USE_CXX11_ABI\" redefined\n<command-line>: note: this is the location of the previous definition\nIn file included from /root/.cache/node-gyp/16.3.0/include/node/v8.h:30,\n                 from /root/.cache/node-gyp/16.3.0/include/node/node.h:63,\n                 from ../src/common.cc:22:\n/root/.cache/node-gyp/16.3.0/include/node/v8-internal.h: In function ‘void v8::internal::PerformCastCheck(T*)’:\n/root/.cache/node-gyp/16.3.0/include/node/v8-internal.h:452:38: error: ‘remove_cv_t’ is not a member of ‘std’\n             !std::is_same<Data, std::remove_cv_t<T>>::value>::Perform(data);\n                                      ^~~~~~~~~~~\n/root/.cache/node-gyp/16.3.0/include/node/v8-internal.h:452:38: note: suggested alternative: ‘remove_cv’\n             !std::is_same<Data, std::remove_cv_t<T>>::value>::Perform(data);\n                                      ^~~~~~~~~~~\n                                      remove_cv\n/root/.cache/node-gyp/16.3.0/include/node/v8-internal.h:4 [... cut ...] 2/4] Fetching packages...", "[3/4] Linking dependencies...", "[4/4] Building fresh packages...", "info Visit https://yarnpkg.com/en/docs/cli/add for documentation about this command."]}

CLARIF: the goal is to upgrade IIAB from Node.j 14.x to 16.x across the board, soon/prior to it becoming LTS in October if possible.  So https://github.com/iiab/iiab/blob/master/roles/internetarchive/tasks/install.yml#L12-L15 (Node.js version assertion) is being commented out at this time (to help with testing).

mitra42, 2021-06-16:
It looks to me like you might have jumped the gun on the upgrade to 14.x  ! This looks like issues with building dependencies of dweb-mirror, but it could be that dweb-mirror is version locking to an earlier version of a dependency that doesnt support node16.  

If that's the case I should be able to fix it once I upgrade one of my boxes to node 16 - I haven't done that yet, but don't mind doing so.  I don't currently have a setup that can run IIAB but if I can replicate this bug on OSX then I should be able to fix it. 

Since node 16 only just came out, so just as dweb-mirror hasn't been upgraded,  its also possible that one of dweb-mirror's dependencies has its own dependency on something that isn't upgraded to node 16 in which case it may not be fixable without waiting for those package maintainers to notice.

holta, 2021-06-16:
Thanks @mitra42 !

(Shout if you need a hand when looking into this.)

mitra42, 2021-06-17:
@holta - I found the problem - needed to bump version on "sharp" which is a dependency to support node 16. 

I have tested that this installs on my laptop, but not in a position to do a IIAB test install (don't have a Raspberry Pi easily available). Could you see if this fixes it... Note I think it needs https://github.com/iiab/iiab/blob/master/roles/internetarchive/tasks/install.yml#L15 changing to include 16 as an option.

holta, 2021-06-17:
> Note I think it needs https://github.com/iiab/iiab/blob/master/roles/internetarchive/tasks/install.yml#L15 changing to include 16 as an option.

@mitra42 yes: I used PR #2816 (`git pull https://github.com/holta/iiab nodejs-16.x`) to accomplish that.

> @holta - I found the problem - needed to bump version on "sharp" which is a dependency to support node 16.
> 
> I have tested that this installs on my laptop, but not in a position to do a IIAB test install (don't have a Raspberry Pi easily available). Could you see if this fixes it...

I tried again, including deleting the contents of `/root/.cache` but no luck unfortunately.  Note that https://github.com/iiab/iiab/blob/master/roles/internetarchive/tasks/install.yml#L39-L42 already wipes `/opt/iiab/internetarchive/node_modules` so my running things like `./runrole --reinstall internetarchive` should grab your latest code each time I try.

_Or is there something else I should try?!_

In consideration of the latest error/output pasted in here:

> TASK [internetarchive : Wipe dir /opt/iiab/internetarchive/node_modules (and its contents, typically 112+ MB) so './runrole --reinstall internetarchive' gets you the latest code] ***
ok: [127.0.0.1]
>
> TASK [internetarchive : Run 'yarn add @internetarchive/dweb-mirror' to download/populate /opt/iiab/internetarchive/node_modules (CAN TAKE ~5 MINUTES)] ***
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "yarn config set child-concurrency 1 && yarn add @internetarchive/dweb-mirror", "delta": "0:00:19.709041", "end": "2021-06-17 05:23:07.158462", "msg": "non-zero return code", "rc": 1, "start": "2021-06-17 05:22:47.449421", "stderr": "warning @internetarchive/dweb-mirror > @internetarchive/dweb-archive-dist > react > fbjs > core-js@1.2.7: core-js@<3.3 is no longer maintained and not recommended for usage due to the number of issues. Because of the V8 engine whims, feature detection in old core-js versions could cause a slowdown up to 100x even if nothing is polyfilled. Please, upgrade your dependencies to the actual version of core-js.\nwarning @internetarchive/dweb-mirror > @futurepress/epubjs-reader > epubjs > @types/localforage@0.0.34: This is a stub types definition for localforage (https://github.com/localForage/localForage). localforage provides its own type definitions, so you don't need @types/localforage installed!\nwarning @internetarchive/dweb-mirror > @futurepress/epubjs-reader > epubjs > xmldom@0.3.0: Deprecated due to CVE-2021-21366 resolved in 0.5.0\nerror /opt/iiab/internetarchive/node_modules/sharp: Command failed.\nExit code: 1\nCommand: (node install/libvips && node install/dll-copy && prebuild-install) || (node-gyp rebuild && node install/dll-copy)\nArguments: \nDirectory: /opt/iiab/internetarchive/node_modules/sharp\nOutput:\ninfo sharp Using cached /root/.npm/_libvips/libvips-8.8.1-linux-x64.tar.gz\nprebuild-install WARN install No prebuilt binaries found (target=16.3.0 runtime=node arch=x64 libc= platform=linux)\ngyp info it worked if it ends with ok\ngyp info using node-gyp@7.1.2\ngyp info using node@16.3.0 | linux | x64\ngyp info find Python using Python version 3.7.3 found at \"/usr/bin/python3\"\ngyp http GET https://nodejs.org/download/release/v16.3.0/node-v16.3.0-headers.tar.gz\ngyp http 200 https://nodejs.org/download/release/v16.3.0/node-v16.3.0-headers.tar.gz\ngyp http GET https://nodejs.org/download/release/v16.3.0/SHASUMS256.txt\ngyp http 200 https://nodejs.org/download/release/v16.3.0/SHASUMS256.txt\n(node:15343) [DEP0150] DeprecationWarning: Setting process.config is deprecated. In the future the property will be read-only.\n(Use `node --trace-deprecation ...` to show where the warning was created)\ngyp info spawn /usr/bin/python3\ngyp info spawn args [\ngyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',\ngyp info spawn args   'binding.gyp',\ngyp info spawn args   '-f',\ngyp info spawn args   'make',\ngyp info spawn args   '-I',\ngyp info spawn args   '/opt/iiab/internetarchive/node_modules/sharp/build/config.gypi',\ngyp info spawn args   '-I',\ngyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',\ngyp info spawn args   '-I',\ngyp info spawn args   '/root/.cache/node-gyp/16.3.0/include/node/common.gypi',\ngyp info spawn args   '-Dlibrary=shared_library',\ngyp info spawn args   '-Dvisibility=default',\ngyp info spawn args   '-Dnode_root_dir=/root/.cache/node-gyp/16.3.0',\ngyp info spawn args   '-Dnode_gyp_dir=/usr/lib/node_modules/npm/node_modules/node-gyp',\ngyp info spawn args   '-Dnode_lib_file=/root/.cache/node-gyp/16.3.0/<(target_arch)/node.lib',\ngyp info spawn args   '-Dmodule_root_dir=/opt/iiab/internetarchive/node_modules/sharp',\ngyp info spawn args   '-Dnode_engine=v8',\ngyp info spawn args   '--depth=.',\ngyp info spawn args   '--no-parallel',\ngyp info spawn args   '--generator-output',\ngyp info spawn args   'build',\ngyp info spawn args   '-Goutput_dir=.'\ngyp info spawn args ]\ngyp info spawn make\ngyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]\nmake: Entering directory '/opt/iiab/internetarchive/node_modules/sharp/build'\n  TOUCH Release/obj.target/libvips-cpp.stamp\n  CXX(target) Release/obj.target/sharp/src/common.o\n<command-line>: warning: \"_GLIBCXX_USE_CXX11_ABI\" redefined\n<command-line>: note: this is the location of the previous definition\nIn file included from /root/.cache/node-gyp/16.3.0/include/node/v8.h:30,\n                 from /root/.cache/node-gyp/16.3 [... cut ...] \" to \"1\".", "Done in 0.05s.", "yarn add v1.22.5", "info No lockfile found.", "[1/4] Resolving packages...", "[2/4] Fetching packages...", "[3/4] Linking dependencies...", "[4/4] Building fresh packages...", "info Visit https://yarnpkg.com/en/docs/cli/add for documentation about this command."]}
>
> PLAY RECAP *********************************************************************
> 127.0.0.1                  : ok=48   changed=8    unreachable=0    failed=1    skipped=22   rescued=0    ignored=0

mitra42, 2021-06-17:
Actually .... its a while since I did this, I needed to do a `npm publish` rather than just pushing it to the github repo, could you try again.

holta, 2021-06-17:
> Actually .... its a while since I did this, I needed to do a `npm publish` rather than just pushing it to the github repo, could you try again.

That worked, Thanks @mitra42!

PR #2816 (2021-06-17): WIP: Node.js 14.x -> 16.x
Seeks to resolved #2808

Testing ongoing with BIG-sized IIAB on Debian 10.9 = 193-deb10-srv-BIG-nodejs16-0616.
closed 2021-06-17 by holta