iiab/iiab #2811: Test offline https/SSL/TLS NGINX with self-signed cert instead of LetsEncrypt.org ?

holta, 2021-06-03:
@tim-moody: @deldesir is interested if you or @jvonau make progress here figuring out what's possible?

For ZIM files from https://youzim.it and in general!

Ref: #2762

holta, 2021-06-03:
Relevant tickets are 546, 599 and 608 here:

https://github.com/kiwix/kiwix-hotspot/issues

tim-moody, 2021-07-28:
I tried the following:

```
openssl req -x509 -nodes -days 365 -newkey rsa:2048 -keyout /etc/ssl/private/nginx-selfsigned.key -out /etc/ssl/certs/nginx-selfsigned.crt

nano server.conf after listen 80;

listen 443 ssl;
      ssl_certificate /etc/ssl/certs/nginx-selfsigned.crt;
      ssl_certificate_key /etc/ssl/private/nginx-selfsigned.key;

```

The following on a medium vm were accessible over https:

•	Home
•	Kiwix
•	Module
•	Maps
•	Sugarizer
•	Nextcloud
•	WP
•	Calibre web

kalite was not accessible via the menu. It is not proxied and probably requires port 80. The problem is js-menu creates a link with the current protocol, so tried https when only http is available.

tim-moody, 2021-07-28:
I'm less sure about the kalite problem as it is not working with http on this vm, so there could be another issue.

jvonau, 2021-07-28:
kalite runs on port 8008 and never requires port 80, I'll go with the curl script was not used, now this machine has an incomplete kalite install https://github.com/iiab/iiab/issues/2863#issuecomment-887426453 due to the kalite role|iiab split of commands needed.

tim-moody, 2021-07-28:
>I'll go with the curl script was not used, now this machine has an incomplete kalite install 

probably right