iiab/iiab #2813: "Menu Item Code" often cut off in Admin Console > Content Menus > Edit Menu Items > Edit Menu Item

holta, 2021-06-06:
@tim-moody can this text field be made wider, so that it's readable without being cut off?

[image]

tim-moody, 2021-06-06:
50-edit-menus change lines 227 and 234 col-md-4 to col-md-6

holta, 2021-06-06:
50-edit_menus.html Lines 227 and 233 are here:

https://github.com/iiab/iiab-admin-console/blob/master/roles/console/files/htmlf/50-edit_menus.html#L227
https://github.com/iiab/iiab-admin-console/blob/master/roles/console/files/htmlf/50-edit_menus.html#L233

holta, 2021-06-06:
@tim-moody while at first glance col-md-7 (on Line 227) and col-md-5 (on Line 233) look good...

...things like `fr-la_chaine_de_maths_et_tiques_fr` are still cut off (not displayed!)

Question: what is the purpose of the "Optional Code Suffix" textfield ?  (Normally it's empty...when is this used and/or should it be pushed onto the next line or eliminated?)

tim-moody, 2021-06-06:
the suffix is to allow country or other variants, like adding mx
8 + 4 might be better

closed 2021-06-15 by tim-moody