iiab/iiab #2817: Following task is failing : TASK [kalite : Run '/usr/local/kalite/venv/bin/kalite manage setup ...'

abhiapsunde, 2021-06-17:
### Expected behavior and actual behavior:
Task should complete and install iiab
### Steps to reproduce the problem:
`sudo iiab`
### IIAB version, OS version, hardware details:
IIAB Version : Using latest that is pulled by one line install command
OS : 
```
pi@box:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
```

Hardware : RPi 4 8GB
### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)
http://sprunge.us/4Sxdwo

holta, 2021-06-17:
Here's a 4-line excerpt from the end of `/opt/iiab/iiab/iiab-install.log` (Lines 1436-1439 of http://sprunge.us/4Sxdwo?en) to help investigate:

> 1436 2021-06-17 11:04:38,152 p=2197 u=root n=ansible | TASK [kalite : Run '/usr/local/kalite/venv/bin/kalite manage setup ...'] *******
1437 2021-06-17 11:05:01,256 p=2197 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"ansible_job_id": "665566210286.3284", "changed": true, "cmd": ["/usr/local/kalite/venv/bin/kalite", "manage", "setup", "--username=Admin", "--password=changeme", "--noinput"], "delta": "0:00:11.992684", "end": "2021-06-17 11:04:51.501378", "finished": 1, "msg": "non-zero return code", "rc": 1, "start": "2021-06-17 11:04:39.508694", "stderr": "Traceback (most recent call last):\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/core/management/base.py\", line 224, in run_from_argv\n    self.execute(*args, **options.__dict__)\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/core/management/base.py\", line 263, in execute\n    output = self.handle(*args, **options)\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/distributed/management/commands/setup.py\", line 477, in handle\n    call_command(\"collectstatic_js_reverse\", interactive=False)\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/core/management/__init__.py\", line 161, in call_command\n    return klass.execute(*args, **defaults)\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/core/management/base.py\", line 263, in execute\n    output = self.handle(*args, **options)\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/dist/django_js_reverse/management/commands/collectstatic_js_reverse.py\", line 26, in handle\n    content = urls_js()\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/dist/django_js_reverse/views.py\", line 50, in urls_js\n    'urls': sorted(list(prepare_url_list(default_urlresolver))),\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/dist/django_js_reverse/views.py\", line 74, in prepare_url_list\n    for url_name in urlresolver.reverse_dict.keys():\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/core/urlresolvers.py\", line 297, in reverse_dict\n    self._populate()\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/core/urlresolvers.py\", line 263, in _populate\n    for pattern in reversed(self.url_patterns):\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/core/urlresolvers.py\", line 347, in url_patterns\n    patterns = getattr(self.urlconf_module, \"urlpatterns\", self.urlconf_module)\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/core/urlresolvers.py\", line 342, in urlconf_module\n    self._urlconf_module = import_module(self.urlconf_name)\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/utils/importlib.py\", line 35, in import_module\n    __import__(name)\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/distributed/urls.py\", line 14, in <module>\n    from . import api_urls\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/distributed/api_urls.py\", line 12, in <module>\n    import kalite.coachreports.api_urls\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/coachreports/api_urls.py\", line 3, in <module>\n    from .api_resources import PlaylistProgressResource, PlaylistProgressDetailResource\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/coachreports/api_resources.py\", line 3, in <module>\n    from tastypie.resources import Resource\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/dist/tastypie/resources.py\", line 20, in <module>\n    from django.contrib.gis.db.models.fields import GeometryField\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/contrib/gis/db/models/__init__.py\", line 5, in <module>\n    from django.contrib.gis.db.models.aggregates import *\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/contrib/gis/db/models/aggregates.py\", line 2, in <module>\n    from django.contrib.gis.db.models.sql import GeomField\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/contrib/gis/db/models/sql/__init__.py\", line 2, in <module>\n    from django.contrib.gis.db.models.sql.query import GeoQuery\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/contrib/gis/db/models/sql/query.py\", line 4, in <module>\n    from django.contrib.gis.db.models.fields import GeometryField\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/contrib/gis/db/models/fields.py\", line 4, in <module>\n    from django.contrib.gis import forms\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/contrib/gis/forms/__init__.py\", line 2, in <module>\n    from django.contrib.gis.forms.fields import GeometryField\n  File [... cut ...] /venv/lib/python2.7/site-packages/kalite/database/templates/content_items/content_khan_en.sqlite", "Not running in interactive mode; skipping content pack download.", "Copying static media..."]}
1438 2021-06-17 11:05:01,260 p=2197 u=root n=ansible | PLAY RECAP *********************************************************************
1439 2021-06-17 11:05:01,260 p=2197 u=root n=ansible | 127.0.0.1                  : ok=35   changed=8    unreachable=0    failed=1    skipped=20   rescued=0    ignored=0

holta, 2021-06-17:
@abhiapsunde did you re-run `sudo iiab` 2 or 3 times to confirm?

@jvonau can you see what might be happening here?

KA Lite installation is normally rock solid.  But obviously something is suddenly different, causing `/usr/local/kalite/venv/bin/kalite manage setup ...` to fail here: (repeatedly?)

https://github.com/iiab/iiab/blob/master/roles/kalite/tasks/install.yml#L84-L89

abhiapsunde, 2021-06-17:
Yes I’ve ran it 2-3 times before creating the issue.
Thinking that I might be running brand new code by using the one line
utility after those 3 attempts and filing this report ,  I also tried one
liner that installs a specific release version (7.1) it also fails with
same error as in original report.

holta, 2021-06-17:
> brand new code

@abhiapsunde just as clarification this KA Lite installation code is extremely well-tested code, and in production over many years.

So this is highly mysterious &mdash; as this has not affected anybody else that we know of.

But thanks for reporting so we can now investigate!

abhiapsunde, 2021-06-17:
Thanks for clarification @holta , feel free to ask me if you need to try something out on this box during the investigation.

tim-moody, 2021-06-17:
was this the first run? not sure why the database would already exist.

`| / _ \ \n | |\ \| | | | | || | || __/ \n \| \/\| |/ \_____/|\\| \n \nhttps://learningequality.org/ka-lite/\n \n version 0.17.5\n \n-------------------------------------------------------------------\nWARNING: You are installing KA-Lite as root user!\n Installing as root may cause some permission problems while running\n as a normal user in the future.\n-------------------------------------------------------------------\n\n-------------------------------------------------------------------\nWARNING: Database file already exists!\n`

holta, 2021-06-17:
I did a full/fresh install of MIN-sized IIAB onto RaspiOS "Stage 4" (Desktop without Additional App, exactly the same) to try to simulate the KA Lite installation failure -- onto a 2GB RAM Raspberry Pi 4 -- **and I cannot reproduce the problem :/**

1) So getting to the bottom of @tim-moody's database question above seems especially relevant!

   @abhiapsunde can you consider `rm -rf /library/ka-lite` to completely wipe KA Lite and its database there &mdash; before trying again `sudo iiab` ?

2) Also can you please send the specs for your 476.9G disk (that's according to `lsblk`) and the specs of your power supply specs driving it and the RPi 4?  As we see repeated failures when large/external disks are being underpowered, e.g. with Raspberry Pi "hats" and/or insufficient electrical power from the charger itself.

abhiapsunde, 2021-06-17:
1. I will try rm -rf sometime today.
2. The disk is a 512GB SD card (OS resides on it, no separate disk), I'm
using a stock power supply that came with rpi, there is nothing else
connected to pi other than keyboard / mouse , no HATs. I dont see any under
voltage / temperature alerts on monitors which normally show up during such
an issue.

holta, 2021-06-18:
> WARNING: Database file already exists!\n-------------------------------------------------------------------\nUpgrading database to KA Lite version 0.17.5\nBaking a fresh database from scratch or upgrading existing database.\nSyncing...\nCreating tables ...

@abhiapsunde thanks again for pushing:

1) Are you customizing Raspberry Pi OS in any way, prior to installing IIAB?

2) As I don't think we've ever seen a similar message — installing KA Lite for about 7 years now — we'd all love to find out: what specifically could be unique and different with your particular setup?

3) FINALLY: do you have access to a 32GB or any other/spare microSD card _(for quick + important comparison!)_ that will allow you install a separate MIN-sized Internet-in-a-Box?

abhiapsunde, 2021-06-18:
1. It was a plain desktop version of the latest raspberry pi os, the only
thing I installed on that was GNU radio, I've been meaning to use that card
for some Software Defined Radio experiments, nothing other than that (no
overclocking , changes to default OS config etc.)
2 & 3 : Unfortunately, that card failed to boot when I tried bringing it up
for rm -rf , so I wiped the same 512G out and installed headless rpi os (no
desktop, just console access). IIAB installed successfully after that. I
didn't see the error and I could see the hotspot getting created and could
access the web UI / admin to download content. Sorry I lost that setup now
so can't do any further tests but looks like it was something with software
setup on that box rather than hardware issues.

To answer Tim's question, no it wasn't first run, it was from 3rd or 4th
attempt after initial failure, the console tips said run "sudo iiab" if
installation fails, so I though its normal for install to fail sometimes as
huge data downloads are involved (looks like that understanding was
incorrect , and data download is actually happening later , initiated from
admin UI as the user selects packages to download) , so didn't think much
about it and didn't capture diagnostic info at initial failure. I only came
to know about it while filing the issue, by that time I had attempted "sudo
iiab" 2-3 times.

holta, 2021-06-18:
Thank you @abhiapsunde for your initiative.

Letting us know if you run into any further issues!

In which case we can re-open this ticket if/as necessary.

closed 2021-06-18 by holta