iiab/iiab #2818: Test IIAB on Ubuntu 21.10 pre-releases (RPi+PC) x (Server+Desktop). Does RPi WiFi firmware 7.45.18.0 work well?

holta, 2021-06-18:
In the short-term https://cdimage.ubuntu.com/daily-live/current/ should be modified by hand to specify PHP 8.0 (and likely later PostgreSQL 13 -> 14) as these versions are different from Ubuntu 21.04

Ubuntu 21.10 (Impish Indri) pre-release daily builds:

- https://cdimage.ubuntu.com/ubuntu-server/daily-live/current/ (Server)
- https://cdimage.ubuntu.com/daily-live/current/ (Desktop)

Refs: #2744, #2814

holta, 2021-06-25:
To get a better picture of PHP 8.0 released 7 months ago (#2814) a BIG-sized IIAB was installed on Ubuntu Server 20.10 pre-release daily build (VM) at 10.8.0.30 = u2110-srv-BIG-0625.  Using these 2 PHP 8.0 workarounds: (WIP)

1. Package `php8.0-xmlrpc` doesn't exist:

   > TASK [www_base : Install ~10 PHP packages (debuntu)] ***************************
   > fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "No package matching 'php8.0-xmlrpc' is available"}

   If it turns out it's still needed by IIAB (?) fyi the "XML-RPC Extension Has Been Moved to PECL in PHP 8.0" :

   - https://lindevs.com/xml-rpc-extension-has-been-moved-to-pecl-in-php-8-0/
   - https://php.watch/versions/8.0/xmlrpc
   - https://pecl.php.net/package/xmlrpc

   For purposes of experimentation, I commented out the installation of `php8.0-xmlrpc` (on Line 18 of https://github.com/iiab/iiab/blob/master/roles/www_base/tasks/main.yml#L18) for now.

2. Package `php8.0-json` does not exist as it's now moved into PHP's core:

   > TASK [nextcloud : Install ffmpeg + libxml2 + 13 php packages] ******************
fatal: [127.0.0.1]: FAILED! => {"cache_update_time": 1624640653, "cache_updated": false, "changed": false, "msg": "'/usr/bin/apt-get -y -o \"Dpkg::Options::=--force-confdef\" -o \"Dpkg::Options::=--force-confold\"       install 'ffmpeg' 'php8.0-bcmath' 'php8.0-bz2' 'php8.0-gmp' 'php8.0-imagick' 'php8.0-json' 'php8.0-zip'' failed: E: Package 'php8.0-json' has no installation candidate\n", "rc": 100, "stderr": "E: Package 'php8.0-json' has no installation candidate\n", "stderr_lines": ["E: Package 'php8.0-json' has no installation candidate"], "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nPackage php8.0-json is a virtual package provided by:\n  php8.0-phpdbg 8.0.5-1ubuntu1\n  php8.0-fpm 8.0.5-1ubuntu1\n  libphp8.0-embed 8.0.5-1ubuntu1\n  php8.0-cli 8.0.5-1ubuntu1\n  php8.0-cgi 8.0.5-1ubuntu1\n  libapache2-mod-php8.0 8.0.5-1ubuntu1\n\n", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information...", "Package php8.0-json is a virtual package provided by:", "  php8.0-phpdbg 8.0.5-1ubuntu1", "  php8.0-fpm 8.0.5-1ubuntu1", "  libphp8.0-embed 8.0.5-1ubuntu1", "  php8.0-cli 8.0.5-1ubuntu1", "  php8.0-cgi 8.0.5-1ubuntu1", "  libapache2-mod-php8.0 8.0.5-1ubuntu1", ""]}

   Aside: Nextcloud 22 should be released on 2021-07-06, to further clarify this, with the successor document to: https://docs.nextcloud.com/server/21/admin_manual/installation/source_installation.html#prerequisites-for-manual-installation

holta, 2021-06-25:
General Reminder from https://github.com/iiab/iiab/wiki/IIAB-Platforms that manually installing Node.js 12.x will do for now:

> Prior to Ubuntu 21.10's 2021-10-14 release, Node.js and npm need to be installed manually, running these 2 lines as root: (for roles/internetarchive, roles/nodered, roles/pbx i.e. Asterisk/FreePBX and/or roles/sugarizer)
> - apt install nodejs npm    # Installs Node.js 12.x.y (for now, 14.x.y or 16.x.y later?) and npm 7.x.y (for now)
> - echo 'nodejs_installed: True' >> /etc/iiab/iiab_state.yml

holta, 2021-06-29:
Ubuntu Impish Indri Release Schedule:
https://discourse.ubuntu.com/t/impish-indri-release-schedule/18540

 (For the OS that will be released in 3.5 months on 2021-10-14)

jvonau, 2021-09-12:
How about an iiab-diagnostics from recent testing install?

holta, 2021-09-13:
> How about an iiab-diagnostics from recent testing install?

/etc/iiab/diag/7.2_ubuntu-21_2021-09-12_23:37:01_EDT_u2110-srv is posted to:

http://sprunge.us/6aeZrn?en

If you need to log in, it's: `198-u2110-srv-MEDIUM` = `10.8.0.50`

holta, 2021-09-13:
> http://sprunge.us/6aeZrn?en
> 
> If you need to log in, it's: `198-u2110-srv-MEDIUM`

Just reminder that Ubuntu 21.10 pre-releases still require 2 manual steps when install installing IIAB (PHP 8.0 and Node.js 12) as explained in Section 2. hereunder:

https://github.com/iiab/iiab/wiki/IIAB-Platforms#operating-systems

These 2 manual steps will go away within the coming month (on or before 2021-10-14).

holta, 2021-10-16:
@deldesir & all: Node 16.x now auto-installs on Ubuntu 21.10 &mdash; thanks for pushing ahead with all 4 testing scenarios when you can!

Thanks to:

- https://deb.nodesource.com/node_16.x/dists/impish/
- https://github.com/nodesource/distributions/issues/1280

holta, 2021-10-16:
@georgejhunt do you recognize this JupyterHub error on Ubuntu Server 21.10 ?

> TASK [jupyterhub : pip install 7 packages into virtual environment: /opt/iiab/jupyterhub (~229 MB)] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": ["/usr/bin/python3", "-m", "venv", "/opt/iiab/jupyterhub", "/opt/iiab/jupyterhub"], "msg": "\n:stderr: Error: Command '['/opt/iiab/jupyterhub/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']' returned non-zero exit status 1.\n"}

The machine (VM) is 10.8.0.34 == 196-u2110-srv-jupyterhub with [sudo iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md) output here: 

http://sprunge.us/DlWJOy?en

holta, 2021-10-16:
@georgejhunt can you help us explain why `./runrole jupyterhub` ran successfully the 2nd time around?  (But failed the 1st time, as seen in the Ansible failure 10 min ago, just above.)  It's almost as if this is becoming an ugly pattern (run everything twice!) as `./runrole lokole` certainly also needs to be run twice, as documented here:

- #2833

A few more details here, just in case this sheds light:

```
root@u2110-srv:~# node -v
v16.11.1

root@u2110-srv:~# /opt/iiab/jupyterhub/bin/python3 -Im ensurepip --upgrade --default-pip
Looking in links: /tmp/tmpkftn_60g
Requirement already satisfied: setuptools in /opt/iiab/jupyterhub/lib/python3.9/site-packages (44.1.1)
Processing /tmp/tmpkftn_60g/setuptools-44.1.1-py2.py3-none-any.whl
Requirement already satisfied: pip in /opt/iiab/jupyterhub/lib/python3.9/site-packages (20.3.4)
Processing /tmp/tmpkftn_60g/pip-20.3.4-py2.py3-none-any.whl
Requirement already satisfied: pkg_resources in /opt/iiab/jupyterhub/lib/python3.9/site-packages (0.0.0)
Processing /tmp/tmpkftn_60g/pkg_resources-0.0.0-py2.py3-none-any.whl
```

holta, 2021-10-16:
ASIDE: I personally cannot recommend Ubuntu Desktop 21.10 on RPi 4 for schools.  As is its graphical desktop freezes completely, with the freshly-installed OS and nothing else installed on it &mdash; freezing regularly, about every hour or so &mdash; whether the machine is being actively used or not :/

PS I'm able to unfreeze the graphical desktop by running `systemctl restart display-manager.service` which may be fine for hackers &mdash; but is completely impractical for teachers who need to teach (and NOT spend their lives fighting the technology ;-)

jvonau, 2021-10-16:
e88c5df3df59f6da0ddf02cc69d73fb69b57e936 introduced a regression by not installing the hard dependence python3-setuptools found in [ansible.core.pip](https://docs.ansible.com/ansible/latest/collections/ansible/builtin/pip_module.html#notes), this apt package may not be dragged in by merely installing pyhon3-pip on Ubuntu.

holta, 2021-10-16:
3. The [Ansible pip failure from 8h ago](https://github.com/iiab/iiab/issues/2818#issuecomment-944843848) with JupyterHub was also not repeatable on Ubuntu **Desktop** 21.10 (VM), as confirmed here: http://sprunge.us/ASlf4F?en

   So if it was an obscure timing/dependency issue of some kind that occurred 9h ago, possibly arising from my use of [runrole](https://github.com/iiab/iiab/blob/master/runrole) during the 1st install (that failed) I guess we just keep an eye on it.

   In any case, re-running `sudo iiab` worked to resolve it then (9h ago) just in case this glitch recurs for others in future.

holta, 2021-10-16:
> https://docs.ansible.com/ansible/latest/collections/ansible/builtin/pip_module.html#notes

:point_up:

@deldesir the above link (that @jvonau sent overnight) is incredibly useful.

Just in case python / pip dependency issues (inevitably!) crop up in future...

deldesir, 2021-10-16:
I found no python dependency or other issue for BIG size IIAB install regarding ansible-core 2.12.0b2 on 4 different VMs and RPI's.

holta, 2021-11-10:
Looks good enough for now &mdash; all 4 BIG-sized tests were re-run in recent days: Ubuntu 21.10 (RPi+PC) x (Server+Desktop).

Thanks to @icebox827 and thanks to everybody who made this possible.

We can re-open this ticket later if absolutely necessary, if/as small issues flare up en route to [Ubuntu "Jammy Jellyfish" 22.04 LTS](https://linuxconfig.org/ubuntu-22-04-features-and-release-date).

closed 2021-11-10 by holta