iiab/iiab #2820: Test Raspberry Pi OS 11 on its release. Does RPi WiFi firmware 7.45.18.0 work well?

holta, 2021-06-19:
Officially this is being released in the coming 10 days.  But it might possibly be delayed a bit in conjunction with the underlying OS (Debian 11, Bullseye) appearing a bit later than usual in its release cycle, right now tentatively scheduled for 2021-07-31.

FYI Raspberry Pi OS is also known as RaspiOS.  And formerly known as Raspbian.

Refs:

- #2749 "Testing IIAB on the new Debian 11 "Bullseye" Release Candidate(s) [and Haiti content discussion]"
- #2818 "Test IIAB on Ubuntu 21.10 pre-releases (RPi+PC) x (Server+Desktop). Does RPi WiFi firmware 7.45.18.0 work well?"
- PR #2819 "Raspberry Pi OS 11 is imminent, expected before Debian 11's own ETA "2021-07-31"

holta, 2021-07-10:
Related:

- https://github.com/moodlebox/moodlebox/issues/210

martignoni, 2021-07-24:
FYI, Bullseye's release is [officialy scheduled for 2021-08-14](https://release.debian.org/).

holta, 2021-09-08:
Quick notes re: testing IIAB on pre-release RaspiOS Bullseye &mdash; @jvonau asked for the pastebins below:

- For now, `pip3 install ansible-core` fails with gcc compile problems.  Temporary hack / workaround is to first run `apt install ansible` (apparently to put cryptography stuff in place).
  - http://sprunge.us/X2I0vI?en
  - http://sprunge.us/8CYAYY?en

- JupyterHub won't install, with similar looking pip compile problem: http://sprunge.us/jRGxrR?en

- Calibre-Web fails to install likewise due to some pip3 issue(s): http://sprunge.us/aZWlGk?en

jvonau, 2021-09-09:
apt [history](http://sprunge.us/7BHyxW)

jvonau, 2021-09-09:
```
Start-Date: 2021-09-07  20:03:56
Commandline: /usr/bin/apt -y install python3-pip
Install: libpython3-dev:armhf (3.9.2-3, automatic), python3-lib2to3:armhf (3.9.2-1, automatic), python-pip-whl:armhf (20.3.4-4, automatic), python3-wheel:armhf (0.34.2-1, automatic), python3-dev:armhf (3.9.2-3, automatic), python3-pip:armhf (20.3.4-4), python3.9-dev:armhf (3.9.2-1+rpi1, automatic), libpython3.9-dev:armhf (3.9.2-1+rpi1, automatic), python3-distutils:armhf (3.9.2-1, automatic), libjs-jquery:armhf (3.5.1+dfsg+~3.5.5-7, automatic), libexpat1-dev:armhf (2.2.10-2, automatic), python3-setuptools:armhf (52.0.0-4, automatic), libjs-sphinxdoc:armhf (3.4.3-2, automatic), javascript-common:armhf (11+nmu1, automatic), libjs-underscore:armhf (1.9.1~dfsg-3, automatic)
End-Date: 2021-09-07  20:04:17

```
>pi@box:/opt/iiab/iiab $ apt list -a python-pip-whl 
Listing... Done
python-pip-whl/unstable,now 20.3.4-4+rpt1 all [installed,automatic]
python-pip-whl/stable 20.3.4-4 all

> apt changelog python-pip-whl 
python-pip (20.3.4-4+rpt1) bullseye; urgency=medium
  /* python-pip-whl - add pip.conf to use piwheels
 -- Serge Schneider <serge@raspberrypi.com>  Fri, 16 Jul 2021 18:04:04 +0100
python-pip (20.3.4-4) unstable; urgency=medium
  /* No-change upload against distlib 0.3.2+really+0.3.1-0.1.
 -- Stefano Rivera <stefanor@debian.org>  Thu, 01 Jul 2021 16:44:29 -0400
python-pip (20.3.4-3) unstable; urgency=medium
  /* Modify hands-off-system-packages.patch to act correctly under PyPy3, which
    shares dist-packages with cPython, but has a different sys.prefix.
 -- Stefano Rivera <stefanor@debian.org>  Mon, 28 Jun 2021 12:20:17 -0400

holta, 2021-09-09:
@jvonau solved this.  It was my mistake for not patching _both_ these files from `buster` to `bullseye` :

- /etc/apt/sources.list
- /etc/apt/sources.list.d/raspi.list

All IIAB roles seem to work perfectly after that, with what will soon be Raspberry Pi OS 11 (Bullseye).

holta, 2021-10-08:
FYI many of us noticed when Raspberry Pi OS 11 (Bullseye) broke dhcpcd5 a bit more than 10 days ago:
https://forums.raspberrypi.com/viewtopic.php?t=317888&start=25#p1918475

Apparently it's since been fixed?  (But I personally have not yet tried the fix.)

Tangentially Related:

- #2978

holta, 2021-10-18:
@deldesir if there is time, please also look into the 64-bit version(s) of Raspberry Pi OS, if that too is release in coming weeks!

https://downloads.raspberrypi.org/raspios_arm64/images/

And in general, let's take advantage of these 2 commands that can be very handy, thanks to @jvonau:

```
dpkg --print-architecture             # RaspiOS-on-PC shows: i386
dpkg --print-foreign-architectures    # RaspiOS-on-PC shows: amd64
```

jvonau, 2021-10-29:
While the world waits for that to happen I have prefabricated bullseye RaspiOS images build with [pi-gen](https://github.com/jvonau/pi-gen/tree/iiab-builder) available but need a place to host them.

holta, 2021-10-29:
> While the world waits for that to happen I have prefabricated bullseye RaspiOS images build with [pi-gen](https://github.com/jvonau/pi-gen/tree/iiab-builder) available but need a place to host them.

What are their sizes?

jvonau, 2021-10-29:
```
-rw-r--r-- 1 pi pi 752M Oct 28 03:35 image_2021-10-28-IIAB-base.zip
-rw-r--r-- 1 pi pi 3.8G Oct 28 04:56 image_2021-10-28-IIAB-full.zip
-rw-r--r-- 1 pi pi 465M Oct 28 03:20 image_2021-10-28-IIAB-lite.zip
-rw-r--r-- 1 pi pi 1.3G Oct 28 03:59 image_2021-10-28-IIAB-min.zip
```
lite - just RaspiOS  stage2
base - iiab with no options and admin-console installed  stage2iiab
min - iiab with min and admin-console installed  stage_iiab3
full - as above with full desktop  stage5

jvonau, 2021-10-29:
see 10.8.0.10/local_content/pi-gen/deploy.hold/

holta, 2021-10-30:
FYI when unzipping, burning then booting either of these two...

- http://10.8.0.10/local_content/pi-gen/deploy.hold/image_2021-10-28-IIAB-lite.zip
- http://10.8.0.10/local_content/pi-gen/deploy/image_2021-10-30-IIAB-lite.zip

...I get the following dhcpcd5 problem:

[image]

jvonau, 2021-10-30:
Have you tried toggling the setting in rasp-config as per https://forums.raspberrypi.com/viewtopic.php?t=317888&start=25#p1919103?

jvonau, 2021-10-30:
I see what I did, just dropped the -q, the line should of been changed to:
ExecStart=/usr/sbin/dhcpcd -w

Spinning up new image.

jvonau, 2021-10-30:
Quick fix would be to edit /etc/systemd/system/dhcpcd.service.d/wait.conf with the above change and reboot.

jvonau, 2021-11-01:
The networking glitch fixed in git with https://github.com/jvonau/pi-gen/commit/f041123d2c5b8329c546c1b27a74020c2c8b2b3d on the iiab-builder branch

The other noted issue was not resizing on first boot, fixed in git with https://github.com/jvonau/pi-gen/commit/edd6890703bb5fb278d797b772aa112abf96cdd8 on the iiab-builder branch. Back story is upstream came up with a new workflow to use less space and speed up the build process, but took a page out of IIAB's playbook making the new option the default for something that is not ready for prime time.

jvonau, 2021-11-02:
https://github.com/RPi-Distro/pi-gen/issues/557

jvonau, 2021-11-09:
https://github.com/RPi-Distro/pi-gen/commit/68fee340f95eb630a706e2dbf75615c6be4e980b

holta, 2021-11-10:
Thanks to @icebox827 who helped spot test (basic functionality testing) a BIG-sized IIAB install on the new 64-bit Raspberry Pi OS Lite (Bullseye) on RPi 4.

This "2021-11-08" (originally from 2021-10-30) OS was downloaded from: https://downloads.raspberrypi.org/raspios_lite_arm64/images/

Its internal WiFi hotspot (with firmware 7.45.18.0 applied by IIAB) appears to work fine.

ASIDE: 64-bit Raspberry Pi OS is _NOT_ mainline.  But it's good to know it appears to work increasingly well, alongside its mainline/official 32-bit OS, as much discussed here: https://www.raspberrypi.com/news/raspberry-pi-os-debian-bullseye/#comment-1565609

jvonau, 2021-11-11:
> Thanks to @icebox827 who helped spot test (basic functionality testing) a BIG-sized IIAB install on the new 64-bit Raspberry Pi OS Lite (Bullseye) on RPi 4.
> 
> This "2021-11-08" OS (originally from 2021-10-30) was downloaded from: https://downloads.raspberrypi.org/raspios_lite_arm64/images/
> 

Data point to check would be if the mongodb apt repos were used or the older version tar file. 
> Its internal WiFi hotspot (with firmware 7.45.18.0 applied by IIAB) appears to work fine.
>
> ASIDE: 64-bit Raspberry Pi OS is _NOT_ mainline. But it's good to know it appears to work increasingly [...]

Docker is easier on arm64.

holta, 2021-11-11:
> > This "2021-11-08" OS (originally from 2021-10-30) was downloaded from: https://downloads.raspberrypi.org/raspios_lite_arm64/images/
> 
> Data point to check would be if the mongodb apt repos were used or the older version tar file.

```
root@box:~# apt list "*mongo*" | grep installed

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

mongodb-database-tools/bionic,now 100.5.1 arm64 [installed,automatic]
mongodb-org-database-tools-extra/bionic,now 4.4.10 arm64 [installed,automatic]
mongodb-org-mongos/bionic,now 4.4.10 arm64 [installed,automatic]
mongodb-org-server/bionic,now 4.4.10 arm64 [installed]
mongodb-org-shell/bionic,now 4.4.10 arm64 [installed,automatic]
mongodb-org-tools/bionic,now 4.4.10 arm64 [installed,automatic]
mongodb-org/bionic,now 4.4.10 arm64 [installed]

root@box:~# grep -rin mongo /etc/apt
grep: /etc/apt/trusted.gpg~: binary file matches
grep: /etc/apt/trusted.gpg: binary file matches
/etc/apt/sources.list.d/mongodb-org.list:1:deb [ arch=amd64,arm64 ] https://repo.mongodb.org/apt/ubuntu bionic/mongodb-org/4.4 multiverse
```

I'm assuming the above `bionic` should be changed to `focal` on Lines 85 & 87 of https://github.com/iiab/iiab/blob/master/roles/mongodb/tasks/install.yml#L84-L90 ?

(To work with... https://repo.mongodb.org/apt/ubuntu/dists/focal/ )

jvonau, 2021-11-11:
As release guru it's up to you.

jvonau, 2022-04-05:
status?

holta, 2022-04-05:
Good enough!

Related:

- PR #3103

closed 2022-04-05 by holta