iiab/iiab #2827: Sugarizer reinstall fan fail: possibly automate the workaround in future?

holta, 2021-06-24:
Example on Ubuntu Desktop 21.04 on RPi 4, when running `./runrole --reinstall sugarizer` :

> TASK [sugarizer : Run 'npm install --allow-root --unsafe-perm=true' to create /opt/iiab/sugarizer-server/node_modules (CAN TAKE ~10 MINUTES)] ***
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": ["npm", "install", "--allow-root", "--unsafe-perm=true"], "delta": "0:00:34.562661", "end": "2021-06-24 07:47:28.104804", "msg": "non-zero return code", "rc": 217, "start": "2021-06-24 07:46:53.542143", "stderr": "npm WARN old lockfile \nnpm WARN old lockfile The package-lock.json file was created with an old version of npm,\nnpm WARN old lockfile so supplemental metadata must be fetched from the registry.\nnpm WARN old lockfile \nnpm WARN old lockfile This is a one-time fix-up, please be patient...\nnpm WARN old lockfile \nnpm ERR! code ENOTEMPTY\nnpm ERR! syscall rename\nnpm ERR! path /opt/iiab/sugarizer-server-1.3.0/node_modules/gifsicle\nnpm ERR! dest /opt/iiab/sugarizer-server-1.3.0/node_modules/.gifsicle-ilfjZBjt\nnpm ERR! errno -39\nnpm ERR! ENOTEMPTY: directory not empty, rename '/opt/iiab/sugarizer-server-1.3.0/node_modules/gifsicle' -> '/opt/iiab/sugarizer-server-1.3.0/node_modules/.gifsicle-ilfjZBjt'\n\nnpm ERR! A complete log of this run can be found in:\nnpm ERR!     /root/.npm/_logs/2021-06-24T11_47_27_991Z-debug.log", "stderr_lines": ["npm WARN old lockfile ", "npm WARN old lockfile The package-lock.json file was created with an old version of npm,", "npm WARN old lockfile so supplemental metadata must be fetched from the registry.", "npm WARN old lockfile ", "npm WARN old lockfile This is a one-time fix-up, please be patient...", "npm WARN old lockfile ", "npm ERR! code ENOTEMPTY", "npm ERR! syscall rename", "npm ERR! path /opt/iiab/sugarizer-server-1.3.0/node_modules/gifsicle", "npm ERR! dest /opt/iiab/sugarizer-server-1.3.0/node_modules/.gifsicle-ilfjZBjt", "npm ERR! errno -39", "npm ERR! ENOTEMPTY: directory not empty, rename '/opt/iiab/sugarizer-server-1.3.0/node_modules/gifsicle' -> '/opt/iiab/sugarizer-server-1.3.0/node_modules/.gifsicle-ilfjZBjt'", "", "npm ERR! A complete log of this run can be found in:", "npm ERR!     /root/.npm/_logs/2021-06-24T11_47_27_991Z-debug.log"], "stdout": "", "stdout_lines": []}
>
> PLAY RECAP *********************************************************************
> 127.0.0.1                  : ok=101  changed=17   unreachable=0    failed=1    skipped=42   rescued=0    ignored=0

However this does work:

```
rm -rf /opt/iiab/sugarizer-server/node_modules    # This is a symlink to /opt/iiab/sugarizer-server-1.3.0/node_modules
./runrole --reinstall sugarizer
```

jvonau, 2021-06-24:
Your refusal to implement any form of versioning makes the situation problematic to move forward with a solution.

holta, 2021-07-21:
Very rare & isolated install failure.

closed 2021-07-21 by holta