iiab/iiab #2828: Is hotspot failing when disconnecting from Internet ?

holta, 2021-06-25:
TK Kang says this is a bug, e.g. with 32-bit Raspberry Pi OS Lite on RPi 4 here:

http://sprunge.us/vYPEPZ?en

TK writes:

> if I do not have an internet upstream (e.g. my mobile hotspot) for the IIAB wifi to connect then the IIAB "Internet in a Box" hotspot will not appear or work.

As an aside Terry Gillette comments that with his similar Digital Library (DL) product:

> we go to some trouble to check if upstream wifi link is working If not, we restart wifi without that interface enabled.

holta, 2021-06-25:
@jvonau has asked TK to try this `host_country_code:` correctly in his [/etc/iiab/local_vars.yml](http://wiki.laptop.org/go/IIAB/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it.3F) :

```
=IIAB==========================================================================
-rw-r--r-- 1 root root 265 Jun 25 02:07 /etc/hostapd/hostapd.conf

# Basic configuration

interface=ap0
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0

ssid=Internet in a Box
channel=6 
bridge=br0

# Hardware configuration
driver=nl80211
country_code=US
# limit emissions to what is legal in country
ieee80211d=1
hw_mode=g

=IIAB==========================================================================
-rw-r--r-- 1 root root 133 Jun 25 02:06 /etc/wpa_supplicant/wpa_supplicant.conf

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=AU

network={
	ssid="Tyne_Network"
	psk=[REDACTED]
}
```

> country_code=US
country=AU
>
> Please set host_country_code: in local_vars to the correct code for your location.

holta, 2021-06-25:
If the above change to `host_country_code:` works, documentation will need to be improved so that many more people realize how very important this is.

jvonau, 2021-06-26:
Documentation should be improved anyway, setting the country code should be the first item customized in the local_vars file to match the local running environment.

jvonau, 2021-06-26:
>
322 ? 00:00:30 wicd 
591 ? 00:00:02 wicd-monitor

This a network management daemon, any idea if this is now part of the stock raspiOS install or has been added manually?

jvonau, 2021-06-26:
>Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage2

>x2gocleansessio

I'm going to go with a do-it-yourself desktop install.... I'm just going to file this under unsupported, not the first time I've said wicd is not supported by me. https://github.com/iiab/iiab/issues/2574

holta, 2021-06-26:
Huge progress overnight thanks to TK and @jvonau:

http://sprunge.us/mcdOoH?en

> What the main issue is the firmware doesn't really play nice when upstream wifi is set to connect but is not present. The one radio is occupied with the scanning for the ssid and keeps scanning every 20-30 seconds, this would cycle the availability of ap0 as scanning results in ap0 being set 'down' by a bug in the firmware. There is code in place to work around most issues but I saw no point in adding more code when running in that configuration would result in a horrible user experience anyway. In short, after installation remove the network entries in wpa_suppliciant.conf if you want to run without upstream wifi active and have the hotspot active, don't leave the entries present when the network is unavailable.

> Don't remove the wpa_suppliciant.conf file [instead] remove the 'network={}' entries within wpa_suppliciant.conf, the country_code is required to be present.

These commands might also be useful, in related situations:

- `sudo journalctl -b 0 --no-pager -u wifi-test.service`
- `sudo journalctl -b 0 --no-pager -u dhcpcd`

jvonau, 2021-06-26:
>2339 Jun 26 08:42:25 box dnsmasq[832]: started, version 2.80 cachesize 150
2340 Jun 26 08:42:25 box dnsmasq[832]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
2341 Jun 26 08:42:25 box dnsmasq-dhcp[832]: DHCP, IP range 172.18.100.1 -- 172.18.126.254, lease time 1h
2342 Jun 26 08:42:25 box dnsmasq-dhcp[832]: DHCP, sockets bound exclusively to interface br0
2343 Jun 26 08:42:25 box dnsmasq[832]: using local addresses only for domain lan
2344 Jun 26 08:42:25 box dnsmasq[832]: no servers found in /run/dnsmasq/resolv.conf, will retry
2345 Jun 26 08:42:25 box dnsmasq[832]: read /etc/hosts.dnsmasq - 1 addresses
2346 Jun 26 08:42:26 box dhcpcd-run-hooks[1002]: wlan0: 50-iiab CARRIER change wlan0
2347 Jun 26 08:42:26 box dhcpcd-run-hooks[1007]: wlan0: 50-iiab set channel 6
2348 Jun 26 08:42:26 box dhcpcd-run-hooks[1014]: wlan0: 50-iiab set ap0 up CARRIER
2349 Jun 26 08:42:32 box dnsmasq[832]: reading /run/dnsmasq/resolv.conf
2350 Jun 26 08:42:32 box dnsmasq[832]: using local addresses only for domain lan
2351 Jun 26 08:42:32 box dnsmasq[832]: using nameserver 192.168.0.1#53
2352 Jun 26 08:42:32 box dhcpcd-run-hooks[1108]: wlan0: 50-iiab set ap0 up BOUND

Boot looks fine 
>2353 Jun 26 08:48:09 box hostapd: ap0: STA cc:af:78:6d:7f:97 IEEE 802.11: associated
2354 Jun 26 08:48:09 box hostapd: ap0: STA cc:af:78:6d:7f:97 RADIUS: starting accounting session 1FCFEC7FFEC9C9F8

Client starts to connect
>2355 Jun 26 08:48:09 box /etc/mysql/debian-start[1152]: Upgrading MySQL tables if necessary.

Guess boot is not finished yet
>2356 Jun 26 08:48:09 box dnsmasq-dhcp[832]: DHCPREQUEST(br0) 172.18.118.35 cc:af:78:6d:7f:97 
2357 Jun 26 08:48:09 box dnsmasq-dhcp[832]: DHCPACK(br0) 172.18.118.35 cc:af:78:6d:7f:97 admintkkang-ACR

dnsmasq ack'd the request.

>2358 Jun 26 08:48:09 box /etc/mysql/debian-start[1155]: /usr/bin/mysql_upgrade: the '--basedir' option is always ignored
2359 Jun 26 08:48:10 box /etc/mysql/debian-start[1155]: Looking for 'mysql' as: /usr/bin/mysql
2360 Jun 26 08:48:10 box /etc/mysql/debian-start[1155]: Looking for 'mysqlcheck' as: /usr/bin/mysqlcheck
2361 Jun 26 08:48:10 box /etc/mysql/debian-start[1155]: This installation of MySQL is already upgraded to 10.3.27-MariaDB, use --force if you still need to run mysql_upgrade
2362 Jun 26 08:48:10 box /etc/mysql/debian-start[1163]: Checking for insecure root accounts.
2363 Jun 26 08:48:10 box /etc/mysql/debian-start[1167]: Triggering myisam-recover for all MyISAM tables and aria-recover for all Aria tables

Continues to load
>364 Jun 26 08:48:13 box vmunix: [   53.913487] Under-voltage detected! (0x00050005)

Not good
>2365 Jun 26 08:48:13 box dnsmasq-dhcp[832]: DHCPINFORM(br0) 172.18.118.35 cc:af:78:6d:7f:97 
2366 Jun 26 08:48:13 box dnsmasq-dhcp[832]: DHCPACK(br0) 172.18.118.35 cc:af:78:6d:7f:97 admintkkang-ACR

Dropped packets 
>2367 Jun 26 08:48:17 box vmunix: [   58.072875] Voltage normalised (0x00000000)

Junk power supply messing with the wifi.
>2368 Jun 26 08:49:31 box dnsmasq-dhcp[832]: DHCPINFORM(br0) 172.18.118.35 cc:af:78:6d:7f:97 
2369 Jun 26 08:49:31 box dnsmasq-dhcp[832]: DHCPACK(br0) 172.18.118.35 cc:af:78:6d:7f:97 admintkkang-ACR

Dropped packets retried.

Unless somebody can duplicate TK report, I'm don't going to spend any more time looks at this.

holta, 2021-08-11:
Let's retest things like this, as soon as Raspberry Pi OS 11"Bullseye" is released in coming days:

- #2820

holta, 2021-08-15:
TK and @jvonau: Are these below at all relevant?

- https://github.com/raspberrypi/documentation/issues/1806
- Raspberry Pi 3 B+ hostapd issues: https://www.raspberrypi.org/forums/viewtopic.php?&t=209818

jvonau, 2021-08-17:
How can that be relevant 3 years later, that is applicable in the single interface pre-ap0 days and we now have working dual mode up|down wifi now.

holta, 2021-08-30:
No bug here (I hope?) but this is an important reminder to improve `host_country_code:` documentation shortly.

And to reach out to TK in general!

jvonau, 2022-04-05:
status?

holta, 2022-04-05:
I have not spoken with TK unfortunately.

This needs to happen.

Tangentially related:

- #3097

holta, 2022-05-23:
Docs & techniques have improved substantially over the past month:

- PR #3179
- PR #3222
- https://wiki.iiab.io/go/FAQ#Can_I_create_a_Wi-Fi_hotspot_using_an_old_laptop%3F

Possibly close this ticket soon?!

Ref:

- #1520

holta, 2022-07-20:
If ongoing issues arise, please shout.

closed 2022-07-20 by holta