iiab/iiab #2829: Moodle failed to install repeatedly: possibly automate the workaround in future?

holta, 2021-06-25:
This error arose repeatedly:

> TASK [moodle : Clone (i.e. use git to download) https://github.com/moodle/moodle.git to /opt/iiab/moodle (~288 MB)] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Failed to set a new url https://github.com/moodle/moodle.git for origin:  fatal: not a git repository (or any of the parent directories): .git\n"}

Despite `/opt/iiab/moodle` appearing to be a valid though minimal git directory, very oddly.

In any case, this workaround gets past the repeated error, allowing IIAB's install to continue:

```
sudo rm -rf /opt/iiab/moodle
sudo iiab
```

Ref: #2827

jvonau, 2021-06-26:
Question is why is 'iiab-install' trying to perform a reinstall if /opt/iiab/moodle exists, there should be an iiab_state file entry for moodle unless there was some tinkering going on or did the first install crash during moodle's install?

Better question is why does sugarizer clone successfully a second time while moodle doesn't, 'force: yes' is used in both but moodle is using 'version:' which may not match upsteam's naming.

holta, 2021-06-26:
The 1st install crashed.

Preventing repeated attempts at `sudo iiab` from continuing.

Very unusual.

holta, 2021-07-21:
Very rare & isolated install failure.

closed 2021-07-21 by holta