iiab/iiab #2831: Do Admin Console logins fail in general with the 'Debian Edu' distro?

holta, 2021-06-27:
@shantibhardwa is unable to login to IIAB's Admin Console here, even after we verified that the username (iiab-admin) is part of group iiab-admin and the correct password is being typed in:

http://sprunge.us/ath2Ad?en

Very strange, as Admin Console logins works on all similar Linux OS's to my knowledge:
- Hypothetically Admin Console may be using a very different technique to verify username/password ?
- Is this password-checking routine somehow broken on this pre-release build of Debian Edu 11 / Bullseye ?  Or within 'Debian Edu' in general?

A bit more about 'Debian Edu' in general: https://blends.debian.org/edu/

jvonau, 2021-06-27:
'Debian Edu' is a LTSP [Linux Terminal Server Project](https://wiki.debian.org/DebianEdu/BeforeGettingStarted) based solution using [LDAP](https://salsa.debian.org/debian-edu/debian-edu-install/-/blob/master/preseed-values/defaults.common) (not the local machine) authentication.

holta, 2021-06-27:
@jvonau has some promising suggestions to try to figure out why this is happening: (if @shanti-bhardwa's machine hopefully comes back online?)

- Seems like [this Debian Edu] machine was not fully setup before the iiab install was started [?]
  - Would have to join an LDAP server or run it's own
  - It's not looking at  /etc/passwd [and /etc/shadow] for the users
  - Should be transparent once LDAP is setup [File a bug upstream?]
- The pam stack could use pam_ldap, that should be transparent to the users of the stack
- If the machine comes back up check /etc/nsswitch.conf

holta, 2021-06-27:
@shanti-bhardwa do you have time to try installing Debian 10's 'Edu' https://get.debian.org/cdimage/release/current/amd64/iso-cd/debian-edu-10.10.0-amd64-netinst.iso (406MB released 2021-06-19) to see if Admin Console logins also fail?

shanti-bhardwa, 2021-06-28:
yes /etc/nsswitch.conf exists and the contents are

cat /etc/nsswitch.conf
# /etc/nsswitch.conf
#
# Example configuration of GNU Name Service Switch functionality.
# If you have the `glibc-doc-reference' and `info' packages installed, try:
# `info libc "Name Service Switch"' for information about this file.

passwd:         files systemd
group:          files systemd
shadow:         files
gshadow:        files

hosts:          files mdns4_minimal [NOTFOUND=return] dns myhostname
networks:       files

protocols:      db files
services:       db files
ethers:         db files
rpc:            db files

netgroup:       nis

shanti-bhardwa, 2021-06-28:
reading through this document, it seems the build needs lot of work and user & password management is not straight forward.
https://jenkins.debian.net/userContent/debian-edu-doc/debian-edu-doc-en/debian-edu-bullseye-manual.pdf
see 6.3.5 and 7.3
if we can get IIAB on the main server, then it is available to the whole school.

jvonau, 2021-06-28:
Given IIAB was installed successfully there might be hope but I'm pretty sure the install would of wiped out access to the builtin web-tools (https://www/gosa) offered by the distro noted in the above link. I'm not inclined to spend any 'free time' to address the issues arising from trying to install on Debian Edu but will entertain offers should somebody wants to fund that project.

holta, 2021-06-28:
Certainly we can mark this as "WONTFIX" if it's too hard to quickly fix.

Still: @shanti-bhardwa I'm still extremely interested in knowing if the Debian 10 version of "Debian Edu" has the same problem i.e. will its installer lead many other people to fall in this same trap?

Also can you confirm with a photo of Debian Edu's installer(s) whether this might be a "usability bug" &mdash; that leaves people (and Python3 !) confused about how to verify passwords using LDAP/alternatives/etc ?

holta, 2021-06-28:
Recap: whose bug this is of course unclear!

But it's certainly a "usability bug" from the perspective of the Debian Edu user community, that we should try to become good neighbors with if possible over coming years.

Whether of not the Norwegian govt (and others!) are giving this initiative the financial support it deserves.

jvonau, 2021-06-28:
> Recap: whose bug this is, is of course unclear!
> 
> But it's certainly a "usability bug" from the perspective of the Debian Edu user community, that we should try to become good neighbors with, if possible over coming years.
> 
> Whether of not the Norwegian govt (and others!) are continuing to give this initiative the financial support it deserves.

You have it backwards, IIAB is a guest on that platform, IIAB must bend not the OS. 

- First off I would disable iiab-admin creation via local_vars.yml, meaning populate a custom file before starting just to simplify things, after it's optional. 
- Second, given Edu's use of ssl out of the box suggests that messing with that combo will break things when IIAB blindly stomps on some config files somewhere. What is needed to know what web backend is serving up stuff... 
- Until that is all sorted out I would suggest to having nginx's config files stored elsewhere, redefine nginx_conf_dir
- supply a custom systemd unit file to start nginx service referencing the new config directory
- add a port directive to nginx to start the service on a different port. (like how apache is rigged now)
- observe, tinker Edu's config to redirect everything but it's own uri to the nginx instance running on the alternate port.

shanti-bhardwa, 2021-06-29:
OK given up on Edu build and had a go using debian-bullseye-DI-rc2-amd6debian-bullseye-DI-rc2-amd64-netinst.iso downloaded from https://cdimage.debian.org/cdimage/bullseye_di_rc2/amd64/iso-cd/.

The build does not have the wifi drivers for wifi interface which I had to install later on. WIFI was set up correctly then. iia 7.2 installed OK, created iiab-admin user with default password and it works from desktop. But still cannot login to Admin Console with same error. hostapd.service does not start. See here for diagnostics
 http://sprunge.us/20EHAt

and VPN is up here 10.8.0.74 (shanti-deb11-29june)

 

VPN is up  on

jvonau, 2021-06-29:
>host_channel = 11
client_wifi_channel = 11
channel 36 (5180 MHz), width: 40 MHz, center1: 5190 MHz

From a glance at the dump when the install was performed the wifi was on channel 11 and hostapd.conf was written same channel=11 now the wifi is on channel 36, we don't handle hotspots on 5GHz. Doesn't explain why ap0 disappeared, or maybe does. I'll log in and have peek.

shanti-bhardwa, 2021-07-01:
I did another build, this time using debian 10. On this build, Admin Console works fine. The install script gives you the option of changing the default password. See Diagnostics here: http://sprunge.us/xCu216 
The main difference I found that with similar debian 11 build ( see http://sprunge.us/20EHAt) I did NOT get the prompt for changing the default password during one line install.

I found this, note which seems to suggest that the way passwords are encrypted in Debian 11 is different from Debian 10.
See here:https://www.debian.org/releases/bullseye/arm64/release-notes/ch-information.en.html#security-archive 

 https://www.debian.org/releases/bullseye/arm64/release-notes/ch-information.en.html#security-archive

On WIFI issue, there is mismatch on the channel used by wifi interface and the configuration used for hostapd. Hostapd fails to start because it compalins about ap0 not starting. See the log in diagnostics file.

jvonau, 2021-07-01:
> On WIFI issue, there is mismatch on the channel used by wifi interface and the configuration used for hostapd. Hostapd fails to start because it compalins about ap0 not starting. See the log in diagnostics file.

host_channel = 6
client_wifi_channel = 36

https://github.com/iiab/iiab/wiki/IIAB-7.2-Release-Notes#known-issues
The consensus is/was not to offer a 5GHz hotspot, one of the reasons is older rpi's don't have the support for that band's use.

shanti-bhardwa, 2021-07-01:
I did this change on Deb11 machine as suggested here
https://www.debian.org/releases/bullseye/arm64/release-notes/ch-information.en.html#security-archive

i.e modify /etc/pam.d/common-password. Find the line that looks like:

password [success=1 default=ignore] pam_unix.so obscure yescrypt
	
and replace yescrypt with sha512.

Then rebooted and changed iiab-admin passwod with

passwd iiab-admin (need su or sudo)

The logged into Admin Console and new password works!!.

I have left the machine on openvpn, 10.8.0.46 and the password is shanti01. Somebody please try and login.

I have also checked ssh iiab-admin@ip-address-of-ethernet and new password also works.

jvonau, 2021-07-01:
>iiab-admin@box:\~$ sudo systemctl status clone-wifi.service
● clone-wifi.service - IIAB ap0 clone wifi device
     Loaded: loaded (/etc/systemd/system/clone-wifi.service; enabled; vendor preset: enabled)
     Active: active (exited) since Thu 2021-07-01 21:12:31 BST; 1h 54min ago
   Main PID: 440 (code=exited, status=1/FAILURE)
      Tasks: 0 (limit: 4293)
     Memory: 0B
        CPU: 0
     CGroup: /system.slice/clone-wifi.service
Jul 01 21:12:31 box.11deb systemd[1]: Starting IIAB ap0 clone wifi device...
Jul 01 21:12:31 box.11deb iw[372]: command failed: No such file or directory (-2)
Jul 01 21:12:31 box.11deb ip[400]: Cannot find device "ap0"
Jul 01 21:12:31 box.11deb ip[440]: Cannot find device "ap0"
Jul 01 21:12:31 box.11deb systemd[1]: Finished IIAB ap0 clone wifi device.

That is pretty early in the boot sequence based on the PID, like to poke further but the machine disappeared...

shanti-bhardwa, 2021-07-02:
the machine back-on on 10.8.0.50. I have switched  off 5GHz on my BT home router and the wifi interface is using 2.4Gz Channel 1. I have set channel 1 on hostapd.conf, but ap0 still does not come up.

tim-moody, 2021-11-12:
fixed in iiab/admin-console#428

closed 2021-11-12 by tim-moody