iiab/iiab #2833: Elgg, Lokole & Node-RED don't yet work on Ubuntu 20.10 ?

holta, 2021-06-29:
Not at all urgent but something to monitor and be aware of as the OS's 2021-10-14 release date approaches.

The reasons are likely entirely disconnected from PHP 8.0 &mdash; these 3 app failures are just early data points &mdash; that arose during the very extensive IIAB testing as part of PR #2832, providing an early glimpse of Ubuntu Server 21.10.

holta, 2021-06-29:
A bit more detail:

1. http://box/elgg - browser shows the following when IIAB's installed on Ubuntu 21.10: _(whereas it works on Ubuntu 21.04)_

   > HTTP ERROR 500

2. http://box/lokole - browser shows the following: _(regardless whether IIAB's installed on Ubuntu 21.04 / PHP 7.4 OR on Ubuntu 21.10 / PHP 8.0)_

   > 502 Bad Gateway
nginx/1.18.0 (Ubuntu)

3. http://box/nodered - shows the following when IIAB's installed on Ubuntu 20.10: _(whereas it works on Ubuntu 21.04)_

   > Service Unavailable
The server is temporarily unable to service your request due to maintenance downtime or capacity problems. Please try again later.
Apache/2.4.46 (Ubuntu) Server at 127.0.0.1 Port 8090

jvonau, 2021-06-29:
nginx logs? apache logs?

holta, 2021-06-29:
Apologies I typo'd the subject line (this is about Ubuntu 21.10 not 20.10).

The original VM is now lost but a new one is now building here:

198-u2110-srv-BIG-0629 = 10.8.0.26

And should hopefully complete well within 45min.  Feel free to login and reboot it as nec.

holta, 2021-06-29:
> 198-u2110-srv-BIG-0629 = 10.8.0.26

Its iiab-diagnostics are here: http://sprunge.us/q2aHVz?en

Don't hesitate to log in if you choose to.

PS all 3 page failures are re-confirmed on this new VM, per the subject line / title of this ticket (i.e. for Elgg, Lokole and Node-RED).

jvonau, 2021-09-27:
#2986 #2739

holta, 2021-09-27:
Moved to the IIAB 7.2 milestone, to help @nzola in his decision-making in coming days, regardless what is decided.

(e.g. even if Lokole cannot be made to work on Ubuntu 21.04 in a short timeframe, as @rickgurmet is hoping for!)

holta, 2021-09-28:
> Given lokole works on the primary OS that nzola has deployed, proven from #2988 with 46-rpi4-lite & 42-u20 across the vpn but what is missing is a current report from a u-21.04 install for a quick no go/go test and easy access to probe further.

@rickgurmet can we give @jvonau remote access to your Ubuntu 21.04 machine where Lokole was attempted?

Email or Text me directly, if possible, to arrange password etc?

(Or if that proves too complicated for you, I can set up an Ubuntu 21.04 VM to do similar.)

holta, 2021-09-30:
FYI @jvonau confirmed on 2 different Ubuntu Server 21.04 machines that *reinstalling* Lokole seems to work, as follows:

```
sudo systemctl stop supervisor.service
sudo rm -rf /home/lokole
cd /opt/iiab/iiab
sudo ./runrole --reinstall lokole
```

Unclear why the original Lokole install does not work &mdash; upon being installed by IIAB (http://box/lokole shows "502 Bad Gateway") &mdash; until the above 4 commands are run.

But the good news is that the above workaround would appear to be repeatable (reproducible) for those who need it.

nzola, 2021-10-05:
> FYI @jvonau confirmed on 2 different Ubuntu Server 21.04 machines that _reinstalling_ Lokole seems to work, as follows:
> 
> ```
> sudo systemctl stop supervisor.service
> sudo rm -rf /home/lokole
> cd /opt/iiab/iiab
> sudo ./runrole --reinstall lokole
> ```
> 
> Unclear why the original Lokole install does not work — upon being installed by IIAB (http://box/lokole shows "502 Bad Gateway") — until the above 4 commands are run.
> 
> But the good news is that the above workaround would appear to be repeatable (reproducible) for those who need it.

Excellent!

deldesir, 2021-10-14:
Lokole workaround works on Ubuntu server 21.10 on RPI4.

holta, 2021-11-10:
@deldesir wrote:
> Lokole workaround works on Ubuntu server 21.10 on RPI4.

I confirm that Lokole's [manual reinstall workaround](https://github.com/iiab/iiab/issues/2833#issuecomment-931559600) also works on these 2 Ubuntu 21.10 VM's (one server, one desktop) here:

- 196-rpi4-U2110-serv-BIG = http://10.8.0.74/lokole = http://sprunge.us/xmoIOs?en
- 197-rpi4-U2110-desk-BIG = http://10.8.0.30/lokole = http://sprunge.us/e8Q4Av?en

holta, 2021-11-28:
> @deldesir wrote:
> 
> > Lokole workaround works on Ubuntu server 21.10 on RPI4.
> 
> I confirm that Lokole's [manual reinstall workaround](https://github.com/iiab/iiab/issues/2833#issuecomment-931559600) also works on these 2 Ubuntu 21.10 VM's (one server, one desktop) here:
> 
> * 196-rpi4-U2110-serv-BIG = http://10.8.0.74/lokole = http://sprunge.us/xmoIOs?en
> * 197-rpi4-U2110-desk-BIG = http://10.8.0.30/lokole = http://sprunge.us/e8Q4Av?en
> 
> cc: @nzola

I've confirmed that Lokole's 4-line manual reinstall workaround also works on Ubuntu Desktop 22.04 (pre-release daily build on x86_64 VM).

jvonau, 2021-11-30:
https://github.com/iiab/iiab/pull/2843#issuecomment-951090944

holta, 2021-11-30:
Clarification of what was mentioned ~2.5 days ago &mdash; http://10.8.0.22/lokole appears to fully work after @jvonau's 2021-09-30 workaround is run: (which boils down to installing Lokole twice)

- https://github.com/iiab/iiab/issues/2833#issuecomment-981079242

holta, 2021-12-01:
Just FYI I've posted a link to the Lokole workaround to:

https://github.com/iiab/iiab/wiki/IIAB-7.2-Release-Notes#known-issues

> Lokole does not work when installed on Ubuntu 21.04 or 21.10, until you apply the [#2833 workaround](https://github.com/iiab/iiab/issues/2833#issuecomment-931559600).

jvonau, 2022-04-05:
Unsupported OS's close

holta, 2022-04-05:
> Unsupported OS's close

Agreed: while 21.10 is tolerated for a little while longer, 22.04 is now in thje spotlight.

No question you and @nzola have put a lot of work into this, and we can only do our best!

closed 2022-04-05 by holta