iiab/iiab #2835: Quick installation script fails on Rasberry Pi 1 B+ (php-stem)

Hermanio, 2021-06-29:
### Expected result
Quick install script finishes the installation successfully on the Raspberry Pi 1.

### Actual result
installation fails when linking a config file related to php-stem:
```
TASK [www_base : Symlink /etc/php/7.3/fpm/conf.d/20-stem.ini -> /etc/php/7.3/mods-available/stem.ini] ****************************************************************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "src file does not exist, use \"force=yes\" if you really want to create the link: /etc/php/7.3/mods-available/stem.ini", "path": "/etc/php/7.3/fpm/conf.d/20-stem.ini", "src": "/etc/php/7.3/mods-available/stem.ini"}
```
That file is indeed missing from the original path.

Going by past discussions that relate to this keyword, I understand that you intend to get rid of this soon, is that correct?

### Steps to reproduce the problem:

Run the installation script on the latest version of Raspberry Pi OS 32-bit: `curl d.iiab.io/install.txt | sudo bash`

### IIAB version, OS version, hardware details:

OS: latest Raspberry Pi OS 32-bit
Hardware: Raspberry Pi 1 B+ (single-core, 1GB RAM), 128GB Samsung microSD card.

IIAB diagnostics output here: http://sprunge.us/d52YZE

### Additional comments

Install script works fine on a ThinkPad X230 running Ubuntu 20.04.2 LTS desktop version so this seems to be Raspberry Pi specific.

Documentation hints at building IIAB on a more powerful Raspberry Pi model first (such as Raspberry Pi 4), and then moving the install to the slower Pi (such as the Zero W), is this recommendation made due to the slow performance or some fundamental incompatibility during the install process?

I'm just testing this out to see how well this project works so no need to treat this with any priority.

holta, 2021-06-29:
@Hermanio thanks for the detailed report — especially given the current dominance of Raspberry Pi 4 — largely due to its speed as you say.  Your report (and followup, if you have time in coming days) stands to give us a better understanding of older CPU's like the `armv6l` in your Raspberry Pi 1 (with 512 MB RAM according to Line 637 of http://sprunge.us/d52YZE?en).

Until we come up with a better answer for you, do consider looking over #2797 and PR #2799 from just a few weeks ago, which I believe is the same issue, towards supporting the WikiHow content module (likely in English or Spanish, normally available from http://oer2go.org but that site is offline for the moment).  That is in fact very popular among secondary schools.

Cross-compiling stem for more platforms might possibly be part of the answer, but in any case this would take quite some time to resolve satisfactorily for all Raspberry Pi CPU's.  A similar issue with Node.js challenges on older CPU's is outlined at the bottom of our "Known Issues" here: https://github.com/iiab/iiab/wiki/IIAB-7.2-Release-Notes#known-issues

Finally, while PR #2799 has not yet led to a clean (tolerable!) solution, it might possibly offer you enough of a workaround, to get your IIAB installation much closer to finish?  In any case, please do keep in touch!

Historical Refs: #983, PR #984, PR #2269

tim-moody, 2021-06-29:
From what I read the rpi 1 b+ uses the same chip as the rpi 0 W, so http://d.iiab.io/packages/php7.3-stem.rpi.tar should work. The problem is that this is a hardware case we have never seen and probably the identification logic has failed.

Hermanio, 2021-06-29:
> your Raspberry Pi 1 (with 512 MB RAM according to Line 637...

My bad, I must've been thinking about a Raspberry Pi 2 that I used to have at one point. 

I can probably take a look at the identification logic and work around it to see how far I can get with this on an underpowered machine. Thanks for the super quick response!

Hermanio, 2021-06-29:
It seems that the identification did mess up a bit here. 

I ran this to get the variable value on the Raspberry Pi 1:
```
root@boxrpi:~# ansible localhost -m setup -a 'filter=ansible_architecture'
[DEPRECATION WARNING]: Ansible will require Python 3.8 or newer on the controller starting with Ansible 2.12. Current version: 3.7.3 (default, Jan 22 2021, 20:04:44) [GCC 8.3.0]. This feature will be removed from ansible-core in version 
2.12. Deprecation warnings can be disabled by setting deprecation_warnings=False in ansible.cfg.
[WARNING]: No inventory was parsed, only implicit localhost is available
localhost | SUCCESS => {
    "ansible_facts": {
        "ansible_architecture": "armv6l"
    },
    "changed": false
```

`php-stem.yml` has this condition for Raspberry Pi at the moment: 
```
  when: ansible_machine == "armv7l" and stem_available is defined
```
I modified it to `when: (ansible_machine == "armv7l" or ansible_machine == "armv6l") and stem_available is defined` and will retry the setup. Going to take a while so I'll report back once I have some news.

Hermanio, 2021-06-30:
Seems like that did the trick. I ran into some networking related issues at the step where the WiFi hotspot is created, since I assume that the current steps are very specific to the integrated WiFi that other Raspberry Pi models have, while I'm using another WiFi adapter over USB. Without the WiFi adapter connected, I've got IIAB working on the Raspberry Pi and am in the process of loading some Wikipedia archives to it to see Kiwix performance. 

It's definitely running, but you need a lot of patience, that single core is very limiting. For example, I could access the admin page but the command server took a couple of minutes to catch up, after that I could log in.

As for the invitation, I don't want to give out promises that I cannot keep, but I will definitely keep an eye on the project, since it intersects with my data hoarding and archival efforts as well.

holta, 2021-07-15:
@Hermanio I'm closing this now assuming your PR #2837 patch is good enough for now!

If you have a moment, please do also help us keep #2864 on track in coming weeks for all ~600 Million Spanish speakers worldwide (:

closed 2021-07-15 by holta