iiab/iiab #2842: WordPress + IIAB : Code Review & Basic Security Audit

holta, 2021-07-06:
It's been a few years!

Whether critical upkeep happens tomorrow or later this year, we should try to make it happen this year, with the guidance of WordPress gurus like @kananigit and @ericnitschke.

e.g. things like https://github.com/iiab/iiab/blob/master/roles/wordpress/templates/wp-config.php.j2 could use fresh sanity checks (if not Spring Cleaning!) half a decade later, starting here:

https://github.com/iiab/iiab/tree/master/roles/wordpress

Also reviewing the latest community norms and requirements around `nginx_high_php_limits` here — in support of Moodle, Nextcloud, roles/pbx's FreePBX and WordPress:

https://github.com/iiab/iiab/blob/master/roles/www_options/tasks/main.yml#L100-L112

In keeping with very recent work like:

- PR #2830
- PR #2832
- PR #2836
- PR #2841

holta, 2021-07-06:
On the side, WordPress 5.8 is bringing about an unusually high number of advances &mdash; as is scheduled for release 2 weeks from today: ("2021-07-20")

https://www.wpbeginner.com/news/whats-coming-in-wordpress-5-8-features-and-screenshots/

jvonau, 2021-07-11:
https://wordpress.org/support/article/editing-wp-config-php/ might want to update to a unix socket path ie: ```define( 'DB_HOST', 'localhost:/var/run/mysqld/mysqld.sock' );```

holta, 2021-07-20:
WordPress 5.8 is released
https://wordpress.org/news/2021/07/tatum/

WordPress 5.8 Field Guide
https://make.wordpress.org/core/2021/07/03/wordpress-5-8-field-guide/
WordPress 5.8 Field Guide
https://make.wordpress.org/core/2021/02/23/wordpress-5-7-field-guide/

https://make.wordpress.org/core/2020/11/20/wordpress-5-6-field-guide/

holta, 2021-07-21:
Certainly WordPress 5.8 works with IIAB, as tested on Debian 11 VM summarized @ https://github.com/iiab/iiab/issues/2749#issuecomment-884360192

But a legit look-over and/or refactor of [roles/wordpress](https://github.com/iiab/iiab/tree/master/roles/wordpress) is still needed &mdash; when time/energy/volunteer spirit allows!