iiab/iiab #2853: Opinions about Pi 4 WiFi features

pelwell, 2021-07-09:
Hi Internet-in-a-Box community,

Please forgive the use of an issue - I wanted this message seen and the Discussions feature doesn't look well used.

We at Raspberry Pi have been given the option of a second 43455 firmware variant which is optimised for use as an AP with the emphasis being on the number of supported clients. Every bit of device RAM occupied by firmware eats into the number of clients when running as an AP, so Cypress have put together a list of features together with how many clients can be reclaimed by omitting that feature.

The good news is that the current firmware includes some features which it is not possible to make use of (this was surprising). Dropping them should gain us space for about 7 additional clients with no loss of functionality, and should be applied to all future firmwares. Dropping some advanced roaming features (802.11k, 802.11v and 802.11r) which are unlikely to help an AP should yield another 4 (total: ~11).

The dfsradar feature allows an AP to operate in channels that may be used by radar systems. Without it you would not be able to configure an AP to use one of those channels but would gain an extra 4 clients (total: ~15).

Finally, ACS (Auto Channel Support) is costing us 3 clients. Using this feature requires the configuration and cooperation of hostapd. If you as a community are happy with manual channel selection than that could increase the number of supported clients by around 18, leading to a total well in excess of 20.

I don't mind if you want to have an open discussion here or go away and reach a concensus via some other means - I suspect both may happen - but it would be great to get some feedback rather than just guessing about what is going to be most useful.

N.B. All numbers here are estimates and not guarantees, but they should be relatively close.

Thanks for your taking the time to read this, for the good work you do, and for supporting Raspberry Pi.

Phil Elwell - Raspberry Pi (Trading)

holta, 2021-07-09:
Profound thanks @pelwell for asking:

So many of the world's poorest schools ask Internet-in-a-Box to support for 32 (or more!) simultaneous WiFi client devices, as a result of the flood of cheap Android phones, that keep appearing even in the most destitute communities worldwide.

But we'll try to gather many of these thoughts & opinions together in one place over the weekend, to help shed more light around these general community/design questions, that affect so many folks at so many levels!

Relevant: we are curious if there will be a successor to the Raspberry Pi Zero W, for the most low-income communities worldwide.  Is that possible?  Or unlikely for many more years?

pelwell, 2021-07-09:
Re: a Zero W successor - I don't blame you for asking, but our product roadmap is a closely guarded secret.

martignoni, 2021-07-09:
Just like @holta I'd like to express my gratitude to @pelwell for considering this issue.

My opinion about this and the [MoodleBox](https://github.com/moodlebox/moodlebox): I don't care about all these (new) features that enable e.g. advanced roaming, radar systems and, to a lesser degree, Auto Channel Support, and would be glad to see them dropped altogether, in exchange of more client devices.

In summary, IMHO, the more clients the better :wink:

holta, 2021-07-09:
@pelwell I definitely agree with @martignoni, in general.  At the same time others will weigh in patiently and deliberately, over the weekend, so let's hear their voices too.

_Side Question:_ are there private NDA (or public community) testing arrangements possible for proven contributors able to help with pre-release testing of Raspberry Pi OS 11 Bullseye?  Possibly similar to [rpi-update](https://github.com/Hexxeh/rpi-update) for quality engineering of WiFi usability questions like this one?

- https://github.com/iiab/iiab/issues/2749

(And/or similar to Debian and Ubuntu Release Candidates and daily build continuous improvement, if a community growth path like that might possibly be considered in future?)

georgejhunt, 2021-07-09:
I didn't know about RPi as radar sensor. I applaud efforts to service the maker space sector. 

But I'm glad that the RPi foundation is considering a dual firmware option. Because I think  client/server educational applications are strategic until worldwide inexpensive internet access is available (at least 10 years in my opinion).

tim-moody, 2021-07-09:
@pelwell I'm very happy to see this issue on the table. I agree that number of clients is the most important capability, though I am somewhat tempted by ACS (Auto Channel Support) .

I would also ask that since we have demonstrated and documented that some old versions of the firmware are capable of 32 clients on current hardware (see #823), what features might those versions be missing that could also be eliminated from an AP oriented firmware?

shanti-bhardwa, 2021-07-10:
@pelwell I would prefer  the wifi parameters to be configurable e.g within raspi-config. Getting wifi working across all the Linux distribution  on various hardware platform is like  "doing black magic".  In my experience, PI 4 running latest IIAB build on Rapbian Lite can support 10 to 12 clients. One problem we have is that we don't have the resources to test larger configurations as we  support IIAB development from home. We would welcome some help on  getting   testing support from your company. Happy to share some ideas.

tim-moody, 2021-07-10:
On the subject of capacity testing, I have successfully simulated 10 wifi clients using a rpi 4 with 9 wifi dongles attached and some rudimentary python code that makes connections, makes continuous http requests, and tracks results. I can run multiple such devices to get more connections. The code is at https://github.com/iiab/iiab-factory/tree/master/testing/wifi-testing.

pelwell, 2021-07-12:
@shanti-bhardwa These features are not things that can be removed at boot time, except by building 2^n firmwares and selecting which one to load based on user preference (and that's not going to happen). The proposal on the table is to choose one other feature set aimed at APs. As such it might not be perfect for all AP users, but the aim is to make it less imperfect than the current one-size-fits-all firmware.

holta, 2021-08-01:
Hopefully August will allow us an opportunity to test the hard work @pelwell has discussed above &mdash; i.e. both kinds of WiFi firmware &mdash; e.g. if Raspberry Pi OS 11 is released during the same month as Debian 11 Bullseye:

- #2749
- #2820

pelwell, 2021-08-02:
Thanks for all the feedback. On balance, I'll be asking for a minimal firmware that support the largest possible number of clients. I'm not sure it will be in our first Bullseye release, and even if the file is there it will still need to be activated manually (i.e. by renaming a file). in which case requiring it to be downloaded specially is just one more command for users or distros.

pelwell, 2021-08-17:
There's a trial minimal 43455 firmware available from here: https://drive.google.com/file/d/19EiC4O3hshQUuFGPcHQ315todXBk9cU6/view?usp=sharing
It must be installed to `/lib/firmware/brcm/brcmfmac43455-sdio.bin`.

leezu, 2021-08-22:
Thank you @pelwell. I tried the minimal firmware, but it breaks the AP feature on my Pi 4B.

Log with `brcmfmac43455-sdio.bin_2015-03-01_7.45.18.0_ub19.10.1`

```
Feb 14 10:12:01 raspberrypi kernel: brcmfmac: F1 signature read @0x18000000=0x15264345
Feb 14 10:12:01 raspberrypi kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Feb 14 10:12:01 raspberrypi kernel: usbcore: registered new interface driver brcmfmac
Feb 14 10:12:01 raspberrypi kernel: brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt failed with error -2
Feb 14 10:12:01 raspberrypi kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Feb 14 10:12:01 raspberrypi kernel: brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  1 2015 07:29:38 version 7.45.18 (r538002) FWID 01-6a2c8ad4
Feb 14 10:12:01 raspberrypi systemd-udevd[236]: Using default interface naming scheme 'v240'.
Feb 14 10:12:01 raspberrypi systemd-udevd[184]: Using default interface naming scheme 'v240'.
Feb 14 10:12:01 raspberrypi systemd-networkd-wait-online[206]: ignoring: lo
Feb 14 10:12:02 raspberrypi systemd[1]: Found device /sys/subsystem/net/devices/ap0.
```

Log with `trial_minimal_43455_fw.bin`

```
Feb 14 10:12:01 raspberrypi kernel: brcmfmac: F1 signature read @0x18000000=0x15264345
Feb 14 10:12:01 raspberrypi kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Feb 14 10:12:01 raspberrypi kernel: usbcore: registered new interface driver brcmfmac
Feb 14 10:12:01 raspberrypi kernel: brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt failed with error -2
Feb 14 10:12:01 raspberrypi kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Feb 14 10:12:01 raspberrypi kernel: brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug  5 2021 12:49:40 version 7.45 (RPI_TEST) (48f55e8 CY) FWID 01-8e379eb9
Feb 14 10:12:02 raspberrypi kernel: ieee80211 phy0: brcmf_cfg80211_request_ap_if: setting ssid failed -52
Feb 14 10:12:02 raspberrypi kernel: ieee80211 phy0: brcmf_cfg80211_add_iface: add iface ap0 type 3 failed: err=-52
Feb 14 10:12:02 raspberrypi kernel: ieee80211 phy0: brcmf_fweh_call_event_handler: no interface object
Feb 14 10:12:02 raspberrypi systemd-udevd[265]: Process '/sbin/iw dev wlan0 interface add ap0 type __ap' failed with exit code 204.
```

Both with the following hostapd config

```
interface=ap0

# SSID to be used in IEEE 802.11 management frames
ssid=Pi
# Driver interface type (hostap/wired/none/nl80211/bsd)
driver=nl80211
# Country code (ISO/IEC 3166-1)
country_code=US

# Operation mode (a = IEEE 802.11a (5 GHz), b = IEEE 802.11b (2.4 GHz)
hw_mode=a
ieee80211n=1
require_ht=1
ieee80211ac=1
require_vht=1
#Recommended but breaks
#vht_oper_chwidth=1
#vht_oper_centr_freq_seg0_idx=42
# Channel number
channel=36
# Maximum number of stations allowed
max_num_sta=5

# Bit field: bit0 = WPA, bit1 = WPA2
wpa=2
# Bit field: 1=wpa, 2=wep, 3=both
auth_algs=1

# Set of accepted cipher suites; disabling insecure TKIP
wpa_pairwise=CCMP
# Set of accepted key management algorithms
wpa_key_mgmt=WPA-PSK
wpa_passphrase=redacted

# hostapd event logger configuration
logger_stdout=-1
logger_stdout_level=2

# QoS support
wmm_enabled=1
# Use "iw list" to show device capabilities and modify ht_capab accordingly
# This prevents using higher channels..
# ht_capab=[HT40+][SHORT-GI-40][DSSS_CCK-40]
```

I'm happy to modify the config if that is causing the error.

pelwell, 2021-08-22:
Could you add "brcmfmac.debug=0x100000" to cmdline.txt, reboot and try again? There will be some additional logging from the firmware present in the kernel log.

holta, 2021-08-22:
Thank you @leezu:

I experienced a similar problem on Raspberry Pi 4 Model B Rev 1.2, with this trial minimal 43455 firmware from 5 days ago.

But didn't have time to write it up.

@jvonau looked into it and may a few more details to share later.

leezu, 2021-08-23:
Thank you @pelwell and @holta. I rebooted both the old and the trial firmware the debug flag:

`dmesg | grep brcmfm` with trial firmware

```
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60,margin_left=48,margin_right=48,margin_top=48,margin_bottom=48 smsc95xx.macaddr=E4:5F:01:30:41:D3 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=a1bec75d-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait brcmfmac.debug=0x100000
[    5.595086] brcmfmac: F1 signature read @0x18000000=0x15264345
[    5.612095] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.613124] usbcore: registered new interface driver brcmfmac
[    5.635259] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt failed with error -2
[    5.863062] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.872659] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug  5 2021 12:49:40 version 7.45 (RPI_TEST) (48f55e8 CY) FWID 01-8e379eb9
[    5.954542] brcmfmac: CONSOLE: ndlocked 0
[    5.954565] brcmfmac: CONSOLE: 000000.062 wl0: Broadcom BCM4345 802.11 Wireless Controller 7.45 (RPI_TEST) (48f55e8 CY)
[    5.954582] brcmfmac: CONSOLE: 000000.062 TCAM: 256 used: 232 exceed:0
[    5.954599] brcmfmac: CONSOLE: 000000.063 reclaim section 1: Returned 111296 bytes to the heap
[    5.954616] brcmfmac: CONSOLE: 000000.063 reclaim section 4: Returned 44 bytes to the heap
[    5.954631] brcmfmac: CONSOLE: 000000.064 sdpcmd_dpc: Enable
[    5.954648] brcmfmac: CONSOLE:
[    5.954664] brcmfmac: CONSOLE: 000000.087 wl0: unable to find iovar "rsdb_mode"
[    5.954681] brcmfmac: CONSOLE: 000000.087 wl0: wlc_iovar_op: rsdb_mode BCME -23 (Unsupported)
[    5.954697] brcmfmac: CONSOLE: 000000.087 wl0: unable to find iovar "tdls_enable"
[    5.954716] brcmfmac: CONSOLE: 000000.087 wl0: wlc_iovar_op: tdls_enable BCME -23 (Unsupported)
[    6.214038] brcmfmac: CONSOLE: 000000.402 dngl_binddev: too many virtual devices: 0, ifidx:0
[    6.214058] brcmfmac: CONSOLE: 000000.402 wl0: wl_alloc_if: bus_binddev failed
[    6.214075] brcmfmac: CONSOLE: 000000.402 wl0: wl_add_if: failed to create BSS interface 2
[    6.214092] brcmfmac: CONSOLE: 000000.402 wl0: _wlc_bsscfg_init: wl_add_if failed for index 2
[    6.214109] brcmfmac: CONSOLE: 000000.402 wl0: wlc_bsscfg_alloc_ext: _wlc_bsscfg_init() failed with -1
[    6.214128] brcmfmac: CONSOLE: 000000.402 wl0: wlc_bsscfg_alloc: wlc_bsscfg_alloc_ext failed with -1
[    6.214144] brcmfmac: CONSOLE: 000000.402 wl0: wlc_quiet_bsscfg_deinit: Quiet info not found
[    6.214160] brcmfmac: CONSOLE: 000000.402 wl0: wlc_csa_bsscfg_deinit: CSA info not found
[    6.214176] brcmfmac: CONSOLE: 000000.402 wl0: wlc_iovar_op: bsscfg:ssid BCME -27 (No Memory)
```

`dmesg | grep brcmfm` with `brcmfmac43455-sdio.bin_2015-03-01_7.45.18.0_ub19.10.1`

```
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60,margin_left=48,margin_right=48,margin_top=48,margin_bottom=48 smsc95xx.macaddr=E4:5F:01:30:41:D3 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=a1bec75d-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait brcmfmac.debug=0x100000
[    5.590915] brcmfmac: F1 signature read @0x18000000=0x15264345
[    5.602498] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.602988] usbcore: registered new interface driver brcmfmac
[    5.620704] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt failed with error -2
[    5.828246] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.837832] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  1 2015 07:29:38 version 7.45.18 (r538002) FWID 01-6a2c8ad4
[    5.903976] brcmfmac: CONSOLE: .041 TCAM: 256 used: 124 exceed:0
[    5.903991] brcmfmac: CONSOLE: 000000.042 reclaim section 1: Returned 113244 bytes to the heap
[    5.904004] brcmfmac: CONSOLE: 000000.042 reclaim section 4: Returned 44 bytes to the heap
[    5.904017] brcmfmac: CONSOLE: 000000.042 sdpcmd_dpc: Enable
[    5.904030] brcmfmac: CONSOLE: 
[    5.904043] brcmfmac: CONSOLE: 000000.062 wl0: wlc_iovar_op: txbf BCME -23 (Unsupported)
[    5.904056] brcmfmac: CONSOLE: 000000.064 wl0: unable to find iovar "rsdb_mode"
[    5.904069] brcmfmac: CONSOLE: 000000.065 wl0: wlc_iovar_op: rsdb_mode BCME -23 (Unsupported)
[    5.904082] brcmfmac: CONSOLE: 000000.067 wl0: unable to find iovar "sup_wpa"
[    5.904096] brcmfmac: CONSOLE: 000000.067 wl0: wlc_iovar_op: sup_wpa BCME -23 (Unsupported)
[    5.904110] brcmfmac: CONSOLE: 000000.109 wl0: wlc_enable_probe_req: state down, deferring setting of host flags
[    5.904124] brcmfmac: CONSOLE: 000000.110 wl0: wlc_enable_probe_req: state down, deferring setting of host flags
[    6.293946] brcmfmac: CONSOLE: 000000.500 wl0: unable to find iovar "toe_ol"
[    6.293968] brcmfmac: CONSOLE: 000000.500 wl0: wlc_iovar_op: toe_ol BCME -23 (Unsupported)
[    6.293984] brcmfmac: CONSOLE: 000000.501 wl0: wl_open
[    6.314348] brcmfmac: CONSOLE: 000000.527 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
[... cut ...]
[    8.204809] brcmfmac: CONSOLE: 000002.396 wl0: wl_open
[    8.303823] brcmfmac: CONSOLE: 000002.489 wl0: link up (wl0.2)
[   17.813909] brcmfmac: CONSOLE: 000011.968 wl0.2: wlc_send_bar: seq 0x1 tid 6
[   17.813926] brcmfmac: CONSOLE: 000011.972 wl0.2: wlc_send_bar: seq 0x4 tid 0
[   18.953859] brcmfmac: CONSOLE: 000013.067 wl0.2: wlc_send_bar: seq 0x1 tid 4
[   18.953877] brcmfmac: CONSOLE: 000013.078 wl0.2: wlc_send_bar: seq 0x2 tid 3
[   23.013741] brcmfmac: CONSOLE: 000017.149 wl0.2: wlc_send_bar: seq 0x1 tid 2
```

martignoni, 2021-08-23:
FYI, this is what I get on a Raspberry Pi 4B (1 GB RAM) with `dmesg | grep brcmfm`, with this minimal firmware:
```
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=DC:A6:32:00:6B:C2 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  console=ttyS0,115200 console=tty1 root=PARTUUID=3c5972b0-02 rootfstype=ext4 elevator=deadline fsck.repair=yes brcmfmac.debug=0x100000 rootwait
[    5.867766] brcmfmac: F1 signature read @0x18000000=0x15264345
[    5.921150] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.921662] usbcore: registered new interface driver brcmfmac
[    6.192707] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    6.192849] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    6.211922] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug  5 2021 12:49:40 version 7.45 (RPI_TEST) (48f55e8 CY) FWID 01-8e379eb9
[    6.307095] brcmfmac: CONSOLE: ndlocked 0
[    6.307117] brcmfmac: CONSOLE: 000000.062 wl0: Broadcom BCM4345 802.11 Wireless Controller 7.45 (RPI_TEST) (48f55e8 CY)
[    6.307133] brcmfmac: CONSOLE: 000000.062 TCAM: 256 used: 232 exceed:0
[    6.307150] brcmfmac: CONSOLE: 000000.063 reclaim section 1: Returned 111296 bytes to the heap
[    6.307167] brcmfmac: CONSOLE: 000000.063 reclaim section 4: Returned 44 bytes to the heap
[    6.307182] brcmfmac: CONSOLE: 000000.064 sdpcmd_dpc: Enable
[    6.307198] brcmfmac: CONSOLE: 
[    6.307214] brcmfmac: CONSOLE: 000000.095 wl0: unable to find iovar "rsdb_mode"
[    6.307230] brcmfmac: CONSOLE: 000000.095 wl0: wlc_iovar_op: rsdb_mode BCME -23 (Unsupported)
[    6.307246] brcmfmac: CONSOLE: 000000.096 wl0: unable to find iovar "tdls_enable"
[    6.307262] brcmfmac: CONSOLE: 000000.096 wl0: wlc_iovar_op: tdls_enable BCME -23 (Unsupported)
[    7.745010] brcmfmac: CONSOLE: 000001.600 wl0: unable to find iovar "toe_ol"
[    7.745031] brcmfmac: CONSOLE: 000001.600 wl0: wlc_iovar_op: toe_ol BCME -23 (Unsupported)
[    7.745047] brcmfmac: CONSOLE: 000001.602 wl0: wl_open
[    7.745063] brcmfmac: CONSOLE: 000001.611 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
[    7.758103] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   19.554709] brcmfmac: CONSOLE: 000013.404 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
[   19.574657] brcmfmac: CONSOLE: 000013.418 wl0: wl_open
[   19.574667] brcmfmac: CONSOLE: 000013.430 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
[   19.705008] brcmfmac: CONSOLE: 000013.530 wl0: link up (wl0)
[   19.705017] brcmfmac: CONSOLE: 000013.530 wl0: link up (wl0)
[   19.705024] brcmfmac: CONSOLE: 000013.542 wl0: unable to find iovar "nd_hostip_clear"
[   19.705031] brcmfmac: CONSOLE: 000013.542 wl0: wlc_iovar_op: nd_hostip_clear BCME -23 (Unsupported)
```
Hostapd config file in attachment.
[hostapd.conf.txt](https://github.com/iiab/iiab/files/7029727/hostapd.conf.txt)

pelwell, 2021-08-23:
@martignoni Did the minimal firmware work with your configuration? The log doesn't appear to show any major errors.

jvonau, 2021-08-23:
@leezu Three things I noticed, "Feb"? Think you might want to add the package apt 'fake-hwclock' to your install to adjust the time forward without an RTC. Second how and when are you cloning ap0? Third the 5Ghz AP support is something I have not tried yet as there is a long standing thought process in 'iiab' that 2.4 is preferred to support the older wifi clients, and 'iiab' supports more than just the PRi hardware base. I would be interested in what setting you found useful on the journey to configure the 5 GHz band.
@holta The error I noted mentioned ap0, tried the cloning command from the cmdline, with the above noted message about ap0 failing, switched hostapd.conf over to the parent device wlan0, restarted hostapd. That is where the test box was left at and you mentioned that the SSID was now visible.
@pelwell I don't think any new firmware will make the members of this crowd happy, their desire is to push the boundaries of the hardware beyond it's limits, with the best engineering habits being tossed out the window. Now if it was my wish list I would ask for the ability to split the bands to be able to run wpa_supplicant on one band and hostapd on the other without the limitation that both be on the same channel.

martignoni, 2021-08-23:
> @martignoni Did the minimal firmware work with your configuration? The log doesn't appear to show any major errors.

Yes it works without problems. However, I could only test with 6 simulteaneous wifi clients (I do not have more devices at home).

pelwell, 2021-08-23:
@jvonau Thank you for your feedback. I understand that it isn't possible to keep everybody happy, but the core goal of a firmware with just enough features to support a simple AP configuration seems within reach.

jvonau, 2021-08-23:
@pelwell I haven't tested yet but does the 'new ap geared' firmware still support client wifi connections using wpa_supplicant?

pelwell, 2021-08-23:
It works fine for basic usage - this reply was sent from a 4B running the trial firmware:
```
[    5.765246] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug  5 2021 12:49:40 version 7.45 (RPI_TEST) (48f55e8 CY) FWID 01-8e379eb9
[    5.835429] brcmfmac: CONSOLE: ndlocked 0
[    5.835450] brcmfmac: CONSOLE: 000000.062 wl0: Broadcom BCM4345 802.11 Wireless Controller 7.45 (RPI_TEST) (48f55e8 CY)
[    5.835467] brcmfmac: CONSOLE: 000000.062 TCAM: 256 used: 232 exceed:0
[    5.835483] brcmfmac: CONSOLE: 000000.063 reclaim section 1: Returned 111296 bytes to the heap
[    5.835500] brcmfmac: CONSOLE: 000000.063 reclaim section 4: Returned 44 bytes to the heap
[    5.835515] brcmfmac: CONSOLE: 000000.064 sdpcmd_dpc: Enable
[    5.835532] brcmfmac: CONSOLE: 
[    5.835548] brcmfmac: CONSOLE: 000000.088 wl0: unable to find iovar "rsdb_mode"
[    5.835564] brcmfmac: CONSOLE: 000000.088 wl0: wlc_iovar_op: rsdb_mode BCME -23 (Unsupported)
[    5.835579] brcmfmac: CONSOLE: 000000.089 wl0: unable to find iovar "tdls_enable"
[    5.835595] brcmfmac: CONSOLE: 000000.089 wl0: wlc_iovar_op: tdls_enable BCME -23 (Unsupported)
[    8.505308] brcmfmac: CONSOLE: 000002.805 wl0: unable to find iovar "toe_ol"
[    8.505334] brcmfmac: CONSOLE: 000002.805 wl0: wlc_iovar_op: toe_ol BCME -23 (Unsupported)
[    8.505350] brcmfmac: CONSOLE: 000002.806 wl0: wl_open
[    8.505366] brcmfmac: CONSOLE: 000002.815 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
[    8.533552] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[    8.735223] brcmfmac: CONSOLE: 000003.030 wl0: wl_open
[    8.735238] brcmfmac: CONSOLE: 000003.039 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
[    8.777466] brcmfmac: CONSOLE: 000003.049 wl0: unable to find iovar "p2p_disc"
[    8.777491] brcmfmac: CONSOLE: 000003.049 wl0: wlc_iovar_op: p2p_disc BCME -23 (Unsupported)
[    8.777512] brcmfmac: CONSOLE: 000003.050 wl0: unable to find iovar "p2p_da_override"
[    8.777529] brcmfmac: CONSOLE: 000003.050 wl0: wlc_iovar_op: p2p_da_override BCME -23 (Unsupported)
[    8.777544] brcmfmac: CONSOLE: 000003.051 wl0: unable to find iovar "p2p_disc"
[    8.777561] brcmfmac: CONSOLE: 000003.051 wl0: wlc_iovar_op: p2p_disc BCME -23 (Unsupported)
[   13.926333] brcmfmac: CONSOLE: 000008.198 wl0: wlc_phy_BSSinit: Chanspec 0xe02a not found in calibration cache
[   13.926344] brcmfmac: CONSOLE: 000008.198 wl0: link up (wl0)
[   14.005112] brcmfmac: CONSOLE: 000008.276 wl0: unable to find iovar "nd_hostip_clear"
[   14.005121] brcmfmac: CONSOLE: 000008.276 wl0: wlc_iovar_op: nd_hostip_clear BCME -23 (Unsupported)
[   14.024967] brcmfmac: CONSOLE: 000008.297 wl0.0: wlc_send_bar: seq 0x1 tid 7
[   14.124978] brcmfmac: CONSOLE: 000008.386 wl0.0: wlc_send_bar: seq 0x3 tid 0
```

martignoni, 2021-08-23:
Just got [another user report with new firmware](https://discuss.moodlebox.net/d/260-plans-for-moodlebox-bullseye-and-64-bit/20): 10 devices connected, no interruptions.

leezu, 2021-08-28:
@pelwell is the minimal firmware supposed to support 5Ghz AP?

@jvonau

> Three things I noticed, "Feb"? Think you might want to add the package apt 'fake-hwclock' to your install to adjust the time forward without an RTC. 

Great, thanks for pointing.

> Second how and when are you cloning ap0?

I've mostly followed https://hackaday.io/project/162164/instructions?page=1 https://hackaday.io/project/162164/instructions?page=2 , though hostapd config is modified as described in https://github.com/iiab/iiab/issues/2853#issuecomment-903327483 and the ap0 is created by systemd-network               
                                                                   
/etc/systemd/network/20-ap0.network                                                           

```
  [Match]                                                                                     
  Name=ap0                                                                                    
                                                                                              
  [Network]                                                                                   
  Address=192.168.11.254/28                                                                   
  DHCPServer=yes                                                                              
  IPMasquerade=yes                                                                            
  IPForward=ipv4                                                                              
                                                                                              
  [DHCPServer]                                                                                
  # PoolOffset=100                                                                            
  # PoolSize=20                                                                               
  # EmitDNS=yes                                                                               
  DNS=9.9.9.9                                                                                 
  DNS=8.8.8.8                                                                                 
```

> Third the 5Ghz AP support is something I have not tried yet as there is a long standing thought process in 'iiab' that 2.4 is preferred to support the older wifi clients, and 'iiab' supports more than just the PRi hardware base. I would be interested in what settings you found useful on the journey to configure the 5 GHz band.

I have disabled 2.4GHz, as I'm running the RPi in a "crowded" space where 2.4GHz would be very slow. 5GHz plays much better in areas with many WiFi networks. The settings are all in https://github.com/iiab/iiab/issues/2853#issuecomment-903327483 or are you looking for another kind of setting? In addition, I've disabled WiFi powersave feature to significantly improve latency https://raspberrypi.stackexchange.com/questions/96606/make-iw-wlan0-set-power-save-off-permanent

pelwell, 2021-08-31:
Yes it should run a 5G AP - is it not working for you?

holta, 2021-08-31:
Bad news first (which appears to be) that trial_minimal_43455_fw.bin (from 2 weeks ago) does not seem to offer much/any improvement at all, seemingly unable to work with more than 8-10 WiFi client devices.

NOTE: that testing was with 2.4GHz (as we need in most low-income countries, with much older client devices).

Let's try to get our testing much more organized in September, to try to understand better why we are not seeing improvement here.  (Apologies several of us have been very busy with other things in August!)

leezu, 2021-08-31:
> Yes it should run a 5G AP - is it not working for you?

@pelwell it's not working for me. Please see the log in https://github.com/iiab/iiab/issues/2853#issuecomment-903327483 and debug log https://github.com/iiab/iiab/issues/2853#issuecomment-903383175

pelwell, 2021-08-31:
Did you try changing your hostapd.conf, perhaps using @martignoni's (https://github.com/iiab/iiab/issues/2853#issuecomment-903492210) as a reference?

beta-tester, 2021-09-30:
where can i get and how to enable the firmware for more wifi clients at the current Raspi OS 10 Buster?

pelwell, 2021-09-30:
You can download it here: https://drive.google.com/file/d/19EiC4O3hshQUuFGPcHQ315todXBk9cU6/view?usp=sharing
It's not been released more formally yet due to a shortage of feedback.

martignoni, 2021-09-30:
@pelwell I'd like to give (or ask for) more feedback. What kind of feedback do you need?

tim-moody, 2021-09-30:
I have done the following tests on a Raspberry Pi 4:

- Raspberry Pi OS Lite from May 7th 2021 with included firmware: Maximum of **8 Connections**
- Raspberry Pi OS Lite from May 7th 2021 with trial_minimal_43455_fw.bin : Maximum of  **8 Connections**
- Raspberry Pi OS Lite from May 7th 2021 with old firmware (7.45.18.0): I stopped at **20 Connections**
- moodlebox-3.12.0-minimal-fw.img.bz2 with trial_minimal_43455_fw.bin : Maximum of  **8 Connections**

In my tests there was no difference between the included firmware and the trial minimal

How I tested:

- Rpi 4 with 2 usb wifi dongles and usb hub with 7 dongles for a total of 10 wifi devices. I used two of these for a maximum of 20 connections.
- Testing code in https://github.com/iiab/iiab-factory/tree/master/testing/wifi-testing

The number of connections in each case was reported by running the following on the server under test:

`/usr/sbin/hostapd_cli all_sta |grep :
`

pelwell, 2021-09-30:
Yes, that kind of feedback. It's not exactly what I had hoped for, but it's something to work with.

Was that using 2.4GHz or 5GHz channels?

tim-moody, 2021-09-30:
I only tested 2.4GHz.

pelwell, 2021-09-30:
Apologies for asking this question, but are you certain that the trial firmware was actually being used?

martignoni, 2021-09-30:
Summary of tests done by some MoodleBox users (https://discuss.moodlebox.net/d/260-plans-for-moodlebox-bullseye-and-64-bit/), with firmware taken here https://github.com/iiab/iiab/issues/2853#issuecomment-900328119:

- Max 10 connections with a RPi 4B, 4GB (5 iOS devices, 4 Linux notebooks, 1 Galaxy Tab)
- Max 9 connections with a RPi 4B, 4GB (6 iOS devices, 2 Linux notebooks, 1 Galaxy Tab). Additional devices got "Could not connect" error

jvonau, 2021-09-30:
@tim-moody I take it that ap0 was not used as @holta and @leezu noted above that the test firmware doesn't play nice with a cloned interface.

tim-moody, 2021-09-30:
cp /root/trial_minimal_43455_fw.bin /lib/firmware/brcm/brcmfmac43455-sdio.bin
reboot
is what I did.

Also, it has been my experience that phones are not always truly connected. They connect but then the connection is suspended. to allow for other connections. So I use 

`/usr/sbin/hostapd_cli all_sta |grep :
`
to determine the number of live connections. The test software makes continuous requests to make sure they are used simultaneously.

Here is the test log. Note that trial minimum size is 543634; stock is 631467; and old is 488193.

Test 1
•	stock firmware
•	start with tester2
•	8 connections came reasonably quickly
•	Says 9th succeeded, but there are only 8 mac addresses
•	10th the same
•	Killed it after 60 url requests (5 min)

Test 2
•	Cp trial to brcm
•	cp /root/trial_minimal_43455_fw.bin /lib/firmware/brcm/brcmfmac43455-sdio.bin
•	Power off tester2
•	Power cycle server
•	-rw-r--r-- 1 root root 543634 Aug 25 16:13 /lib/firmware/brcm/brcmfmac43455-sdio.bin
•	-rw-r--r-- 1 root root 488193 Aug 25 15:32 /lib/firmware/brcm/brcmfmac43455-sdio.bin.iiab
•	-rw-r--r-- 1 root root 631467 Aug 25 15:31 /lib/firmware/brcm/brcmfmac43455-sdio.bin.orig
•	-rw-r--r-- 1 root root   2074 Jun 23 13:02 /lib/firmware/brcm/brcmfmac43455-sdio.txt
•	-rw-r--r-- 1 root root   7163 Jun 23 13:02 /lib/firmware/brcm/brcmfmac43455-sdio.clm_blob
•	-rw-r--r-- 1 root root  14036 Aug 25 15:32 /lib/firmware/brcm/brcmfmac43455-sdio.clm_blob.iiab
•	-rw-r--r-- 1 root root   7163 Aug 25 15:31 /lib/firmware/brcm/brcmfmac43455-sdio.clm_blob.orig
•	Start with tester1
•	Again 8 connections came reasonably quickly
•	Says 9th succeeded, but there are only 8 mac addresses
•	10th the same
•	Time Started Wed Aug 25 15:17:55 2021. Elapsed 00:14:04
•	Total Connections on Server: 8
•	
•	These are our interfaces:
•	wlan0 mac: f8:1a:67:29:f5:4f ip: 172.18.114.43 connection: connected to server urls accessed: 34
•	wlan1 mac: c4:e9:84:09:f0:1c ip: 172.18.120.193 connection: connected to server urls accessed: 34
•	wlan2 mac: dc:a6:32:95:ad:07 ip: 172.18.112.240 connection: connected to server urls accessed: 33
•	wlan3 mac: d0:37:45:1b:da:1c ip: 172.18.106.204 connection: connected to server urls accessed: 34
•	wlan4 mac: d0:37:45:1b:d9:53 ip: 172.18.120.176 connection: connected to server urls accessed: 34
•	wlan5 mac: d0:37:45:1d:10:a9 ip: 172.18.121.87 connection: connected to server urls accessed: 34
•	wlan6 mac: d0:37:45:16:46:5f ip: 172.18.108.112 connection: connected to server urls accessed: 34
•	wlan7 mac: d0:37:45:1e:15:35 ip: 172.18.122.85 connection: connected to server urls accessed: 34
•	wlan8 mac: d0:37:45:1d:e6:56 ip: None connection: no connection urls accessed: 0
•	wlan9 mac: d0:37:45:1b:01:11 ip: None connection: no connection urls accessed: 0
•	
•	These mac addresses seen on the server:
•	d0:37:45:1e:15:35 ( wlan7 ) connected to server
•	d0:37:45:16:46:5f ( wlan6 ) connected to server
•	d0:37:45:1d:10:a9 ( wlan5 ) connected to server
•	d0:37:45:1b:d9:53 ( wlan4 ) connected to server
•	d0:37:45:1b:da:1c ( wlan3 ) connected to server
•	dc:a6:32:95:ad:07 ( wlan2 ) connected to server
•	c4:e9:84:09:f0:1c ( wlan1 ) connected to server
•	f8:1a:67:29:f5:4f ( wlan0 ) connected to server
•	Time Started Wed Aug 25 15:17:55 2021. Elapsed 02:02:16
•	Total Connections on Server: 8
•	
•	These are our interfaces:
•	wlan0 mac: f8:1a:67:29:f5:4f ip: 172.18.114.43 connection: connected to server urls accessed: 1316
•	wlan1 mac: c4:e9:84:09:f0:1c ip: 172.18.120.193 connection: connected to server urls accessed: 1330
•	wlan2 mac: dc:a6:32:95:ad:07 ip: 172.18.112.240 connection: connected to server urls accessed: 1301
•	wlan3 mac: d0:37:45:1b:da:1c ip: 172.18.106.204 connection: connected to server urls accessed: 1329
•	wlan4 mac: d0:37:45:1b:d9:53 ip: 172.18.120.176 connection: connected to server urls accessed: 1321
•	wlan5 mac: d0:37:45:1d:10:a9 ip: 172.18.121.87 connection: connected to server urls accessed: 1329
[... 46 lines cut ...]
•	wlan9 mac: d0:37:45:8b:61:d9 ip: 172.18.108.92 connection: connected to server urls accessed: 22
•	
•	These mac addresses seen on the server:
•	d0:37:45:8b:61:d9 ( wlan9 ) connected to server
•	d0:37:45:8a:37:11 ( wlan8 ) connected to server
•	d0:37:45:8c:89:15 ( wlan7 ) connected to server
•	d0:37:45:89:0d:dd ( wlan6 ) connected to server
•	d0:37:45:8c:48:ff ( wlan5 ) connected to server
•	d0:37:45:87:12:7c ( wlan4 ) connected to server
•	d0:37:45:8c:88:cf ( wlan3 ) connected to server
•	dc:a6:32:09:fd:85 ( wlan2 ) connected to server
•	d0:37:45:8b:3d:d8 ( wlan1 ) connected to server
•	d0:37:45:87:0f:af ( wlan0 ) connected to server
•	d0:37:45:1b:01:11  on server is not ours
•	d0:37:45:1d:e6:56  on server is not ours
•	d0:37:45:1e:15:35  on server is not ours
•	d0:37:45:16:46:5f  on server is not ours
•	d0:37:45:1d:10:a9  on server is not ours
•	d0:37:45:1b:d9:53  on server is not ours
•	d0:37:45:1b:da:1c  on server is not ours
•	dc:a6:32:95:ad:07  on server is not ours
•	c4:e9:84:09:f0:1c  on server is not ours
•	f8:1a:67:29:f5:4f  on server is not ours
•	20 connections for over an hour
•

tim-moody, 2021-09-30:
From an iiab point of view a min local vars was used and the following were set to False 
•	wifi_hotspot_capacity_rpi_fix
•	wifi_up_down
•	kalite_install
•	captiveportal_install

iiab was installed and not admin console

tim-moody, 2021-09-30:
Note that when I copied trial I did nothing to any stock clm_blob or txt, but I copied those for the old firmware when testing that.

jvonau, 2021-09-30:
In summary, setting wifi_up_down to False disabled the use of ap0 with hostapd declaring the use of the parent device, meaning  interface= was being set to wlan0.

pelwell, 2021-10-01:
I have an update to the trial firmware: https://drive.google.com/file/d/1XfDfX6gqqJOvUr-S7reOQDGAIYIIfQYF/view?usp=sharing

Cypress/Infineon haven't said what the change is, but they think it should be an improvement.

beta-tester, 2021-10-02:
i there a way to prevent apt-get from overwriting the trail-firmware, when a new firmware-brcm80211 was rolled out?

1. `original`:
   brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan  4 2021 19:56:29 version 7.45.229 (617f1f5 CY) FWID 01-2dbd9d2e

2. `trial_minimal_43455_fw.bin`:
   brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug  5 2021 12:49:40 version 7.45 (RPI_TEST) (48f55e8 CY) FWID 01-8e379eb9

3. `2nd_trial_minimal_43455_fw.bin`:
   brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Sep 30 2021 16:35:35 version 7.45 (RPI_TEST) (48f55e8 CY) FWID 01-c9539fe7

4. `apt upgrade` -> **firmware-brcm80211**
   `2nd_trial_minimal_43455_fw.bin` was overwritten be **original**:
   brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan  4 2021 19:56:29 version 7.45.229 (617f1f5 CY) FWID 01-2dbd9d2e

ralf-krause, 2021-10-02:
I installed the image moodlebox-3.12.0-minimal-fw.img.bz2
I can connect 8 clients to the MoodleBox with Raspberry Pi 4B (4GB RAM). 
The next 3 clients got a network connection error and could not connect.

[image]

pelwell, 2021-10-02:
@beta-tester You can `sudo apt-mark hold firmware-brcm80211` (or perhaps `sudo apt hold firmware-brcm80211` nowadays).

beta-tester, 2021-10-02:
> @beta-tester You can `sudo apt-mark hold firmware-brcm80211` (or perhaps `sudo apt hold firmware-brcm80211` nowadays).

thanks

tim-moody, 2021-10-03:
Though I reported 20 Connections on Raspberry Pi OS Lite from May 7th 2021 with old firmware (7.45.18.0), I want to make it clear that I stopped at that point as a matter of convenience as did not have enough test devices to go to 32. I expect that the actual limit would have been 32 connections as reported in https://github.com/iiab/iiab/issues/823#issuecomment-662285202.

I say this because I was able to get a maximum of 20 connections with 2nd_trial_minimal_43455_fw.bin and this was a hard limit as I failed to get a 21st connection with either my phone or my laptop. Here is a screen shot of the two running testing processes.

[image]

leezu, 2021-10-04:
@pelwell I tried the new version and virtual interface (ap0) still does not work with the new version. The error remains as in https://github.com/iiab/iiab/issues/2853#issuecomment-903383175

pelwell, 2021-10-04:
@tim-moody 20 connections? That's more like it - the trial images do still have support for 802.11k, 802.11v and 802.11r (I'll double-check why), so 20 clients sounds about right.

@leezu - @martignoni was kind enough to provide his hostapd.conf for comparison: https://github.com/iiab/iiab/files/7029727/hostapd.conf.txt
I suggest you start with his version, update the SSID and password fields. and see if it works for you. If you find it is lacking some features that you need, change the relevant configuration settings one at a time (or at least keeping the steps as small as possible) until it breaks - then we can try to work out why that setting is a problem.

leezu, 2021-10-04:
@pelwell I tried @martignoni's config. I think the issue is not with the hostapd config, but that I'm using a virtual interface (ap0) whereas @martignoni uses the wlan0 interface. The virtual interface is added via:

```
cat /etc/udev/rules.d/70-ap-interface.rules
SUBSYSTEM=="net", KERNEL=="wlan*", ACTION=="add", RUN+="/sbin/iw dev %k interface add ap%n type __ap"
```
via https://hackaday.io/project/162164/instructions?page=2

beta-tester, 2021-10-05:
@pelwell
> Was that using 2.4GHz or 5GHz channels?

is 5Ghz still fully supported by the 2nd_trial_firmware.bin ?

because it looks like some 5GHz channels are disabled with the firmware.
```
# original firmware:
sudo iw phy#0 info > fw_0.txt

# 2nd_trial_firmware.bin:
sudo iw phy#0 info > fw_2.txt

# show the differences:
diff fw_0.txt fw_2.txt 
10d9
< 	Device supports T-DLS.
120,134c119,133
< 			* 5260 MHz [52] (20.0 dBm) (no IR, radar detection)
< 			* 5280 MHz [56] (20.0 dBm) (no IR, radar detection)
< 			* 5300 MHz [60] (20.0 dBm) (no IR, radar detection)
< 			* 5320 MHz [64] (20.0 dBm) (no IR, radar detection)
< 			* 5500 MHz [100] (20.0 dBm) (no IR, radar detection)
< 			* 5520 MHz [104] (20.0 dBm) (no IR, radar detection)
< 			* 5540 MHz [108] (20.0 dBm) (no IR, radar detection)
< 			* 5560 MHz [112] (20.0 dBm) (no IR, radar detection)
< 			* 5580 MHz [116] (20.0 dBm) (no IR, radar detection)
< 			* 5600 MHz [120] (20.0 dBm) (no IR, radar detection)
< 			* 5620 MHz [124] (20.0 dBm) (no IR, radar detection)
< 			* 5640 MHz [128] (20.0 dBm) (no IR, radar detection)
< 			* 5660 MHz [132] (20.0 dBm) (no IR, radar detection)
< 			* 5680 MHz [136] (20.0 dBm) (no IR, radar detection)
< 			* 5700 MHz [140] (20.0 dBm) (no IR, radar detection)
---
> 			* 5260 MHz [52] (disabled)
> 			* 5280 MHz [56] (disabled)
> 			* 5300 MHz [60] (disabled)
> 			* 5320 MHz [64] (disabled)
> 			* 5500 MHz [100] (disabled)
> 			* 5520 MHz [104] (disabled)
> 			* 5540 MHz [108] (disabled)
> 			* 5560 MHz [112] (disabled)
> 			* 5580 MHz [116] (disabled)
> 			* 5600 MHz [120] (disabled)
> 			* 5620 MHz [124] (disabled)
> 			* 5640 MHz [128] (disabled)
> 			* 5660 MHz [132] (disabled)
> 			* 5680 MHz [136] (disabled)
> 			* 5700 MHz [140] (disabled)
136,140c135,139
< 			* 5745 MHz [149] (disabled)
< 			* 5765 MHz [153] (disabled)
< 			* 5785 MHz [157] (disabled)
< 			* 5805 MHz [161] (disabled)
< 			* 5825 MHz [165] (disabled)
---
> 			* 5745 MHz [149] (13.0 dBm)
> 			* 5765 MHz [153] (13.0 dBm)
> 			* 5785 MHz [157] (13.0 dBm)
> 			* 5805 MHz [161] (13.0 dBm)
> 			* 5825 MHz [165] (13.0 dBm)
154d152
< 		 * tdls_oper
165c163
< 		   total <= 3, #channels <= 2
---
> 		   total <= 3, #channels <= 1
```

or is there something going on with the country_code=DE settings?
```
########################################
#/etc/hostapd/hostapd.conf
interface=wlan0
driver=nl80211

##### IEEE 802.11 related configuration #######################################
country_code=DE
ieee80211d=1

##### 802.11b
hw_mode=b
channel=1

ignore_broadcast_ssid=0

# QoS support (gives different packages different priority)
#wmm_enabled=1

##### IEEE 802.11n related configuration ######################################
ieee80211n=1

##### access control for MAC configuration #######################################
macaddr_acl=1
accept_mac_file=/etc/hostapd.accept

##### WPA/IEEE 802.11i configuration ##########################################
# bit 0 (1) = WPA; bit 1 (2) = WEP; (3) = WPA + WEP
auth_algs=1
# bit 0 (1) = WPA; bit 1 (2) = WPA2 (IEEE822.11i/RSN); (3) = WPA + WPA2
wpa=2
wpa_key_mgmt=WPA-PSK
# WPA2 encryption algorithm
rsn_pairwise=CCMP

wpa_passphrase=<password>
ssid=my-wifi

## optional: create virtual wlan adapter
#auth_algs=1
#wpa=2
#wpa_key_mgmt=WPA-PSK
#rsn_pairwise=CCMP
##wpa_passphrase=<password>
#wpa_psk=...
#ssid=my-wifi-2
#bss=wlan0x

### https://github.com/iiab/iiab/issues/823
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0

```

pelwell, 2021-10-05:
Cypress are on a roll - there's another trial firmware, this time without the 802.11kvr support, meaning it should support 24 STAs: https://drive.google.com/file/d/1XEj61eZTeOBoz2YlQmKwUxwAnlrTilSp/view?usp=sharing
Unless there are problems found with this image it will be prepared for release.

holta, 2021-10-05:
> Cypress are on a roll - there's another trial firmware, this time without the 802.11kvr support, meaning it should support 24 STAs: https://drive.google.com/file/d/1XEj61eZTeOBoz2YlQmKwUxwAnlrTilSp/view?usp=sharing Unless there are problems found with this image it will be prepared for release.

Great News!

@pelwell can you ask Cypress if they are aware that these firmwares appear not to support ap0 (virtual interface) ?

Specifically:

Internet-in-a-Box (and others) currently use ap0 to allow 32 students access to their classroom's learning hotspot — _while at the very same time that (this very same Raspberry Pi's) wlan0 connects to the school building's (WiFi) Internet router._

pelwell, 2021-10-05:
Yes - I'll ask.

pelwell, 2021-10-05:
The suspicion (it's awaiting confirmation) is that support for the AP+STA feature was dropped to free up more memory, and that it might cost 2 clients to restore it. Is this a hard requirement?

holta, 2021-10-05:
> The suspicion (it's awaiting confirmation) is that support for the AP+STA feature was dropped to free up more memory, and that it might cost 2 clients to restore it. Is this a hard requirement?

AP+STA is very strongly desired by Internet-in-a-Box implementers everywhere.
It would be quite tragic to lose this quite critical usability feature (particularly if it only costs 1-2 client connections).

@pelwell we will help test this very carefully for you &mdash; e.g. if Cypress can hopefully provide such AP+STA firmware in coming days?

Huge Thanks!

pelwell, 2021-10-05:
Thanks - I hadn't appreciated that, given the number of positive comments.

leezu, 2021-10-05:
>  Unless there are problems found with this image it will be prepared for release.

@pelwell as reported above, creating a virtual interface causes the firmware to crash. Could Cypress handle support this functionality or at least handle failure in a graceful way?

pelwell, 2021-10-08:
FYI I'm waiting to hear back from Cypress/Infineon about that request, but I think it's a matter of when they can find the time rather than if it is possible.

[11 more replies]