iiab/iiab #2855: Installation of pbx fails while attempting to install apache2 because of possibly old/incorrect apache2 install path

lemueldsouza, 2021-07-11:
### Expected behavior:
Freepbx should be installed

### Actual behavior:
Installation of freepbx fails while installing apache2. Pasting just one line of the log line that shows failure
```
failed: E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/a/apache2/apache2-bin_2.4.38-3+deb10u4_armhf.deb 404 Not Found [IP: 93.93.128.193 80]
```

Looks like the URL for apache2 is incorrect or has been changed. When I took a look at http://raspbian.raspberrypi.org/raspbian/pool/main/a/apache2, I noticed that apache2-bin_2.4.38-3+deb10u4_armhf.deb is not present.

The file seems to have been updated/changed on July 9, 2021. The filename should end with `u5_armhf.deb` instead of u4_armhf.deb

### Steps to reproduce the problem:
1. Install IIAB with one line installer on Raspberry Pi 3 Model B Plus Rev 1.3

2. Update the file /etc/iiab/local_vars.yml to make the following changes:
pbx_install: True
pbx_enabled: True

3. Execute iiab-configure 
cd /opt/iiab/iiab 
./iiab-configure

(I had also tried ./runrole pbx earlier but that failed as well)

### IIAB version, OS version, hardware details:
IIAB version: 7.2
OS version: Latest Raspberry Pi OS 32 bit
Hardware: Raspberry Pi 3 Model B Plus Rev 1.3

### Diagnositcs output here:
http://sprunge.us/Z3Lktc

### Additional comments
I'm new to IIAB world, and interested in getting freepbx to work on it , sorry if I've missed reporting or adding any more info here.
If you can tell me where to change the apache2 install path, I could update it and give it a try.

jvonau, 2021-07-11:
>"cache_updated": false,
>E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

Back story, the 'iiab' script does `apt update` during the initial installation process and is usually enough without having each role that is run asking for fresh copy of the upstream cache, for speed considerations, that usually doesn't change during the initial installation. Given you have deduced that the upstream updated the apt package but the local machine is not aware of the availability of the newer package I suggest that you run `sudo apt update` to freshen the local cache and continue with `sudo ./iiab-configure`.

lemueldsouza, 2021-07-11:
Oops , my bad. Thanks for the quick response.

closed 2021-07-11 by lemueldsouza