iiab/iiab #2856: PBX installation fails at the task - Patch FreePBX source - disable get_magic_quotes_gpc()

lemueldsouza, 2021-07-11:
### Expected behavior:
freepbx installation should be successful

### Actual behavior:
freepbx installation failure 

Reason for failure - incorrect 'dest' path for the Patch FreePBX source in roles/pbx/tasks/freepbx.yml (notice the additional 'freepbx' after freepbx_install_dir)

```
2021-07-11 18:46:35,466 p=2366 u=root n=ansible | TASK [pbx : FreePBX - Patch FreePBX source - disable get_magic_quotes_gpc()] ***

2021-07-11 18:46:36,842 p=2366 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "dest /var/www/html/freepbx/freepbx/admin/libraries/view.functions.php doesn't exist or not writable"}

2021-07-11 18:46:36,848 p=2366 u=root n=ansible | PLAY RECAP *********************************************************************

2021-07-11 18:46:36,850 p=2366 u=root n=ansible | 127.0.0.1 : ok=158 changed=65 unreachable=0 failed=1 skipped=53 rescued=0 ignored=0 
```

Installation succeeded after I manually updated the path in freepbx.yml on my machine

### Steps to reproduce
1. Update the file /etc/iiab/local_vars.yml to make the following changes:
pbx_install: True
pbx_enabled: True

2. Execute iiab-configure 
cd /opt/iiab/iiab 
./iiab-configure

### IIAB version, OS version, hardware details:
IIAB version: 7.2
OS version: Latest Raspberry Pi OS
Hardware: Raspberry Pi 3 Model B Plus Rev 1.3

### Diagnositcs output here:
http://sprunge.us/R8lriW

holta, 2021-07-11:
Thanks @lemueldsouza for the report, and also for #2855.

> Installation succeeded after I manually updated the path in freepbx.yml on my machine

Are you able to offer a possible PR (pull request) towards fixing https://github.com/iiab/iiab/tree/master/roles/pbx ?

lemueldsouza, 2021-07-12:
> Are you able to offer a possible PR (pull request) towards fixing https://github.com/iiab/iiab/tree/master/roles/pbx ?

Sure @holta, I could get to it in a day or two if its alright.
Could you direct me to any preferred PR workflow or guidelines for IIAB, if there's any?

holta, 2021-07-12:
@lemueldsouza does #2843 help, as a possible recent PR example?

Pseudocode is absolutely fine too (i.e. explaining in English, your concrete recommendation[s] for [/opt/iiab/iiab/roles/pbx/tasks/freepbx.yml](https://github.com/iiab/iiab/blob/master/roles/pbx/tasks/freepbx.yml)) if you prefer!

Conversely: if you have time to take FreePBX-on-IIAB very seriously in coming days, many of these Apache dependencies below (in [IIAB's pbx role](https://github.com/iiab/iiab/tree/master/roles/pbx)) urgently need to be stripped out &mdash; or converted to [NGINX](https://github.com/iiab/iiab/tree/master/roles/nginx):

```
root@box:/opt/iiab/iiab/roles/pbx# grep -rin apache

templates/freepbx.conf.j2:18:ErrorLog ${APACHE_LOG_DIR}/pbx-error.log
templates/freepbx.conf.j2:19:CustomLog ${APACHE_LOG_DIR}/pbx-access.log combined
tasks/freepbx_enable.yml:17:- name: FreePBX - Enable http://box:{{ pbx_http_port }} via Apache, if pbx_enabled    # http://box:83
tasks/freepbx_enable.yml:21:- name: FreePBX - Disable http://box:{{ pbx_http_port }} via Apache, if not pbx_enabled    # http://box:83
tasks/freepbx_enable.yml:25:- name: FreePBX - Restart Apache service ({{ apache_service }})
tasks/freepbx_enable.yml:27:    name: "{{ apache_service }}"    # httpd or apache2
tasks/freepbx.yml:129:- name: FreePBX - Install Apache's sites-available/freepbx.conf from template
tasks/freepbx.yml:132:    dest: /etc/apache2/sites-available/freepbx.conf
tasks/freepbx.yml:133:    owner: "{{ apache_user }}"
tasks/freepbx.yml:134:    group: "{{ apache_user }}"
tasks/freepbx.yml:137:- name: FreePBX - Add {{ pbx_http_port }} as a Listen directive to apache ports.conf
tasks/freepbx.yml:139:    path: /etc/apache2/ports.conf
tasks/freepbx_dependencies.yml:27:      - libapache2-mod-php
tasks/freepbx_dependencies.yml:29:      - libapache2-mpm-itk    # To serve FreePBX through a VirtualHost as asterisk user
tasks/install.yml:20:- name: "Set 'apache_install: True' and 'apache_enabled: True'"
tasks/install.yml:22:    apache_install: True
tasks/install.yml:23:    apache_enabled: True
tasks/install.yml:25:- name: APACHE - run 'httpd' role
```

In short: profound thanks for any patch suggestion or any PR you might offer, either way~

_(Towards helping get the FreePBX-on-IIAB community on track, oriented, and heading in the right direction &mdash; as we wrap up IIAB's transition from Apache to NGINX [PR #2850] very shortly now!)_

PR #2861 (2021-07-14): Update freepbx.yml
Fixes #2856, pbx installation succeeded on latest raspberry pi os (raspbian-10)

### Fixes bug:
#2856 

### Description of changes proposed in this pull request:
File: iiab/roles/pbx/tasks/freepbx.yml
'dest' path has been updated to get rid of additional 'freepbx' included in the path

### Smoke-tested on which OS or OS's:
freepbx installation succeeded on latest Raspberry Pi OS (raspbian-10)

### Mention a team member @username e.g. to help with code review:
@holta, please review
commit e9cbf248 (2021-07-14): Update freepbx.yml

Fixes #2856, pbx installation succeeded on latest raspberry pi os (raspbian-10)
closed 2021-07-14 by holta