iiab/iiab #2863: Unable to change WiFi or open WiFi Menu

deltacloud9, 2021-07-15:
### Expected behavior and actual behavior:

In the Rasberry Pi OS, when clicking on the WiFi icon on the panel bar the only option I receive is to "Click here to set WiFi country". This should give me the option to change my WiFi or to turn my WiFi on/off.

### Steps to reproduce the problem:

In Rasberry Pi OS I tried creating a new panel and adding the Wireless applet to the panel to see if the same issue would occur and it did.

### IIAB version, OS version, hardware details:

Using a Rasberry Pi 400 running Rasberry Pi OS with iiab installed

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)

This is: /etc/iiab/diag/7.2_raspbian-10_2021-07-14_20:52:46_EDT_mattos

0. GIT HASHES + RASPBERRY PI MODEL + OS

iiab commit:               9955aeddd4549272d4fcee683dbd1562d1042998
iiab-admin-console commit: 3ed240b5298d49165be777c5f17a58f2d43b1efb

=IIAB==========================================================================
FILE DOES NOT EXIST: /etc/iiab/pr-list-pulled
=IIAB==========================================================================
-r--r--r-- 1 root root 25 Jul 14 20:47 /sys/firmware/devicetree/base/model

Raspberry Pi 400 Rev 1.0 
=IIAB==========================================================================
-rw-r--r-- 1 root root 145 May  7 11:07 /etc/rpi-issue

Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage4

-IIAB-EXPLANATION-OF-THE-ABOVE-------------------------------------------------

stage2 = Raspberry Pi OS Lite
stage4 = Raspberry Pi OS with desktop
stage5 = Raspberry Pi OS with desktop + recommended software

SEE https://github.com/RPi-Distro/pi-gen#stage-anatomy

=IIAB==========================================================================
-rw-r--r-- 1 root root 22 Mar 29 21:16 /etc/issue.net

Raspbian GNU/Linux 10

=IIAB==========================================================================
-rw-r--r-- 1 root root 6 Jun 24 14:40 /etc/debian_version

10.10

1. FILES SPECIALLY REQUESTED (FROM "iiab-diagnostics PATH/FILE1 PATH/FILE2")

2. REGULAR FILES

=IIAB==========================================================================
-rw-r--r-- 1 root root 34 Jul 13 17:30 /etc/iiab/openvpn_handle

Medical IIAB - Put Your Name Here

=IIAB==========================================================================
FILE DOES NOT EXIST: /.iiab-image
=IIAB==========================================================================
-rw-r--r-- 1 root root 402 Jul 13 17:45 /etc/iiab/iiab.env

# This is a configuration file for IIAB
# It can be sourced in a shell script or read into an application
IIAB_BASE_PATH=/opt/iiab
IIAB_DIR=/opt/iiab/iiab
IIAB_RELEASE=7.2
[... 1744 lines cut ...]
Jul 15 00:47:07 box rtkit-daemon[1754]: Successfully made thread 1767 of process 1746 (n/a) owned by '1000' RT at priority 5. 
Jul 15 00:47:07 box rtkit-daemon[1754]: Supervising 2 threads of 1 processes of 1 users. 
Jul 14 20:47:07 box bluetoothd[1098]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSource
Jul 14 20:47:07 box bluetoothd[1098]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSink
Jul 14 20:47:08 box vmunix: [   18.415132] Bluetooth: RFCOMM TTY layer initialized
Jul 14 20:47:08 box vmunix: [   18.415148] Bluetooth: RFCOMM socket layer initialized
Jul 14 20:47:08 box vmunix: [   18.415169] Bluetooth: RFCOMM ver 1.11
Jul 14 20:47:08 box dbus-daemon[1522]: [session uid=1000 pid=1522] Activating via systemd: service name='org.gtk.vfs.UDisks2VolumeMonitor' unit='gvfs-udisks2-volume-monitor.service' requested by ':1.7' (uid=1000 pid=1691 comm="pcmanfm --desktop --profile LXDE-pi ")
Jul 14 20:47:08 box dbus-daemon[1522]: [session uid=1000 pid=1522] Successfully activated service 'org.gtk.vfs.UDisks2VolumeMonitor'
Jul 14 20:47:08 box dbus-daemon[1522]: [session uid=1000 pid=1522] Activating via systemd: service name='org.gtk.vfs.GPhoto2VolumeMonitor' unit='gvfs-gphoto2-volume-monitor.service' requested by ':1.7' (uid=1000 pid=1691 comm="pcmanfm --desktop --profile LXDE-pi ")
Jul 14 20:47:08 box dbus-daemon[1522]: [session uid=1000 pid=1522] Successfully activated service 'org.gtk.vfs.GPhoto2VolumeMonitor'
Jul 14 20:47:08 box dbus-daemon[1522]: [session uid=1000 pid=1522] Activating via systemd: service name='org.gtk.vfs.GoaVolumeMonitor' unit='gvfs-goa-volume-monitor.service' requested by ':1.7' (uid=1000 pid=1691 comm="pcmanfm --desktop --profile LXDE-pi ")
Jul 14 20:47:08 box dbus-daemon[1522]: [session uid=1000 pid=1522] Successfully activated service 'org.gtk.vfs.GoaVolumeMonitor'
Jul 14 20:47:08 box dbus-daemon[1522]: [session uid=1000 pid=1522] Activating via systemd: service name='org.gtk.vfs.MTPVolumeMonitor' unit='gvfs-mtp-volume-monitor.service' requested by ':1.7' (uid=1000 pid=1691 comm="pcmanfm --desktop --profile LXDE-pi ")
Jul 14 20:47:08 box dbus-daemon[1522]: [session uid=1000 pid=1522] Successfully activated service 'org.gtk.vfs.MTPVolumeMonitor'
Jul 14 20:47:08 box dbus-daemon[1522]: [session uid=1000 pid=1522] Activating via systemd: service name='org.gtk.vfs.AfcVolumeMonitor' unit='gvfs-afc-volume-monitor.service' requested by ':1.7' (uid=1000 pid=1691 comm="pcmanfm --desktop --profile LXDE-pi ")
Jul 14 20:47:08 box dbus-daemon[1522]: [session uid=1000 pid=1522] Successfully activated service 'org.gtk.vfs.AfcVolumeMonitor'
Jul 14 20:47:09 box dhcpcd-run-hooks[1862]: wlan0: 50-iiab CARRIER change wlan0
Jul 14 20:47:09 box dhcpcd-run-hooks[1869]: wlan0: 50-iiab set channel 3
Jul 14 20:47:09 box dhcpcd-run-hooks[1874]: wlan0: 50-iiab set ap0 up CARRIER
Jul 14 20:47:15 box dnsmasq[1567]: reading /run/dnsmasq/resolv.conf
Jul 14 20:47:15 box dnsmasq[1567]: using local addresses only for domain lan
Jul 14 20:47:15 box dnsmasq[1567]: using nameserver 10.0.0.11#53
Jul 14 20:47:15 box dhcpcd-run-hooks[2191]: wlan0: 50-iiab set ap0 up BOUND
Jul 14 20:48:16 box vmunix: [   73.404500] v3d fec00000.v3d: MMU error from client L2T (0) at 0x6721000, pte invalid

holta, 2021-07-15:
@deltacloud9 there are several issues here:

1) Did you try `iiab-hotspot-off` then `raspi-config` (to [dis]associate with any upstream WiFi access point) then `iiab-hotspot-on` as mentioned at https://github.com/iiab/iiab/wiki/IIAB-7.2-Release-Notes#whats-upgraded ?  Running as root of course.

2) In the case of "Raspberry Pi OS with desktop" @jvonau mentions that they've documented this cannot work when hostapd is running (i.e. when IIAB's hotspot is live, for up-to-32 students in a classroom).

3) Almost every Graphical/Desktop OS uses a different interactive widget (typically in the top-right of the OS's graphical desktop) to connect to WiFi access points &mdash; that tend to clash with the OS's low-level networking internals &mdash; e.g. when hostapd is turned on, etc!

SUMMARY: IIAB _cannot_ work around all these gotchas &mdash; but we definitely _should_ document the very most important of these "UX Confusions" &mdash; _so that everyday users can quickly Google for practical and pragmatic workarounds._

FUTURE: @jvonau can say more if there's a particular bug you're chasing down, that you need help with (please do send details if so).  Also keep in mind: all of this might change when Raspberry Pi OS 11 is released very shortly now, quite likely even before Debian 11 is released on/around its "2021-07-31" ETA.

deltacloud9, 2021-07-15:
> @deltacloud9 there are several issues here:
> 
> 1. Did you try `iiab-hotspot-off` then `raspi-config` (to [dis]associate with any upstream WiFi access point) then `iiab-hotspot-on` as mentioned at https://github.com/iiab/iiab/wiki/IIAB-7.2-Release-Notes#whats-upgraded ?  Running as root of course.
> 2. In the case of "Raspberry Pi OS with desktop" @jvonau mentions that they've documented this cannot work when hostapd is running (i.e. when IIAB's hotspot is live, for up-to-32 students in a classroom).
> 3. Almost every Graphical/Desktop OS uses a different interactive widget (typically in the top-right of the OS's graphical desktop) to connect to WiFi access points — that tend to clash with the OS's low-level networking internals — e.g. when hostapd is turned on, etc!
> 
> SUMMARY: IIAB _cannot_ work around all these gotchas — but we definitely _should_ document the very most important of these "UX Confusions" — _so that everyday users can quickly Google for practical and pragmatic workarounds._
> 
> FUTURE: @jvonau can say more if there's a particular bug you're chasing down, that you need help with (please do send details if so). Also keep in mind: all of this might change when Raspberry Pi OS 11 is released very shortly now, quite likely even before Debian 11 is released on/around its "2021-07-31" ETA.

@holta looks like turning on/off the iiab-hotspot did the trick! I am now able to see my WiFi networks list.

Thank you!!

holta, 2021-07-16:
@deltacloud9 can you review http://FAQ.IIAB.IO -> "23 Any other networking tips?"

Any focused revisions required there?

deltacloud9, 2021-07-16:
Not sure if we should add my issue or not?

Don't think it is as necessary since it may have been a glitch within the
OS.

Still, it may be worth having there.

holta, 2021-07-16:
> Not sure if we should add my issue or not?

Can you force a hard reload of http://FAQ.IIAB.IO ?

Currently the 3rd bullet point reads as follows:

> If your IIAB Raspberry Pi is connected to the Internet over Wi-Fi, and you need to change its upstream Wi-Fi connection: run "sudo iiab-hotspot-off" then "sudo raspi-config" (to [dis]associate from/to any upstream WiFi access point) then "sudo iiab-hotspot-on" as outlined at: https://github.com/iiab/iiab/wiki/IIAB-7.2-Release-Notes#whats-upgraded

deltacloud9, 2021-07-16:
Yes, this is perfect! I may have just gotten lucky but hoping this will
help anyone in the future with this problem!

jvonau, 2021-07-27:
Hum... judging by the appearance of kalite after the network stanza in iiab.ini suggests that kalite was added post-install for some reason, depending on the path that was chosen for this post-install action kalite may not be fully configured as some of the kalite install functions are performed in ['iiab'](https://github.com/iiab/iiab-factory/pull/175) with no mention of these additional steps within the kalite role's [readme.](https://github.com/iiab/iiab/tree/master/roles/kalite#readme) Now I'm curious @deltacloud9 on how this additional install of kalite was performed and with what steps you used given this machine stated off as a "Medical IIAB' install.

holta, 2021-07-27:
Is it possible the installation of KA Lite was triggered by Content Pack 'en-kalite_health' ?

@tim-moody & @deltacloud9 can you explain a bit more the sequence of events, as to how this likely arose?

jvonau, 2021-07-27:
The 2 kalite functions should be abstracted into other scrips that could be called from within 'iiab' or by the admin-console as needed.

jvonau, 2021-08-08:
Is there still a 'bug' here or have the docs been cleaned up enough?

deltacloud9, 2021-08-08:
I believe there is still a bug here. Let me test today and I will get back
to you.

holta, 2021-08-08:
Thanks Matt!

Just FYI a major PR was merged about an hour ago:

- PR #2850

jvonau, 2021-08-16:
I've seen "pinned issues" used elsewhere

jvonau, 2021-08-17:
Let's close this one and if something new pop up open a new issue for the specific problem.

holta, 2021-08-17:
> Let's close this one and if something new pop up open a new issue for the specific problem

@deltacloud9: Good enough?

deltacloud9, 2021-08-17:
Yeah go ahead and close this for now, i’ll advise if the issue occurs again.

jvonau, 2021-08-28:
>[Graphical desktop OS's need googlable docs, so people try: iiab-hotspot-off -> raspi-config -> iiab-hotspot-on]

The idea behind 'pinning' to reserve some real estate on the issue's landing page so end users can easily find this issue for future reference like the title of this issue suggests and not used to promote the latest issues that tickle TPTB fancy at this moment in time. Why is this issue still open anyway?

holta, 2021-08-28:
I'll work on better and more Googlable documentation in coming weeks.

If @deltacloud9 can also help along the way, that would be great.

(Community Documentation may never be perfect, when we're at the mercy of desktop OS's especially, but we definitely have room for improvement here.)

deltacloud9, 2021-08-28:
Yeah, absolutely I can assist with documentation. I do a lot of
documentation for my full time job so I am used to it.

Let me know what you would like me to do.

holta, 2022-05-23:
@deltacloud9

Docs & techniques have improved substantially over the past month:

- PR #3179
- PR #3222
- https://wiki.iiab.io/go/FAQ#Can_I_create_a_Wi-Fi_hotspot_using_an_old_laptop%3F

Close this ticket soon?!

jvonau, 2025-01-27:
status?

closed 2025-01-28 by holta