iiab/iiab #2864: php-stem for PHP 7.4 (e.g. Raspberry Pi OS 11, Debian 11, Ubuntu 21.04) and PHP 8.0 (e.g. Ubuntu 21.10)

holta, 2021-07-15:
As discussed on this week's http://minutes.iiab.io call-

We'll work on adding these soon (e.g. when Raspberry Pi OS 11 is released in not so many days) which will need this or similar:

- http://d.iiab.io/packages/php7.4-stem.rpi.tar

We should probably also compile php-stem files for PHP 8.0 (released November 2020) as IIAB has been well-tested on Ubuntu 21.10 daily builds over the past month:

- #2814
- #2818

(Creating these should be easier than in the past thanks to https://github.com/iiab/php-stem that @tim-moody set up recently!)

Recap: http://oer2go.org/viewmod/es-wikihow needs this for full-text searching.

jvonau, 2021-08-17:
How about using the pinned issue for FAQ type subjects in place of frivolous issues that are targeted at a single fix of a 5 year old php website scraping. Use this real estate wisely.

jvonau, 2021-12-10:
Need to build and provide php7.4-stem.rpi.tar now that a 32bit version of php7.4 is installed in RaspiOS bullseye, there are only arm64 and amd64 available to download that are used to support Ubuntu but the arm64 version should support the 64bit version of RaspiOS bullseye as is.

holta, 2021-12-10:
> Need to build and provide php7.4-stem.rpi.tar now that a 32bit version of php7.4 is installed in RaspiOS bullseye

Yes as the extremely popular 3.2 GB https://rachel.worldpossible.org/viewmod/es-GCF2015 will again be searchable across ~600 million Spanish-speakers in ~30 countries.

If https://github.com/iiab/php-stem can be made to work without compile errors ??

Reason:

1) The 20GB http://download.kiwix.org/zim/wikihow/wikihow_es_maxi_2021-10.zim bodes well but is too big for many schools :/

2) Likewise @tim-moody's http://iiab.me/modules/es-gcf_learn_2021/edu.gcfglobal.org/es/topics/ bodes extremely well.

   But it's not searchable last I checked (hopefully Tim can update us if that's changed?)

   FYI this es-gcf_learn_2021 appears to be about 15GB according to: https://github.com/iiab-share/iiab-content/blob/main/catalogs/iiab-module-cat.json#L724 so a subsetting strategy...might also make a lot of sense.

holta, 2021-12-26:
PHP 7.4 is finally taken care of!  Thanks to:

- PR iiab/php-stem#3
- PR #3080
- PR #3081 
- PR #3082

_Compiling the [3 stem.so files](https://github.com/iiab/php-stem/tree/main/so) for PHP 8.0 and/or 8.1 might be harder ?_

In any case, that question can be addressed if they are needed after 2022-04-21, when Ubuntu 22.04 LTS ("Jammy Jellyfish") is released.

holta, 2022-01-03:
FYI there was speculation during Dec 30th's http://minutes.iiab.io call that [stem-php might not be that hard to compile](https://github.com/iiab/php-stem) for PHP 8.0 and/or 8.1 after all?

So we'll try again in a few months — by/around the time Ubuntu 22.04 (with PHP 8.1) is released on 2022-04-21.

holta, 2022-12-08:
Just FYI PHP 8.2 is released.

We'll monitor the consequences as OS's like Debian 12 "Bookworm" and/or Ubuntu 23.04 "Lunar Lobster" begin to incorporate this.

holta, 2022-12-22:
FYI new IIAB installs now auto-detect the OS's PHP version (whether or not PHP is installed) by parsing `apt list php` :

- PR #3439

holta, 2022-12-30:
Huge thanks to @randytate and @tim-moody who brought us this far.

PHP 7.4 and 8.1 are the most important versions if we support this, as Ubuntu-like OS's bypassed PHP 8.0 entirely.

Then PHP 8.2 (released earlier this month) may well dominate in a year from now, if in the end that's what Debian 12 Bookworm and Raspberry Pi OS Bookworm finalize upon in Q1 or Q2 of 2023.

holta, 2023-01-16:
> PHP 7.4 and 8.1 are the most important versions if we support this

Just FYI PHP 7.4 was end-of-life'd 2 months ago in November 2022 ("there will be no bug or security fixes made to any of the PHP 7 versions") as mentioned here:

- #3467

PRAGMATICALLY: RasPiOS Bullseye will of course continue to be very popular with PHP 7.4 for a few more months, even despite :smile:

jvonau, 2023-01-20:
@holta @tim-moody @avni Given this entire php-stem issue involves one old wikihow Spanish OER module have you considered giving the latest [zim](https://mirrors.dotsrc.org/kiwix/zim/wikihow/?C=M&O=D) version a whirl? Going with the zim file, should it work, would negate the need for maintaining php-stem going forward.

holta, 2023-01-20:
Indeed a good idea:

While wikiHow ZIM files had many serious problems in 2022, hopefully these problems are fully going away in 2023.

holta, 2023-03-31:
Nextcloud has moved away from the EOL'd PHP 7.4 and we should soon too...

(Especially now that wikiHow ZIM files are likely good enough going forward.)

closed 2023-03-31 by holta