iiab/iiab #2868: Validate just-released Node-RED 2.0.0: did dependencies slightly changed?

holta, 2021-07-20:
Node-RED 2.0.0 is now part of IIAB, and its dependencies/requirements should be quickly verified here:

- https://nodered.org/blog/2021/07/20/version-2-0-released
- https://github.com/node-red/node-red/blob/master/CHANGELOG.md
- https://nodered.org/docs/

To check that everything looks solid here: https://github.com/iiab/iiab/blob/master/roles/nodejs/tasks/install.yml !

holta, 2021-08-11:
All seems to work.

But a more formal checkup into Node-RED's official/latest requirements would also be wise in the coming year.

jvonau, 2022-04-05:
status?

holta, 2022-04-05:
Node-RED is probably working on many more OS's now (?) but should be verified on Ubuntu 22.04 pre-releases in coming days/weeks.

(Ubuntu 21.10 is no longer the priority...)

holta, 2022-05-23:
Seems good...with Node-RED 3.0.0 landing shortly!

closed 2022-05-23 by holta