iiab/iiab #2873: JupyterHub + Node-RED + Moodle + Internet Archive install problems on RaspiOS-for-PC's Node.js 10.x -- lacking Node.js 16.x, which requires amd64 not i386 python3

holta, 2021-07-21:
Strangely I succeeded in installing most of BIG-sized IIAB onto RaspiOS-for-PC/VM (graphical desktop version, latest ISO is [2021-01-12](https://downloads.raspberrypi.org/rpd_x86/images/rpd_x86-2021-01-12/2021-01-11-raspios-buster-i386.iso)) _BUT_ Sugarizer was the only Node.js role that I cared to actually make work.  As explained below (just for the record!) as it's probably NOT worth solving these issues.

iiab-diagnostics output: http://sprunge.us/xuBc6j?en

FWIW I did not try IIAB's 5th Node.js role (pbx) as it's not currently part of [BIG-sized IIAB](http://FAQ.IIAB.IO#What_services_.28IIAB_apps.29_are_suggested_during_installation.3F).

## Node.js 10.x PROBLEM NUMBER 1 OF 4: JupyterHub

```
TASK [jupyterhub : Use pip to install into a virtual environment: /opt/iiab/jupyterhub (~229 MB)] **********************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": ["/opt/iiab/jupyterhub/bin/pip3", "install", "--no-cache-dir", "pip", "wheel", "ipywidgets", "jupyterhub", "jupyterlab", "jupyterhub_firstuseauthenticator", "jupyterhub-systemdspawner"], "msg": "stdout: Already using interpreter /usr/bin/python3\nUsing base prefix '/usr'\nNew python executable in /opt/iiab/jupyterhub/bin/python3\nAlso creating executable in /opt/iiab/jupyterhub/bin/python\nInstalling setuptools, pkg_resources, pip, wheel...done.\nLooking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple\nRequirement already satisfied: pip in /opt/iiab/jupyterhub/lib/python3.7/site-packages (21.1.3)\nRequirement already satisfied: wheel in /opt/iiab/jupyterhub/lib/python3.7/site-packages (0.36.2)\nCollecting ipywidgets\n  Downloading https://www.piwheels.org/simple/ipywidgets/ipywidgets-7.6.3-py2.py3-none-any.whl (121 kB)\nCollecting jupyterhub\n  Downloading https://www.piwheels.org/simple/jupyterhub/jupyterhub-1.4.2-py3-none-any.whl (3.6 MB)\nCollecting jupyterlab\n  Downloading https://www.piwheels.org/simple/jupyterlab/jupyterlab-3.0.16-py3-none-any.whl (8.2 MB)\nCollecting jupyterhub_firstuseauthenticator\n  Downloading https://www.piwheels.org/simple/jupyterhub-firstuseauthenticator/jupyterhub_firstuseauthenticator-0.14.1-py3-none-any.whl (7.6 kB)\nCollecting jupyterhub-systemdspawner\n  Downloading https://www.piwheels.org/simple/jupyterhub-systemdspawner/jupyterhub_systemdspawner-0.15.0-py3-none-any.whl (8.4 kB)\nCollecting jupyterlab-widgets>=1.0.0\n  Downloading https://www.piwheels.org/simple/jupyterlab-widgets/jupyterlab_widgets-1.0.0-py3-none-any.whl (243 kB)\nCollecting traitlets>=4.3.1\n  Downloading https://www.piwheels.org/simple/traitlets/traitlets-5.0.5-py3-none-any.whl (100 kB)\nCollecting ipykernel>=4.5.1\n  Downloading https://www.piwheels.org/simple/ipykernel/ipykernel-6.0.3-py3-none-any.whl (122 kB)\nCollecting ipython>=4.0.0\n  Downloading https://www.piwheels.org/simple/ipython/ipython-7.25.0-py3-none-any.whl (786 kB)\nCollecting nbformat>=4.2.0\n  Downloading https://www.piwheels.org/simple/nbformat/nbformat-5.1.3-py3-none-any.whl (178 kB)\nCollecting widgetsnbextension~=3.5.0\n  Downloading https://www.piwheels.org/simple/widgetsnbextension/widgetsnbextension-3.5.1-py2.py3-none-any.whl (2.2 MB)\nCollecting matplotlib-inline<0.2.0,>=0.1.0\n  Downloading https://www.piwheels.org/simple/matplotlib-inline/matplotlib_inline-0.1.2-py3-none-any.whl (8.2 kB)\nCollecting jupyter-client<7.0\n  Downloading https://www.piwheels.org/simple/jupyter-client/jupyter_client-6.1.12-py3-none-any.whl (112 kB)\nCollecting importlib-metadata<4\n  Downloading https://www.piwheels.org/simple/importlib-metadata/importlib_metadata-3.10.1-py3-none-any.whl (14 kB)\nCollecting tornado<7.0,>=4.2\n  Downloading tornado-6.1-cp37-cp37m-manylinux2010_i686.whl (428 kB)\nCollecting debugpy<2.0,>=1.0.0\n  Downloading debugpy-1.3.0-cp37-cp37m-manylinux2014_i686.whl (4.5 MB)\nCollecting zipp>=0.5\n  Downloading https://www.piwheels.org/simple/zipp/zipp-3.5.0-py3-none-any.whl (5.7 kB)\nCollecting typing-extensions>=3.6.4\n  Downloading https://www.piwheels.org/simple/typing-extensions/typing_extensions-3.10.0.0-py3-none-any.whl (26 kB)\nCollecting prompt-toolkit!=3.0.0,!=3.0.1,<3.1.0,>=2.0.0\n  Downloading https://www.piwheels.org/simple/prompt-toolkit/prompt_toolkit-3.0.19-py3-none-any.whl (368 kB)\nCollecting jedi>=0.16\n  Downloading https://www.piwheels.org/simple/jedi/jedi-0.18.0-py2.py3-none-any.whl (1.4 MB)\nRequirement already satisfied: setuptools>=18.5 in /opt/iiab/jupyterhub/lib/python3.7/site-packages (from ipython>=4.0.0->ipywidgets) (57.4.0)\nCollecting pygments\n  Downloading https://www.piwheels.org/simple/pygments/Pygments-2.9.0-py3-none-any.whl (1.0 MB)\nCollecting pexpect>4.3\n  Downloading https://www.piwheels.org/simple/pexpect/pexpect-4.8.0-py2.py3-none-any.whl (59 kB)\nCollecting pickleshare\n  Downloading https://www.piwheels.org/simple/pickleshare/pickleshare-0.7.5-py2.py3-none-any.whl (6.9 kB)\nCollecting decorator\n  Downloading https://www.piwheels.org/simple/decorator/decorator-5.0.9-py3-none-any.whl (8.9 kB)\nCollecting backcall\n  Downloading https://www.piwheels.org/simple/backcall/backcall-0.2.0-py2.py3-none-any.whl (11 kB)\nCollecting parso<0.9.0,>=0.8.0\n  Downloading https://www.piwheels.org/simple/parso/parso-0.8.2-py2.py3-none-any.whl (94 kB)\nCollecting pyzmq>=13\n  Downloading pyzmq-22.1.0-cp37-cp37m-manylinux1_i686.whl (1.1 MB)\nCollecting python-dateutil>=2.1\n  Downloading https://www.piwheels.org/simple/python-dateutil/python_dateutil-2.8.2-py2.py3-none-any.whl (247 kB)\nCollecting jupyter-core>=4.6.0\n  Downloading https://www.piwheels.org/simple/jupyter-core/jupyter_core-4.7.1-py3-none-any.whl (82 kB)\nCollecting jsonschema!=2.5.0,>=2.4\n  Downloading https://www.piwheels.org/simple/jsonschema/jsonschema-3.2.0-py2.py3-none-any.whl (56 kB)\nCollecting ipython-genutils\n  Downloading https://www.piwheels.org/simple/ipython-genutils/ipython_genutils-0.2.0-py2.py3-none-any.whl (26 kB)\nCollecting attrs>=17.4.0\n  Downloading https://www.piwheels.org/simple/attrs/attrs-21.2.0-py2.py3-none-any.whl (53 kB)\nCollecting six>=1.11.0\n  Downloading https://www.piwheels.org/simple/six/six-1.16.0-py2.py3-none-any.whl (11 kB)\nCollecting pyrsistent>=0.14.0\n  Downloading pyrsistent-0.18.0-cp37-cp37m-manylinux1_i686.whl (112 kB)\nCollecting ptyprocess>=0.5\n  Downloading https://www.piwheels.org/simple/ptyprocess/ptyprocess-0.7.0-py2.py3-none-any.whl (13 kB)\nCollecting wcwidth\n  Downloading https://www.piwheels.org/simple/wcwidth/wcwidth-0.2.5-py2.py3-none-any.whl (30 kB)\nCollecting notebook>=4.4.1\n  Downloading https://www.piwheels.org/simple/notebook/notebook-6.4.0-py3-none-any.whl (9.5 MB)\nCollecting Send2Trash>=1.5.0\n  Downloading https://www.piwheels.org/simple/send2trash/Send2Trash-1.7.1-py3-none-any.whl (17 kB)\nCollecting terminado>=0.8.3\n  Downloading https://www.piwheels.org/simple/terminado/terminado-0.10.1-py3-none-any.whl (14 kB)\nCollecting jinja2\n  Downloading https://www.piwheels.org/simple/jinja2/Jinja2-3.0.1-py3-none-any.whl (133 kB)\nCollecting nbconvert\n  Downloading https://www.piwheels.org/simple/nbconvert/nbconvert-6.1.0-py3-none-any.whl (551 kB)\nCollecting prometheus-client\n  Downloading https://www.piwheels.org/simple/prometheus-client/prometheus_client-0.11.0-py2.py3-none-any.whl (56 kB)\nCollecting argon2-cffi\n  Downloading argon2-cffi-20.1.0.tar.gz (1.8 MB)\n  Installing build dependencies: started\n  Installing build dependencies: finished with status 'done'\n  Getting requirements to build wheel: started\n  Getting requirements to build wheel: finished with status 'done'\n    Preparing wheel metadata: started\n    Preparing wheel metadata: finished with status 'done'\nCollecting requests\n  Downloading https://www.piwheels.org/simple/requests/requests-2.26.0-py2.py3-none-any.whl (62 kB)\nCollecting oauthlib>=3.0\n  Downloading https://www.piwheels.org/simple/oauthlib/oauthlib-3.1.1-py2.py3-none-any.whl (146 kB)\nCollecting entrypoints\n  Downloading https://www.piwheels.org/simple/entrypoints/entrypoints-0.3-py2.py3-none-any.whl (11 kB)\nCollecting pamela\n  Downloading https://www.piwheels.org/simple/pamela/pamela-1.0.0-py2.py3-none-any.whl (5.8 kB)\nCollecting alembic>=1.4\n  Downloading https://www.piwheels.org/simple/alembic/alembic-1.6.5-py2.py3-none-any.whl (164 kB)\nCollecting certipy>=0.1.2\n  Downloading https://www.piwheels.org/simple/certipy/certipy-0.1.3-py3-none-any.whl (22 kB)\nCollecting jupyter-telemetry>=0.1.0\n  Downloading https [... cut ...]
## CONCLUSION: I set the following in [/etc/iiab/local_vars.yml](http://FAQ.IIAB.IO#What_is_local_vars.yml_and_how_do_I_customize_it.3F) &mdash; and completed the IIAB install experiment, by repeatedly running `sudo iiab`

```
jupyterhub_install: False
jupyterhub_enabled: False

nodered_install: False
nodered_enabled: False

internetarchive_install: False
internetarchive_enabled: False
```

This experiment is thanks to @deltacloud9 and @jvonau's work yesterday on:

- #2749
- #2820
- PR #2862
- PR #2870 
- PR #2871
- #2872

holta, 2021-07-21:
An "ENTIRELY AT YOUR OWN RISK" WARNING summarizing this entire RaspiOS-on-PC situation is now posted at the bottom of SECTION 2 here: https://github.com/iiab/iiab/wiki/IIAB-Platforms

Specifically:

> [Debian Buster with Raspberry Pi Desktop](https://downloads.raspberrypi.org/rpd_x86/images/) (i.e. for x86_64 PC/VM) will NOT work with modern Node.js apps — as a result of this distro's hybrid 32-bit and 64-bit CPU arch legacy (specifically, i386-instead-of-amd64 versions of python3 packages) which lead to serious Node.js 10.x problems across the board. Sugarizer can be made to work if absolutely necessary. But this is ENTIRELY AT YOUR OWN RISK. Extensive details written up on 2021-07-21: #2863

closed 2021-07-21 by holta

jvonau, 2021-07-21:
What does `dpkg --print-architecture` and `dpkg --print-foreign-architectures` return? Think that image was build with a i386 userspace see https://github.com/RPi-Distro/pi-gen/pull/415/files. On a side note the /etc/rpi-issue's git hash is nowhere to be found on github

holta, 2021-07-21:
> What does `dpkg --print-architecture` and `dpkg --print-foreign-architectures` return?

```
root@raspios10-desk-vm:~# dpkg --print-architecture
i386
```

```
root@raspios10-desk-vm:~# dpkg --print-foreign-architectures
amd64
```

Interesting!

jvonau, 2021-07-27:
Please add both `dpkg --print-architecture` `dpkg --print-foreign-architectures` to iiab-diagnostics to add future debugging of other goofy oddball combinations that show up out of the blue.

holta, 2021-07-27:
> Please add both `dpkg --print-architecture` `dpkg --print-foreign-architectures` to iiab-diagnostics to add future debugging of other goofy oddball combinations that show up out of the blue.

Done.  See:

- PR #2880 

If you would like the ordering changed (within the pastebins) please LMK.