iiab/iiab #2904: "IIAB destroys the Internet connection"

holta, 2021-08-03:
@darkenvy reports:

http://download.iiab.io/install.txt on option 1 (small) fails pretty early.

I'm not sure what it does so early on, but it destroys the internet connection. It won't connect via WiFi nor Ethernet and establish an ip address.

I'm not an expert at analyzing ansible... but it fails when grabbing download.iiab.io/packages/kiwix-tools_linux-armhf-3.1.2-5.tar.gz because there is suddenly no internet :P

(I'm connected to the pi via serial. That's how I initiated the script).

(The Pi is not broadcasting "Internet in a box" wifi yet. So it's stuck in this bad networking state. Can't fix it with dhcpcd/wpa_supplicant restarting. However, I was able to get a 169.254.xxx.xxx IP)

jvonau, 2021-08-04:
> @darkenvy reports:
> 
> http://download.iiab.io/install.txt on option 1 (small) fails pretty early.
> 
> I'm not sure what it does so early on, but it destroys the internet connection. It won't connect via WiFi nor Ethernet and establish an ip address.
>

Filed under too vague.. 
 > I'm not an expert at analyzing ansible... but it fails when grabbing download.iiab.io/packages/kiwix-tools_linux-armhf-3.1.2-5.tar.gz because there is suddenly no internet :P
> 

which implies that stages 1-5 installed with the internet just fine, the cause of the internet outage is the point of debate.

> (I'm connected to the pi via serial. That's how I initiated the script).
> 

How much scroll back does you terminal have? Can you grab the last few stanzas before the failure or summit iiab-diagnostics

> (The Pi is not broadcasting "Internet in a box" wifi yet. So it's stuck in this bad networking state. Can't fix it with dhcpcd/wpa_supplicant restarting. However, I was able to get a 169.254.xxx.xxx IP)

If the install is only at stage 6 the wifi is not configured yet. The question is why did the internet go away... Does rebooting restore the internet?

darkenvy, 2021-08-04:
http://sprunge.us/EKK6XS Line 1363.
I don't see any network changes prior to this point. Perhaps it changed way early and just happened to avoid network activity until it tried to download kiwix-tools?

jvonau, 2021-08-04:
```
2021-08-03 20:31:30,539 p=2804 u=root n=ansible | TASK [kiwix : Force Ansible to exit (FAIL) if kiwix-tools appears unavailable for your architecture (armv7l)] ***
2021-08-03 20:31:30,586 p=2804 u=root n=ansible | skipping: [127.0.0.1]
2021-08-03 20:31:30,626 p=2804 u=root n=ansible | TASK [kiwix : Set fact 'kiwix_src_file: kiwix-tools_linux-armhf-3.1.2-5.tar.gz'] ***
2021-08-03 20:31:30,682 p=2804 u=root n=ansible | ok: [127.0.0.1]
2021-08-03 20:31:30,721 p=2804 u=root n=ansible | TASK [kiwix : Download http://download.iiab.io/packages/kiwix-tools_linux-armhf-3.1.2-5.tar.gz to /opt/iiab/downloads
2021-08-03 20:31:34,858 p=2804 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": false, "dest": "/opt/iiab/downloads/kiwix-tools_linux-armhf-3.1.2-5.tar.gz", "elapsed": 3, "msg": "Request failed: <urlopen error [Errno 113] No route to host>", "url": "http://download.iiab.io/packages/kiwix-tools_linux-armhf-3.1.2-5.tar.gz"}
2021-08-03 20:31:34,861 p=2804 u=root n=ansible | PLAY RECAP *********************************************************************
2021-08-03 20:31:34,861 p=2804 u=root n=ansible | 127.0.0.1                  : ok=260  changed=143  unreachable=0    failed=1    skipped=81   rescued=0    ignored=2   

```
>No route to host

Something caused your wifi connection to go away and that is an outside issue, suggest just `sudo iiab` after painfully watching ansible reinstall it should pickup where you left off, that is the whole point of having an interruptible installer..

holta, 2021-08-04:
> http://sprunge.us/EKK6XS Line 1363

Just FYI this is web-browsable by tacking 3-characters on the end:

http://sprunge.us/EKK6XS?en

jvonau, 2021-08-04:
>  316 ?        00:00:00   sh
  318 ?        00:00:00     ifup
  575 ?        00:00:00       sh
  576 ?        00:00:00         dhclient
  577 ?        00:00:00           dhclient

So what script are these commands coming from? that is not part of the stock dhcpcd layout...

>Aug  4 01:27:40 box dhcpcd[1016]: wlan0: NAK: from 10.0.0.1
Aug  4 01:27:42 box dhcpcd[1016]: wlan0: NAK: from 10.0.0.1
Aug  4 01:27:45 box dhcpcd[1016]: wlan0: NAK: from 10.0.0.1
Aug  4 01:27:49 box dhclient[577]: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 5
Aug  4 01:27:49 box sh[316]: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 5
Aug  4 01:27:49 box dhclient[577]: DHCPOFFER of 10.0.0.19 from 10.0.0.1
Aug  4 01:27:49 box sh[316]: DHCPOFFER of 10.0.0.19 from 10.0.0.1
Aug  4 01:27:49 box sh[316]: DHCPREQUEST for 10.0.0.19 on wlan0 to 255.255.255.255 port 67
Aug  4 01:27:49 box dhclient[577]: DHCPREQUEST for 10.0.0.19 on wlan0 to 255.255.255.255 port 67
Aug  4 01:27:57 box dhclient[577]: DHCPREQUEST for 10.0.0.19 on wlan0 to 255.255.255.255 port 67
Aug  4 01:27:57 box sh[316]: DHCPREQUEST for 10.0.0.19 on wlan0 to 255.255.255.255 port 67

darkenvy, 2021-08-04:
Yes... `sudo iiab` would pick back up.... if I had internet. (I did try it btw. Same result, but it fails on the initial step of apt update).

I have spent the better half of the afternoon trying to get the internet back. It seems dhcpcd is getting a ip offered from the router (10.0.0.19 specifically) but it wont take it. Instead it assigns a bogus `169.254.xx.xx` address (indicating that there was a problem). `sudo service dhcpcd status` returns green. No error.

I checked wpa_supplicant too for good measure. Nothing changed. I assumed iiab toasted the internet.

jvonau, 2021-08-04:
whatever is running dhclient is toasting the network...

darkenvy, 2021-08-04:
I just finished burning a second SD card. So I'll be up and running on a second try in a minute. 
I figure that installation is kapput. I thought it was weird though because even ethernet failed. The only reason I was able to communicate with the Pi was because Im using serial.

jvonau, 2021-08-04:
Before you install iiab grab `ps -AH` something looks a bit funny if dhclient is running... that would be in competition with dhcpcd for control of the interface...  Something else is at play any other steps you take to configure the network on the rpi?

darkenvy, 2021-08-04:
on the former setup, I performed `sudo raspi-config` to configure the WiFi. `curl d.iiab.io/install.txt | sudo bash` was the command that followed.

jvonau, 2021-08-04:
or even favorite packages or a custom desktop on top of the stock install?

darkenvy, 2021-08-04:
Nothing. I didn't even install vim (to my regret. It really sucks using vi XD). But I don't have the `history` anymore to prove it LOL.
You know, I had a SD card corrupt on me while using it before. Caused a kernel panic. Im not ruling this out as a possibility just yet.

`ps -AH` on a fresh install: http://sprunge.us/iwEqAN

darkenvy, 2021-08-04:
Oh it also occurred to me that dhclient could be a red-herring. Me trying to fix the internet AFTER the issue arose. Although, I didn't [intentionally] start dhclient directly.

jvonau, 2021-08-04:
ps -AH ordering looks normal how about at `ip a` the only thing I could think of is ipv6 being disabled in 2-common and that taking awhile to pop up when the lease renews...

darkenvy, 2021-08-04:
I started `sudo iiab` before being able to run `ip a` first. But the bug happened again. 

Here is `ip a` after the bug presented itself:
```
pi@raspberrypi:~$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether dc:a6:32:08:52:86 brd ff:ff:ff:ff:ff:ff
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether dc:a6:32:08:52:87 brd ff:ff:ff:ff:ff:ff
4: ap0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast master br0 state UP group default qlen 1000
    link/ether 02:3c:d5:a8:5a:be brd ff:ff:ff:ff:ff:ff
5: br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 02:3c:d5:a8:5a:be brd ff:ff:ff:ff:ff:ff
    inet 172.18.96.1/19 brd 172.18.127.255 scope global noprefixroute br0
       valid_lft forever preferred_lft forever
```

history:
```
pi@raspberrypi:~$ history
    1  clear
    2  ps -AH
    3  ip a
    4  ip a | less
    5  ps -AH | less
    6  ps -AH | curl -F 'sprunge=<-' http://sprunge.us
    7  sudo iiab
    8  ip a
    9  history
```

jvonau, 2021-08-04:
> I started `sudo iiab` before being able to run `ip a` first. But the bug happened again.
>
> Here is `ip a` after the bug presented itself:
>
> ```
> pi@raspberrypi:~$ ip a
> 1: lo: <LOOPBACK,UP,LOWER_UP> mtu [...]
The above suggests iiab-install was completed as the network was configured this time as ap0 is present as opposed to crapping out earlier.. The bug is in a crappy upstream wifi connection or that connection is ipv6...

jvonau, 2021-08-04:
On a positive note the hotspot should be active, but without iiab-diagnostics I can't tell the current state of the installation.

darkenvy, 2021-08-04:
To rule out wifi issues, the pi4 was plugged into ethernet this installation.

darkenvy, 2021-08-04:
PS: I can connect to "Internet in a Box" AP, but can't access https://www.box.lan/. So it may have crapped out too early

jvonau, 2021-08-04:
> To rule out wifi issues, the pi4 was plugged into ethernet this installation.

Now that is real odd, wired ethernet failing to grab a lease, that is why I was leaning towards being ipv6 related... or maybe a real slow dhcp server on your LAN. From the serial console `sudo systemctl status dhcpcd` should shed some light... and `sudo systemctl restart dhcpcd` should just work to put it back online...

jvonau, 2021-08-04:
> PS: I can connect to "Internet in a Box" AP, but can't access https://www.box.lan/. So it may have crapped out too early

Most likely, just noted about where the install progressed to based on the observed facts. Now if restarting dhcpcd or rebooting doesn't clear up the wired interface's default gateway status then there some merit at looking at the changes done in 2-common... What is really needed is a snapshot of `ip a` before starting the install, then we might know for sure your upstream connection looks like.

jvonau, 2021-08-04:
https://github.com/iiab/iiab/pull/2883#discussion_r678091391 https://github.com/iiab/iiab/issues/2650#issuecomment-734899110 https://github.com/iiab/iiab/issues/2650#issuecomment-735313747

darkenvy, 2021-08-04:
New setup. First command:
```
pi@raspberrypi:~$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether dc:a6:32:08:52:86 brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.25/24 brd 10.0.0.255 scope global dynamic noprefixroute eth0
       valid_lft 86314sec preferred_lft 75514sec
    inet6 fe80::de79:93d9:51f2:b8f5/64 scope link 
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether dc:a6:32:08:52:87 brd ff:ff:ff:ff:ff:ff
```

holta, 2021-08-04:
@darkenvy: Please a fresh [sudo iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md) pastebin URL when this failure next occurs, Thanks!

jvonau, 2021-08-04:
Ethernet going away and not recovering suggests an issue with the [power supply](https://kalitut.com/raspberry-pi-network-error/#Raspberry_pi_ethernet_not_working), so I'm going to ask how are you powering your RPi4?

darkenvy, 2021-08-04:
I'm using a 2A Samsung Galaxy Quick Charger. But the cable very well could be the problem; it's not the cable that came with the wall-wart. I have a USB volt meter somewhere. Ill look into it and get back to you.

darkenvy, 2021-08-04:
Actually, I'm going to swap in my 30W Macbook USBC charger. That should give it enough juice! (Although no volt-meter. Its USB-A :man_shrugging: . Ill just trust the Apple Charger)

jvonau, 2021-08-04:
Please change the title to "Weak power supply destroys the Internet connection" so this issue can become searchable in the future, kind of tired of chasing down hardware misadventures.

holta, 2021-08-04:
@darkenvy can this be explained by Amazon and Microsoft using up all available electricity in Washington State &mdash; leaving nothing left for you?

darkenvy, 2021-08-04:
Haha. I forgot Microsoft was here. They are eclipsed merely by Amazon's shadow.

I got a new plug and cable. Got 2A solid on my volt-meter when charging my phone. The Pi4 stays around 0.6A when performing `sudo iiab`. The installation failed again with the networking cut out.

For good measure, I tried a brand new SD card as well.

iiab-diagnostics: http://sprunge.us/sR3wT9
`ps -AH` (after): http://sprunge.us/bgSF2o

`ip a` (after)
```
pi@raspberrypi:~$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether dc:a6:32:08:52:86 brd ff:ff:ff:ff:ff:ff
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether dc:a6:32:08:52:87 brd ff:ff:ff:ff:ff:ff
4: ap0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast master br0 state UP group default qlen 1000           link/ether 02:e2:b6:b1:2e:ee brd ff:ff:ff:ff:ff:ff                                                                5: br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000                         link/ether 02:e2:b6:b1:2e:ee brd ff:ff:ff:ff:ff:ff                                                                    inet 172.18.96.1/19 brd 172.18.127.255 scope global noprefixroute br0                                                    valid_lft forever preferred_lft forever
```

darkenvy, 2021-08-04:
Im curious if anyone else would have the issue performing the following steps:

1) Pi4
2) `2021-05-07-raspios-buster-armhf-lite.img` flashed to SD card
3) add `enable_uart=1` to last line of `/boot/config.conf`
4) connect via serial (as apposed to ssh)
5) `curl d.iiab.io/install.txt | sudo bash`

Im curious because if you can then that means it must be another variable. Such as SD card, network, power, broken Pi4, doesn't like installation through serial, ect.

jvonau, 2021-08-04:
enable_uart=1 may not be enough see https://www.raspberrypi.org/documentation/configuration/uart.md and we do mess with the bluetooth config in that role maybe try 'bluetooth_install: False' in local_vars.yml?

jvonau, 2021-08-04:
>host_channel = 6
client_wifi_channel = 34

Now that is a bit strange given there is no configured upstream in wpa_suppliciant.conf...
>[    6.908587] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan  4 2021 19:56:29 version 7.45.229 (617f1f5 CY) FWID 01-2dbd9d2e
[ 2790.523455] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[ 2796.116962] ieee80211 phy0: brcmf_cfg80211_add_iface: iface validation failed: err=-16
[ 2799.086258] ieee80211 phy0: brcmf_cfg80211_stop_ap: setting AP mode failed -52
[ 2801.407367] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled

Related to below
>2021-08-04 20:58:01,938 p=2665 u=root n=ansible | TASK [network : New Raspbian requires country code -- check for it] ************
2021-08-04 20:58:02,589 p=2665 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "grep country /etc/wpa_supplicant/wpa_supplicant.conf", "delta": "0:00:00.013316", "end": "2021-08-04 20:58:02.471662", "msg": "non-zero return code", "rc": 1, "start": "2021-08-04 20:58:02.458346", "stderr": "", "stderr_lines": [], "stdout": "", "stdout_lines": []}
2021-08-04 20:58:02,590 p=2665 u=root n=ansible | ...ignoring
2021-08-04 20:58:02,626 p=2665 u=root n=ansible | TASK [network : Put country code (US) in /etc/wpa_supplicant/wpa_supplicant.conf if nec] ***
2021-08-04 20:58:03,236 p=2665 u=root n=ansible | changed: [127.0.0.1]
2021-08-04 20:58:03,275 p=2665 u=root n=ansible | TASK [network : Enable the WiFi with rfkill] ***********************************
2021-08-04 20:58:03,872 p=2665 u=root n=ansible | changed: [127.0.0.1]

raspi-config was not used to pre-configure the wifi, not recommend but should work anyway

>Aug  4 20:58:47 raspberrypi dhcpcd[4104]: eth0: soliciting a DHCP lease
Aug  4 20:58:47 raspberrypi dhcpcd[4104]: eth0: offered 10.0.0.25 from 10.0.0.1
Aug  4 20:58:47 raspberrypi dhcpcd[4104]: eth0: NAK: from 10.0.0.1
Aug  4 21:00:19 raspberrypi dhcpcd[4104]: eth0: soliciting a DHCP lease
Aug  4 21:00:19 raspberrypi dhcpcd[4104]: eth0: offered 10.0.0.25 from 10.0.0.1
Aug  4 21:00:19 raspberrypi dhcpcd[4104]: eth0: NAK: from 10.0.0.1
Aug  4 21:01:19 raspberrypi dhcpcd[4104]: eth0: soliciting a DHCP lease
Aug  4 21:01:19 raspberrypi dhcpcd[4104]: eth0: offered 10.0.0.25 from 10.0.0.1
Aug  4 21:01:19 raspberrypi dhcpcd[4104]: eth0: NAK: from 10.0.0.1
Aug  4 21:02:20 raspberrypi dhcpcd[4104]: eth0: soliciting a DHCP lease
Aug  4 21:02:20 raspberrypi dhcpcd[4104]: eth0: offered 10.0.0.25 from 10.0.0.1
Aug  4 21:02:20 raspberrypi dhcpcd[4104]: eth0: NAK: from 10.0.0.1
Aug  4 21:03:20 raspberrypi dhcpcd[4104]: eth0: soliciting a DHCP lease
Aug  4 21:03:20 raspberrypi dhcpcd[4104]: eth0: offered 10.0.0.25 from 10.0.0.1
Aug  4 21:03:20 raspberrypi dhcpcd[4104]: eth0: NAK: from 10.0.0.1
Aug  4 21:04:20 raspberrypi dhcpcd[4104]: eth0: soliciting a DHCP lease
Aug  4 21:04:20 raspberrypi dhcpcd[4104]: eth0: offered 10.0.0.25 from 10.0.0.1
Aug  4 21:04:20 raspberrypi dhcpcd[4104]: eth0: NAK: from 10.0.0.1
Aug  4 21:05:20 raspberrypi dhcpcd[4104]: eth0: soliciting a DHCP lease
Aug  4 21:05:20 raspberrypi dhcpcd[4104]: eth0: offered 10.0.0.25 from 10.0.0.1
Aug  4 21:05:20 raspberrypi dhcpcd[4104]: eth0: NAK: from 10.0.0.1
Aug  4 21:06:20 raspberrypi dhcpcd[4104]: eth0: soliciting a DHCP lease
Aug  4 21:06:20 raspberrypi dhcpcd[4104]: eth0: offered 10.0.0.25 from 10.0.0.1
Aug  4 21:06:20 raspberrypi dhcpcd[4104]: eth0: NAK: from 10.0.0.1
Aug  4 21:07:20 raspberrypi dhcpcd[4104]: eth0: soliciting a DHCP lease
Aug  4 21:07:20 raspberrypi dhcpcd[4104]: eth0: offered 10.0.0.25 from 10.0.0.1
Aug  4 21:07:20 raspberrypi dhcpcd[4104]: eth0: NAK: from 10.0.0.1
Aug  4 21:08:20 raspberrypi dhcpcd[4104]: eth0: soliciting a DHCP lease
Aug  4 21:08:20 raspberrypi dhcpcd[4104]: eth0: offered 10.0.0.25 from 10.0.0.1
Aug  4 21:08:20 raspberrypi dhcpcd[4104]: eth0: NAK: from 10.0.0.1

It appear restarting the network may have contributed, but these dhcp like errors were noted in EKK6XS where hand mucking was unsuccessful to raise the wired network, which begs for the question why is the dhcp server passing NAK to its own lease offer of 10.0.0.25.... 2 dhcp servers on this LAN? What is the upstream dhcp server?

darkenvy, 2021-08-05:
10.0.0.1 is the DHCP server. There should not be two DHCP servers.
Being a good scientist, how can I verify my network only has one DHCP server? The ONLY thing that comes to mind that could give out DHCP leases is my "Pi-Hole". But it isn't set up to give out leases.

jvonau, 2021-08-05:
What is the dhcp server? It's the one sending the NAK to the lease offing of 10.0.0.25 that itself just sent out? Whatever, be a good scientist and dig some more. Good luck with your local issue, be interest to know what you uncover. Gut feeling power supply.

darkenvy, 2021-08-06:
I figured out the issue! 

The network drops around the time that `/etc/hostname` gets set to `box.lan`. If I set "Allow special characters" under my router's DHCP settings, then I no longer get DHCPNAKs. I immediately get a IP address lease. To verify this finding, I disabled the "Allow special characters" setting in the router and rebooted the Pi. I received DHCPNAKs once again. It should be noted that this setting is off by default on consumer routers (Centurylink ActionTec)

Looking up valid characters for hostnames, it seems to be "A-Z", "0-9" and "-". The period in `box.lan` is the culprit as the period is not a valid character. Upon changing `box.lan` to `box` in `/etc/hostname` and performing a reboot got me a DHCP lease (with "Allow special characters off").

(I used dhcpdump to watch this network activity. Very useful tool)
Setting in router: https://dl.dropboxusercontent.com/s/yxclxw2g18m1q35/Screenshot%20at%2014-07-21.png

holta, 2021-08-06:
@jvonau can you confirm where https://serverfault.com/questions/229331/can-i-have-dots-in-a-hostname is relevant here?

> Due to how DNS works, the "alpha" component of "a.alpha" is considered a discrete 'label' in DNS. Using a hostname with a dot in it will cause inconsistent results from any system that consumes DNS.
>
> Avahi does interact with DNS names, and specifically the <host-name> directive needs to have the DNS FQDN of the service in it, so it's also subject to DNS inconsistency with dotted names.
>
> Don't use dotted name.

darkenvy, 2021-08-06:
> will cause inconsistent results from any system that consumes DNS
I think is the more accurate statement. Considering we have had `box.lan` for years(?) and this is the first time someone has noticed, it may be a very small issue. 

Why do we use box.lan anyways? `.lan` is just as obscure/arbutrary as `.local` (both are "in-knowledge" in computing) but `.local` works out of the box. Im guessing we use `.lan` because the use of a TLD forced browsers to go to the domain instead of a web search. (If I type just `box` in my address bar, it tries to google search "box").

Going forward, if we really want `.lan`, I thing it would be worth verifying that we have a dhcp lease after changing the hostname. And fallback to just `box` (and therefore `box.local`). Ofc, it would be simpler to just use `box` ;)

holta, 2021-08-06:
A few phones require http://box.lan/ instead of http://box/

I don't know the reasons.

But I've seen it with my own eyes, several time in very poor countries.

(Which doesn't mean we can't reassess ~5 years later, as the above quirk is rare, even if it's still out there!)

darkenvy, 2021-08-06:
yea, I actually just edited my previous post. I see the reason for box.lan. And the admin panel offers changing the TLD. 

I was just mentioning that we should just add a check in ansible (ping) after a hostname refresh. If it fails, a detailed output would be nice. One that mentions unsupportive-routers/hostnames as a possible cause.

jvonau, 2021-08-07:
> I figured out the issue!
>
> The network drops around the time that `/etc/hostname` gets set to `box.lan`. If I set "Allow special characters" under my router's DHCP settings, then I no longer get [...]

Interesting find, that would imply your router is providing reverse dns lookups, what domain name, search string, dns server is the dhcp server passing to the clients to use? What is the default setting out of the box Block or Allow? Did you toggle this setting at anytime in the past?

> Looking up valid characters for hostnames, it seems to be "A-Z", "0-9" and "-". The period in `box.lan` is the culprit as the period is not a valid character. Upon changing `box.lan` to `box` in `/etc/hostname` and performing a reboot got me a DHCP lease (with "Allow special characters off").
> 

Interesting, `man hostname` suggest everything to the left of the first dot is coincided the hostname and a FQDN that resolves is acceptable. Think that is more dhcpcd related in handing of the hostname passed to the dhcp server during the lease request, if that fixes it yay.. However hostname is set very early (0-init) in the ansible run and your network outages are doc'd at various times at a point in time much later, is the failure at lease renewal time? If you leave /etc/hostname as is when you reboot do you obtain an ip address or is the network dead at that point?

> ubuntu@box:/etc/asterisk$ hostname -d
lan
ubuntu@box:/etc/asterisk$ hostname -f
box.lan
ubuntu@box:/etc/asterisk$ hostname 
box.lan
ubuntu@box:/etc/asterisk$ hostname -a
localhost.localdomain localhost box box box.lan
ubuntu@box:/etc/asterisk$ hostname -d
lan
ubuntu@box:/etc/asterisk$ hostname -s
box

Interesting, The box.lan as a  hostname has been used in an attempt to keep the docs the same over time. Funny thing 'way back when' .local was used.
> (I used dhcpdump to watch this network activity. Very useful tool)

Doesn't have a crystal ball option... lol ;)
> Setting in router: https://dl.dropboxusercontent.com/s/yxclxw2g18m1q35/Screenshot%20at%2014-07-21.png

Seems like the dhcp server is expecting the FQDN of the client to be set to the same domain name string it is sending out. Changing the 'TLD' is available by changing 'iiab_domain: lan' in local_vars.yml and 'iiab_hostname: box' is available also, these combined would make up the first entry in /etc/hosts where `hostname -f` should return the first entry, ansible is noted to rely on this entry for the ansible_fqdn variable.

darkenvy, 2021-08-07:
> Interesting find, that would imply your router is providing reverse dns lookups
Would it imply that? Considering the option is "Block Special Characters", there is hardly any implication here. It is explicity blocking special characters. Not because of some failed reverse-dns-lookup, but because the setting told it to.

> What is the default setting out of the box Block or Allow? 
Block is the default

> Did you toggle this setting at anytime in the past?
No

> Interesting, man hostname suggest everything to the left of the first dot is coincided the hostname and a FQDN that resolves is acceptable. Think that is more dhcpcd related in handing of the hostname passed to the dhcp server during the lease request, if that fixes it yay.. 
The intercepted packed from `dhcpdump` shows that the request includes the `.lan`. If it could be made so dhcpcd requests only left of the `.`, then that would be ideal :).

> is the failure at lease renewal time?
The failure occurs when the network is restarted via `sudo iiab` task `Restart the networking service if appropriate`. During that last failed attempt. (PS: `sudo iiab` installs all the way now successfully, just as long as I have my hostname set to `box`.)

> If you leave /etc/hostname as is when you reboot do you obtain an ip address or is the network dead at that point?
As is? With the `box.lan` in it? If it has `box.lan` in /etc/hostname, I will never get an IP address. NAKs only. Even after reboot.
But If I remove the `.` and have `boxlan` or `box`, then upon reboot, I get an IP address.

> Seems like the dhcp server is expecting the FQDN of the client to be set to the same domain name string it is sending out. Changing the 'TLD' is available by changing 'iiab_domain: lan' in [...]
The domain name string it is sending out is `box.lan` though. According to the packet dump from dhcpdump. 

:man_shrugging: Finding the problem took a lot of time. I think we should at least document this under a FAQ or something. Preferably as detailed output on the 'restart network' task if it fails (can that be done with Ansible?)

darkenvy, 2021-08-07:
I did test with with my desktop and saw that the dhcp request ONLY had the name to the left of the `.` and therefor was a valid hostname in the eyes of my fussy router. `hostname -d` showed `.lan` but the dchp request did NOT have the `.lan` in it.

So it seems some change COULD happen on the request side. But this is outside of my knowledge (dhcpcd) on how to make it work on the Pi.

PS: My desktop is Ubuntu 20.10

holta, 2021-08-07:
> > is the failure at lease renewal time?
> 
> The failure occurs when the network is restarted via `sudo iiab` task `Restart the networking service if appropriate`. During that last failed attempt. (PS: `sudo iiab` installs all the way now successfully, just as long as I have my hostname set to `box`.)

Interesting.  Just FYI this code snippet (https://github.com/iiab/iiab/blob/master/roles/network/tasks/rpi_debian.yml#L67-L72 pasted in below) is run by [roles/network](https://github.com/iiab/iiab/tree/master/roles/network) (i.e. the command `./iiab-network`) every time:

```
- name: Restart the networking service if appropriate
  systemd:
    name: networking
    enabled: yes
    state: restarted
  when: iiab_wired_lan_iface is defined
```

jvonau, 2021-08-08:
> Would it imply that? Considering the option is "Block Special Characters", there is hardly any implication here. It is explicity blocking special characters. Not because of some failed [...]

There is the possibility that the dhcp server is reserving an ip address based on hostname/mac_address, don't really have the means to pop the top off the box and peer into its inner workings so you and I don't really know that is going on behind the curtain. On a fresh boot of a new image the contents /etc/resolv.conf would be useful to know what has been sent from the dhcp server for the client to use. 

> Block is the default
>
>
> No
>
>
> The intercepted packed from `dhcpdump` shows that the request includes the `.lan`. If it could be made so dhcpcd requests only left of the `.`, then that would be ideal :).

Good you can see the packet exchange, what is sent from the server for the client to use? things like nameserver, domain, search_domain? Those would make up what normally would be in /etc/resolv.conf if you can catch the info.

> > is the failure at lease renewal time?
> 
> The failure occurs when the network is restarted via `sudo iiab` task `Restart the networking service if appropriate`. During that last failed attempt. (PS: `sudo iiab` installs all the way now successfully, just as long as I have my hostname set to `box`.)
> 

That is weird there is a conditional that is normally not met, the service is not normally restarted at that point, only when trying to setup a wired bridge, what is restarted is dhcpcd a couple of stanzas below.    

> As is? With the `box.lan` in it? If it has `box.lan` in /etc/hostname, I will never get an IP address. NAKs only. Even after reboot.
> But If I remove the `.` and have `boxlan` or `box`, then upon [...]

> The domain name string it is sending out is `box.lan` though. According to the packet dump from dhcpdump.

Can you see what the server sent out as domain name|search? Not going to dig to deep that would imply that dhcpcd is just using /etc/hostname directly, I'm suggesting try making /etc/hostname box.from.dhcp where from.dhcp is same string that might be returned by your dhcp server as domain or search that would endup in /etc/resolve.conf. At least that would help further figure out what the dhcp server expects.

> man_shrugging Finding the problem took a lot of time. I think we should at least document this under a FAQ or something. Preferably as detailed output on the 'restart network' task if it fails (can that be done with Ansible?)

Don't think there has been an dump of the iiab-install.log with the failure at that location in networking because what you are seeing is dhcpcd not getting an ip address some point later.

jvonau, 2021-08-08:
> I did test with with my desktop and saw that the dhcp request ONLY had the name to the left of the `.` and therefor was a valid hostname in the eyes of my fussy router. `hostname -d` showed `.lan` [...]

Could you post the /etc/resolv.conf from that machine assuming iiab is not installed on that one yet.

jvonau, 2021-08-08:
> Interesting. Just FYI this code snippet (https://github.com/iiab/iiab/blob/master/roles/network/tasks/rpi_debian.yml#L67-L72 pasted in below) is run by [...]

[iiab_wired_lan_iface](https://github.com/iiab/iiab/blob/master/roles/network/tasks/detected_network.yml#L178-L181) should not be defined on RasPiOS ever. Or would have to declare the variable in local_vars for this value to be set.

holta, 2021-08-08:
@darkenvy can you also post your latest IIAB's sprunge.us pastebin URL from [sudo iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md), just so we're all on the same wavelength?

jvonau, 2021-08-08:
One could ask the TPTB to add a variable to suppress the hostname modifications like what I proposed in #2834 with is_guest or some other sort of PR but it is beyond my powers to press the commit button so no PR coming from me to fix a non-issue for what appears to be a single use case.

jvonau, 2021-08-08:
Think #2921 would be helpful in this situation and meets my objective of making iiab less 'closed blackbox' like.

jvonau, 2021-08-17:
What is the call? Add the suppression variable, re-evaluate the use of box.lan or close this with 'information requested with no response'?

darkenvy, 2021-08-17:
> Could you post the /etc/resolv.conf from that machine assuming iiab is not installed on that one yet.

My Desktop machine's /etc/resolv.conf.
```
# This file is managed by man:systemd-resolved(8). Do not edit.
#
# This is a dynamic resolv.conf file for connecting local clients to the
# internal DNS stub resolver of systemd-resolved. This file lists all
# configured search domains.
#
# Run "resolvectl status" to see details about the uplink DNS servers
# currently in use.
#
# Third party programs should typically not access this file directly, but only
# through the symlink at /etc/resolv.conf. To manage man:resolv.conf(5) in a
# different way, replace this symlink by a static file or a different symlink.
#
# See man:systemd-resolved.service(8) for details about the supported modes of
# operation for /etc/resolv.conf.

nameserver 127.0.0.53
options edns0 trust-ad
search Home
```

darkenvy, 2021-08-17:
> IIAB

Adam, did you want the sudo iiab-diagnostics of the successfully installed IIAB? Because it finishes all-the-way if I change that hostname and has no problem.

holta, 2021-08-17:
> @holta , did you want the sudo iiab-diagnostics of the successfully installed IIAB? Because it finishes all-the-way if I change that hostname and has no problem.

Please do post a fresh iiab-diagnostics URL (e.g. sprunge.us) for the benefit for all, if possible!

jvonau, 2021-08-17:
> > Could you post the /etc/resolv.conf from that machine assuming iiab is not installed on that one yet.
> 
> My Desktop machine's /etc/resolv.conf.
> 
Shoot forgot Ubuntu has systemd-resolved running use `sudo resolvectl --no-pager`
```
ubuntu@box:~/git/iiab-factory$ sudo resolvectl --no-pager
Global
       LLMNR setting: no            
MulticastDNS setting: no            
  DNSOverTLS setting: no            
      DNSSEC setting: no            
    DNSSEC supported: no            
          DNS Domain: hitronhub.home
                      box.lan       

Link 2 (eth0)
      Current Scopes: DNS           
DefaultRoute setting: yes           
       LLMNR setting: yes           
MulticastDNS setting: no            
  DNSOverTLS setting: no            
      DNSSEC setting: no            
    DNSSEC supported: no            
         DNS Servers: 192.168.0.1   
          DNS Domain: ~.            
                      hitronhub.home

Link 3 (wlan0)
      Current Scopes: none
DefaultRoute setting: no  
       LLMNR setting: yes 
MulticastDNS setting: no  
  DNSOverTLS setting: no  
      DNSSEC setting: no  
    DNSSEC supported: no  

Link 4 (br0)
      Current Scopes: none
DefaultRoute setting: no  
       LLMNR setting: yes 
MulticastDNS setting: no  
  DNSOverTLS setting: no  
      DNSSEC setting: no  
    DNSSEC supported: no  
          DNS Domain: lan 

Link 5 (ap0)
      Current Scopes: none
DefaultRoute setting: no  
       LLMNR setting: yes 
MulticastDNS setting: no  
  DNSOverTLS setting: no  
      DNSSEC setting: no  
    DNSSEC supported: no  

Link 6 (tun0)
      Current Scopes: none
DefaultRoute setting: no  
       LLMNR setting: yes 
MulticastDNS setting: no  
  DNSOverTLS setting: no  
      DNSSEC setting: no  
    DNSSEC supported: no  

Link 7 (docker0)
      Current Scopes: none
DefaultRoute setting: no  
       LLMNR setting: yes 
MulticastDNS setting: no  
  DNSOverTLS setting: no  
      DNSSEC setting: no  
    DNSSEC supported: no  

```

jvonau, 2021-08-17:
@darkenvy  "search Home" is what is was looking for, just want to test a theory could you try setting  /etc/hostname to box.Home and see if that gets an ipaddress please.

jvonau, 2021-08-17:
```
Global
       LLMNR setting: no            
MulticastDNS setting: no            
  DNSOverTLS setting: no            
      DNSSEC setting: no            
    DNSSEC supported: no            
          DNS Domain: hitronhub.home
                      box.lan       
```
Clearly systemd-resolved thinks box.lan is the domain part of a fqdn. That would point to using just 'box' has the hostname and the historical use of 'box.lan' should go away. Keeping the **alias** box.lan in /etc/hosts should be fine.

jvonau, 2021-08-17:
Hum.... maybe 'iiab_domain: .lan' could become iiab.lan? Then all the wifi clients would need to use box.iiab.lan. Whatever @holta current whim is will be used.

darkenvy, 2021-08-17:
> @darkenvy "search Home" is what is was looking for, just want to test a theory could you try setting /etc/hostname to box.Home on the RPi machine and see if that gets an ipaddress please.

Using `box.Home` doesn't work :(

jvonau, 2021-08-17:
> > @darkenvy "search Home" is what is was looking for, just want to test a theory could you try setting /etc/hostname to box.Home on the RPi machine and see if that gets an ipaddress please.
> 
> Using `box.Home` doesn't work :(

Thanks for the data point, is 'Home' the out of the box domain setting of the router or did you change that? From your Ubuntu desktop mentioned above can you gather the output of `sudo resolvectl --no-pager` please. I would like to see how 'Home' is being handled by systemd-resolved, should be the similar to using .lan in my example above.

The historical use of .lan dates back to the real early [XSCE days](https://github.com/XSCE/xsce/commit/ed0b00e6440af5e96172f76b4e13ed18fca9394b) and has be that way since. Given the intended use is offline not much attention has been given to 'best practices' over the years.

[16 more replies]

closed 2021-09-01 by holta