iiab/iiab #2914: FreePBX 15 + NGINX/IIAB quick recipe that worked for @lemueldsouza

holta, 2021-08-05:
Huge thanks to @lemueldsouza and @jvonau who've pushed so hard here &mdash; it seems like we're close:

1. Installed Asterisk 16.x and FreePBX 15.x and nodejs 12.x on Raspbian 10.
Followed steps from https://computingforgeeks.com/how-to-install-asterisk-16-with-freepbx-15-on-ubuntu-debian/
This worked fine, but was without IIAB, and nginx user was asterisk and not www-data.

2. Installed min version of IIAB on the same Raspberry Pi using the one line installer (I did not enable pbx as I had already installed it in step 1).

3. Added www-data to the asterisk group: `sudo usermod -a -G asterisk www-data`

4. Added another server entry/virtual host in /etc/nginx/server.conf to have freepbx listen on port 83:

   ```
   server {
       listen 83;
       listen [::]:83;

       root /var/www/html;
       index index.html index.htm index.nginx-debian.html index.php;
       server_name localhost;

       location / {
           try_files $uri $uri/ =404;
       }

       location ~ [^/]\.php(/|$) {
           fastcgi_split_path_info ^(.+?\.php)(/.*)$;

           if (!-f $document_root$fastcgi_script_name) {
               return 404;
           }

           fastcgi_pass unix:/run/php/php7.3-fpm.sock;
           fastcgi_index index.php;

           include fastcgi_params;
           fastcgi_param SCRIPT_FILENAME $document_root$fastcgi_script_name;
           fastcgi_param PATH_INFO $fastcgi_path_info;
           fastcgi_param PATH_TRANSLATED $document_root$fastcgi_path_info;
           fastcgi_param HTACCESS on;
           #fastcgi_param SERVER_NAME $host;
       }

       # disallows the things that the FreePBX .htaccess files disallow
       location ~ (/\.ht|/\.git|\.ini$|/libraries|/helpers|/i18n|/node|/views/.+php$) {
           deny all;
       }

       # from the api module .htaccess file
       rewrite ^/admin/api/([^/]*)/([^/]*)/?(.*)?$ /admin/api/api.php?module=$1&command=$2&route=$3 last;
   }
   ```

5. Changed back the entries in /etc/php/7.3/fpm/pool.d/www.conf to:

   ```
   user = www-data
   group = www-data
   ```

   ```
   listen.owner = www-data
   listen.group = www-data
   ```

6. I didn't need to change any permissions, but here's mine for freepbx.conf

   ```
   pi@box:~ $ ls -l /etc/freepbx.conf
   -rw-rw---- 1 asterisk asterisk 350 Jul 21 18:23 /etc/freepbx.conf
   ```

Related:

- PR #2850
- PR #2862
- PR #2905
- PR #2911

jvonau, 2021-08-05:
3 & 5 is what is missing from #2913

holta, 2021-08-05:
> 3 & 5 is what is missing from #2913

Are we already able to dump port 83?  (Or should we keep that in the interim, until everything's tested?)

Looks like NGINX probably needs 1-2 remaining things from our Apache legacy .conf file?
https://github.com/iiab/iiab/blob/master/roles/pbx/templates/freepbx.conf.j2

e.g. the `/var/lib/php/asterisk_sessions/` directory mentioned here: https://github.com/iiab/iiab/blob/master/roles/pbx/tasks/freepbx.yml#L88

jvonau, 2021-08-05:
Adding a port requires a new /etc/nginx/server.conf while the code I used just drops into place. Depends on how nginx handles php sessions as they differ.

jvonau, 2021-08-05:
scratch number 5, just 3 the having www-data user in the asterisk group

holta, 2021-08-05:
Later we'll also need cleaner resolution to this repeated warning from `journalctl -u freepbx` :

> [NOTIFICATION]-[core]-[MEMLIMIT] - Memory Limit Changed (Your memory_limit, 128M, is set too low and has been increased to 256M. You may want to change this in you php.ini config file)

A separate roles/php-limits role might even be best, so this stops falling thru the cracks:

Ref: https://github.com/iiab/iiab/blob/master/roles/www_options/tasks/main.yml#L53-L133

holta, 2021-08-05:
> mariadb-connector-odbc mentioned above might need to be compiled also see http://mghadam.blogspot.com/2021/03/install-asterisk-18-freepbx-15-on.html

Per https://github.com/iiab/iiab/pull/2913#issuecomment-893666059

holta, 2021-08-06:
Apologies I'm away today helping a friend with a very serious medical condition.

I don't know the most practical next step, but hopefully @lemueldsouza and @jvonau who've done the very hard work can recommend.

So PR #2850 (or similar!) can be merged shortly: https://github.com/iiab/iiab/tree/master/roles/nginx#transition-to-nginx

holta, 2021-08-07:
> Looks like NGINX probably needs 1-2 remaining things from our Apache legacy .conf file?
> https://github.com/iiab/iiab/blob/master/roles/pbx/templates/freepbx.conf.j2
>
> e.g. the [...]

Question: Do we also need something like this package for NGINX ?  [freepbx_dependencies.yml around Line 29](https://github.com/iiab/iiab/blob/master/roles/pbx/tasks/freepbx_dependencies.yml#L29) shows...

```
- libapache2-mpm-itk    # To serve FreePBX through a VirtualHost as asterisk user
```

holta, 2021-08-07:
FYI this is now merged, to make FreePBX-on-NGINX community testing very easy, pushing us all forward:

- PR #2916

holta, 2021-08-11:
FreePBX 16 Beta appears to now work very well with NGINX or Apache (using Asterisk 18 in either case!)  As a result of PR's in recent days like the following...

- PR #2916
- PR #2929
- PR #2931
- PR #2932

_Still, we should keep this ticket open to broadly gather community feedback if possible!_

@lemueldsouza's README evolving from PR #2905 should help here, to make sure that IIAB-with-Asterisk/FreePBX is inviting and quite easy to understand (and quickly make work with regular Android phones, running SIP apps like Linphone etc!)

holta, 2021-08-14:
FYI FreePBX 16 Beta on NGINX is now in really good shape!  Thanks to PRs #2929, #2930, #2931 etc.  Please see the **new instructions** here and try it out:

- https://github.com/iiab/iiab/tree/master/roles/pbx#pbx-readme

Just a couple issues now remain:

- https://github.com/iiab/iiab/tree/master/roles/pbx#known-issues
  - Should `/var/lib/php/asterisk_sessions/` (or similar) be used with NGINX? https://github.com/iiab/iiab/blob/master/roles/pbx/tasks/freepbx.yml#L147-L168
  - `/etc/odbc.ini` for CDR (Call Details Records) is being worked on at: PR #2942

holta, 2021-08-22:
@lemueldsouza and @jvonau should we close this higher-level ticket, that has help keep us oriented?

(Or do you prefer to keep it open as an umbrella Asterisk/FreePBX tracking/monitoring ticket over coming months?)