iiab/iiab #2924: Review phpMyAdmin security + practicality

holta, 2021-08-08:
Who can help review https://github.com/iiab/iiab/tree/master/roles/phpmyadmin so that it serves a clear + pragmatic public purpose?

Thanks!

(Worst case if nobody is willing, this role should be deprecated in coming weeks.)

Refs:

- PR #2923
- http://wiki.laptop.org/go/IIAB/Security

cc: @georgejhunt, @tim-moody, @jvonau, @deltacloud9

holta, 2021-08-12:
Does IIAB really need this?

_"phpMyAdmin is a free software tool written in PHP, intended to handle the administration of MySQL over the Web.  phpMyAdmin supports a wide range of operations on MySQL and MariaDB.  Frequently used operations (managing databases, tables, columns, relations, indexes, users, permissions, etc) can be performed via the user interface, while you still have the ability to directly execute any SQL statement."_

Please weigh in.

I've heard ambiguous sentiments from @georgejhunt and @jvonau.

Either decision is fine, but we need a decision.

georgejhunt, 2021-08-13:
I have found it useful. But security concerns make it hazardous. Let's drop
it.

jvonau, 2021-08-17:
What does 2948 have to do with this subject? Keep the cross linking directly relevant please.

holta, 2021-08-17:
Deprecating stale software appears to be the very hardest work of all.

In that sense, these are intimately related, from a software lifecycle perspective:

- #2948

jvonau, 2021-08-17:
#2122 #2712 My personal choice is to never install the role anyway, made a note about maybe using the socket for the connection in the PR. If nobody want to keep an eye on the role the lets drop it and move on.

holta, 2021-08-18:
As a start, I've removed phpMyAdmin from https://github.com/iiab/iiab/wiki/IIAB-7.2-Release-Notes

holta, 2021-09-28:
Villagers in remote northern India apparently need this.

Please see the ongoing day here:

- #2712
- #2812