iiab/iiab #2934: Test Asterisk 19.0 on its release "2021-09-28", with FreePBX 16 if possible!

holta, 2021-08-11:
- https://wiki.asterisk.org/wiki/display/AST/Asterisk+Versions
- PR #2914
- https://github.com/iiab/iiab/pull/2931#issuecomment-896320509

jvonau, 2021-08-12:
Don't like the idea, the working combos seem to have the same asterisk and freepbx versions, I suggest you roll back asterisk to version 16 to match the version of freebpx and start testing all over.

holta, 2021-09-03:
FYI Asterisk 19 was delayed 4 weeks from "2021-09-28" to approx "2021-10-26" (still a Tuesday!)

holta, 2021-11-03:
@lemueldsouza now that [Asterisk 19.0 is released](http://lists.digium.com/pipermail/asterisk-announce/2021-November/000832.html), can you try installing it with IIAB as [described above](https://github.com/iiab/iiab/issues/2934#issuecomment-948736549) ?

holta, 2021-11-05:
@lemueldsouza the following install is now underway to begin testing Asterisk 19.0 with IIAB: (just FYI Asterisk's `install_prereq install` completed in 5min54sec and `./configure --with-pjproject-bundled --with-jansson-bundled` completed in 2min24sec, so approximately 30min remain ?)

```
cd /opt/iiab/iiab
sudo ./runrole pbx
```
 
Can you help evaluate its solidness &mdash; by logging into this Raspberry Pi 4 (2GB RAM) that is running a pre-release Raspberry Pi OS 11 "Bullseye" (with Node.js 16.x) here?

- 182-rpi4-lite-11-MIN-asterisk19 = 10.8.0.46

And if the above looks good, also helping to evaluate these 2 URL's?

- http://10.8.0.46:83/freepbx (Apache)
- http://10.8.0.46/freepbx (NGINX)

_Thank You If Possible!_

Refs:

- http://downloads.asterisk.org/pub/telephony/asterisk/asterisk-19-current.tar.gz (27 MB)
- #2914 
- PR #2916

holta, 2021-11-05:
> 182-rpi4-lite-11-MIN-asterisk19 = 10.8.0.46

FYI the above install failed, as follows:

- Compiling Asterisk 19 (the [make](https://github.com/iiab/iiab/blob/master/roles/pbx/tasks/asterisk.yml#L91-L95) step) took 18min52sec.
- Running [scripts/mk-odbc-connector](https://github.com/iiab/iiab/blob/master/roles/pbx/tasks/freepbx.yml#L184-L187) took 1min59sec.
- Step [FreePBX - 2-step install - won't run if /var/www/html/freepbx already exists - CAN TAKE 3-12 MIN OR LONGER!](https://github.com/iiab/iiab/blob/master/roles/pbx/tasks/freepbx.yml#L204-L212) failed as follows:

> 2021-11-05 18:18:05,771 p=1847 u=root n=ansible | TASK [pbx : FreePBX - 2-step install - won't run if /var/www/html/freepbx already exists - CAN TAKE 3-12 MIN OR LONGER!] ***
2021-11-05 18:18:13,606 p=1847 u=root n=ansible | changed: [127.0.0.1] => (item=./start_asterisk start)
2021-11-05 18:18:14,495 p=1847 u=root n=ansible | failed: [127.0.0.1] (item=./install -n --webroot /var/www/html/freepbx --dbuser asterisk --dbpass asterisk) => {"ansible_loop_var": "item", "changed": true, "cmd": ["./install", "-n", "--webroot", "/var/www/html/freepbx", "--dbuser", "asterisk", "--dbpass", "asterisk"], "delta": "0:00:00.299658", "end": "2021-11-05 18:18:14.424794", "item": "./install -n --webroot /var/www/html/freepbx --dbuser asterisk --dbpass asterisk", "msg": "non-zero return code", "rc": 1, "start": "2021-11-05 18:18:14.125136", "stderr": "", "stderr_lines": [], "stdout": "Checking if SELinux is enabled...Its not (good)!\nReading /etc/asterisk/asterisk.conf...Done\nChecking if Asterisk is running and we can talk to it as the 'asterisk' user...Error!\nUnsupported Version of 19.0.0\nSupported Asterisk versions: 13, 14, 15, 16, 17, 18", "stdout_lines": ["Checking if SELinux is enabled...Its not (good)!", "Reading /etc/asterisk/asterisk.conf...Done", "Checking if Asterisk is running and we can talk to it as the 'asterisk' user...Error!", "Unsupported Version of 19.0.0", "Supported Asterisk versions: 13, 14, 15, 16, 17, 18"]}
2021-11-05 18:18:14,502 p=1847 u=root n=ansible | PLAY RECAP *********************************************************************
2021-11-05 18:18:14,503 p=1847 u=root n=ansible | 127.0.0.1                  : ok=90   changed=52   unreachable=0    failed=1    skipped=9    rescued=0    ignored=0

CONCLUSION: [/opt/iiab/freepbx/install.php (Line 38)](https://github.com/FreePBX/framework/blob/release/16.0/install.php#L38) could possibly be **surgically patched** until FreePBX 16 releases a point release or similar?  If so, changing this...

```
if (version_compare($astversion, "13", "lt") || version_compare($astversion, "19", "ge")) {
```

...to this...

```
if (version_compare($astversion, "13", "lt") || version_compare($astversion, "20", "ge")) {
```

holta, 2021-11-05:
> [/opt/iiab/freepbx/install.php (Line 38)](https://github.com/FreePBX/framework/blob/release/16.0/install.php#L38) could possibly be **surgically patched** until FreePBX 16 releases a point release [...]

To test this out, I also modified [/opt/iiab/freepbx/installlib/installcommand.class.php (Line 297)](https://github.com/FreePBX/framework/blob/release/16.0/installlib/installcommand.class.php#L297) (with the very same logic as above).

This allowed FreePBX 16 to install.  This **bold red error** appeared, but it did not stop FreePBX from installing:

```
Unable to install module certman:
 - Cannot find module
 ```

Ansible output, for the record:

```
TASK [pbx : FreePBX - 2-step install - won't run if /var/www/html/freepbx already exists - CAN TAKE 3-12 MIN OR LONGER!] ***********************************************
changed: [127.0.0.1] => (item=./start_asterisk start)
Updating tables admin, ampusers, cronmanager, featurecodes, freepbx_log, freepbx_settings, globals, module_xml, modules, notifications, cron_jobs...Done
Determined Asterisk version to be: 19.0.0
No directory /var/www/html/freepbx/admin/modules/framework/amp_conf/htdocs, install script not needed
Generating CSS...Done
Module framework version 16.0.10.33 successfully installed
Updating Hooks...Done
Chowning directories...Done
Updating tables trunks, pjsip, sip, dahdi, iax, indications_zonelist, devices, users, incoming, dahdichandids, outbound_route_patterns, outbound_route_sequence, outbound_route_trunks, outbound_routes, outbound_route_email, trunk_dialpatterns...Done
Adding emergencydevices table..Done
Migrating pickup groups to named pickup groups
Migrating call groups to named call groups
Checking for possibly invalid emergency caller id fields..none found
Migrating old media encryption values...done
Removing encoding on incoming routes alertinfo values...done
Generating CSS...Done
Module core version 16.0.56.14 successfully installed
Generating CSS...Done
Module dashboard version 16.0.8 successfully installed
Updating tables sipsettings...Done
New SIPSettings installation detected. Initializing default settings
populating default codecs..ulaw, alaw, gsm, g726 added
Updating kvstore_Sipsettings
Migrating Audio Codecs
Generating CSS...Done
Module sipsettings version 16.0.15 successfully installed
Updating tables voicemail_admin...Done
OPERATOR_XTN added
VM_OPTS added
VM_GAIN added
VM_DDTYPE added
VMX_OPTS_LOOP added
VMX_OPTS_DOVM added
VMX_TIMEOUT added
VMX_REPEAT added
VMX_LOOPS added
Generating CSS...Done
Module voicemail version 16.0.21 successfully installed
Unable to install module certman:
 - Cannot find module
Updating Hooks...Done
Chowning directories...Done
Setting Permissions...
Setting base permissions...Done in 2 seconds
Setting specific permissions...
 24124 [============================]
Finished setting permissions
changed: [127.0.0.1] => (item=./install -n --webroot /var/www/html/freepbx --dbuser asterisk --dbpass asterisk)

TASK [pbx : FreePBX - Run 'fwconsole stop', 'killall -9 safe_asterisk' to stop both main Asterisk processes - this avoids "Unable to run Pre-Asterisk hooks, because Asterisk is already running" in 'journalctl -u freepbx' logs] ***
changed: [127.0.0.1] => (item=fwconsole stop)
failed: [127.0.0.1] (item=killall -9 safe_asterisk) => {"ansible_loop_var": "item", "changed": true, "cmd": ["killall", "-9", "safe_asterisk"], "delta": "0:00:00.019208", "end": "2021-11-05 18:46:41.219923", "item": "killall -9 safe_asterisk", "msg": "non-zero return code", "rc": 1, "start": "2021-11-05 18:46:41.200715", "stderr": "safe_asterisk: no process found", "stderr_lines": ["safe_asterisk: no process found"], "stdout": "", "stdout_lines": []}
...ignoring

TASK [pbx : FreePBX - Run 'killall -9 "PM2 v4.5.0: God"' for good measure] *********************************************************************************************
changed: [127.0.0.1]
```

[iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md) output: http://sprunge.us/QjoLon

FINAL CONCLUSION &mdash; @lemueldsouza can you please test both of these (live!) URL's below?

- http://10.8.0.46:83/freepbx (Apache)
- http://10.8.0.46/freepbx (NGINX)

THANKS!

holta, 2021-11-05:
Interesting that Toronto-based Sangoma announced Asterisk 19 and FreePBX 16 together earlier this week, as if they're supposed to work together: (ahem, hopefully soon!)

- https://www.sangoma.com/press-releases/sangoma-announces-asterisk-19-and-freepbx-16/

holta, 2021-11-06:
FYI FreePBX point releases occur almost every week:
https://github.com/FreePBX/framework/tags

So hopefully its next point release(s) (and also freepbx-16.0-latest.tgz) will contain the "2-line fix" for Asterisk 19 very shortly:
https://github.com/FreePBX/framework/tree/release/16.0.10.40
https://git.freepbx.org/projects/FREEPBX/repos/framework/branches?base=release%2F16.0.10.40

holta, 2021-11-23:
Further testing can't hurt but it looks like...IIAB's temporary patch is still necessary...for now:

- PR #3019

That's based on a quick look at [Line 38 of install.php](https://github.com/FreePBX/framework/blob/5af305909f9f8e7e4a0303d780dfe1622a3114bd/install.php#L38) and [Line 297 of installlib/installcommand.class.php](https://github.com/FreePBX/framework/blob/5af305909f9f8e7e4a0303d780dfe1622a3114bd/installlib/installcommand.class.php#L297) within the brand new FreePBX 16.0.10.42 here:

https://github.com/FreePBX/framework/tree/5af305909f9f8e7e4a0303d780dfe1622a3114bd

holta, 2021-12-21:
Still not quite there yet, but FYI FreePBX 16 is at least making motions to start supporting Asterisk 19 more officially soon:

- https://github.com/FreePBX/framework/ommit/050c96e9166d90d7091c62c63e775f051f0033e4
- https://github.com/FreePBX/framework/blob/release/16.0/install.php#L38
- https://github.com/FreePBX/framework/blob/release/16.0/installlib/installcommand.class.php#L297

Towards catching up with their press release from almost 2 months ago: (ahem ;)

https://www.sangoma.com/press-releases/sangoma-announces-asterisk-19-and-freepbx-16/

holta, 2022-06-23:
Good enough.

Let's open a new ticket to deal with future versions of Asterisk 19.x and FreePBX (e.g. pre-releases of version 17).