iiab/iiab #2935: /kiwix/wikipedia_en_medicine_2021-07/A/Book on IIAB 7.2 does not work

shanti-bhardwa, 2021-08-11:
I have built IIAB 7.2 on PI4 using the current raspbian-lite image. The idea is to create IIAB with medical content
and then run it on PI Zero W as a "headless server"

I installed wikipedia_en_medicine_2021-07 but get the following error. 

Is there a problem we know about?

Home Page is
http://192.168.1.170/kiwix/wikipedia_en_medicine_2021-07/A/Wikipedia:WikiProject_Medicine/Open_Textbook_of_Medicine2

from Home Page for 2nd Item 

http://192.168.1.170/kiwix/wikipedia_en_medicine_2021-07/A/Book%3AChildren's_health

get this error

Not Found, The requested URL "/kiwix/wikipedia_en_medicine_2021-07/A/Book:Children's_health" was not found on this server.

But doing a serach with say Malaria gives the page

http://192.168.1.170/kiwix/wikipedia_en_medicine_2021-07/A/Malaria

which tells me that the zim module installed OK. 

The diagnostics is here
http://sprunge.us/4uwtvT?en

tim-moody, 2021-08-12:
@WikiDocJames confirms that the links are missing at the source wikipedia. Short term we need a new home page. Long term we need the mdwiki zim.

WikiDocJames, 2021-08-12:
Hoping to just get the new ZIM working from MDWiki and than to deprecate the Wikipedia based on. Not sure if we will be able to do this in a week or so or not...

shanti-bhardwa, 2021-08-12:
There is no urgency. I can use a working version of the medical wikipedia

tim-moody, 2021-08-12:
I'm pretty sure wikipedia_en_medicine_2021-04 works. Not sure about 05, 06.

holta, 2021-09-12:
@shanti-bhardwa can you update us a month later, on where we stand here?

WikiDocJames, 2021-09-12:
Things are slow on our end. A couple of bugs we are still trying to fix. Tim is away until the end of September so will not be until Oct until we get this working.

jvonau, 2022-04-05:
status?

WikiDocJames, 2022-04-05:
We now have the MDWiki zims out. Would recommend you go with those.

https://download.kiwix.org/zim/other/

James

holta, 2022-05-23:
@shanti-bhardwa can we close this now?

shanti-bhardwa, 2022-05-23:
yes can be closed

closed 2022-05-23 by holta