iiab/iiab #2943: Azuracast installation issue on IIAB

m-anish, 2021-08-14:
### Expected behavior and actual behavior:
Do a BIG sized IIAB install with Azuracast enabled. It should work and it doesn't.

### Steps to reproduce the problem:
When doing a BIG sized install on intel NUC, with azuracast install set to True, the installer fails. Manually trying to install azuracast also fails as it complains about some conflict on the port (10080). Since I haven't worked on IIAB recently, i'm unaware of underlying changes to the server setup, so another pair of eyes would be appreciated. 

Maybe one idea could be to do a small sized IIAB install and then manually install azuracast and seeing what happens?

### IIAB version, OS version, hardware details:
Version: Latest from Git as of 13th Aug 2021
Hardware: 11th gen i5 Intel NUC
OS: Ubuntu server 20.04

holta, 2021-08-14:
> Maybe one idea could be to do a small sized IIAB install

Yes.  Even better, we should also try the unadvertised option zero '0' offered by http://download.iiab.io &mdash; instead of advertised options {'1', '2', '3', and 'm'} &mdash; this new unadvertised option is meant for unit-testing exactly this this.  It installs this extremely minimal /etc/iiab/local_vars.yml :

https://github.com/iiab/iiab/blob/master/vars/local_vars_unittest.yml

> and then manually install azuracast and seeing what happens?

Yes.  Manually settng Lines 182-183 to `True` as follows:

```
azuracast_install: True
azuracast_enabled: True    # This var is currently IGNORED.
```

And then proceeding to run:

```
sudo iiab
```

holta, 2021-08-14:
FYI @m-anish and Gurmet:

[AzuraCast's README](https://github.com/iiab/iiab/tree/master/roles/azuracast#azuracast-readme) has broken/stale links &mdash; and should be converted from .rst (reStructuredText) to the much-friendlier .adoc (AsciiDoc, as Raspberry Pi Foundation is also doing) if this playbook hopefully converges!

An example is here, if this helps in the conversion:
- OLD: https://github.com/iiab/iiab/blob/master/roles/pbx/README.rst.unused
- NEW: https://github.com/iiab/iiab/blob/master/roles/pbx/README.adoc

PS conversion is almost entirely automatic if you run:

```
pandoc README.rst -o README.adoc
```

Preferably with the latest pandoc (`dpkg -i $DEB`) from: https://github.com/jgm/pandoc/releases

jvonau, 2021-08-14:
> ### Expected behavior and actual behavior:
> 
> Do a BIG sized IIAB install with Azuracast enabled. It should work and it doesn't.
> ### Steps to reproduce the problem:
> 
> When doing a BIG sized install on intel NUC, with azuracast install set to True, the installer fails. 

Where does it fail at? Need the info from [iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md) 

> Manually trying to install azuracast also fails as it complains about some conflict on the port (10080). 

Sounds like part iof the iiab install worked. A link to the instructions that you are trying would be useful to know.

holta, 2021-08-14:
> A link to the instructions that you are trying would be useful to know.

Yes.

Until @m-anish & Gurmet can explain more, the installer did not fail (it completed) during this basic install (unit test of roles/azuracast) onto an Ubuntu Server 20.04 VM:

- [iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md) output is here: http://sprunge.us/tpzb7t?en

Feel free to log in to this VM (10.8.0.34 = 191-u20-srv-2943) if that helps.

jvonau, 2021-08-14:
I'm going to repeat myself see: https://github.com/iiab/iiab/pull/2865#issuecomment-882538964. I'd omit docker-compose.override.yml.j2 from the play to better understand what the stock network layout is without the override in place and somehow suppress the network role from running to get that baseline information.

jvonau, 2021-08-15:
`sudo systemctl stop  docker.socket` 
`sudo systemctl start  docker.socket`

dropped the running docker-proxy processes

jvonau, 2021-08-16:
@m-anish I was able to bring up the containers and the networking with #2946 but the ports were not altered can you have a look at that part of the code please. Think the 10080 port clash was an upstream issue, try `./docker.sh update` to rebuild the containers with the latest available.

jvonau, 2021-08-29:
@holta given the lack of response from the original sponsor and lack of progress on #2946 think demoting the role is in order.

holta, 2021-08-29:
It's possible @m-anish is disconnected from the Internet for a few weeks, in the far north of India?

If so @m-anish please clarify on your return: what are your recommendations for AzuraCast?

jvonau, 2021-08-30:
> Until @m-anish & Gurmet can explain more, the installer did not fail (it completed) during this basic install (unit test of roles/azuracast) onto an Ubuntu Server 20.04 VM:
> 

Points to the role completing without error, I suspect what m-anish ran into what was an upstream [issue](https://github.com/AzuraCast/AzuraCast/issues/4491), the port reassignment not working is a new wrinkle I noticed in the role but does not cause the role to fail, just not deployed as intended and is a different issue that PR 2946 might address.

jvonau, 2022-06-25:
0871592ed61704f9cb95048bd6ebf8760ebdf6be

holta, 2022-09-04:
Appears solved by:

- PR #2946

Thanks @m-anish, @rickgurmet & @jvonau!

closed 2022-09-04 by holta