iiab/iiab #2949: Upgrade IIAB's password-tester to support 'yescrypt' not just SHA-515

holta, 2021-08-14:
As a result of Debian 11 Bullseye[1] released an hour ago, yescrypt[2][3] is now the default:

1. https://www.debian.org/News/2021/20210814
2. https://www.debian.org/releases/bullseye/amd64/release-notes/ch-information.en.html#pam-default-password
3. https://manpages.debian.org/testing/libcrypt-dev/crypt.5.en.html#yescrypt

However invoking `yescrypt` from Python is not so easy.  Converting [our code](https://github.com/iiab/iiab-factory/blob/master/iiab#L58-L68) may take some time:

```bash
    [ $(python3 -c "import crypt; print(crypt.crypt('$2', '\$$meth\$$salt'))") == "\$$meth\$$salt\$$hash" ]
```

Why?  Maintainers of Python's crypt.crypt do not seem eager to add additional methods as of 2021-06-05: https://bugs.python.org/issue44309

The claim is that `libcrypt` and [libxcrypt](https://github.com/besser82/libxcrypt) are unreliable &mdash; _and not only that..._

> "We still plan to deprecate and remove the crypt module because it's not reliable"

jvonau, 2021-08-16:
You could drop pre-creation the iiab-admin user in the 'iiab' script and just let the ansible [role](https://github.com/iiab/iiab/tree/master/roles/iiab-admin) handle the situation for you.

holta, 2021-08-16:
> You could drop pre-creation of the iiab-admin user in the 'iiab' script and just let the ansible [role](https://github.com/iiab/iiab/tree/master/roles/iiab-admin) handle the situation for you.

That approach is non-optimal &mdash; as the implementer's attention is very short-lived &mdash; before they walk away from the keyboard (for dinner, etc, as IIAB installs).

So best to get them to deal with published passwords _immediately_ during those very first few seconds.

jvonau, 2021-08-17:
Or not install iiab-admin by default is another option but that just moves the issue to within iiab-support to deal with but that is a better option to install when asked rather than the false sense that the user is forced to change the password but can be bypassed by hitting enter twice. Move that action to where we have the user's complete attention, they want help and are now asking for that help making the out of the box install more secure with one less account on the machine.

jvonau, 2021-08-28:
#2739 https://github.com/iiab/iiab/issues/2749#issuecomment-883898489 Think this issue is a by-product from 2749 which is tagged 7.2 suggesting this issue needs to be dealt with sooner than post release.

tim-moody, 2021-08-28:
I think a better solution is

https://www.debian.org/releases/bullseye/amd64/release-notes/ch-information.en.html#pam-default-password

> If compatibility is required for password hashes between bullseye and buster, modify /etc/pam.d/common-password. Find the line that looks like:
> 
> password [success=1 default=ignore] pam_unix.so obscure yescrypt
> 	
> and replace yescrypt with sha512.

@shanti-bhardwa has tried this with success.

jvonau, 2021-08-28:
I'll just point out Ubuntu-21.04 contents of /etc/pam.d/common-password differ, looking forward yescrypt might be creeping into Ubuntu at some point in the future anyway. Right now 'iiab' really has no concept of OS embedded in the logic yet to know when to edit this very critical system file that I'd [prefer that we stick with OS maintainers' choices](https://github.com/iiab/iiab/pull/2948#issuecomment-899817591) given /etc/shadow is involved.

holta, 2021-08-28:
Solved by:

- PR iiab/iiab-factory#183

closed 2021-08-28 by holta