iiab/iiab #2953: Look at per app setting of php via /etc/php/x.x/fpm/pool.d

jvonau, 2021-08-18:
As demonstrated with freepbx tuning each app's php profile can be done with a drop-in file.

holta, 2021-08-18:
I don't fully understand this per-app settings will overlap and interact &mdash's e.g. when setting these 6 common PHP max/min resource footprint vars: https://github.com/iiab/iiab/blob/master/roles/www_options/tasks/main.yml#L88-L112

But definitely promising, if we can sort out these sometimes hairy detail~

jvonau, 2021-08-18:
Might be getting a bit fancy, not sure if I want to really spend the time. It would involve giving each php based app it's own private php socket to use rather than just the one in www.conf and the single definition of 'php' in /etc/nginx/nginx.conf that gets used with fastcgi_pass. That is how I pulled off freepbx, it's the cookie cutter. Each app could have it's own 'pool' file to setup a private socket for the app's exclusive use and tune the socket setting as need, just point fastcgi_pass to the app's private socket.

jvonau, 2021-09-09:
```
jvonau@jvonau-desktop:~/git/iiab$ grep -r fastcgi_pass roles/*/templates/ | sort 
roles/awstats/templates/awstats-nginx.conf:    fastcgi_pass    php;
roles/mediawiki/templates/mediawiki-nginx.conf.j2:#        fastcgi_pass 127.0.0.1:9000; # or whatever port your PHP-FPM listens on
roles/mediawiki/templates/mediawiki-nginx.conf.j2:    fastcgi_pass php; # or whatever port your PHP-FPM listens on
roles/moodle/templates/moodle-nginx.conf.j2:    fastcgi_pass php;
roles/nextcloud/templates/nextcloud-nginx.conf.j2:        fastcgi_pass php;
roles/nginx/templates/admin-console-nginx.conf.unused:    fastcgi_pass php;
roles/nginx/templates/server.conf.j2:        fastcgi_pass php;
roles/osm-vector-maps/templates/osm-vector-maps-nginx.conf.j2:    fastcgi_pass php;
roles/pbx/templates/freepbx-nginx.conf.j2:        fastcgi_pass unix:/run/php/php-asterisk.sock;
roles/wordpress/templates/wordpress-nginx.conf.j2:        fastcgi_pass php;
roles/wordpress/templates/wordpress-nginx.conf.j2:        fastcgi_pass php;
```