iiab/iiab #2958: More informative messages than IIAB DOES NOT SUPPORT UNINSTALLS; Ticket template; FAQ

dagelf, 2021-08-19:
Run iiab-install. 
It gives an ansible error. 
Change /etc/iiab/local_vars.yml to disable the thing that gave the error. 
Now a new error comes up, the one named about this ticket. 

### Expected behavior and actual behavior:
A message pointing the user to a solution, rather than contradictory messages or new errors. Even if it is "rm -rf /opt/iiab and restart"

### Steps to reproduce the problem:
Run iiab-install enabling something that's broken on a certain platform. 
Disable that item in /etc/iiab/local_vars.yml. 
Re-run `cd /opt/iiab/iiab; iiab-install` or `sudo iiab` script as instructed upon first run. 

### IIAB version, OS version, hardware details:
Latest as of this ticket. If you're reading this, please tell me where to check this...? I've grepped the whole tree and I don't see anything that looks like a version number or issue file. Maybe put instructions to find this in the ticket template too?

There's also nothing mentioned in the FAQ on how to proceed if you run into errors. At least a pointer to the supported hardware and OSes page would help, along with instructions on how to restart a failed installation with a new configuration that might work.

jvonau, 2021-08-19:
related #2525 https://github.com/iiab/iiab/issues/2502#issuecomment-686481523

jvonau, 2021-08-19:
> Run iiab-install.
> It gives an ansible error.
> Change /etc/iiab/local_vars.yml to disable the thing that gave the error.
> Now a new error comes up, the one named about this ticket.

Think you may have 'undone' the wrong variable in relation to whatever error you ran into, now the error you are seeing is because that role as already been recorded as installed and you changed local_vars.yml to *_install: False after the role was installed with *_install: True where * is the name of the role involved. Yes viewing the error is a pain there is no summary that shows up at the bottom, here is a mockup 
```
TASK [0-init : DISALLOW "XYZ_install: False" WHEN "XYZ_installed is defined" IN /etc/iiab/iiab_state.yml ...for up-to-43 var pairs] ***
ok: [127.0.0.1] => (item=hostapd)
ok: [127.0.0.1] => (item=dhcpd)
ok: [127.0.0.1] => (item=named)
ok: [127.0.0.1] => (item=dnsmasq)
ok: [127.0.0.1] => (item=bluetooth)
ok: [127.0.0.1] => (item=sshd)
ok: [127.0.0.1] => (item=openvpn)
ok: [127.0.0.1] => (item=admin_console)
failed: [127.0.0.1] (item=squid) => {"ansible_loop_var": "item", "assertion": "squid_install or squid_installed is undefined", "changed": false, "evaluated_to": false, "item": "squid", "msg": "DISALLOWED: 'squid_install: False' (e.g. in /etc/iiab/local_vars.yml) WHEN 'squid_installed' is defined (e.g. in /etc/iiab/iiab_state.yml) -- IIAB DOES NOT SUPPORT UNINSTALLS -- please verify those 2 files especially, and other places variables are defined?"}
ok: [127.0.0.1] => (item=cups)
ok: [127.0.0.1] => (item=samba)
ok: [127.0.0.1] => (item=usb_lib)
ok: [127.0.0.1] => (item=azuracast)
ok: [127.0.0.1] => (item=gitea)
ok: [127.0.0.1] => (item=jupyterhub)
ok: [127.0.0.1] => (item=lokole)
ok: [127.0.0.1] => (item=mediawiki)
ok: [127.0.0.1] => (item=mosquitto)
skipping: [127.0.0.1] => (item=nodejs) 
ok: [127.0.0.1] => (item=nodered)
ok: [127.0.0.1] => (item=nextcloud)
ok: [127.0.0.1] => (item=wordpress)
ok: [127.0.0.1] => (item=kalite)
ok: [127.0.0.1] => (item=kolibri)
ok: [127.0.0.1] => (item=kiwix)
skipping: [127.0.0.1] => (item=postgresql) 
ok: [127.0.0.1] => (item=moodle)
skipping: [127.0.0.1] => (item=mongodb) 
ok: [127.0.0.1] => (item=sugarizer)
ok: [127.0.0.1] => (item=osm_vector_maps)
ok: [127.0.0.1] => (item=transmission)
ok: [127.0.0.1] => (item=awstats)
ok: [127.0.0.1] => (item=monit)
ok: [127.0.0.1] => (item=munin)
ok: [127.0.0.1] => (item=phpmyadmin)
ok: [127.0.0.1] => (item=vnstat)
skipping: [127.0.0.1] => (item=yarn) 
ok: [127.0.0.1] => (item=captiveportal)
ok: [127.0.0.1] => (item=internetarchive)
ok: [127.0.0.1] => (item=minetest)
ok: [127.0.0.1] => (item=calibreweb)
ok: [127.0.0.1] => (item=calibre)
ok: [127.0.0.1] => (item=pbx)

PLAY RECAP ************************************************************************************************************************
127.0.0.1                  : ok=12   changed=0    unreachable=0    failed=1    skipped=2    rescued=0    ignored=0  
```
>DISALLOWED: 'squid_install: False' (e.g. in /etc/iiab/local_vars.yml) WHEN 'squid_installed' is defined (e.g. in /etc/iiab/iiab_state.ym -- IIAB DOES NOT SUPPORT UNINSTALLS 

You can get a better view of the error from iiab-install.log, the error might of scrolled out of view on the monitor if you are at the keyboard and not ssh'd in.

dagelf, 2021-08-19:
The only question is what is the correct thing to do here? Or the best way
out. If it is to reformat and reinstall everything from scratch, then a
better message would be "YOU HAVE TO START OVER" or if there is just one
file that you need to delete, or one directory, then a better message would
be "DELETE ... TO REINSTALL". The problem is that I think most people, even
very technical, would get stuck here.

The question is how can we direct people to be unstuck. The log only shows
exactly the same. Is it possible to remove a role, or to just remove all
the scripts related to a role? Is it really necessary for everyone who want
to use IIAB to become an ansible novice?

jvonau, 2021-08-19:
@dagelf Think the correct thing to do would be to ask for help, you asked, I explained where to look for your error in editing the main config file. Let's cut to the chase here I'm on your side but you don't know that yet see https://github.com/iiab/iiab/issues/2525#issue-702000006, I foretold of this then unreported issue months ago. If you think you have to start over that is up to you but that is where iiab-diagnostics is you friend and we can help avoid starting over if you want.

dagelf, 2021-08-19:
Thanks, I think you misunderstand the point of this ticket - next I want to
do a PR that changes the message to something like:

"Changing local_vars.yml after install is not supported, check
iiab-diagnostics, read xxx or reinstall."

I have worked my way through
https://github.com/iiab/iiab/wiki/IIAB-Contributors-Guide - that's a great
introduction. My intent here is to see what I can do to make things even
friendlier to new devops- and developers who can contribute. Where "xxx"
would be either a link to https://github.com/iiab/iiab/wiki/IIAB-Platforms,
or to
https://github.com/iiab/iiab/wiki/IIAB-Installation#do-everything-from-scratch
- do we have a short url for either of these pages?

holta, 2021-08-19:
Thanks @dagelf for pushing.  Just a couple quick points just for now:

> "Changing local_vars.yml after install is not supported, check
iiab-diagnostics, read xxx or reinstall."

1. Actually /etc/iiab/local_vars.yml can be changed after IIAB's installed, but within limits!

   Personally I use `cd /opt/iiab/iiab` then `sudo ./runrole <SOME ROLE>` often, after having changed local_vars.yml

2) See also this new FAQ, that is not yet released &mdash; but you are dedicated so best to use the latest tips as you explore:

   https://wiki.iiab.io/go/FAQ

jvonau, 2021-08-19:
Think the general point would be that adding the extra "IIAB DOES NOT SUPPORT UNINSTALLS" is distracting the attention away from the error message being returned by taking up to much visual real estate.

jvonau, 2021-08-19:
Think the best summation is in the error message ""DISALLOWED: 'squid_install: False' (e.g. in /etc/iiab/local_vars.yml) WHEN 'squid_installed' is defined"

dagelf, 2021-08-19:
Okay, I see that that is actually what it says before that line. While
technically correct, I also agree with your previous point that it's
distracting attention away.

The previous message is useful to those versed in ansible. The last message
don't help much - so my proposal here is to just change the part in all
caps:
    iiab/roles/0-init/tasks/validate_vars.yml:    fail_msg: "DISALLOWED:
'{{ item }}_install: False' WITH '{{ item }}_enabled: True' -- IIAB DOES
NOT SUPPORT UNINSTALLS -- please verify those 2 variable values e.g. in
/etc/iiab/local_vars.yml, and other places variables are defined?"
    iiab/roles/0-init/tasks/validate_vars.yml:    fail_msg: "DISALLOWED:
'{{ item }}_install: False' (e.g. in /etc/iiab/local_vars.yml) WHEN '{{
item }}_installed' is defined (e.g. in /etc/iiab/iiab_state.yml) -- IIAB
DOES NOT SUPPORT UNINSTALLS -- please verify those 2 files especially, and
other places variables are defined?"

Would you accept a PR that simply replaces the all caps part with "Please
take the time to read
https://github.com/iiab/iiab/wiki/IIAB-Installation#do-everything-from-scratch"
- although I am still working through this myself and I am hoping to come
up with a more targeted place to lead the any newcomer to this project to.
Give me a day or so to work through this.

(I am preparing to install IIAB on 3 new platforms but I don't want to bug
you with all the issues, just want to contribute here to something more
general that would've saved helped me on my journey so far - I have had to
start from scratch 7 times now... the purpose of the message would be a
message that could've helped someone like me go from 0 to 1 without having
to start over at all... because I was hoping that I could have been helping
curate content by now, but am still struggling just to start. I know I
could just use a supported platform, and I usually do, but if I can get
this working I think I will be able to take IIAB to a lot more people.)

holta, 2021-08-19:
@dagelf have you looked over [/etc/iiab/local_vars.yml](https://wiki.iiab.io/go/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it.3F) itself?

I very strongly recommend reading through it, even if just very quickly!

Just as an example, these 2 lines near the very top, are critically important:

https://github.com/iiab/iiab/blob/42d4d70367ca1e983fe961fc50a38e9ac3ae10bb/vars/local_vars_medium.yml#L7-L8

holta, 2021-08-19:
> I am preparing to install IIAB on 3 new platforms [...]

Which 3 new platforms?

Thanks @dagelf for explaining a bit, when you have a sec!

dagelf, 2021-08-19:
I have read through it in detail. I have even Googled it and saw that almost no other projects use the same filename ☺️. (The confusing part about "not supporting uninstalling" is that it hasn't installed anything yet... I am still trying to do the first install. 

I have finally managed to find what I broke before this started happening... problem with working with a new unfamiliar system, is that you tend to look for problems where there aren't any, because you don't know the system. It's great working with something new and then finding that it's very easy and friendly to use and that it even shows you your own mistakes too... that inspires confidence that builds trust and respect, and invites more contributions.

jvonau, 2021-08-19:
Always nice to meet redistributors of "internet in a box", nice [video](https://www.youtube.com/watch?v=FJljDlPyuDk) and [user guide](https://www.internetsociety.org/wp-content/uploads/2020/11/South-Africa-Chapter-Manual-Internet-In-A-Box-Setup-for-Everyone.pdf), what other technical difficulties have you encountered during your rollout of IIAB?

dagelf, 2021-08-19:
I am super inspired by the foundations you've built. We're but a handful of volunteers and it's been slow going, but I think we are just getting started because there is so much potential here.

Our main goal is to make IIAB more user friendly for non technical users. We are trying to bring technology to within easier reach of townships and rural villages, in a low cost way, as a stepping stone to empowering people to use technology and to make it their own. Our biggest problems here are social problems and a dependency mindset of "things are made and come from elsewhere" - and that is the hardest thing that we are focused on changing and want to play a part with this; anything technical really takes a back seat or falls on the one or two people who can do it, while the people who know the least about technology have the highest expectations and the least patience, unnecessarily, and ironically the same people who could stand to gain the most - and whom we seek to empower - if we can get the technical things out of the way.  

We have a $2000 grant from Internet Society, which we plan on spending on as many IIAB devices as possible - to distribute and gain feedback about how people think it can be improved in this setting, so that we can see what we can do - and also as a way to help us find communities who can benefit from building their own "community networks", to connect them with the right help, so they can build themselves more affordable internet. 

Also, we had quite a lot of trouble finding Raspberry Pi's, even Pi Zero W's - it took me a year to find a supplier who will sell us more than 1 at a time, without a 100% mark-up... and in quantities of less than 50 at a time. 

While I am waiting for them to arrive, I am trying to get to know the basics by getting it to run on a VM, Banana Pi M2 Zero and then want to explore options to make it more portable with docker or systemd-nspawn. But what I really want to get to is easier curation (and creation) of content, and interoperability with smartphones, maybe an F-Droid server or something similar. I've always been of the opinion that it's less effort to just promote building out better internet networking, but there's still a HUGE role for IIAB to play here, besides the schools - which all have non-functioning government-sponsored internet (because they don't have electricity or tech savvy teachers) - there's also a whole non-cash economy in Africa, where every cent and bit of knowledge really really goes a long way... it just needs to be packaged to be accessible, and local. The way you remember the internet from the 80's? That's reality, today, right here. Except if can link people to the cutting edge of today. Incredible, right?! That's the thought that keeps me going! And then there's such amazing content being made, finally... but still a lot that can be done. There are also new local community TV- and radio stations popping up everywhere. It's chaos really, but the material and commercial world needs to be balanced out somehow, so that people can see that technology is not just about apps and products - and I can't think of anything better than a library to do that.

jvonau, 2022-04-05:
#2739

closed 2022-04-05 by holta