iiab/iiab #2960: [WARNING]: error loading facts as JSON or ini - please check content: /etc/ansible/facts.d/local_facts.fact

dagelf, 2021-08-19:
### Expected behavior and actual behavior:
Following one click install, with unit-test, minimal, medium or big install, and manual steps from https://github.com/iiab/iiab/wiki/IIAB-Installation#do-everything-from-scratch, all lead to the same dead end below. It seems that `bash.env` does not get generated or is missing. Where does it get generated? Or did I miss a step? Which one? I have done this at least 10 times now, so I am sure I am not missing a step.

I've never used ansible before - but I have read the manual and the introduction, but I am still missing something. This is where I am at: From my reading of https://github.com/iiab/iiab/blob/master/vars/default_vars.yml it seems there is a catch-22 - the `/etc/ansible/facts.d/local_facts.fact` mentioned contains environment variables that aren't set, and seems to be set form `iiab.env`, but that only gets generated much later, from the facts file, by the looks of it. 

### Steps to reproduce the problem:

```
root@box:/opt/iiab/iiab# sudo ./iiab-install

./iiab-install  BEGUN IN /opt/iiab/iiab

Placed /etc/ansible/facts.d/local_facts.fact into position.
Found Kernel 5.4.0-81-generic
Found Ansible [core 2.11.4]

TRY TO RERUN './iiab-install' IF IT FAILS DUE TO CONNECTIVITY ISSUES ETC!

Running local Ansible playbooks...
...Stage 0 will now run
...followed by Stages 1-9
...and then the Network Role.

[WARNING]: error loading facts as JSON or ini - please check content:
/etc/ansible/facts.d/local_facts.fact
        "ansible_python": {
            "executable": "/usr/bin/python3",
            "type": "cpython",
        "ansible_python_version": "3.8.10",
        "ansible_selinux_python_present": true,
[WARNING]: error loading facts as JSON or ini - please check content:
/etc/ansible/facts.d/local_facts.fact
[WARNING]: While constructing a mapping from /etc/iiab/local_vars.yml, line 16, column 1, found a duplicate dict key (iiab_hostname). Using last defined value only.

PLAY [all] ********************************************************************************************************************************************************************************************

TASK [Gathering Facts] ********************************************************************************************************************************************************************************
[WARNING]: error loading facts as JSON or ini - please check content: /etc/ansible/facts.d/local_facts.fact
ok: [127.0.0.1]
ERROR! an undefined variable was found when attempting to template the vars_files item 'vars/{{ ansible_local.local_facts.os_ver }}.yml'

The error appears to be in '/opt/iiab/iiab/iiab-stages.yml': line 8, column 5, but may
be elsewhere in the file depending on the exact syntax problem.

The offending line appears to be:

  - vars/default_vars.yml
  - vars/{{ ansible_local.local_facts.os_ver }}.yml
    ^ here
We could be wrong, but this one looks like it might be an issue with
missing quotes. Always quote template expression brackets when they
start a value. For instance:

    with_items:
      - {{ foo }}

Should be written as:

    with_items:
      - "{{ foo }}"
root@box:/opt/iiab/iiab#

```

My `/etc/ansible/facts.d/local_facts.fact` is identical to https://github.com/iiab/iiab/blob/eabd521200be088fd484edaaa8bc430851297adc/scripts/local_facts.fact

### IIAB version, OS version, hardware details:
Version: git
OS: Ubuntu 20.04
Hardware: qemu_system_x86_64 kvm

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)

Even this tries to reference the `iiab.env` file which I seem to be missing (where and how did I lose it?):
```
root@box:/opt/iiab/iiab# scripts/iiab-diagnostics
cat: /etc/iiab/iiab.env: No such file or directory
cat: /etc/iiab/iiab.env: No such file or directory

Gathers IIAB diagnostics into 1 file, to accelerate troubleshooting.  USAGE:

   iiab-diagnostics
   sudo iiab-diagnostics                                # USE 'sudo' FOR MORE
   sudo iiab-diagnostics PATH/FILE1 PATH/FILE2 ...      # COMPLETE RESULTS !!

Can you provide a short public nickname: (no spaces!) isa1

Compiling diagnostics...

  0. Filename Header + Git Hashes + Raspberry Pi Model + OS
scripts/iiab-diagnostics: line 110: /etc/iiab/diag/__2021-08-19_21:08:21_UTC_isa1: No such file or directory
scripts/iiab-diagnostics: line 111: /etc/iiab/diag/__2021-08-19_21:08:21_UTC_isa1: No such file or directory
scripts/iiab-diagnostics: line 112: /etc/iiab/diag/__2021-08-19_21:08:21_UTC_isa1: No such file or directory
scripts/iiab-diagnostics: line 113: /etc/iiab/diag/__2021-08-19_21:08:21_UTC_isa1: No such file or directory
scripts/iiab-diagnostics: line 114: /etc/iiab/diag/__2021-08-19_21:08:21_UTC_isa1: No such file or directory
scripts/iiab-diagnostics: line 115: /etc/iiab/diag/__2021-08-19_21:08:21_UTC_isa1: No such file or directory
scripts/iiab-diagnostics: line 116: /etc/iiab/diag/__2021-08-19_21:08:21_UTC_isa1: No such file or directory
   ... these errors continue for everything onwards

```

dagelf, 2021-08-19:
Re iiab-diagnostics: I am so confused. Line 110 of `scripts/iiab-diagnostics` is an echo command - and there is no directory named `/etc/iiab/diag` on the system.

dagelf, 2021-08-19:
I now see that `/etc/ansible/facts.d/local_facts.fact` is a script that plays a role in generating the missing file. 

It failed because I made a typo in a pre-setup script that overwrote Solution found: a typo in another script that broke `systemctl` 🤦

Sanity checking the output of `systemctl` commands and errorring as part of the local_facts.fact file, could've saved me a few hours here... will try do a PR later. Although the chances of someone else doing what I just did are slim!

jvonau, 2021-08-20:
Are you trying to introduce a new OS or trying to setup a build pipeline? Being a bash script you can just run it and see the returned values
```
ubuntu@box:/opt/iiab/iiab$ /opt/iiab/iiab/scripts/local_facts.fact 
/opt/iiab/iiab/scripts/local_facts.fact: line 94: warning: command substitution: ignored null byte in input
Failed to get unit file state for dhcpcd.service: No such file or directory
{"stage"                  : "9",
"dhcpcd"                  : "none",
"network_manager"         : "enabled",
"systemd_networkd"        : "enabled",
"iiab_branch"             : "no-net-exit",
"iiab_commit"             : "1e2096e50c1f778e5ebc5aaa13817226a50a424a",
"rpi_model"               : "Raspberry Pi 400 Rev 1.0",
"ansible_version"         : "2.10.12",
"os"                      : "ubuntu",
"os_ver"                  : "ubuntu-20"}
```
The error you noted is about os detection that would be derived from the contents of /etc/os-release on your target VM, to tell ansible about your OS.

jvonau, 2021-08-20:
https://github.com/iiab/iiab-factory/pull/136