iiab/iiab #2977: How do I get all menu items activated?

holta, 2021-09-05:
### Discussed in https://github.com/iiab/iiab/discussions/2969

<div type='discussions-op-text'>

<sup>Originally posted by **bernardroth** August 25, 2021</sup>
After I install the largest option I install the medical version and try to add the other menu items, but they do not work. Only the medical and map options work. The school options fail. When I click on Wikipedia I get "Undefined Zim in a Menu Definition". Radiolab "404 Not Found".

Is there a way I can get every option (medical and education) to work at the same time?</div>

holta, 2021-09-05:
See both:

- #2969
- https://github.com/iiab/iiab-admin-console/issues/427

tim-moody, 2021-09-05:
This is a duplicate of the mentioned #2969.

holta, 2021-09-05:
This ticket is very much a duplicate tracking ticket on purpose, as discussion tickets are not trackable as issues and PR's are.