iiab/iiab #2980: Trying to get in touch regarding a security issue

JamieSlome, 2021-09-09:
Hey there!

I'd like to report a security issue but cannot find contact instructions on your repository.

If not a hassle, might you kindly add a `SECURITY.md` file with an email, or another contact method? GitHub [recommends](https://docs.github.com/en/code-security/getting-started/adding-a-security-policy-to-your-repository) this best practice to ensure security issues are responsibly disclosed, and it would serve as a simple instruction for security researchers in the future.

Thank you for your consideration, and I look forward to hearing from you!

(cc @huntr-helper)

holta, 2021-09-09:
Sure.  And in the meantime, please consider contacting us via https://internet-in-a-box.org/pages/contributing.html

Thanks!

holta, 2021-09-10:
FYI security@huntr.dev (https://huntr.dev/users/vikychoi/ and https://huntr.dev/users/zidingz/) emailed:

> Path Traversal in iiab/iiab
 Unverified Reported on Sep 9th 2021
>
> ✍️ Description
A path traversal attack (also known as directory traversal) aims to access files and directories that are stored outside the web root folder. By manipulating variables that reference files with “dot-dot-slash (../)” sequences and its variations or by using absolute file paths, it may be possible to access arbitrary files and directories stored on file system including application source code or configuration and critical system files. It should be noted that access to files is limited by system operational access control (such as in the case of locked or in-use files on the Microsoft Windows operating system).
>
> NGINX is a web server which can also be used as a reverse proxy, load balancer, mail proxy and HTTP cache. The NGINX alias directive defines a replacement for the specified location.
>
> The incorrect configuration of the alias could allow an attacker to read file stored outside the target folder.
>
> In particular, the locations /static, /media, /content is configured to have aliases $kalite_home/httpsrv/static/, $kalite_home/httpsrv/media/, $kalite_home/content/, which the locations doesn't ends with directory separator (/) in nginx configuration file kalite-nginx.conf.unused.
>
> Remediation
Find all NGINX alias directives and make sure that the parent prefixed location ends with directory separator. 📍 Location kalite-nginx.conf.unused#L23 kalite-nginx.conf.unused#L27 kalite-nginx.conf.unused#L31 📝 References Path traversal via misconfigured NGINX alias
>
> 📍 Location
kalite-nginx.conf.unused#L23
kalite-nginx.conf.unused#L27
kalite-nginx.conf.unused#L31
>
> 📝 References
Path traversal via misconfigured NGINX alias
https://www.acunetix.com/vulnerabilities/web/path-traversal-via-misconfigured-nginx-alias/

> https://cwe.mitre.org/data/definitions/22.html