iiab/iiab #2985: Unable to log in to http://box/admin/ with user iiab-admin password set during installation or default password

boholder, 2021-09-25:
### Expected behavior and actual behavior:

Use the 1-line installer to install. 
The installation process prompted for a user password for `iiab-admin`, so I entered a custom password. 

After a successful installation and reboot, I logged into the administration interface on the host using http://box/admin, 
but both the custom password entered during installation and the default password (`g0adm1n`) showed `invalid login` (username as `iiab-admin`).

After searched for issues, I found that it seems that the password uses the password of the unix user system directly, so I tried to change the password of the user `iiab-admin` on the system (use `passwd` command) and reboot, but it still did not work.

Any suggestions, or how I can change that password directly using the command line?

(BTW, the installation process was rocky (due to poor internet connection and ISP active blocking policies), I manually configured various software images and proxies, but I kept running the single line install script over and over again, so maybe I didn't screwed up iiab itself?)

### Steps to reproduce the problem:

N/A

### IIAB version, OS version, hardware details:

IIAB: 7.2, Debian 11 amd64 (bullseye) (in virtualbox), host is Windows10

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)

http://sprunge.us/jLQuPl

holta, 2021-09-25:
Thanks @boholder for reporting this!

Admin Console does not support Debian 11 Bullseye's default password hashing scheme (yescrypt) at this time, as used to record passwords in `/etc/shadow`

As a temporary workaround, can you force the Linux user to hash its password using SHA-512 instead of yescrypt ?

Here is one way to do this:

1. Modify `/etc/pam.d/common-password`

   Find the line that looks like:

    ```password [success=1 default=ignore] pam_unix.so obscure yescrypt```
	
    ...and replace `yescrypt` with `sha512`

2. Change the password for user `iiab-admin` (or any user that is a member of *group* `iiab-admin`) as explained here:

    - https://github.com/iiab/iiab/tree/master/roles/iiab-admin
    - https://github.com/iiab/iiab-admin-console/blob/master/Authentication.md

3. OPTIONAL: Verify its password is hashed using SHA-512 instead of yescrypt in `/etc/shadow`

4. Log in to Admin Console

Apologies!  This will be fixed, likely soon in coming weeks.

boholder, 2021-09-25:
Yep, thanks for the detailed guide, I solved the problem by changing the password hashing scheme and the password of user `iiab-admin`.
(I was really relieved when I saw that it was a bug with a clear cause, thanks again for your prompt reply.)

holta, 2021-09-25:
@boholder it's useful to keep this ticket open, to help others track the issue until it's solved, if that's Ok!

holta, 2021-09-26:
@boholder & @tim-moody:

An interim fix is here:

- https://github.com/iiab/iiab-admin-console/pull/428

holta, 2021-09-26:
Thanks @boholder & @tim-moody for helping to fix this for many others, Done!

(This PR / patch below is now tested quite carefully, and merged...)

- https://github.com/iiab/iiab-admin-console/pull/428

closed 2021-09-26 by holta

portna, 2026-04-19:
Yes! Thanks for your prompt reply. https://paste.centos.org/view/0df9fb9e