iiab/iiab #2986: Lokole not working, showing 502 Gateway error

rickgurmet, 2021-09-27:
### Expected behavior and actual behavior:
I tried accessing lokole from my newly set up IIAB but unfortunately it showed Gateway error. I tried every possible solution for gateway error like closing my browser, clearing my cache, etc but nothing worked and it still showed 502 Gateway error.
### Steps to reproduce the problem:

### IIAB version, OS version, hardware details:

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)

holta, 2021-09-27:
@rickgurmet please post the **sprunge.us** URL that is output when you run `sudo iiab-diagnostics` on your IIAB.

Thank You!

Ref:

- #2833

rickgurmet, 2021-09-27:
@holta sprunge.us URL: http://sprunge.us/pKgbjJ

Thanks.

jvonau, 2021-09-28:
Just another half truth report, there would be no way "pbx_installed" should appear, the play was not even reached that stage yet. 
>rw-r--r-- 1 root root 1091 Sep 12 08:59 /etc/iiab/iiab_state.yml
sshd_installed: True
openvpn_installed: True
iiab_admin_installed: True
mysql_installed: True
nginx_installed: True
www_base_installed: True
pylibs_installed: True
usb_lib_installed: True
cups_installed: True
samba_installed: True
www_options_installed: True
azuracast_installed: True
gitea_installed: True
nodejs_installed: True
jupyterhub_installed: True
lokole_installed: True
mediawiki_installed: True
mosquitto_installed: True
nodered_installed: True
nextcloud_installed: True
wordpress_installed: True
kalite_installed: True
kolibri_installed: True
kiwix_installed: True
postgresql_installed: True
moodle_installed: True
osm_vector_maps_installed: True
mongodb_installed: True
sugarizer_installed: True
transmission_installed: True
awstats_installed: True
munin_installed: True
phpmyadmin_installed: True
vnstat_installed: True
captiveportal_installed: True
yarn_installed: True
internetarchive_installed: True
minetest_installed: True
calibreweb_installed: True
pbx_installed: True

>2021-09-27 10:15:28,883 p=45810 u=root n=ansible | TASK [internetarchive : UPGRADE: Stop 'internetarchive' systemd service, if internetarchive_installed is defined] ***
2021-09-27 10:15:29,446 p=45810 u=root n=ansible | changed: [127.0.0.1]
2021-09-27 10:15:29,456 p=45810 u=root n=ansible | TASK [internetarchive : UPGRADE: Run 'yarn upgrade' in /opt/iiab/internetarchive, if internetarchive_installed is defined] ***
2021-09-27 10:15:30,295 p=45810 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "yarn config set child-concurrency 1 && yarn install && yarn upgrade", "delta": "0:00:00.683247", "end": "2021-09-27 10:15:30.278898", "msg": "non-zero return code", "rc": 128, "start": "2021-09-27 10:15:29.595651", "stderr": "warning package.json: No license field\nwarning package.json: No license field\nwarning No license field\nwarning Downloading the git repo \"git://github.com/mitra42/readable-stream-clone\" over plain git without a commit hash\nerror Command failed.\nExit code: 128\nCommand: git\nArguments: ls-remote --tags --heads git://github.com/mitra42/readable-stream-clone\nDirectory: /opt/iiab/internetarchive\nOutput:\nfatal: unable to look up github.com (port 9418) (Temporary failure in name resolution)", "stderr_lines": ["warning package.json: No license field", "warning package.json: No license field", "warning No license field", "warning Downloading the git repo \"git://github.com/mitra42/readable-stream-clone\" over plain git without a commit hash", "error Command failed.", "Exit code: 128", "Command: git", "Arguments: ls-remote --tags --heads git://github.com/mitra42/readable-stream-clone", "Directory: /opt/iiab/internetarchive", "Output:", "fatal: unable to look up github.com (port 9418) (Temporary failure in name resolution)"], "stdout": "yarn config v1.22.5\nsuccess Set \"child-concurrency\" to \"1\".\nDone in 0.07s.\nyarn install v1.22.5\n[1/4] Resolving packages...\ninfo Visit https://yarnpkg.com/en/docs/cli/install for documentation about this command.", "stdout_lines": ["yarn config v1.22.5", "success Set \"child-concurrency\" to \"1\".", "Done in 0.07s.", "yarn install v1.22.5", "[1/4] Resolving packages...", "info Visit https://yarnpkg.com/en/docs/cli/install for documentation about this command."]}
2021-09-27 10:15:30,297 p=45810 u=root n=ansible | PLAY RECAP *********************************************************************
2021-09-27 10:15:30,297 p=45810 u=root n=ansible | 127.0.0.1                  : ok=38   changed=9    unreachable=0    failed=1    skipped=21   rescued=0    ignored=1   

Oh hey lookey here...
>[version]
iiab_commit = 3a9a68f62e4b9e1ed0eb4802ef6967def5980b3e
install_date = 2021-09-12T08:22:21Z
rpi_model = none
[runtime]
iiab_stage = 8
iiab_base_ver = 7.2
iiab_revision = 0
runtime_branch = master
runtime_commit = 3a9a68f62e4b9e1ed0eb4802ef6967def5980b3e
runtime_date = 2021-09-27T10:15:15Z

>Sep 27 10:15 /opt/iiab/iiab/iiab-install.log
Sep 26 07:53 /opt/iiab/iiab/iiab-network.log

The reinstall of internetarchive never completed after a change to captiveportal_enabled? Next iiab-network was run for some reason but the install did complete at some point in the past as other stage 9 apps are marked with installed and internetarchive wants to reinstall (why? installs are a one shot affair for this very reason... reinstalls should be explicit not implied). On a side note why is the flaky azuracast role not in stage 9 given the past grief? 

Ok how about detailing all the gory errors and steps you chose to use to get into this broken state.