iiab/iiab #2991: Azuracast not able to access though Web

rickgurmet, 2021-09-30:
### Expected behavior and actual behavior:

Azuracast is working within the server when accessed through the server using (lynx localhost:10080) but when I access the same on a browser using the URL http://box.lan:10080 it doesn't work.

### Steps to reproduce the problem:

### IIAB version, OS version, hardware details:
IIAB version: Cloned the version as of 15-Sep-2021
Hardware: Intel NUC i3, 10th Gen
OS version: Ubuntu 21.04

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)

holta, 2021-09-30:
@m-anish are you familiar with this issue, or do you understand it?

holta, 2021-10-01:
@rickgurmet it's very important that the `sprunge.us` URL be included, that is output when you run `iiab-diagnostics` on the particular machine where you are experiencing this AzuraCast problem.

Can you please reply with that?

Thanks!

rickgurmet, 2021-10-01:
I have updated it. 
Here is the sprunge.us URL http://sprunge.us/rI2zR9

holta, 2022-08-15:
Just FYI the problem occurs with all browsers that @rickgurmet tried.

He just sent me this screenshot from Chrome:

[image]

jvonau, 2022-08-16:
client side [issue](https://www.google.com/search?q=err_unsafe_port)

holta, 2022-08-16:
Interesting that 10080 is now banned:

Forefox's `gBadPortList` appears to include these 81 values:
https://hg.mozilla.org/mozilla-central/file/tip/netwerk/base/nsIOService.cpp#l106

Chrome's equivalent list:
https://superuser.com/questions/188058/which-ports-are-considered-unsafe-by-chrome/188070#188070
https://chromium.googlesource.com/chromium/src.git/+/refs/heads/master/net/base/port_util.cc#23

A more "official" list appears to be:
https://fetch.spec.whatwg.org/#port-blocking

holta, 2022-09-04:
Appears solved by:

- PR #2946

Thanks @m-anish, @rickgurmet & @jvonau!

closed 2022-09-04 by holta