iiab/iiab #2993: Admin Console "Init failed"

holta, 2021-10-01:
How can we help @rickgurmet work through this failure to access Admin Console?

Output of `sudo iiab-diagnostics` is here: http://sprunge.us/rI2zR9?en

[image]

tim-moody, 2021-10-01:
reboot and try again. if still fails to initialize:

1. ssh into server as root
2. systemctl stop iiab-cmdsrv
3. cd /opt/admin/cmdsrv
4. ./iiab-cmdsrv3.py
5. try again and note any errors

rickgurmet, 2021-10-01:
I did all the steps you mention but after step 4, there is a line that says "in job_minder_thread" and nothing happens after that, just stays there. 
Here is the sprunge.us URL, http://sprunge.us/rgkho1

tim-moody, 2021-10-01:
and you reran admin console?

tim-moody, 2021-10-01:
"in job_minder_thread" is waiting for a command

tim-moody, 2021-10-01:
This was in js console

jquery.min.js:2 jQuery.Deferred exception: Cannot read properties of undefined (reading 'locname') TypeError: Cannot read properties of undefined (reading 'locname')
    at String.<anonymous> (http://10.8.0.38/admin/js/zim_functions.js:139:39)
    at Function.each (http://10.8.0.38/common/js/jquery.min.js:2:3003)
    at procZimGroups (http://10.8.0.38/admin/js/zim_functions.js:135:5)
    at procZimCatalog (http://10.8.0.38/admin/js/zim_functions.js:409:3)
    at Array.initDone (http://10.8.0.38/admin/js/admin_console.js:2307:5)
    at e (http://10.8.0.38/common/js/jquery.min.js:2:30038)
    at t (http://10.8.0.38/common/js/jquery.min.js:2:30340) undefined

lbj is missing from lang_codes.json

holta, 2021-10-01:
Should we close this now, or keep it as a placeholder for other languages that also need fixing?

holta, 2021-10-03:
Here are 36 other missing languages that @tim-moody mentioned:

(that might also need to be addressed in future)

```
Language code lbj not in langCodes.
admin_console.js:2156 Language code ban not in langCodes.
admin_console.js:2156 Language code lfn not in langCodes.
admin_console.js:2156 Language code din not in langCodes.
admin_console.js:2156 Language code gor not in langCodes.
admin_console.js:2156 Language code nai not in langCodes.
admin_console.js:2156 Language code sgs not in langCodes.
admin_console.js:2156 Language code grc not in langCodes.
admin_console.js:2156 Language code rup not in langCodes.
admin_console.js:2156 Language code azb not in langCodes.
admin_console.js:2156 Language code nan not in langCodes.
admin_console.js:2156 Language code arp not in langCodes.
admin_console.js:2156 Language code rmr not in langCodes.
admin_console.js:2156 Language code cbk not in langCodes.
admin_console.js:2156 Language code kbp not in langCodes.
admin_console.js:2156 Language code ady not in langCodes.
admin_console.js:2156 Language code brx not in langCodes.
admin_console.js:2156 Language code crh-latn not in langCodes.
admin_console.js:2156 Language code lzh not in langCodes.
admin_console.js:2156 Language code be-tarask not in langCodes.
admin_console.js:2156 Language code inh not in langCodes.
admin_console.js:2156 Language code vro not in langCodes.
admin_console.js:2156 Language code sat not in langCodes.
admin_console.js:2156 Language code hbs not in langCodes.
admin_console.js:2156 Language code atj not in langCodes.
admin_console.js:2156 Language code lrc not in langCodes.
admin_console.js:2156 Language code jam not in langCodes.
admin_console.js:2156 Language code myn not in langCodes.
admin_console.js:2156 Language code oji not in langCodes.
admin_console.js:2156 Language code enm not in langCodes.
admin_console.js:2156 Language code bgs not in langCodes.
admin_console.js:2156 Language code olo not in langCodes.
admin_console.js:2156 Language code kld not in langCodes.
admin_console.js:2156 Language code ale not in langCodes.
admin_console.js:2156 Language code yue not in langCodes.
admin_console.js:2156 Language code shn not in langCodes.
admin_console.js:2156 Language code mnw not in langCodes.

holta, 2021-10-03:
Also @m-anish will likely benefit from this Ladakhi-related language, if this might be added as well:

> zau – Zangskari

jvonau, 2022-04-05:
status of this issue is?