iiab/iiab #2995: Hot to change IIAB Home Page to WordPress?

holta, 2021-10-03:
@rickgurmet asks:

Can you please tell me where can i change the the home page setting, I have installed another WordPress and i want the homepage to redirect it to that page. 🙏

I don't know nginx much so need to know which file has the homepage setting

holta, 2021-10-03:
@rickgurmet

Did you try changing Line 37 of [/etc/iiab/local_vars.yml](http://FAQ.IIAB.IO#What_is_local_vars.yml_and_how_do_I_customize_it.3F) as explained hereunder?

https://github.com/iiab/iiab/blob/master/vars/local_vars_min.yml#L36-L38

That is, did you try changing...

```
iiab_home_url: /home
```

...to...

```
iiab_home_url: /wordpress
```

holta, 2021-10-03:
@rickgurmet

Just a clarification that you should (try!) to set up custom configurations in [/etc/iiab/local_vars.yml](http://faq.iiab.io/#What_is_local_vars.yml_and_how_do_I_customize_it.3F) before installing IIAB.

If however IIAB is already installed, in this case you might also need to then do:

```
cd /opt/iiab/iiab-admin-console
sudo ./install

holta, 2021-10-03:
Related:

- http://FAQ.IIAB.IO -> ["36 WordPress & Moodle Administration: What tips & tricks exist?"](http://wiki.laptop.org/go/IIAB/FAQ#WordPress_.26_Moodle_Administration:_What_tips_.26_tricks_exist.3F)

tim-moody, 2021-10-03:
Haven't tried it in awhile, but Admin Console Configure->Server Portal is supposed to allow this.

jvonau, 2021-12-11:
The correct answer would be to edit local_vars.yml 's to iiab_home_url: /wordpress then `sudo ./runrole www_options`

holta, 2021-12-11:
https://github.com/iiab/iiab/blob/3b4a8589a140e197b214582941fa0aa2e3486d3d/vars/local_vars_small.yml#L36-L38

jvonau, 2021-12-11:
Allow greater flexibility as the field is free form than the 4 tickbox choices of

Select Server Home Page
/home
/xs-portal
/wordpress
/wiki  

Selecting /wordpress option in admin-console will work with 'save settings' then ICO.

jvonau, 2022-05-25:
> Haven't tried it in awhile, but Admin Console Configure->Server Portal is supposed to allow this.

Note, current iiab-admin-console/master removed this functionality, the only way is via www_options now.

holta, 2022-05-25:
> Note, current iiab-admin-console/master removed this functionality, the only way is via www_options now.

Thanks @jvonau for clarifying.

I'll close this ticket.

(Feel free to re-open if there are outstanding questions!)

closed 2022-05-25 by holta