iiab/iiab #3027: #3023 breaks existing installs

tim-moody, 2021-11-11:
Anyone with an install on raspian 10 who does a pull will now get ERROR! vars file vars/{{ ansible_local.local_facts.os_ver }}.yml was not found when doing anything that involves ansible install.

holta, 2021-11-11:
Debian Buster was desupported 4 months ago.

Very Intentionally So: after Debian Bullseye was very carefully tested, and to keep our community from balkanizing.

Raspbian Buster was desupported 3 days ago.

Very Intentionally So: after Raspberry Pi Bullseye was very carefully tested, and to keep our community from balkanizing.

IIAB supports 9 OS versions within https://github.com/iiab/iiab/blob/master/scripts/local_facts.fact#L72-L85 which is already too many.

Please focus on the positive **(people who've spent hundreds of hours in recent months making sure IIAB runs well on Raspberry Pi OS Bullseye).**

Finally any individual that absolutely must attempt an older, legacy OS is of course free to do so with a 1-line change to the file mentioned above, as is mentioned repeatedly in: https://github.com/iiab/iiab/wiki/IIAB-Platforms

holta, 2021-11-11:
If the "OS Not Supported" message from Line 83 of https://github.com/iiab/iiab/blob/master/scripts/local_facts.fact is not clear, I will work on making that more prominent, more visible and more explanatory.

tim-moody, 2021-11-11:
>Debian Buster was desupported 4 months ago.
Very Intentionally So: after Debian Bullseye was very carefully tested, and to keep our community from balkanizing.
Raspberry Pi OS Buster was desupported 3 days ago.

>IIAB supports 9 OS versions

so what is the recommendation for an existing install, throw it away and start over?

at the very least, be careful never to do a pull

You're only looking at this from the standpoint of doing an install, not an existing server.

>IIAB supports 9 OS versions

Ubuntu from 2020, but not raspios that was only superseded 3 days ago. Even ubuntu-2104 is older than May raspios. And linux mint from 2020.

holta, 2021-11-11:
http://FAQ.IIAB.IO has for many years made clear that IIAB software upgrades are entirely "At Your Own Risk" (e.g. `git pull` is definitely not supported as a mainline practice or ongoing recommendation).

For this reason: those who want to use older, legacy OS's [*] as always need to manually patch `/opt/iiab/iiab/scripts/local_facts.fact` as is mentioned (twice) in https://github.com/iiab/iiab/wiki/IIAB-Platforms

[*] Raspberry Pi OS Buster that was released 29 months ago is just one example, as was deliberately desupported 3 days ago as part of PR #3023, so community usage & testing remain strongly focused on the present not the past.

jvonau, 2021-11-12:
You could have both worlds, the test for NEW installs in the wrong place, something like https://github.com/iiab/iiab-factory/pull/136/files could block older OS at the start of new installs and reverting 3023 could support existing installs where replacing 500GB of content is a PITA. An in place upgrade might be possible explored in #2978 #2820.

tim-moody, 2021-11-12:
The issue is not fresh installs; for those, sure, use the latest os. If I have an existing install on raspios-10 and happen to do a pull just because I want to be current, I lose the ability to enable or disable services because ansible can't run. This just happened to someone from KidsOnComputers. Stopping support for raspios 10 3 days after the release of 11 makes no sense, especially when mint 20 and ubuntu 2004 are still supported.

holta, 2021-11-12:
> happen to do a pull just because I want to

"just because I want to" is fine for people that want to do things At Their Own Risk (as stated prominently in http://FAQ.IIAB.IO -> "27 Can I upgrade IIAB software?" this is not supported — and for good reason: asking for years LTS of support updates for offline devices from IIAB is simply unrealistic in 2021.

As explained in https://github.com/iiab/iiab/wiki/IIAB-Platforms, those individuals who are attached to years-old OS's (Raspberry Pi OS Buster will soon be 3 years old) should make the 1-line change to `/opt/iiab/iiab/scripts/local-facts.fact` — and of course carefully make backups of their IIAB to be sure — if legacy support of years-old software is in fact necessary in some select cases, even while this is damaging IIAB (this inherently balkanizes community focus + quality).

New/experimental options can be better documented for sure, and definitely should be increasingly, to offer cleaner alternatives and community pathways, year by year, and we will do that.

We can do better for sure.  These kinds of questions will surely evolve in the coming years, especially for multi-week implementation efforts evolving in different directions in the short-term — while at the same time keeping ourselves 100% honest that extending support to almost 3-year-old OS's is neither healthy nor viable at this time.

_Especially_ for a small community organization like ours.

**Scope Creep of this nature (implicitly demanding that IIAB must suddenly offer extended LTS support or similar to all existing installs) becomes deeply demotivating and corrosive to testing folk who want to focus on Quality — effectively this is pushing others to Dirty Work against their will, implicitly asking others to test Far Too Many Scenarios than is realistic — in ways that are plainly unsustainable.**

And also becomes inherently very damaging to keeping software quality high, if we accidentally alienate IIAB's testing community, which we cannot afford to do :-(

tim-moody, 2021-11-12:
>asking for years LTS of support updates for offline devices from IIAB is simply unrealistic in 2021

how about days of support. 10 was killed a few days after 11 came out

icebox827, 2021-11-12:
As someone who’s tested IIAB (on many OS’s) in recent weeks, I think that is logical, and it will be a good idea to have a better de-supporting policy that will benefit almost everyone. Three days was too short IIAB should consider including in that policy to not de-support OS’s for a full 10 - 15 days, or whatever duration in the future, to ease the transition. Suggestion for the Future: 10 or 15 days of advance notice? Ansible (26 months) and Nextcloud (14 months) support their software for about two years maximum. Raspberry Pi OS Buster will soon be three years old, and in my opinion, IIAB should focus its energies on OSs that are generally less than two years old. We can have all of those things wrapped up in an agreement to consolidate our legacy and focus on our future.
[6_L'humilité.pdf](https://github.com/iiab/iiab/files/7528610/6_L.humilite.pdf)
[7_L'agilité.pdf](https://github.com/iiab/iiab/files/7528612/7_L.agilite.pdf)

holta, 2021-11-12:
> [6_L'humilité.pdf](https://github.com/iiab/iiab/files/7528610/6_L.humilite.pdf)
> [7_L'agilité.pdf](https://github.com/iiab/iiab/files/7528612/7_L.agilite.pdf)

Thanks @icebox827 for your perspective.

As an aside an IIAB Content Pack on these topics of _empathetic community-building (grounded in reality, not just poetry of course)_ would be fantastic if you possibly have time for scraping/curating sometime in the future.

avni, 2021-11-12:
+1 on having a community-accepted/understood timeframe for de-supporting OSs (10-15 days). 

Tim, can Mason make changes to " /opt/iiab/iiab/scripts/local_facts.fact" to do what he needs to for Buster to move forward?

holta, 2021-11-12:
Please see this PR I just merged, in support of @icebox827's recommendation of a 15-day public transitional/informational period, especially at the end/beginning of each 2-year LTS-like OS release cycle: (whether it's Debian, Ubuntu, or Raspberry Pi OS)

- PR #3034

So that IIAB is not burdened with the impossible task (and unfunded mandate) of supporting an ever-growing panoply of years-old-OS's.

While at the very same time deliberately providing public notice over 2 weeks at the end of each of these LTS-like OS release cycles.

To help all kinds of IIAB communities bring our extremely limited community resources together in common, as quality/testing investments of our lives building out concrete -> common -> humanitarian goals.

tim-moody, 2021-11-12:
>+1 on having a community-accepted/understood timeframe for de-supporting OSs (e.g., 10-15 days).

@avni to be clear, you think that IIAB should end support for existing IIAB installations within as little as 15 days of their creation?

@icebox827 we need to distinguish between support for a new install on an os and supporting an existing install on an older os.

holta, 2021-11-12:
> IIAB should end support for existing IIAB installations within as little as 15 days of their creation?

The above mis-characterizes the longstanding (and very prominently advertised) reality that IIAB does not support "git pull".

To be 100% clear: there is definitely no ending of support for something that was definitely never supported.

As is spelled out with capital letters for many years ("At Your Own Risk") in http://FAQ.IIAB.IO -> "27 Can I upgrade IIAB software?"

holta, 2021-11-12:
> @icebox827 we need to distinguish between support for a new install on an os and supporting an existing install on an older os.

In the future, yes (maybe).  Particularly when offline installs of IIAB become more hybridized and quasi-permanently or more frequently (even intermittently) online in coming years.

That's not yet the case at the moment in ~2021.

(In 5 years it likely will be!?)

Still: even if that does become the new reality over the coming ~5 years, we need to be honest about the infra quality desired + required for this to scale such things meaningfully.  When (or if) field communities truly ask for this in significant and growing numbers.

Android and Windows (despite multi-billion dollar corporate investments) very regularly fail to provide cleanly-QA'd updates (Google messed up and my Android phone accidentally downloaded Android 12 twice in recent weeks, thankfully/apparently not damaging the phone, and Windows updates damage PC's unintentionally on a somewhat regular basis unfortunately).  Not to blame them, but it's not an entirely pretty picture.  So it's not realistic (at this time anyway) to assume IIAB can outperform Google (Android) and Microsoft (Windows) providing a comprehensive updating infra &mdash; for what remains principally an offline tool, as of ~2021.

tim-moody, 2021-11-12:
> longstanding (and very prominently advertised) reality that IIAB does not support "git pull".

I wasn't aware. Where is that documented so prominently

In the faq I found

> PRO TIP: if you've accidentally made changes that prevent "cd /opt/iiab/iiab; git pull" from merging the latest from GitHub's repo cleanly onto your local machine, consider cloning a fresh copy of repo "iiab". To do this, you'd typically run "cd /opt/iiab; mv iiab iiab.old; git clone https://github.com/iiab/iiab". Then if your /etc/iiab/local_vars.yml is correct, you'd typically run "cd /opt/iiab/iiab; ./iiab-install --reinstall".

OK I also found

>Please understand this can DESTROY your Internet-in-a-Box if attempted on an older IIAB !
If your IIAB is more than a few weeks old, it's generally much better to install IIAB from scratch at http://download.iiab.io

Have to read the fine print to know this is about git pull, but I guess it's there further down.

I don't recall participating in a community discussion about any of this. Still feels pretty customer unfriendly to me.

I should add that 3 days or 15 days doesn't matter to me. A miss is as good as a mile. My interest is backwards compatibility, not whether you get 3 days or 15 to do something.

jvonau, 2021-11-13:
I find it very perplexing that https://github.com/iiab/iiab/issues/3006 gets rammed into place while being completely half baked in the beginning breaking master just over 2 weeks ago with a call for testing despite "Structural merges at the end of a release cycle are Not Ok #2531" being bantered about in the past. How does one test the newest bits to be able to use https://github.com/iiab/iiab/tree/master/roles/remoteit/ without a "git pull"? There has been great effort on @holta part to suppress a proper change log practice that would be useful for advanced users to know what has changed from what was in place at the time of their install verses what new latest code offers today, seems that information is being compartmentalized for a reason. I'm also interested in backwards compatibility and safe upgrade paths. I see no valid reason somebody has to start completely over to get the latest bits when there can be branching strategies that could keep everybody happy but the same old same old routine keeps progress stalling. The BS have been going on for over a year now with the targets moving at a whim. When is 7.2 going to be released so things can move forward?

avni, 2021-11-13:
Trying to help find a path forward. Clear documentation and aligned upon timelines for transitions is the suggestion, not a prescribed timeframe. There may be use cases where different durations make sense (as pointed out [here](https://github.com/iiab/iiab/pull/3034#issuecomment-967384768)). 

Mason's experience is not ideal yet there are valid reasons why Buster was desupported. I see/observe the following area where some improvement is needed with respect to support/de-supporting. 

**(Potentially) Can be addressed shorter-term:** 

1. Documentation: From Tim's comment [here](https://github.com//iiab/iiab/issues/3027#issuecomment-967519822), it needs to be more clear that cloning/git pull older version of IIAB can destroy your instance. How do we make it more clear? 
2. Decision Making Criteria: What is the decision making criteria used to de-support a particular OS? The justification given [here](https://github.com//iiab/iiab/issues/3027#issuecomment-966597573) is reasonable. Were folks not aware or not aligned? If the latter, what criteria is preferred? If the former, see #3 below. 
3. Communication: What is the preferred communication when IIAB decides to de-support a particular OS version. Is there a way to notify developers/users proactively - a push notification of some sort? What timeframe/advance notice is needed? 

**Longer-term**

3. Maintenance: It's not trivial to maintain support for multiple OS's as both Adam and Tim have called out. What criteria is used to determined what IIAB supports. Is there a healthy ratio of active IIAB contributors to # of OSs? That may be too prescriptive, but a thought on how to consider what is the "maximum" number.  
4. Change Management: A technical solution to address Jerry's [request](https://github.com//iiab/iiab/issues/3027#issuecomment-967751248) may be helpful to developers. For Kids on Computers, on past trips, we generally reinstalled IIAB, but for someone actively developing, being able to get incremental updates easily may be helpful. Not sure of the time required to make this happen, but worth considering if active developers are requesting it.  

I'm happy to help with #1 if folks have suggestions.

jvonau, 2021-11-13:
Going to put my project management hat on here, there are several issues that are bubbling up to the surface and I'm going to be blunt.

-  The first touch point for the average Joe user would be using the curl routine, that is the logical place to warn the end user their OS is out of date and exit before trying to install anything. This is where you guide the masses to use the latest, not failing after enduring an apt upgrade, reboot then an ansible install only to puke at iiab-install. It's a cop-out if you can't provide this end user experience upgrade as wasting people's time does not leave a warm and fuzzy feeling.

-  With users/developers wishing to preserve backwards compatibility in the short and/or long term there needs to be a clear break point from frozen/working on given OS and the general movement forward. There is a postponed milestone #2601 in respect to being PC in naming of the git branches, this would be the golden opportunity to make the clean break between releases and get with the times. Done right users of now de-supported OS could continue to work on 'master' while work moving forward shifts to 'main'. Given that the basic change in support is RaspiOS-10 current users of any other flavor of OS  should be able to update to 'main' and continue forward. I have tested code 2 years ago that handles the shift nicely but needs updating to what would be the current names in use. Do not act as a lone wolf, this needs to be discussed and organized first.

-  Change logs & versioning, is it really that hard to roll up a series of commits, bump https://github.com/iiab/iiab/blob/master/vars/default_vars.yml#L17 and tag that as a periodic release? Done on a regular bases the ability to use the github's builtin diff tools becomes much more useful and easier to gauge how the recent changes might impact a given use case.

-  I've got more but I just find myself repeating myself cause nobody is listening, why waste my time....

holta, 2021-11-13:
> logical place to warn the end user their OS is out of date and exit before trying to install anything

A) This was very intentionally avoided in the past.  A quick recap of recent years:

For years this was explicitly avoided to permit and encourage wider IIAB experimentation, inviting a broader array of distro developers.  (Anish Mangal's recommendations are now changing some of those community conditions, possibly allowing IIAB to consolidate more strictly around particular apt distros in future.)

So in the future, now that IIAB mandates apt OS's more strictly, and is de-supporting Red Hat OS's on Anish Mangal's recommendation, even further reducing this tolerance for distro diversity and experimentation might well be an option.

To be clear: this would be a 1-line code change (run `local-facts.fact` or similar in the very beginning of IIAB's https://download.iiab.io installer) that certainly did not make sense in the past — so as not to alienate a broader swath of distro experimenters — but indeed might make sense in the future.

<BR>

B) Closely related is the fact that OS requirements are 
carefully spelled, and very prominently so, in places like:

- http://FAQ.IIAB.IO
and
- https://github.com/iiab/iiab/wiki/IIAB-Platforms

There's no confusion here.

At best it's a red herring to suggest otherwise.

At worst, it's yet another in the long line of ad hominem personal attacks that damage absolutely everyone to the very core.

Instead of treating people as authentic human beings.

As @icebox827 courageously chose to do ~24h ago: https://github.com/iiab/iiab/issues/3027#issuecomment-967199938

<BR>

C) With regards to IIAB suddenly changing its mission and purpose from an offline learning tool to transform into a regularly updated online platform (Kolibri "Channels" are just the tip of the iceberg, FOMO being a wider question for civilization) with LTS-like support of older IIAB installs, this is not a decision to be taken at all lightly:

1. Who will finance this?

2. What test plan ensures this will scale?

3. What broad field community needs validate which kinds of content updating and/or app updating are truly required?  What other kinds of updating are NOT required until later years, being more complex and expensive to get right?

4. What modern architectures and techniques will ensure that all updating is resilient for many years to come, respecting versioning SSOT (DRY, etc, etc) ?  How would experienced cloud engineers be hired to avoid the worst mistakes of Android and Windows, that despite billions of dollars of investment, continually struggle (to this day) to get their updating architectures reliable?

These are massive societal and sustainability questions. 

That would need to be tackled professionally over the coming decade (with a business plan).

One year at a time.

tim-moody, 2021-11-13:
Thanks to @jvonau and @avni for thoughtful and insightful comments. Both point to more fundamental problems than a glitch in creating an install that was actually solved by simply creating a new base image.

6 months ago we all agreed that upon the release of 7.2 we would rotate the release manager role, as we have done in the past when I took over from George and Adam took over from me. This would be an opportunity to revisit our process.

holta, 2021-11-13:
> 6 months ago we all agreed that upon the release of 7.2 we would rotate the release manager role

Clarification: This was 14 months ago in September 2020, and in fact we did not agree (the above claim is incorrect).

tim-moody, 2021-11-13:
Sorry, that was my recollection. I think I proposed the rotation and thought everyone agreed. Who was it that didn't agree?

icebox827, 2021-11-13:
> Trying to help find a path forward. Clear documentation and aligned upon timelines for transitions is the suggestion, not a prescribed timeframe. There may be use cases where different durations make sense (as pointed out [here](https://github.com/iiab/iiab/pull/3034#issuecomment-967384768)).
> 
> Mason's experience is not ideal yet there are valid reasons why Buster was desupported. I see/observe the following area where some improvement is needed with respect to support/de-supporting.
> 
> **Can be addressed shorter-term:**
> 
> 1. Documentation: From Tim's comment [here](https://github.com//iiab/iiab/issues/3027#issuecomment-967519822), it needs to be more clear that cloning/git pull older version of IIAB can destroy your instance. How do we make it more clear?
> 2. Decision Making Criteria: What is the decision making criteria used to de-support a particular OS? The justification given [here](https://github.com//iiab/iiab/issues/3027#issuecomment-966597573) is reasonable. Were folks not aware or not aligned? If the latter, what criteria is preferred? If the former, see number 3 below.
> 3. Communication: What is the preferred communication when IIAB decides to de-support a particular OS version. Is there a way to notify developers/users proactively - a push notification of some sort? What timeframe/advance notice is needed?
> 
> **Longer-term:**
> 
> 1. Maintenance: It's not trivial to maintain support for multiple OS's as both Adam and Tim have called out. What criteria is used to determined what IIAB supports. Is there a healthy ratio of active IIAB contributors to # of OSs? That may be too prescriptive, but a thought on how to consider what is the "maximum number" of OS's IIAB can support.
> 2. Change Management: A technical solution to address Jerry's [request](https://github.com//iiab/iiab/issues/3027#issuecomment-967751248) may be helpful to developers. For Kids on Computers, on past trips, we generally reinstalled IIAB, but for someone actively developing, being able to get incremental updates easily may be helpful. Not sure of the time required to make this happen, but worth considering if there is demand and active developers are requesting it.
> 
> I'm happy to help with number 1 above if someone wants to work on this together.

@avni Sure! I can help you with that.

jvonau, 2021-11-14:
> > logical place to warn the end user their OS is out of date and exit before trying to install anything
> 
> A) This was very intentionally avoided in the past. A quick recap of recent years:
> 
> For years this was explicitly avoided to permit and encourage wider IIAB experimentation, inviting a broader array of distro developers. (Anish Mangal's recommendations are now changing some of those community conditions, possibly allowing IIAB to consolidate more strictly around particular apt distros in future.)
> 
> So in the future, now that IIAB mandates apt OS's more strictly, and is de-supporting Red Hat OS's on Anish Mangal's recommendation, even further reducing this tolerance for distro diversity and experimentation might well be an option.
> 
> To be clear: this would be a 1-line code change (run `local-facts.fact` or similar in the very beginning of IIAB's https://download.iiab.io installer) that certainly did not make sense in the past — so as not to alienate a broader swath of distro experimenters — but indeed might make sense in the future.
>
> 
> B) Closely related is the fact that OS requirements are carefully spelled out, and prominently so, in places like:
> 
>     * http://FAQ.IIAB.IO
>       and
> 
>     * https://github.com/iiab/iiab/wiki/IIAB-Platforms
> 
> 
> There's no confusion here.
> 
> At best it's a red herring to suggest otherwise.
> 

I don't see why there is no connection between the two, I'm suggesting a supported OS test very early in install.txt where you have the end-user's attention just 1 second after the first curl action is performed before there is a potential reboot and several minutes before the point where this will end the install in a cryptic message as found in #2960. Stop the install and exit with a messages to check to the FAQ and/or IIAB-Platforms if the OS conditions are not met, just like the better curl based installers do. 
```
Can't detect supported Distro
Please see http://FAQ.IIAB.IO
https://github.com/iiab/iiab/wiki/IIAB-Platforms
whatever else that may be helpful
```
The bump in ansible's version is the blocker on RaspiOS-10, this aspect deals with new installs going forward and separation between what 'iiab-admin' CAN support with the older ansible version installed a week ago. The code base as of right now works with recent installs, keep it like that until there is no longer clear support updates being released by upstream, even then keep it in place just to be able to toggle services as needed in the field for backwards compatibility.
> At worst, it's yet another in the long line of ad hominem personal attacks that damage absolutely everyone to the very core.
> 
> Instead of treating people as authentic human beings.
> 

What is the above suppose to imply? OK I get it, any difference of opinion is viewed as a personal attack. If anything I see the above as a personal attack with a large dose of gaslighting tossed in, I'll just ignore the statement. 

> As @icebox827 courageously chose to do ~24h ago: [#3027 (comment)](https://github.com/iiab/iiab/issues/3027#issuecomment-967199938)
> 
> C) With regards to IIAB suddenly changing its mission and purpose from an offline learning tool to transform into a regularly updated online platform (Kolibri "Channels" are just the tip of the iceberg, FOMO being a wider question for civilization) with LTS-like support of older IIAB installs, this is not a decision to be taken at all lightly:
>
The FUD is starting to get real thick now. None of the optional roles(kolibri, wordpress, moodle, nextcloud) provide any additional support for any of the advanced features that are available upstream, all that the roles accomplish is a skeleton install to get the end-user out of the gate and test the waters. If the objective of the end-user is to use the heavy hitters like moodle or nextcloud the best course of action I could suggest is to start with the dedicated images of those projects. The fly in the ointment 
is wanting to add iiab-admin based content to such an install, like every other installer the assumption is made that the current installer is running the entire show and willfully stomps on all other configurations. I have shown this could programmatically worked around ["very easily (just complete the soft-coding) "](https://github.com/iiab/iiab/pull/2834) to allow the introduction of iiab content to test the waters on any other install base, working towards solving https://github.com/iiab/iiab/issues/2831#issuecomment-869836646
 
>     1. Who will finance this?
>

Who does now? I donate my time. 
>     2. What test plan ensures this will scale?
>

What test plan is in place now? 
>     3. What broad field community needs validate which kinds of content updating and/or app updating are truly required?  What other kinds of updating are NOT required until later years, being more complex and expensive to get right?
> 

What kind of community feedback is received that is not filtered first?  
> 4. What modern architectures and techniques will ensure that all updating is resilient for many years to come, respecting versioning SSOT (DRY, etc, etc) ?  How would experienced cloud engineers be [...]

Life is a dance that you learn as you go. 
> These are massive societal and sustainability questions.
>

How massive is the user base? 
> That would need to be tackled professionally over the coming decade (with a business plan).
> 
> One year at a time.

Why start now, tossing stuff over the fence has worked so well thus far.

> Sorry, that was my recollection. I think I proposed the rotation and thought everyone agreed. Who was it that didn't agree?

I share the same recollection, but lets put that in the past. What is the path forward?

holta, 2021-11-14:
> What is the path forward?

- [6_L'humilité.pdf](https://github.com/iiab/iiab/files/7528610/6_L.humilite.pdf)
- [7_L'agilité.pdf](https://github.com/iiab/iiab/files/7528612/7_L.agilite.pdf)

jvonau, 2021-11-14:
> > What is the path forward?
> 
>     * [6_L'humilité.pdf](https://github.com/iiab/iiab/files/7528610/6_L.humilite.pdf)
> 
>     * [7_L'agilité.pdf](https://github.com/iiab/iiab/files/7528612/7_L.agilite.pdf)
When did the primary language of commutation for this project change? Where can I find the English equivalent for this new direction?

tim-moody, 2021-11-14:
>What is the path forward?

सर्वमिदमग्नीषोममयम्

My French is not good enough to read the chapters cited; perhaps someone could summarize the sections relevant to the current discussion. I have to say that when someone says I should read a chapter on humility it feels like an ad hominem attack.

I would be happy to have @avni and @icebox827 provide documentation on best practices for this group. Perhaps you can decide between you who will drive the effort and let us know what you can deliver. I am content at this point to wait for that.

tim-moody, 2021-11-14:
as another data point, wget appears limited to 2G downloads on raspios-11, so zims larger than that fail to download completely.
tried on en wp and gutenberg.

avni, 2021-11-15:
Hello! We connected this morning and will start with updating the [Contributor's guide](https://github.com/iiab/iiab/wiki/IIAB-Contributors-Guide) over the coming weeks. We will subsequently tackle updating the [FAQs](http://wiki.laptop.org/go/IIAB/FAQ) one-by-one. Will keep this thread posted as we reach notable milestones.

tim-moody, 2021-11-23:
as another data point the fix to wget, which for me was a show stopper, occurred 7 days ago.

tim-moody, 2021-11-23:
In light of the merger of #3045 with no further discussion or documentation I think this can be closed.

closed 2021-11-23 by tim-moody

jvonau, 2021-11-24:
> as another data point the fix to wget, which for me was a show stopper, occurred 7 days ago.

http://raspbian.raspberrypi.org/raspbian/pool/main/w/wget/?C=M;O=D

avni, 2021-12-06:
>Hello! We connected this morning and will start with updating the Contributor's guide over the coming weeks. We will subsequently tackle updating the FAQs one-by-one. Will keep this thread posted as we reach notable milestones.

Following up here that @holta, @icebox827, and I have updated and continue to update the [Technical Contributor's Guide](https://github.com/iiab/iiab/wiki/Technical-Contributors-Guide). There's still a lot to cleanup and update, but we've made a lot of progress in the last 3 weeks testing and updating the page. We welcome any input! Thank you!