iiab/iiab #3039: kiwix-serve doesn't start on Ubuntu 20.04 ARM64 (UTM VM on OSX 12.0.1)

avni, 2021-11-21:
### Expected behavior 
- Have Kiwix-serve start successfully after installation via sudo iiab

### System Information

- **Host OS:** OSX Monterey v12.0.1 M1 processor
- **VM:** UTM VM: https://getutm.app
- **uname -a**
`Linux box 5.4.0-90-generic #101-Ubuntu SMP Fri Oct 15 20:02:26 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux`
- **dpkg --print-architecture** 
`arm64`
- **ansible_machine**
`aarch64`
- **Kiwix version:** Kiwix was downloaded from: http://download.iiab.io/packages/kiwix-tools_linux-armhf-3.1.2-5.tar.gz

### Actual Behavior: When running, sudo iiab, I get the following error related to kiwix

- **iiab-diagnostics output:** http://sprunge.us/GykyNH?en
- **systemctl status kiwix-serve.service** 
```
root@box:/opt/iiab/iiab# systemctl status kiwix-serve.service 
kiwix-serve.service - Provides the Kiwix ZIM File Server
     Loaded: loaded (/etc/systemd/system/kiwix-serve.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sun 2021-11-21 12:57:26 EST; 25min ago
    Process: 49546 ExecStart=/opt/iiab/kiwix/bin/kiwix-serve --daemon --port 3000 --nolibrarybutton --library /library/zims/library.xml --urlRootLocation=/kiwix/ --threads 4 (cod>

Nov 21 12:57:26 box systemd[1]: Starting Provides the Kiwix ZIM File Server...
Nov 21 12:57:26 box systemd[49546]: kiwix-serve.service: Failed to execute command: Exec format error
Nov 21 12:57:26 box systemd[49546]: kiwix-serve.service: Failed at step EXEC spawning /opt/iiab/kiwix/bin/kiwix-serve: Exec format error
Nov 21 12:57:26 box systemd[1]: kiwix-serve.service: Control process exited, code=exited, status=203/EXEC
Nov 21 12:57:26 box systemd[1]: kiwix-serve.service: Failed with result 'exit-code'.
Nov 21 12:57:26 box systemd[1]: Failed to start Provides the Kiwix ZIM File Server.
```
- **Error:** 
```
TASK [kiwix : Enable & Ensure 'kiwix-serve.service' is running] 
**************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service kiwix-serve: 
Job for kiwix-serve.service failed because the control process exited with error code.
\nSee \"systemctl status kiwix-serve.service\" and \"journalctl -xe\" for details.\n"}

PLAY RECAP
**************
127.0.0.1                  : ok=37   changed=10   unreachable=0    failed=1    skipped=15   rescued=0    ignored=0   
```

holta, 2021-11-22:
@avni as an aside, note that your VM disk is only 8.4GB and almost full, and so the `sudo iiab` IIAB install would have failed shortly afterwards, even if kiwix-serve was starting properly.  Please see line 712 of http://sprunge.us/GykyNH?en :

```
COMMAND: /usr/bin/df -h    # Disk usage

Filesystem                         Size  Used Avail Use% Mounted on
...
 /dev/mapper/ubuntu--vg-ubuntu--lv  8.4G  7.3G  642M  93% /
```

Please consider allocating 25GB for VM disks in future, to allow for many different testing scenarios?!

holta, 2021-11-22:
@avni please see @tim-moody's question from 2019 here on a similar topic:

- https://github.com/kiwix/kiwix-tools/issues/346

@jvonau and @kelson do you know more — about what else she might be able to try to get kiwix-serve running in any virtualizated environment, running on top of Apple's M1 CPU ?

avni, 2021-11-22:
> @avni as an aside, note that your VM disk is only 8.4GB and almost full, and so the `sudo iiab` IIAB install would have failed shortly afterwards, even if kiwix-serve was starting properly. Please see line 712 of http://sprunge.us/GykyNH?en :
> 

Ah thank you! I will increase the size.

kelson42, 2021-11-22:
kiwix-serve should be started manually once without any argument to see what goes wrong in detail. If it does not then I would recomment to open a bug report with the `core` file at kiwix/kiwix-tools.

jvonau, 2021-11-22:
```
  32 =IIAB==========================================================================
  33 COMMAND: /usr/bin/dpkg --print-architecture    # RaspiOS-on-PC shows: i386
  34 
  35 arm64
  36 
  37 =IIAB==========================================================================
  38 COMMAND: /usr/bin/dpkg --print-foreign-architectures    # RaspiOS-on-PC shows: amd64
  39 
  40
  1464 Nov 21 16:06:35 iiab-ubuntu-2004 systemd[43161]: kiwix-serve.service: Failed to execute command: Exec format error
  1465 Nov 21 16:06:35 iiab-ubuntu-2004 systemd[43161]: kiwix-serve.service: Failed at step EXEC spawning /opt/iiab/kiwix/bin/kiwix-serve: Exec format error
``` 
Given I've seen kiwix work under arm64 on Ubuntu in the past with 20.04 21.04 on bare arm hardware, @holta confirm working on bare metal with current Ubuntu 21.10-arm64 and RaspiOS-64bit. "Exec format error" suggest a mismatch in architecture that up till now has not been seen when using the armhf binary on a 64bit user-space OS. I check the selinux & app amour settings in the VM to be double sure the issue is not the ready made image if that was the starting point.

holta, 2021-11-22:
> Given I've seen kiwix work under arm64 on Ubuntu in the past with 20.04 21.04 on bare arm hardware, @holta confirm working on bare metal with current Ubuntu 21.10-arm64 and RaspiOS-64bit. "Exec [...]

Makes sense.

There are (claims) that Apple's new M1 CPU's slightly break compatibility with mainline arm64 / aarch64 (?)

So possibly [qemu](https://www.qemu.org/) (that underlies the [UTM virtualization app](https://mac.getutm.app/) @avni is using on macOS) does not yet fully support translating to this newer arm64 instruction set?

Then again I'm just speculating :/

jvonau, 2021-11-22:
My money is on selinux being on the Image, `apt list selinux-utils ` if that is installed check the the setting with `setenforce`, other good Refs with helpful hints:
https://unix.stackexchange.com/questions/472950/systemd-status-203-exec-error-when-creating-new-service 
https://unix.stackexchange.com/questions/664811/systemd-service-failing-with-exit-code-status-203-exec

Otherwise sure the M1 might be the issue, too costly for my budget to find out for myself but the project is at 
https://github.com/utmapp/UTM/ have a look the issues there...

avni, 2021-11-24:
avni@box:~$ apt list selinux-utils
Listing... Done
selinux-utils/focal 3.0-1build2 arm64

avni@box:~$ setenforce
Command 'setenforce' not found, but can be installed with:
sudo apt install selinux-utils

I installed it. 

avni@box:~$ getenforce
Disabled

avni, 2021-11-24:
> kiwix-serve should be started manually once without any argument to see what goes wrong in detail. If it does not then I would recomment to open a bug report with the `core` file at kiwix/kiwix-tools.

**root@box:/etc/systemd/system# systemctl start kiwix-serve.service**
```
Job for kiwix-serve.service failed because the control process exited with error code.
See "systemctl status kiwix-serve.service" and "journalctl -xe" for details.
root@box:/etc/systemd/system# systemctl status kiwix-serve.service

Job for kiwix-serve.service failed because the control process exited with error code.
See "systemctl status kiwix-serve.service" and "journalctl -xe" for details.
```

**root@box:/etc/systemd/system# journalctl -xe**
```
-- 
-- The process' exit code is 'exited' and its exit status is 203.
Nov 23 21:37:52 box systemd[1]: kiwix-serve.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- The unit kiwix-serve.service has entered the 'failed' state with result 'exit-code'.
Nov 23 21:37:52 box systemd[1]: Failed to start Provides the Kiwix ZIM File Server.
-- Subject: A start job for unit kiwix-serve.service has failed
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- A start job for unit kiwix-serve.service has finished with a failure.
```

I can't find kiwix-tools: 

**root@box:/# find . -name "kiwix\*"** 
```
./home/avni/kiwix-error.txt
find: ‘./run/user/1000/gvfs’: Permission denied
./run/lock/kiwix
./run/lock/kiwix/kiwix-lib-access.LCK
./etc/systemd/system/multi-user.target.wants/kiwix-serve.service
./etc/systemd/system/kiwix-serve.service
./opt/iiab/downloads/kiwix-tools_linux-armhf-3.1.2-5.tar.gz
./opt/iiab/iiab-admin-console/roles/cmdsrv/files/json/kiwix_catalog.json
./opt/iiab/iiab/roles/kiwix
./opt/iiab/iiab/roles/kiwix/templates/kiwix.conf.j2.unused
./opt/iiab/iiab/roles/kiwix/templates/kiwix-serve.service.j2
./opt/iiab/iiab/roles/kiwix/templates/kiwix-nginx.conf.j2
./opt/iiab/iiab/roles/kiwix/tasks/kiwix-apk.yml
./opt/iiab/kiwix
./opt/iiab/kiwix/bin/kiwix-serve
./opt/iiab/kiwix/bin/kiwix-search
./opt/iiab/kiwix/bin/kiwix-read
./opt/iiab/kiwix/bin/kiwix-manage
./opt/iiab/iiab-factory/testing/kiwix-load-test.js
./opt/iiab/iiab-factory/content/kiwix
./opt/iiab/iiab-factory/content/kiwix-apks
root@box:/# cd /opt/iiab/kiwix/
```

jvonau, 2021-11-24:
@kelson42 should take over with the steps needed to grab the debugging info from a terminal, /opt/iiab/kiwix/bin/kiwix-serve should be the binary, I can't repo this on rpi hardware on a later version of arm64 Ubuntu.

jvonau, 2021-11-24:
>./run/lock/kiwix
./run/lock/kiwix/kiwix-lib-access.LCK

Can you confirm if /library/zims/library.xml exists or any tmp files?

avni, 2021-11-24:
**avni@box:/library/zims$ ls -al**
```
total 16
drwxr-xr-x 4 root root 4096 Nov 21 10:47 .
drwxr-xr-x 8 root root 4096 Nov 21 10:51 ..
drwxr-xr-x 2 root root 4096 Nov 21 10:52 content
drwxr-xr-x 2 root root 4096 Nov 21 10:47 index
```

**avni@box:/library/zims/content$ ls**
```
test.zim
avni@box:/library/zims/content$ 
```

I am now: 
- creating a new VM of Ubuntu 20.04 arm64 with a larger hard drive (50GB)
- creating a new VM of Ubuntu 14.04 x86 

and will test installing iiab on both to see if I can replicate the issue or successfull install.

avni, 2021-11-24:
Same error on new VM of Ubuntu 20.04 arm64 with a larger hard drive (50GB)

avni, 2021-11-24:
Same issue on new VM with Ubuntu 21.10 arm64, 20GB hard drive.

**root@box:~# systemctl status kiwix-serve.service**
```
× kiwix-serve.service - Provides the Kiwix ZIM File Server
     Loaded: loaded (/etc/systemd/system/kiwix-serve.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2021-11-24 04:50:56 UTC; 1min 21s ago
    Process: 43896 ExecStart=/opt/iiab/kiwix/bin/kiwix-serve --daemon --port 3000 --nolibrarybutton --library /library/zims/library.xm>
        CPU: 1ms

Nov 24 04:50:56 box systemd[1]: Starting Provides the Kiwix ZIM File Server...
Nov 24 04:50:56 box systemd[43896]: kiwix-serve.service: Failed to execute /opt/iiab/kiwix/bin/kiwix-serve: Exec format error
Nov 24 04:50:56 box systemd[43896]: kiwix-serve.service: Failed at step EXEC spawning /opt/iiab/kiwix/bin/kiwix-serve: Exec format err>
Nov 24 04:50:56 box systemd[1]: kiwix-serve.service: Control process exited, code=exited, status=203/EXEC
Nov 24 04:50:56 box systemd[1]: kiwix-serve.service: Failed with result 'exit-code'.
Nov 24 04:50:56 box systemd[1]: Failed to start Provides the Kiwix ZIM File Server.
```

**root@box:~# iiab-diagnostics:** http://sprunge.us/hbLpOf?en

avni, 2021-11-24:
>> creating a new VM of Ubuntu 14.04 x86
Having trouble making this VM functional on UTM. It's glacially slow.

jvonau, 2021-11-24:
Slow down, @holta /library/zims/library.xml from the previous step in the role looks to be missing,  @avni can you from a terminal try `sudo /usr/bin/iiab-make-kiwix-lib` please.

avni, 2021-11-24:
@kelson42 

> kiwix-serve should be started manually once without any argument to see what goes wrong in detail. If it does not then I would recomment to open a bug report with the core file at kiwix/kiwix-tools.

Ran the following: 
root@box:/opt/iiab/kiwix/bin# ./kiwix-serve
bash: ./kiwix-serve: cannot execute binary file: Exec format error

Can't find the "core" file. Can you advise on next steps?

avni, 2021-11-24:
@jvonau 

**root@box:/opt/iiab/kiwix/bin# sudo /usr/bin/iiab-make-kiwix-lib**

```
Waiting to run iiab-make-kiwix-lib.py
Now running iiab-make-kiwix-lib.py
cp: cannot stat '/library/zims/library.xml': No such file or directory
rm: cannot remove '/library/zims/library.xml': No such file or directory
mv: cannot stat '/library/zims/library.xml.tmp': No such file or directory
Job for kiwix-serve.service failed because the control process exited with error code.
See "systemctl status kiwix-serve.service" and "journalctl -xeu kiwix-serve.service" for details.
Finished making Kiwix library.xml
```

Still no "/library/zims/library.xml" file though...

jvonau, 2021-11-24:
Ok, we have the failure point, but why on UTM is the $64 question... 
>jvonau@box:/home/jvonau/$ sudo iiab-make-kiwix-lib
[sudo] password for jvonau: 
Waiting to run iiab-make-kiwix-lib.py
Now running iiab-make-kiwix-lib.py
Finished making Kiwix library.xml

holta, 2021-11-24:
> Ok, we have the failure point, but why on UTM is the $64 question...

If Avni has time over the Thanksgiving long wkd (i.e. the coming 4-5 days) she might possibly try the Trial Version of "Parallels 17" (https://www.parallels.com/pd/virtual-machines-for-mac/) to see if its M1 emulation/virtualization is superior to UTM ~= QEMU ?

jvonau, 2021-11-24:
From hbLpOf?en
```
 683 COMMAND: /usr/bin/lscpu    # CPU details
 684 
 685 Architecture:                    aarch64
 686 CPU op-mode(s):                  64-bit
 687 Byte Order:                      Little Endian
 688 CPU(s):                          4
 689 On-line CPU(s) list:             0-3
 690 Thread(s) per core:              1
 691 Core(s) per socket:              4
 692 Socket(s):                       1
 693 NUMA node(s):                    1
 694 Vendor ID:                       ARM
 695 Model:                           3
 696 Model name:                      Cortex-A72
 697 Stepping:                        r0p3
 698 BogoMIPS:                        48.00
 699 NUMA node0 CPU(s):               0-3
 700 Vulnerability Itlb multihit:     Not affected
 701 Vulnerability L1tf:              Not affected
 702 Vulnerability Mds:               Not affected
 703 Vulnerability Meltdown:          Not affected
 704 Vulnerability Spec store bypass: Vulnerable
 705 Vulnerability Spectre v1:        Mitigation; __user pointer sanitization
 706 Vulnerability Spectre v2:        Not affected
 707 Vulnerability Srbds:             Not affected
 708 Vulnerability Tsx async abort:   Not affected
 709 Flags:                           fp asimd evtstrm aes pmull sha1 sha2 crc32 fphp asimdhp cpuid dit
```
From U21.04 on Rpi-400
```
/usr/bin/lscpu
Architecture:                    aarch64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
CPU(s):                          4
On-line CPU(s) list:             0-3
Thread(s) per core:              1
Core(s) per socket:              4
Socket(s):                       1
Vendor ID:                       ARM
Model:                           3
Model name:                      Cortex-A72
Stepping:                        r0p3
CPU max MHz:                     1800.0000
CPU min MHz:                     600.0000
BogoMIPS:                        108.00
Vulnerability Itlb multihit:     Not affected
Vulnerability L1tf:              Not affected
Vulnerability Mds:               Not affected
Vulnerability Meltdown:          Not affected
Vulnerability Spec store bypass: Vulnerable
Vulnerability Spectre v1:        Mitigation; __user pointer sanitization
Vulnerability Spectre v2:        Vulnerable
Vulnerability Srbds:             Not affected
Vulnerability Tsx async abort:   Not affected
Flags:                           fp asimd evtstrm crc32 cpuid
```

holta, 2021-11-24:
@avni is Ubuntu/Canonical's own new "VMs done right"-on-M1 approach...hypothetically/possibly an improvement over UTM/QEMU on M1 CPUs?

- 'Canonical, Ubuntu’s publisher, announced today “the quickest way” to run Linux cross-platforms on M1 Macs. With Multipass, users can launch a virtual machine image with one command and have Linux running on an M1 Mac in as little as 20 seconds.'
   https://9to5mac.com/2021/11/09/ubuntu-publisher-brings-linux-support-to-m1-mac-with-multipass/
- Multipass 1.8 was just announced 2021-11-09...and claims to offer "VMs done right" ?
   https://ubuntu.com/blog/canonical-transforms-linux-on-mac
- https://github.com/canonical/multipass/releases/tag/v1.8.0

jvonau, 2021-11-24:
> @avni is Ubuntu/Canonical's own new "VMs done right"-on-M1 approach...hypothetically/possibly an improvement over UTM/QEMU on M1 CPUs?
> 
>     * https://github.com/canonical/multipass/releases/tag/v1.8.0

I installed the snap version, seems pretty good. I have a 20.04 VM (running an min install, should be on the vpn now) on my U21.04 arm64 rpi-400 host.

jvonau, 2021-11-24:
http://sprunge.us/o63HoX

holta, 2021-11-25:
Ubuntu's https://Multipass.run looks to have a lot of potential &mdash; arguably it's better (and simpler) than VirtualBox for many everyday purposes.

_However see this warning from November 9-10: (2 weeks ago)_

Q: "Works great on M1!. I have the opposite need, though. I need to run x64 images on my M1. Can I do that via the included qemu?"

A: "no - the included qemu will only run images matching the host architecture.
We’re looking into cross-architecture runs for the future, but it’s not going to happen very soon.
You could try and build qemu upstream, though - you need qemu-x86_64-static."

https://discourse.ubuntu.com/t/announcing-the-first-release-candidate-for-apple-m1-support/24445/11

kelson42, 2021-11-25:
> root@box:/opt/iiab/kiwix/bin# ./kiwix-serve
> bash: ./kiwix-serve: cannot execute binary file: Exec format error

I think the message is clear, the architecture of the system is not the one the binary is compiled for.

avni, 2021-11-26:
Same error while installing kiwix on multipass

**avni@Avnis-MBP ~ % multipass version**
```
multipass   1.8.1+mac
multipassd  1.8.1+mac
```

```
...
TASK [kiwix : Enable & Ensure 'kiwix-serve.service' is running] ******************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service kiwix-serve: Job for kiwix-serve.service
failed because the control process exited with error code.\nSee \"systemctl status kiwix-serve.service\" and \"journalctl -xe\" for details.\n"}

PLAY RECAP ***********************************************************************************************************
127.0.0.1                  : ok=306  changed=169  unreachable=0    failed=1    skipped=63   rescued=0    ignored=2  
``` 

**[ubuntu@iiab-test:~$ systemctl status kiwix-serve.service**
```
kiwix-serve.service - Provides the Kiwix ZIM File Server
     Loaded: loaded (/etc/systemd/system/kiwix-serve.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Fri 2021-11-26 12:02:44 EST; 10s ago
    Process: 53065 ExecStart=/opt/iiab/kiwix/bin/kiwix-serve --daemon --port 3000 --nolibrarybutton --library /librar>

Nov 26 12:02:44 box systemd[1]: Starting Provides the Kiwix ZIM File Server...
Nov 26 12:02:44 box systemd[53065]: kiwix-serve.service: Failed to execute command: Exec format error
Nov 26 12:02:44 box systemd[53065]: kiwix-serve.service: Failed at step EXEC spawning /opt/iiab/kiwix/bin/kiwix-serve>
Nov 26 12:02:44 box systemd[1]: kiwix-serve.service: Control process exited, code=exited, status=203/EXEC
Nov 26 12:02:44 box systemd[1]: kiwix-serve.service: Failed with result 'exit-code'.
Nov 26 12:02:44 box systemd[1]: Failed to start Provides the Kiwix ZIM File Server.
```

avni, 2021-11-26:
and, unfortunately, same result on trial version of Parallels: 

iiab-diagnostics file: http://sprunge.us/jdHlFC?en

```
...
TASK [kiwix : Enable & Ensure 'kiwix-serve.service' is running] **************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service kiwix-serve: Job for kiwix-serve.service
failed because the control process exited with error code.\nSee \"systemctl status kiwix-serve.service\" and \"journalctl -xe\" for details.\n"}

PLAY RECAP *******************************************************************************************************************
127.0.0.1                  : ok=37   changed=10   unreachable=0    failed=1    skipped=15   rescued=0    ignored=0   
```

**root@box:/opt/iiab/iiab# systemctl status kiwix-serve.service**
```
kiwix-serve.service - Provides the Kiwix ZIM File Server
     Loaded: loaded (/etc/systemd/system/kiwix-serve.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Fri 2021-11-26 14:54:40 EST; 29min ago
    Process: 53908 ExecStart=/opt/iiab/kiwix/bin/kiwix-serve --daemon --port 3000 --nolibrarybutton --library /library/zims/l>

Nov 26 14:54:40 box systemd[1]: Starting Provides the Kiwix ZIM File Server...
Nov 26 14:54:40 box systemd[53908]: kiwix-serve.service: Failed to execute command: Exec format error
Nov 26 14:54:40 box systemd[53908]: kiwix-serve.service: Failed at step EXEC spawning /opt/iiab/kiwix/bin/kiwix-serve: Exec f>
Nov 26 14:54:40 box systemd[1]: kiwix-serve.service: Control process exited, code=exited, status=203/EXEC
Nov 26 14:54:40 box systemd[1]: kiwix-serve.service: Failed with result 'exit-code'.
Nov 26 14:54:40 box systemd[1]: Failed to start Provides the Kiwix ZIM File Server.
```

jvonau, 2021-11-26:
multi-pass be0U1s?en
```
 686 COMMAND: /usr/bin/lscpu    # CPU details
 687 
 688 Architecture:                    aarch64
 689 CPU op-mode(s):                  64-bit
 690 Byte Order:                      Little Endian
 691 CPU(s):                          1
 692 On-line CPU(s) list:             0
 693 Thread(s) per core:              1
 694 Core(s) per socket:              1
 695 Socket(s):                       1
 696 NUMA node(s):                    1
 697 Vendor ID:                       ARM
 698 Model:                           3
 699 Model name:                      Cortex-A72
 700 Stepping:                        r0p3
 701 BogoMIPS:                        48.00
 702 NUMA node0 CPU(s):               0
 703 Vulnerability Itlb multihit:     Not affected
 704 Vulnerability L1tf:              Not affected
 705 Vulnerability Mds:               Not affected
 706 Vulnerability Meltdown:          Not affected
 707 Vulnerability Spec store bypass: Vulnerable
 708 Vulnerability Spectre v1:        Mitigation; __user pointer sanitization
 709 Vulnerability Spectre v2:        Not affected
 710 Vulnerability Srbds:             Not affected
 711 Vulnerability Tsx async abort:   Not affected
 712 Flags:                           fp asimd evtstrm aes pmull sha1 sha2 crc32 fphp asimdhp cpuid dit
 713 
```
Parallels jdHlFC?en
```
 684 COMMAND: /usr/bin/lscpu    # CPU details
 685 
 686 Architecture:                    aarch64
 687 CPU op-mode(s):                  64-bit
 688 Byte Order:                      Little Endian
 689 CPU(s):                          2
 690 On-line CPU(s) list:             0,1
 691 Thread(s) per core:              1
 692 Core(s) per socket:              2
 693 Socket(s):                       1
 694 NUMA node(s):                    1
 695 Vendor ID:                       ARM
 696 Model:                           0
 697 Stepping:                        r0p0
 698 BogoMIPS:                        48.00
 699 NUMA node0 CPU(s):               0,1
 700 Vulnerability Itlb multihit:     Not affected
 701 Vulnerability L1tf:              Not affected
 702 Vulnerability Mds:               Not affected
 703 Vulnerability Meltdown:          Not affected
 704 Vulnerability Spec store bypass: Mitigation; Speculative Store Bypass disabled via prctl
 705 Vulnerability Spectre v1:        Mitigation; __user pointer sanitization
 706 Vulnerability Spectre v2:        Not affected
 707 Vulnerability Srbds:             Not affected
 708 Vulnerability Tsx async abort:   Not affected
 709 Flags:                           fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm jscvt fcma lrcpc dcpop sha3 asimddp sha512 asimdfhm dit uscat ilrcpc flagm ssbs sb paca pacg dcpodp flagm2 frint
 710 
```

Think you have to find a way to make qemu lie about and downgrade what cpu is presented to the VM from what is the underling M1 CPU is actually capable of.

jvonau, 2021-11-26:
Seems there should be a file that has the settings for how the VM is configured that would be passed to qemu on the command line. https://qemu-project.gitlab.io/qemu/system/arm/cpu-features.html#using-cpu-features I'm trying to figure out if/where multipass stores this kind of information.

jvonau, 2021-11-27:
> @jvonau
>
> **root@box:/opt/iiab/kiwix/bin# sudo /usr/bin/iiab-make-kiwix-lib**
>
> ```
> Waiting to run iiab-make-kiwix-lib.py
> Now running iiab-make-kiwix-lib.py
> cp: cannot stat [...]

I suggest trying transferring a library.xml that only has the test.zim installed from another machine into place at /library/zims/ and restarting with `sudo systemctl restart kiwix-serve` as a data point, what is failing is the creation of the library.xml as it is absent preventing the normal startup of kiwix-serve.

From o63HoX
```
COMMAND: /usr/bin/lscpu    # CPU details

Architecture:                    aarch64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
CPU(s):                          1
On-line CPU(s) list:             0
Thread(s) per core:              1
Core(s) per socket:              1
Socket(s):                       1
NUMA node(s):                    1
Vendor ID:                       ARM
Model:                           3
Model name:                      Cortex-A72
Stepping:                        r0p3
BogoMIPS:                        108.00
NUMA node0 CPU(s):               0
Vulnerability Itlb multihit:     Not affected
Vulnerability L1tf:              Not affected
Vulnerability Mds:               Not affected
Vulnerability Meltdown:          Not affected
Vulnerability Spec store bypass: Vulnerable
Vulnerability Spectre v1:        Mitigation; __user pointer sanitization
Vulnerability Spectre v2:        Vulnerable
Vulnerability Srbds:             Not affected
Vulnerability Tsx async abort:   Not affected
Flags:                           fp asimd evtstrm crc32 cpuid

```
The difference between what doesn't work and working VMs are the lines CPU op-mode(s) where 32-bit is present in the working setups and the various Flags that are returned.

avni, 2021-11-28:
Successfully installed iiab by setting the following parameters in /etc/iiab/local_vars.yml to False: 

```
kiwix_install: False
kiwix_enabled: False
```

- **iiab-diagnostics output: http://sprunge.us/GruRt1?en**

jvonau, 2021-11-28:
If you need need a library.xml file to try with the above test let me know, you have shown that the file is missing before the kiwix failure as noted above.

holta, 2021-11-28:
Just FYI @avni Admin Console no longer (fully) requires Kiwix to be installed: (as of ~7 months ago)

- https://github.com/iiab/iiab-admin-console/pull/390 says "don't fail to start Admin Console if kiwix not installed. this will cause other problems." (according to @tim-moody on 2021-04-05, but possibly things have changed a bit since then?)

jvonau, 2021-11-28:
```
 103 devicetree_model = none
 118 devicetree_model = none
 880 sda       8:0    0   20G  0 disk 
 881 ├─sda1    8:1    0 19.9G  0 part /
 882 └─sda15   8:15   0   99M  0 part /boot/efi
```
and what kind of VM is this one?

avni, 2021-11-28:
> If you need need a library.xml file to try with the above test let me know, you have shown that the file is missing before the kiwix failure as noted above.

Yes please, that would be great. 

> and what kind of VM is this one?
**avni@Avnis-MBP ~ % multipass --version**
```
multipass   1.8.1+mac
multipassd  1.8.1+mac
```

**avni@Avnis-MBP ~ % multipass list**
u2004-no-kiwix       Ubuntu 20.04 LTS

jvonau, 2021-11-28:
Opps I see multipass, but install false is not what I would call successful but a workaround. Next data point is the replacement library.xml test, not sure why a python3 routine fails to create the needed library.xml only on this M1 platform under a VM, seems like a bug or a more refined config is need in the VM setup somewhere.

avni, 2021-11-28:
agree, this is a workaround. Would appreciate a library.xml file to test - that would be great.

jvonau, 2021-11-28:
`sudo wget http://10.8.0.42/local_content/library.xml -O /library/zims/library.xml`

holta, 2021-11-28:
> `sudo wget http://10.8.0.42/local_content/library.xml -O /library/zims/library.xml`

Alternatives?

- /opt/iiab/iiab/roles/kiwix/templates/library.xml.deprecated (that every IIAB already has...might be stale...and is also online below...)
- https://github.com/iiab/iiab/blob/master/roles/kiwix/templates/library.xml.deprecated

jvonau, 2021-11-28:
FWIW the .42 machine is a multipass 20.04 VM just like what avni is running but working under ubunutu 21.04 Mate 64bit desktop host on a RPi 400.

jvonau, 2021-11-28:
That old file is missing the favicon= entry present in a produced library.xml file.

tim-moody, 2021-11-28:
@kelson42 wrote:

>>root@box:/opt/iiab/kiwix/bin# ./kiwix-serve
bash: ./kiwix-serve: cannot execute binary file: Exec format error

>I think the message is clear: the architecture of the system is not the one the binary is compiled for.

Is there a reason to think he is wrong? If kiwix-serve won't run then library.xml is irrelevant. Also, if kiwix-serve won't run then probably kiwix-manage, used to create library.xml, won't run either as both are part of kiwix-tools.

holta, 2021-11-28:
@tim-moody wrote:

> if kiwix-serve won't run then probably kiwix-manage, used to create library.xml, won't run either as both are part of kiwix-tools.

Likely.  Anyway that was my assumption earlier this week.

But maybe @avni & @jvonau can now try it to confirm.

jvonau, 2021-11-28:
> @kelson42 wrote:
>
>
>
> Is there a reason to think he is wrong? If kiwix-serve won't run then library.xml is irrelevant. Also, if kiwix-serve won't run then probably kiwix-manage, used to create [...]

Ah right, that logic has been buried within iiab_lib.py and shells out to run kiwix-manage, compiled at the same time as kiwix-serve, that would explain the lack of library.xml with a silent failure with no explanation as to why. 
 
```
ubuntu@box:/opt/iiab/iiab$ systemctl status kiwix-serve.service 
● kiwix-serve.service - Provides the Kiwix ZIM File Server
     Loaded: loaded (/etc/systemd/system/kiwix-serve.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sun 2021-11-28 08:51:00 CST; 1min 33s ago
    Process: 13213 ExecStart=/opt/iiab/kiwix/bin/kiwix-serve --daemon --port 3000 --nolibrarybutton --library /library/zims/library.xml --urlRootLocation=/kiwix/ --threads >

Nov 28 08:51:00 box systemd[1]: Starting Provides the Kiwix ZIM File Server...
Nov 28 08:51:00 box kiwix-serve[13213]: Unable to open the XML library file '/library/zims/library.xml'.
Nov 28 08:51:00 box systemd[1]: kiwix-serve.service: Control process exited, code=exited, status=1/FAILURE
Nov 28 08:51:00 box systemd[1]: kiwix-serve.service: Failed with result 'exit-code'.
Nov 28 08:51:00 box systemd[1]: Failed to start Provides the Kiwix ZIM File Server.

```
I'll bet the 203 error message code doesn't change, systemd looks to be testing before launching the binary and never gets to evaluate the missing library.xml part. The perplexing part is multipass on ubunt works while multipass on M1 doesn't, think the difference with ubuntu on a RPi host, the host is in mixed mode with `CPU op-mode(s):  32-bit, 64-bit` while the M1 is in 64-bit only. For the M1 I see 2 ways forward find a way to 'downgrade' the VM cpu if that is even possible or wait for a proper 64-bit kiwix binary as more 64-bit only hardware becomes available.

> Likely. Anyway that was my assumption earlier this week.

@holta You sat there with your mouth shut, wasting people's time chasing an unsupported combination? Make me reevaluate the amount of time I will allocate to future issues when they pop up.

> But maybe @avni & @jvonau can now try it to confirm.

For me close this as "won't fix" as this has nothing to do with iiab but rather an upstream issue. Now would be the time to update the notes with a known issue for M1 based VMs.

jvonau, 2021-11-28:
Just to complete some more of the '64bit' question I booted RaspiOS with arm_64bit=1 in /boot/config.txt on the 32bit version of the OS, that is the documented way to enable the 64bit kernel, while ubuntu has this already present within what would be the same file on the first partition /boot/firmware/config.txt.
```
pi@box:~ $ uname -a
Linux box 5.10.63-v8+ #1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 GNU/Linux

pi@box:~ $ sudo /usr/bin/lscpu
Architecture:                    aarch64
Byte Order:                      Little Endian
CPU(s):                          4
On-line CPU(s) list:             0-3
Thread(s) per core:              1
Core(s) per socket:              4
Socket(s):                       1
Vendor ID:                       ARM
Model:                           3
Model name:                      Cortex-A72
Stepping:                        r0p3
CPU max MHz:                     1500.0000
CPU min MHz:                     600.0000
BogoMIPS:                        108.00
Vulnerability Itlb multihit:     Not affected
Vulnerability L1tf:              Not affected
Vulnerability Mds:               Not affected
Vulnerability Meltdown:          Not affected
Vulnerability Spec store bypass: Vulnerable
Vulnerability Spectre v1:        Mitigation; __user pointer sanitization
Vulnerability Spectre v2:        Vulnerable
Vulnerability Srbds:             Not affected
Vulnerability Tsx async abort:   Not affected
Flags:                           fp asimd evtstrm crc32 cpuid
```
and note that `CPU op-mode(s):` is missing from the output and kiwix works.

avni, 2021-11-29:
>> Just FYI @avni Admin Console no longer (fully) requires Kiwix to be installed: (as of ~7 months ago)

Confirmed, I'm able to login into /admin/ from a browser and I successfully installed a couple of content packs including the Central America map region on an IIAB instance where kiwix-serve is not installed.

avni, 2021-11-29:
> sudo wget http://10.8.0.42/local_content/library.xml -O /library/zims/library.xml
> But maybe @avni & @jvonau can now try it to confirm.
I get asked for a password when trying to grab this file? I'll ping you on Skype.

> wasting people's time chasing an unsupported combination?
@jvonau I disagree the effort outlined in the ticket was a waste of my time. We learned a week ago that kiwix-serve didn't start on Ubuntu 20.04 arm64 UTM VM on an M1 Mac. I could have probably guessed that since kiwix-serve failed on UTM, it wouldn't work on multipass or Parallels either, but UTM was a new find, so it felt worth the effort to try different VMs. 

This issue also gave me a chance to do 10x+ rapid IIAB installs on different VMs making me  far more comfortable with IIAB installs than I have ever been. If the install had worked as expected on Sunday afternoon, I would be a lot less comfortable. And I appreciate you taking the time to help. You gave me things to try and keey exploring, which are invaluable for someone who's not as familiar with iiab. 

> Now would be the time to update the notes with a known issue for M1 based VMs.
Made a note on this WIP guide that Denis and I are iterating on per our discussion in #3027: 
https://github.com/iiab/iiab/wiki/Technical-Contributors-Guide. The guide still needs further editing but we are making progress.

jvonau, 2021-11-29:
> > sudo wget http://10.8.0.42/local_content/library.xml -O /library/zims/library.xml
> > But maybe @avni & @jvonau can now try it to confirm.
> 
> I get asked for a password when trying to grab this file? I'll ping you on Skype.

I shut the instance down, might be a different machine now. A different binary built with the same tool-chain should have the same result, doesn't execute wrong architecture. I just forgot where kiwix-manage was called from cause some old info gets flushed with time while new learning is on going.
 
> > wasting people's time chasing an unsupported combination?
> 
> @jvonau I disagree the effort outlined in this issue was a waste of my time. We learned a week ago that kiwix-serve didn't start on Ubuntu 20.04 arm64 UTM VM on an M1 Mac. I could have probably guessed that since kiwix-serve failed on UTM, it wouldn't work on multipass or Parallels either, but UTM was a new find, so it felt worth the effort to try different VMs.
> 

Given the effort you put into looking for a quick, easy solution when you tried mulipass I saw an opportunity for a 1 to 1 testing variable and installed multipass, amazingly with relative ease on my ubuntu RPi host machine. I invested this time to have a control variable with the same VM in play to gather further information for further comparison notes against the M1, it's not just your time. The discovery of library.xml not being created was met with no response when I tagged Adam when pointed out, guess I should of tagged Tim as that is his code his domain for a faster response but whatever, silence speaks volumes.
 
> This issue also gave me a chance to do 10x+ rapid IIAB installs on different VMs making me far more comfortable with IIAB installs than I have ever been. If the install had worked as expected on Sunday afternoon 7 days ago, I would be a lot less comfortable. 

The 'that was too easy' part gets me every time I see a successful install on a code base evolution that is pushing 10 years years now. It when you hear silence from the user base I wonder if it is **that** trouble free or if nobody is using the code. lol..

> And I appreciate you taking the time to help. You gave me things to try and keep exploring, which are invaluable for someone who's not as familiar with iiab.
> 

On a very positive note what you have running without iiab being installed is a very good base for a builder to recompile kiwix-tools for arm64 within that VM. Personally, now that I see that multipass is fairly snappy in response it might make a good builder with Ubuntu on a RPi4 to recompile kiwix-tools for arm64 before trying on the M1 VM, just to eliminate the M1 as a variable. Now what would be missing would be an arm64 build target within kiwix-tools and some guidance, perhaps @kelson42 could arrange a branch with some changes with the best guess changes that might suitable as a starting point to bang on to make that happen? That feedback loop should be at kixix-tools.

> Made a note on this WIP guide that Denis and I are iterating on per our discussion in #3027: https://github.com/iiab/iiab/wiki/Technical-Contributors-Guide. The guide still needs further editing but [...]

More fingers on the keyboard and eyes on the monitor the better. :)

holta, 2021-12-02:
Somewhat related, on the topic of Raspberry Pi Imager not running on M1 Mac CPU's:

- https://github.com/raspberrypi/rpi-imager/issues/235

avni, 2021-12-04:
@tim-moody wrote: 
Is there a reason to think he is wrong? If kiwix-serve won't run then library.xml is irrelevant. Also, if kiwix-serve won't run then probably kiwix-manage, used to create library.xml, won't run either as both are part of kiwix-tools.

Confirmed for completeness sake. Tested with a library.xml from @holta on a Multipass VM with Ubuntu 20.04 ARM64 on M1 Mac.

**ubuntu@box:/opt/iiab/iiab$ sudo ./runrole kiwix**
....
```
TASK [kiwix : Enable & Ensure 'kiwix-serve.service' is running] ********************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service kiwix-serve: Job for kiwix-serve.service failed because the control process exited with error code.\nSee \"systemctl status kiwix-serve.service\" and \"journalctl -xe\" for details.\n"}

PLAY RECAP **********************************************************************************************************
127.0.0.1                  : ok=47   changed=15   unreachable=0    failed=1    skipped=9    rescued=0    ignored=0   
```

jvonau, 2021-12-05:
@avni Using runrole executes iiab-make-kiwix-lib, within that routine the old file is moved out of the way while the new file being created is known as library.xml.tmp before being moved into place as library.xml, the creation of the new file is failing. https://github.com/iiab/iiab/issues/3039#issuecomment-977844051. What I suggested was to copy the library.xml file into place and `sudo systemctl restart kiwix-serve` bypassing what the role does, then check the service with `sudo systemctl status kiwix-serve`, don't think return the 203 error returned will change.

avni, 2021-12-05:
Thanks for the clarif @jvonau. Here's what I got - let me know if I should do/try anything else. 

Recopied library.xml into /library/zims/

```
[root@box:/library/zims# sudo systemctl restart kiwix-serve
Job for kiwix-serve.service failed because the control process exited with error code.
See "systemctl status kiwix-serve.service" and "journalctl -xe" for details.
root@box:/library/zims# sudo systemctl status kiwix-serve
kiwix-serve.service - Provides the Kiwix ZIM File Server
     Loaded: loaded (/etc/systemd/system/kiwix-serve.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sat 2021-12-04 17:20:10 EST; 12s ago
    Process: 26502 ExecStart=/opt/iiab/kiwix/bin/kiwix-serve --daemon --port 3000 --nolibrarybutton --library /library/zims/library.xml --urlRootLocation=/kiwix/ --threads 4 (code=exited, st>

Dec 04 17:20:10 box systemd[1]: Starting Provides the Kiwix ZIM File Server...
Dec 04 17:20:10 box systemd[1]: kiwix-serve.service: Control process exited, code=exited, status=203/EXEC
Dec 04 17:20:10 box systemd[1]: kiwix-serve.service: Failed with result 'exit-code'.
Dec 04 17:20:10 box systemd[1]: Failed to start Provides the Kiwix ZIM File Server.
```

tim-moody, 2021-12-05:
Keep thinking this will come to a conclusion, but the need to test library.xml in the assumption that iiab-make-kiwix-lib is the problem persists.

If I wanted to know whether kiwix-serve is able to run I would issue 

/opt/iiab/kiwix/bin/kiwix-serve --version

If that did not yield an error I would run 

/opt/iiab/kiwix/bin/kiwix-manage

avni, 2021-12-05:
I keep thinking this will come to a conclusion too but I like troubleshooting! 😄 

I think I did the kiwix-serve command 11 days ago but I ran it again in case the populated library.xml file helped. No luck: 

```
root@box:~# /opt/iiab/kiwix/bin/kiwix-serve --version
bash: /opt/iiab/kiwix/bin/kiwix-serve: cannot execute binary file: Exec format error

root@box:~# /opt/iiab/kiwix/bin/kiwix-manage
bash: /opt/iiab/kiwix/bin/kiwix-manage: cannot execute binary file: Exec format error
```

For now, given that Rosetta 2 emulation works on M1 Macs, suggest, we recommend that as the path forward in the near-term. @holta and I updated the documentation here yesterday: https://github.com/iiab/iiab/wiki/Technical-Contributors-Guide#install-iiab-on-a-virtual-machine

Thank you!

jvonau, 2021-12-05:
@avni Now that you can boot x86_64 based iso files for the VM are you going to continue to try Ubuntu server? if so go with 21.10. Now with a VM that could handle a shared directory between the VM and the HOST where there is ample space for images of all sorts, this VM could be turned into a content generator. The content is packed up and ready for use in other machines quite quickly with what I have in mind in #3051 with a proof of concept running http://sprunge.us/K9OPQd

jvonau, 2022-04-05:
Did this come to a conclusion at some point?

holta, 2022-04-05:
> Did this come to a conclusion at some point?

@avni put a lot of work into this ~4 months ago.  She should summarize, but I think the conclusion is that kiwix-serve does not have enough engineering resources to compile and maintain kiwix-tools for M1 Macs:

- https://github.com/kiwix/kiwix-build/issues/396

This issue will probably not go away.  Example: a  growing number of M1 Mac users keep complaining to the Raspberry Pi Foundation that they cannot easily run rpi-imager on their fancy M1 Macs, without stumbling through not-fully-supported emulation hacks:

- https://github.com/raspberrypi/rpi-imager/search?q=M1&type=issues

It's a dilemma, affecting Kiwix, Raspberry Pi Imager, and VirtualBox here too:

- https://apple.stackexchange.com/questions/422565/does-virtualbox-run-on-apple-silicon/422566#422566

My own opinion: Apple. Inc should start to THINK DIFFERENT and start to take seriously the Global Public &mdash; providing financial grants to responsible community compilation efforts similar to those above &mdash; with so many of these efforts remain stuck 17 months after M1 Macs went on sale in November 2020.

(Large platforms cannot + should not expect unpaid grassroots FOSS maintainers to do their gardening for them...)

holta, 2024-08-10:
@avni can you confirm almost 3 years later — whether or not .zim files might possibly now work on M1 Macs — in any Ubuntu 24.04 VM?

In other words, ensure these 2 lines are included in the VM's /etc/iiab/local_vars.yml (best to do this prior to installing IIAB...)

```
kiwix_install: True
kiwix_enabled: True
```

...and then confirm (e.g. after IIAB is installed in the VM) by running:

```
systemctl status kiwix-serve
```

(...and if so of course reconfirm by installing any tiny Wikipedia-or-similar using IIAB's **Admin Console > Install Content > Get ZIM Files from Kiwix**.)

holta, 2024-08-10:
1. @avni better yet, if you also hard-code the following line into your M1 Mac VM's /etc/iiab/local_vars.yml
```
kiwix_tar_gz: https://download.kiwix.org/release/kiwix-tools/kiwix-tools_macos-arm64.tar.gz
```
2. Then install IIAB.

3. It's possible `systemctl status kiwix-serve` and .zim files **_might_** then work ?! :pray:

PS if you have good luck here, I'd then refine our `kiwix_arch_dict` lookup table — e.g. based on the output of things like `dpkg --print-architecture` in your VM:

https://github.com/iiab/iiab/blob/ee3781d38a66e353f45ff36ebda5f5815705f297/roles/kiwix/defaults/main.yml#L28-L34

Which (if it works!) would also require overriding `linux` to be `macos` in this one particular case:

https://github.com/iiab/iiab/blob/ee3781d38a66e353f45ff36ebda5f5815705f297/roles/kiwix/defaults/main.yml#L46

[14 more replies]

closed 2024-08-13 by holta