iiab/iiab #3047: Nextcloud 22 won't install on PHP 8.1.0RC4 which is part of Ubuntu 22.04 pre-releases

holta, 2021-11-25:
FYI this LARGE-sized IIAB install attempt a daily build pre-release of Ubuntu Desktop 22.04 is just serving as a basic sanity-check for what's coming in future months (198-u2204-desk-LARGE = 10.8.0.22).

Thanks to @icebox827 who looked into this.

And no surprise Nextcloud isn't quite ready &mdash; this is just something to keep an eye on in coming weeks/months, with PHP 8.1 coincidentally just released today:

- https://www.php.net/releases/8.1/en.php
- https://www.php.net/archive/2021.php#2021-11-25-1

So [Nextcloud 23](https://github.com/nextcloud/server/wiki/Maintenance-and-Release-Schedule#nextcloud-23) might even fix this in less than a week from now?  See the error at bottom here:

> 2021-11-25 14:10:05,374 p=58549 u=root n=ansible | TASK [nextcloud : Allow Nextcloud access from all hosts and IP addresses -- SEE ALSO THE OLD WAY: https://github.com/iiab/iiab/blob/master/roles/nextcloud/templates/nextcloud.conf.j2.unused] ***
2021-11-25 14:10:05,856 p=58549 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": ["php", "/library/www/nextcloud/occ", "config:system:set", "trusted_domains", "1", "--value=*"], "delta": "0:00:00.043385", "end": "2021-11-25 14:10:05.814422", "msg": "non-zero return code", "rc": 255, "start": "2021-11-25 14:10:05.771037", "stderr": "PHP Warning:  PHP Startup: Unable to load dynamic library 'imagick.so' (tried: /usr/lib/php/20210902/imagick.so (/usr/lib/php/20210902/imagick.so: undefined symbol: ZVAL_NEW_ARR), /usr/lib/php/20210902/imagick.so.so (/usr/lib/php/20210902/imagick.so.so: cannot open shared object file: No such file or directory)) in Unknown on line 0", "stderr_lines": ["PHP Warning:  PHP Startup: Unable to load dynamic library 'imagick.so' (tried: /usr/lib/php/20210902/imagick.so (/usr/lib/php/20210902/imagick.so: undefined symbol: ZVAL_NEW_ARR), /usr/lib/php/20210902/imagick.so.so (/usr/lib/php/20210902/imagick.so.so: cannot open shared object file: No such file or directory)) in Unknown on line 0"], "stdout": "This version of Nextcloud is not compatible with > PHP 8.0.<br/>You are currently running 8.1.0RC4.", "stdout_lines": ["**This version of Nextcloud is not compatible with > PHP 8.0.<br/>You are currently running 8.1.0RC4.**"]}
2021-11-25 14:10:05,859 p=58549 u=root n=ansible | PLAY RECAP **********************************************************************************************************************************************************************************
2021-11-25 14:10:05,859 p=58549 u=root n=ansible | 127.0.0.1                  : ok=157  changed=79   unreachable=0    failed=1    skipped=34   rescued=0    ignored=0

[iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md) output:

http://sprunge.us/InwZ4w?en

Refs:

- #2818

holta, 2021-11-25:
Just as an experiment I put Nextcloud 23.0.0 Release Candidate 3 in [/etc/iiab/local_vars.yml](https://FAQ.IIAB.IO#What_is_local_vars.yml_and_how_do_I_customize_it.3F) :

```
nextcloud_dl_url: https://download.nextcloud.com/server/prereleases/nextcloud-23.0.0rc3.tar.bz2
```

Then I ran:

```
sudo su -
cd /opt/iiab/iiab
./runrole nextcloud
```

Error is the same as ~2 hours ago:

> This version of Nextcloud is not compatible with > PHP 8.0.<br/>You are currently running 8.1.0RC4.

So likely Nextcloud "23.1" (or 24?) will be needed with OS's (like Ubuntu 22.04) that contain PHP 8.1...

holta, 2021-11-26:
Another PHP 8.1 issue to monitor in coming weeks/months...with Moodle in this case:

> 2021-11-25 17:36:01,235 p=88499 u=root n=ansible | TASK [moodle : Execut
e /opt/iiab/moodle/moodle_installer IF /opt/iiab/moodle/config.php doesn
't yet exist -- REQUIRES 'max_input_vars = 5000' (or higher) in /etc/php
/8.1/cli/php.ini IF PHP 8+ (as set up by www_options/tasks/main.yml) --
WHEREAS LATER ON Moodle uses /etc/php/8.1/fpm/php.ini during regular ope
ration] ***
2021-11-25 17:36:01,854 p=88499 u=root n=ansible | fatal: [127.0.0.1]: F
AILED! => {"changed": true, "cmd": "/opt/iiab/moodle/moodle_installer",
"delta": "0:00:00.265275", "end": "2021-11-25 17:36:01.812396", "msg": "
non-zero return code", "rc": 255, "start": "2021-11-25 17:36:01.547121",
 "stderr": "+ sudo -u www-data /usr/bin/php /opt/iiab/moodle/admin/cli/i
nstall.php --wwwroot=http://box.lan/moodle --dataroot=/library/moodle --
dbtype=pgsql --dbname=moodle --dbuser=Admin --dbpass=changeme --fullname
=Your_School --shortname=School --adminuser=admin --adminpass=changeme -
-non-interactive --agree-license --allow-unstable\nPHP Warning:  PHP Sta
rtup: Unable to load dynamic library 'imagick.so' (tried: /usr/lib/php/2
0210902/imagick.so (/usr/lib/php/20210902/imagick.so: undefined symbol:
ZVAL_NEW_ARR), /usr/lib/php/20210902/imagick.so.so (/usr/lib/php/2021090
2/imagick.so.so: cannot open shared object file: No such file or directo
ry)) in Unknown on line 0\nPHP Deprecated:  Return type of moodle_record
set::rewind() should either be compatible with Iterator::rewind(): void,
 or the #[\\ReturnTypeWillChange] attribute should be used to temporaril
y suppress the notice in /opt/iiab/moodle/lib/dml/moodle_recordset.php o
n line 61\nPHP Fatal error:  Type of xml_format_exception::$line must be
 int (as in class Exception) in /opt/iiab/moodle/lib/xmlize.php on line
42", "stderr_lines": ["+ sudo -u www-data /usr/bin/php /opt/iiab/moodle/
admin/cli/install.php --wwwroot=http://box.lan/moodle --dataroot=/librar
y/moodle --dbtype=pgsql --dbname=moodle --dbuser=Admin --dbpass=changeme
 --fullname=Your_School --shortname=School --adminuser=admin --adminpass
=changeme --non-interactive --agree-license --allow-unstable", "PHP Warn
ing:  PHP Startup: Unable to load dynamic library 'imagick.so' (tried: /
usr/lib/php/20210902/imagick.so (/usr/lib/php/20210902/imagick.so: undef
ined symbol: ZVAL_NEW_ARR), /usr/lib/php/20210902/imagick.so.so (/usr/li
b/php/20210902/imagick.so.so: cannot open shared object file: No such fi
le or directory)) in Unknown on line 0", "PHP Deprecated:  Return type o
f moodle_recordset::rewind() should either be compatible with Iterator::
rewind(): void, or the #[\\ReturnTypeWillChange] attribute should be use
d to temporarily suppress the notice in /opt/iiab/moodle/lib/dml/moodle_
recordset.php on line 61", "PHP Fatal error:  Type of xml_format_excepti
on::$line must be int (as in class Exception) in /opt/iiab/moodle/lib/xm
lize.php on line 42"], "stdout": "\nDeprecated: Return type of moodle_re
cordset::rewind() should either be compatible with Iterator::rewind(): v
oid, or the #[\\ReturnTypeWillChange] attribute should be used to tempor
arily suppress the notice in /opt/iiab/moodle/lib/dml/moodle_recordset.p
hp on line 61\n\nFatal error: Type of xml_format_exception::$line must b
e int (as in class Exception) in /opt/iiab/moodle/lib/xmlize.php on line
 42", "stdout_lines": ["", "Deprecated: Return type of moodle_recordset:
:rewind() should either be compatible with Iterator::rewind(): void, or
the #[\\ReturnTypeWillChange] attribute should be used to temporarily su
ppress the notice in /opt/iiab/moodle/lib/dml/moodle_recordset.php on li
ne 61", "", "Fatal error: Type of xml_format_exception::$line must be in
t (as in class Exception) in /opt/iiab/moodle/lib/xmlize.php on line 42"
]}

holta, 2021-11-26:
The LARGE-sized IIAB install is essentially done, onto 198-u2204-desk-LARGE.

Most all links on its IIAB main page (http://10.8.0.22) appear to work...with these 3 notable exceptions:

- http://10.8.0.22/lokole (502 Bad Gateway)
- http://10.8.0.22/nodered (502 Bad Gateway)
- http://10.8.0.22/wordpress (HTTP ERROR 500)

... possibly related to Ubuntu Desktop 22.04's PHP 8.1 RC4 ?

In any case: Node-RED and WordPress will quite likely fix themselves (to work with PHP 8.1 etc) in coming months...

jvonau, 2021-11-27:
With the expected release of 7.2 before this year's end this milestone should be tagged as 8.0 to match the April release of 22.04.

jvonau, 2021-11-30:
> The LARGE-sized IIAB install is essentially done, onto 198-u2204-desk-LARGE.
> 
> Most all links on its IIAB main page (http://10.8.0.22) appear to work...with these 3 notable exceptions:
> 
>     * http://10.8.0.22/lokole (502 Bad Gateway)
> 
>     * http://10.8.0.22/nodered (502 Bad Gateway)
> 

Try these with a tailing slash ie 10.8.0.22/lokole/ in the next report.

jvonau, 2021-11-30:
https://github.com/iiab/iiab/pull/2843#issuecomment-951090944

holta, 2021-11-30:
Oops as mentioned ~2.5 days ago, http://10.8.0.22/lokole works after @jvonau's workaround is run: (essentially installing Lokole twice)

- https://github.com/iiab/iiab/issues/2833#issuecomment-981079242

jvonau, 2021-12-01:
Try the the training `/` before re-installing, the need for a reinstall might be superficial because the nginx template is a bit off, but I'll leave that up to the assembled cast of experts to fully qualify.

jvonau, 2021-12-01:
https://moodle.org/mod/forum/discuss.php?d=429511

holta, 2021-12-01:
FYI Nextcloud 23 is now released here:

- https://download.nextcloud.com/server/releases/

As mentioned here:

- https://github.com/iiab/iiab/pull/2684#issuecomment-983080981

jvonau, 2022-04-05:
where is this at with current 22.04?

jvonau, 2022-04-14:
The usual nodejs and mongo holding off till official release.
Nextcloud refuses to install:
>2022-04-14 16:19:34,443 p=29557 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": ["php", "/library/www/nextcloud/occ", "config:system:set", "trusted_domains", "1", "--value=*"], "delta": "0:00:00.014458", "end": "2022-04-14 16:19:34.429886", "msg": "non-zero return code", "rc": 1, "start": "2022-04-14 16:19:34.415428", "stderr": "", "stderr_lines": [], "stdout": "This version of Nextcloud is not compatible with > PHP 8.0.<br/>You are currently running 8.1.2.", "stdout_lines": ["This version of Nextcloud is not compatible with > PHP 8.0.<br/>You are currently running 8.1.2."]}

Kolibri doesn't support python3.10 https://github.com/learningequality/kolibri/issues/9224
>2022-04-14 16:24:43,888 p=36282 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "export KOLIBRI_HOME=\"/library/kolibri\" && \"/usr/bin/kolibri\" manage provisiondevice --facility \"Kolibri-in-a-Box\" --superusername \"Admin\" --superuserpassword \"changeme\" --preset \"formal\" --language_id \"en\" #--preset \"formal\" --language_id \"en\" --verbosity 0 --noinput\n", "delta": "0:00:00.249374", "end": "2022-04-14 16:24:43.875094", "msg": "non-zero return code", "rc": 1, "start": "2022-04-14 16:24:43.625720", "stderr": "INFO: No C extensions are available for this platform\nTraceback (most recent call last):\n  File \"/usr/bin/kolibri\", line 11, in <module>\n    load_entry_point('kolibri==0.15.1', 'console_scripts', 'kolibri')()\n  File \"/usr/lib/python3/dist-packages/pkg_resources/__init__.py\", line 479, in load_entry_point\n    return get_distribution(dist).load_entry_point(group, name)\n  File \"/usr/lib/python3/dist-packages/pkg_resources/__init__.py\", line 2861, in load_entry_point\n    return ep.load()\n  File \"/usr/lib/python3/dist-packages/pkg_resources/__init__.py\", line 2465, in load\n    return self.resolve()\n  File \"/usr/lib/python3/dist-packages/pkg_resources/__init__.py\", line 2471, in resolve\n    module = __import__(self.module_name, fromlist=['__name__'], level=0)\n  File \"/usr/lib/python3/dist-packages/kolibri/utils/cli.py\", line 12, in <module>\n    from django.core.management import execute_from_command_line\n  File \"/usr/lib/python3/dist-packages/kolibri/dist/django/core/management/__init__.py\", line 13, in <module>\n    from django.core.management.base import (\n  File \"/usr/lib/python3/dist-packages/kolibri/dist/django/core/management/base.py\", line 17, in <module>\n    from django.db.migrations.exceptions import MigrationSchemaMissing\n  File \"/usr/lib/python3/dist-packages/kolibri/dist/django/db/migrations/__init__.py\", line 2, in <module>\n    from .operations import *  # NOQA\n  File \"/usr/lib/python3/dist-packages/kolibri/dist/django/db/migrations/operations/__init__.py\", line 1, in <module>\n    from .fields import AddField, AlterField, RemoveField, RenameField\n  File \"/usr/lib/python3/dist-packages/kolibri/dist/django/db/migrations/operations/fields.py\", line 4, in <module>\n    from django.db.models.fields import NOT_PROVIDED\n  File \"/usr/lib/python3/dist-packages/kolibri/dist/django/db/models/__init__.py\", line 5, in <module>\n    from django.db.models.deletion import (\n  File \"/usr/lib/python3/dist-packages/kolibri/dist/django/db/models/deletion.py\", line 5, in <module>\n    from django.db.models import signals, sql\n  File \"/usr/lib/python3/dist-packages/kolibri/dist/django/db/models/sql/__init__.py\", line 2, in <module>\n    from django.db.models.sql.query import *  # NOQA\n  File \"/usr/lib/python3/dist-packages/kolibri/dist/django/db/models/sql/query.py\", line 11, in <module>\n    from collections import Counter, Iterator, Mapping, OrderedDict\nImportError: cannot import name 'Iterator' from 'collections' (/usr/lib/python3.10/collections/__init__.py)", "stderr_lines": ["INFO: No C extensions are available for this platform", "Traceback (most recent call last):", "  File \"/usr/bin/kolibri\", line 11, in <module>", "    load_entry_point('kolibri==0.15.1', 'console_scripts', 'kolibri')()", "  File \"/usr/lib/python3/dist-packages/pkg_resources/__init__.py\", line 479, in load_entry_point", "    return get_distribution(dist).load_entry_point(group, name)", "  File \"/usr/lib/python3/dist-packages/pkg_resources/__init__.py\", line 2861, in load_entry_point", "    return ep.load()", "  File \"/usr/lib/python3/dist-packages/pkg_resources/__init__.py\", line 2465, in load", "    return self.resolve()", "  File \"/usr/lib/python3/dist-packages/pkg_resources/__init__.py\", line 2471, in resolve", "    module = __import__(self.module_name, fromlist=['__name__'], level=0)", "  File \"/usr/lib/python3/dist-packages/kolibri/utils/cli.py\", line 12, in <module>", "    from django.core.management import execute_from_command_line", "  File \"/usr/lib/python3/dist-packages/kolibri/dist/django/core/management/__init__.py\", line 13, in <module>", "    from django.core.management.base import (", "  File \"/usr/lib/python3/dist-packages/kolibri/dist/django/core/management/base.py\", line 17, in <module>", "    from django.db.migrations.exceptions import MigrationSchemaMissing", "  File \"/usr/lib/python3/dist-packages/kolibri/dist/django/db/migrations/__init__.py\", line 2, in <module>", "    from .operations import *  # NOQA", "  File \"/usr/lib/python3/dist-packages/kolibri/dist/django/db/migrations/operations/__init__.py\", line 1, in <module [... cut ...] __.py\", line 2, in <module>", "    from django.db.models.sql.query import *  # NOQA", "  File \"/usr/lib/python3/dist-packages/kolibri/dist/django/db/models/sql/query.py\", line 11, in <module>", "    from collections import Counter, Iterator, Mapping, OrderedDict", "ImportError: cannot import name 'Iterator' from 'collections' (/usr/lib/python3.10/collections/__init__.py)"], "stdout": "", "stdout_lines": []}

potential fix https://github.com/Azure/azure-cli/issues/20444#issuecomment-1009648464

holta, 2022-05-23:
Discussion of developments moved to:

- PR #3189
- #3190

closed 2022-05-23 by holta