iiab/iiab #3049: Python 3.10 (released 2021-10-04) and PostgreSQL 14 (released 2021-09-30) are not quite yet in Ubuntu 22.04 daily builds

holta, 2021-11-25:
FYI Ubuntu 22.04 daily build remain on Python 3.9.9 and PostgreSQL 13 for now.

It's claimed that Ubuntu 22.04 will very likely ship (on 2022-04-21) with Python 3.10 and PostgreSQL 14.

So for now, manually reverting these 2 lines during test-installs of IIAB on Ubuntu 22.04 might have to be sufficient:

- https://github.com/iiab/iiab/blob/master/vars/ubuntu-2204.yml#L27-L29

holta, 2021-11-25:
@icebox827 & All:

I've documented this interim need for manual change(s) to [/opt/iiab/iiab/vars/ubuntu-2204.yml](https://github.com/iiab/iiab/blob/master/vars/ubuntu-2204.yml#L27-L29) here:

https://github.com/iiab/iiab/wiki/IIAB-Platforms -> Section 2. -> Ubuntu 22.04

holta, 2021-12-11:
PostgreSQL 14 is now part of Ubuntu 22.04 (Jammy Jellyfish) daily builds.

Presumably Python 3.10 will be rolled in sometime very soon.

(One down, one to go...)

jvonau, 2022-04-05:
close python3.10 is part of jammy

closed 2022-04-05 by holta