iiab/iiab #3056: Improvements in /info

jvonau, 2021-12-02:
http//localhost/info should have all available docs for the offline version of "To install Internet-in-a-Box (IIAB)", the hyperlink displayed is pointing to the live https://internet-in-a-box.org/, this site needs to be scraped for the videos and adjusted to be viewed from here.

IIAB-Installation.html mentions older local_var.yml names.

holta, 2021-12-03:
Indeed.  The [/usr/bin/iiab-refresh-wiki-docs scraper](https://github.com/iiab/iiab/blob/master/roles/www_base/templates/iiab-refresh-wiki-docs.sh) that automatically creates http://box/info ("offline docs") within every IIAB...could use certain improvements now.  After a few solid years on the road, it needs a tune-up going forward, absolutely.

> IIAB-Installation.html mentions older local_var.yml names.

https://wiki.iiab.io/go/IIAB-Installation is an inflationary/legacy doc that's definitely no longer mainline, nor bearing fruit (unless the pinata contains fruit?)

Still, despite this doc being obsolescent and ripe beyond any smell test, I've removed the worst/most obvious of its errors:

https://github.com/iiab/iiab/wiki/IIAB-Installation/_compare/620861d7a27816b6994fd7f94b27ec18de08c7f7...552bbd886724dfb2d8697561d70e9654f7ace976

Such that the composting process can at least lead to organic/new growth &mdash; when this doc is in fact later reincarnated, into something far more purposeful &mdash; e.g. a genuinely illustrative video sequence of key/common screen caps possibly...

jvonau, 2025-04-06:
Don't think anything improved here